DIY-Flow-Bench/DIY-Flow-Bench #20: Verify Pressure sensor types

DeeEmm, 2019-12-23:
Need to validate / verify pressure sensor choice and code for:

- Reference pressure sensor (should use absolute pressure sensor)
- Baro pressure sensor (requires absolute pressure sensor)

ModchipMan, 2019-12-23:
MPXV7007DP as ref pressure, works, but the maths needs work, raw value at atmosphere is 510-520, so will need to calibrated for zero reading, probably at bench start time or when you do bench calibration..

ModchipMan, 2019-12-23:
ilibBMP280 as intake temp and Baro pressure, works, but needs these two lines in each section or it dosnt update values, just stays on startup value, 
ilibBmp280.awaitMeasurement();
ilibBmp280.triggerMeasurement(); 
Temp dosnt really need rounding..

DeeEmm, 2019-12-26:
I've uncommented the lines in the code in the next release.

MPXV7007DP _should_ give a zero value as it is a differential sensor. It should not need calibrating as it's only displaying the difference between the two ports.

Will take a further look.

ModchipMan, 2019-12-26:
MPXV7007DP has a raw value of around 512 - 2.5v, which is about centre of 1024 or 5v, the 2 x ports both take vac or pressure, the only difference is  if you get less or more raw value with vac or pressure, hence differential pressure between the ports
So im using the port that raw value rises with pressure and reduces with vacuum, leaving the other port open to atmosphere.
with weather change and between both the MPXV7007DP's  the other being pilot pressure they can be between 510-520 raw value difference.
I think because they are si fine in resolution they see weather pressure change as well.
You should be up and running soon and can have a look..

DeeEmm, 2019-12-26:
There were some issues with the formula used to decode the input for the DP sensor. Should be good now.

closed 2020-08-03 by DeeEmm