DIY-Flow-Bench/DIY-Flow-Bench #24: Erratic Rotary Encoder

ModchipMan, 2019-12-24:
This could be my wiring issue.
I have the 2 x Caps 150n attached
fly lead length is about 2ft long from encoder to arduino.
Pin A = D2
Pin B = D3
Pin C= GND
switchP1 = GND
switchP2 = D5
 
Issue is it will only go in a down direction with counter clockwise rotation or clockwise rotation, The push button for OK is very hit and miss to function.

DeeEmm, 2019-12-26:
Do you get the same results if the caps are removed?

ModchipMan, 2019-12-26:
Its the same with or with out caps..

ModchipMan, 2019-12-29:
I have found out what is causing this, 
its the the delays in program
I created a simple menu with tcMenu and it worked a treat, as soon as i added delay(1000); in to program, the encoder become very hit and miss..

ModchipMan, 2019-12-29:
Also, the encoder needs the 150n caps, or it also becomes erratic.

DeeEmm, 2019-12-30:
Cool. Good catch. I will remove the delay from the next update. Just set it to zero for the time being.

The delay was only there to give us an easy way to stabilise the display readout. We can also achieve that by other methods. (eg sample average).

DeeEmm, 2019-12-30:
What build are you using?

The delay has already been removed in the latest version

DeeEmm, 2020-08-03:
Any update on this?

DeeEmm, 2020-08-06:
I'm going to close this one out. I cannot reproduce it.

closed 2020-08-06 by DeeEmm