DIY-Flow-Bench/DIY-Flow-Bench #28: [BUG]

930MAF, 2020-07-29:
Just loaded beta9 for some reason the CFM only reads zero the MAF voltage does change
 with airflow. I may be doing something wrong.
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Do this '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Arduino (please complete the following information):**
 - Firmware Version: [e.g. 1.0.20042001]
 - Hardware Type [e.g. Mega 2560]

**Additional context**
Add any other context about the problem here.

DeeEmm, 2020-07-29:
Check your configuration file to make sure that the pin assignments and configuration match that which you previously used. Some of the defaults may have changed now that the ilib BME280 library has been removed.

930MAF, 2020-07-29:
Using A0 the CFM stays at 0.00 but the MAF voltage changes from 0.00 to 4.00 volts when I use A1 the MAF voltage stays at 4.00 volts and CFM reads 0.00

 

This in the pin config in configuration.h

// Define Physical Pins

    #define VOLTAGE_PIN A0

    #define MAF_PIN A1

    #define REF_PRESSURE_PIN A2

    #define REF_VAC_PIN A3

   #define PITOT_PIN A4

    #define TEMPERATURE_PIN A5

    #define REF_BARO_PIN A6

    #define HUMIDITY_PIN A7

 

This email has been checked for viruses by AVG.
https://www.avg.com

930MAF, 2020-07-30:
did some messing around and changed from this

//int numRows = sizeof(mafMapData)/sizeof(mafMapData[0]);

int numRows = sizeof(mafMapData[0])/sizeof(mafMapData[0][0]);

to this

int numRows = sizeof(mafMapData)/sizeof(mafMapData[0]); 

//int numRows = sizeof(mafMapData[0])/sizeof(mafMapData[0][0]);

now it displays CFM

not sure if this dosen't cause other issues

 

From: Val Marshall [mailto:vmarshall@suddenlink.net] 
Sent: Wednesday, July 29, 2020 1:07 PM
To: 'DeeEmm/DIY-Flow-Bench' <reply@reply.github.com>
Subject: RE: [DeeEmm/DIY-Flow-Bench] [BUG] (#28)

 

Using A0 the CFM stays at 0.00 but the MAF voltage changes from 0.00 to 4.00 volts when I use A1 the MAF voltage stays at 4.00 volts and CFM reads 0.00

 

This in the pin config in configuration.h

// Define Physical Pins

    #define VOLTAGE_PIN A0

    #define MAF_PIN A1

    #define REF_PRESSURE_PIN A2

    #define REF_VAC_PIN A3

   #define PITOT_PIN A4

    #define TEMPERATURE_PIN A5

    #define REF_BARO_PIN A6

    #define HUMIDITY_PIN A7

 

https://www.avg.com

DeeEmm, 2020-07-31:
Val, I'll have to do some testing to determine what I was doing there. It's basically setting the size of the array based on the number of data points in the mafData file. If it was previously not responding, and now you are getting valid figures then perhaps it is now correct. 

It's been six months or so since I last looked at the code, It is possible I was half way through testing something that I failed to finish. (and have subsequently forgotten).

Keep me posted with your testing. When I have had a chance to do some testing myself I will report back here.

DeeEmm, 2020-07-31:
BTW. Anyone else seeing this same issue?

DeeEmm, 2020-07-31:
It is possible that this is related to bug #14 which also relates to mafData files. I note that this is still open and was last updated in December. I will do a review on mafData file importing and try to close both of these out. It's important to get this right as everything depends on it.

DeeEmm, 2020-07-31:
BTW @930MAF what MAF are you using?

930MAF, 2020-07-31:
5WK9605 and a 74-9505 For Ford MAF also I am going to use a 1K potentiometer as a voltage divider to check to see my MAF data is reading from low to high and increment properly.

 

https://www.avg.com

DeeEmm, 2020-07-31:
Do you have data for the 74-9505 sensor? is this in volts / kg/h?

It might interest you to know that I've just added some data for a Ford Ford 1L3F-12B579-AB sensor in the forums. It is possible that this might be shared with your sensor? Might be worth a part number comparison / checkup.

https://diyflowbench.com/forum/mafdata/15-ford-1l3f-12b579-ab

This is in RAW analog data point format (1024 data points)

The data originated from the megasquirt site, so I assume it is accurate.

DeeEmm, 2020-07-31:
BTW, if you have MAF data, please share it in the MAF data forum - https://diyflowbench.com/forum/mafdata

930MAF, 2020-07-31:
Also another thing I notice if a use one of the included MAF samples such as DELPHI_AF10118.h the CFM increase up to.5 volts anything above .5v on the maf pin the CFM will decrease and becomes unstable. So I made a test maf data file all I did was add a zero to the MV column so the Left millivolts column start at100 and ends at 5000 the right KGH/1000 column starts at 100 and ends at 660000. This gives me 0 CFM to 324CFM to if I set the 5000mv much above 660000 KGH/1000   the CFM displays 326 and  then the CFM displays a negative number and starts counting back down. This just a test maf file has no calibration value. Just want if the CFM would increment smoothly for ov to 5v. 

 

https://www.avg.com

DeeEmm, 2020-07-31:
That's a good validation method.

So when you use test values in the MAF file that exceed 660000 the behaviour is that the CFM displays a negative number and starts counting back down?

This could indicate that we are overflowing a variable somewhere.

930MAF, 2020-07-31:
Yes that is what I am seeing.

 

https://www.avg.com

DeeEmm, 2020-08-01:
So I've been testing and made some changes to the code that I'm pretty happy fix this issue. 

There's a few changes, including a bit of a schoolboy error in the mafData files where instead of 0-5000mv (5v) I had the scaling at 0-500mv (0.5v) which had a bit of an obvious effect (oops).

In testing this all out I also added some additional debug commands to the serial API. 

To see a realtime printout of the MAF voltage + kg/h value and calculated cfm value, just issue a 'D' command via the serial interface. To turn it off again, issue a 'd'. Full listing of the API functions are in the WIKI - https://github.com/DeeEmm/DIY-Flow-Bench/wiki/API

Let me know how you get on.

930MAF, 2020-08-02:
Nice!

 

https://www.avg.com

DeeEmm, 2020-08-03:
If there's no objections I'm going to close this one out.

closed 2020-08-03 by DeeEmm