DIY-Flow-Bench/DIY-Flow-Bench #35: Validation

DeeEmm, 2020-08-04:
- [x] Creation of calibration plates
- [ ] 3rd party calibration of test orifices
- [ ] Calibration of bench to test orifice
- [ ] Verification of correct flow rates for other test orifice sizes
- [ ] Results obtained from more than one system

DeeEmm, 2020-08-04:
Need to validate system operation on different benches

DeeEmm, 2021-07-19:
I have closed out the V1 project #54 as the new ESP32 development has now taken over where that left off.

Whilst this task was associated with the V1 project it still remains open as validation is still required.

I still need people to get on board with building and evaluating the DIYFB controllers. 

My aim is now to package up the ESP32 boards along with the DIYFB Shield and offer them to the community for testing.

DeeEmm, 2023-02-01:
I've moved this to the V2 release as calibration orifices and a method for commonising test orifices. Originally the proposed method was to use a CD as this had a common and universal size that could easily be found worldwide. The issue found with this during testing was that the CD deformed under pressure which broke the seal to the bench receptacle. Whilst it was okay for lower test depression values, the issue became evident for higher test depressions.

I did make some orifice plates to a known dimension and calculate the flow rates, which was good for testing but this is not a great solution for everyone else.

The current best method is to use a saw blade as these are flat, made from steel and have a common universal bore size. 

Further testing is required to determine what the flow rates are for the commonly available bore sizes

DeeEmm, 2024-11-04:
(moved comment from #76)

I want to open this up again and discuss some methods for flow validation so that we can agree on a set of universal tests.

Ultimately we need to establish the accuracy of DIYFB based systems. More specifically we need to establish the accuracy of sensor based systems running the DIYFB code. So rather than look at the system as a whole, we do need to look at each one as the sum-total of it's parts.  

It is easy to do some bench to bench comparisons to compare what we all get flowing a common reference sample. But what we really need to do is record ALL sensor / data streams when doing so. This way we can compare not only the result, but how each individual bench has determined that measurement. It will give a better idea for the reasons for any differences in flow

Down the track, this methodology can also be used by new bench builders when setting up their benches to provide a kind of benchmark snapshot that can be analysed by the community. A bit like a hardware log file that can be used to help fix issues or validate builds.

We've been using the CD orifice tests for while now, and whilst these are super accessible, they fall over at higher depressions as they tend to warp and deform, leading to leaks. What I proposed a while back was to expand this test to include circular saw blades. These are also found in standard sizes (10 / 20 / 30mm) and also in imperial sizes (1/4, 1/2, 5/8").

So firstly we need to determine the flow values for these items. I did already purchase a bunch of blades and give them to a local fella to get them flowed, but despite following up several times I never received any results. I have access to another local bench, as well as potentially one stateside, so my next task is to populate a matrix of flow values.

I will update the code to generate an file that includes all of the necessary data. This way the file can be generated and then either posted here, or emailed to me so that the data can be analysed. I plan to keep all of the data in a spreadsheet or database so that we can keep track of the performance of the systems running the DIYFB code as well as undertake statistical analysis.

But. What I need is to determine a range of tests, or a testing regimen that can be carried out. Here's what I am thinking so far....

1 - RAW open test. This is with no flow calibration values or leak offsets. Just open system.
2 - RAW leak test. Same but with closed system
3 - RAW orifice test. Same with nominated test orifice (can be repeated for several different orifice sizes)
4 - Calibrated open test 
5 - Calibrated leak test
6 - Calibrated orifice test(s). Calibrated system flowing nominated orifice(s)

Would be great to do this across 10/20/30mm orifice range and for range of depressions. Maybe 5 / 10 / 15"

Test 3 will be interesting as this will demonstrate the difference between the calibrated value and actual. In a perfect world 
the MAF will correctly show the flow.

Test 6 should theoretically show no difference between the calibrated value and actual as it has been calibrated.

What other tests should we consider?? Please discuss!

DeeEmm, 2025-02-13:
Based on observation by @black-top, we are seeing some deviation between the results recorded by different MAF sensors in a side-by-side comparison - https://github.com/DeeEmm/DIY-Flow-Bench/discussions/141#

The reasons for this variation are at this point unknown. We have largely been taking the accuracy of the MAF data at face value, as focus has been elsewhere with the code development. But now that the code development is essentially finished for the current version, we no need to shift our attention to accuracy.

The fundamental basis for accuracy of the system is the accuracy of the MAF data used to decode the MAF signal. If we have bad MAF data, then the decoded flow value will be inaccurate. Whilst we have tested the system to compare the generated results against the results from the MAF data tables, we have not yet validated the MAF Data tables themselves.

So this is where we currently stand. We need to devise a way to be able to test the MAF data for any given MAF sensor to validate it. This only needs to be done once for each MAF sensor.

----------------------------------------------------

My thoughts copied from other thread - https://github.com/DeeEmm/DIY-Flow-Bench/discussions/141#

So we calibrate at a chosen data point. i.e. Ideally approx 80% of published MAF flow. If we calibrate at the higher end of the flow scale, any error should theoretically decrease as the flow reduces. If we calibrate at the lower end of the flow scale (for any given MAF) then any error could potentially increase as flow increases. 

Next we test at various other data points along the scale - perhaps 10% increments. Maybe ideally at 5% increments

This means that we need multiple calibrated orifices, or a way to approximate them.

I did purchase this device some time ago... 

[image]

It's a drafting device for drawing circles. A quick way to approximate different orifices.

We could calculate the idea sharp edged orifice diameter for any given flow and reference pressure and then set this device to that diameter. This would allow a quick and easy validation test but would obviously not be as accurate as a proper manufactured and calibrated orifice. But I would expect that it could get to within a few CFM of target. Enough to be able to validate the MAF to within a reasonable error.

----------------------------------------------------

With some care and accuracy, it would even be possible to map a MAF sensor across its entire range. This may actually generate more accurate results than using a generic MAF data table as the characteristics would be 'tuned' to the actual flow bench. i.e. the MAF data would be compensated for any losses or deviations arising from the design or integration of the MAF sensor.

This is definitely possible now that the method for converting flow data into 6th order polynomial data has been cracked. We could theoretically do this with just a handful of data points - much like the published Bosch MAF data.

Using this method ANY MAF sensor could be used, whether it had data or not...

- Install MAF
- Determine required flow value at predefined reference pressure
- Calculate ideal sharp edged orifice diameter
- Take voltage readings with orifice installed
- Repeat at various data points throughout MAFs published (or assumed) range
- Pump voltage + flow values into polynomial calculator to generate polynomials

DeeEmm, 2025-02-13:
Thinking further on this...

Lets say we adopt this as the preferred way to set up a new bench. We can provide an online calculator that allows the user to input the measured values and generate the polynomials. We can then save that data in a database. This way we can store multiple different snapshots of the same MAF, which will highlight differences between benches. It would also store a characterisation for each MAF of actual measured data, nit just published data.

It would also provide a quick lookup for anyone who wants to use a MAF - enter the MAF part number and it will show the polynomials / max flow / etc.

Of course this would depend on the accuracy of the orifuce used. So it would be good to do some tests to try and work out how accurate this could be

DeeEmm, 2025-02-13:
(Some food for thought)

black-top, 2025-02-13:
My system is made with 100mm diameter pipe, and MAF adapters are bellmouth velocity stack shape that gradually match MAF housing size. For 067 MAF I can make a 100mm housing to match the pipe, but 5wk MAF are one piece design, so I can't remove the flow element from it, so the size is fixed for it and a lot of MAFS are like that.
Yes, I agree that it is a single observation, and we can not draw conclusions from it, because the actual cause is not entirely clear, but I still think that using % error correction, to match as close as possible lets say 3 or 4 orifice flow, is far better than subtracting offset from single orifice flow. Even if the results are not match 100% on all orifices, it is still more accurate than match 100% at only one point.
The option to generate polynomials from measured points looks like the best solution so far, because it eliminates variables. But we get additional problems like orifice sets. That variable "orifice" looks like an interesting solution, but it might be flexing different amounts depending on diameter, etc. Maybe the better solution will be a single metal plate with multiple holes, similar to D.W. calibration plate shown in his book. The main problem with D.W. plate is that all orifices have different angles. If we decide optimal 6 or 8 hole diameters (all need to fit to a single plate) and we do it with the same angle, say 45°, I think it will be possible to order a batch of these plates at a reasonable price. Laser cut will be the cheapest option, but holes are not perfect, and the chamfer needs to be machined afterward, so maybe CNC milling from lets say 2 or 3mm mild steel plate will be optimal.

black-top, 2025-02-13:
And BTW, capturing all available data at the same time for data validation is great to have, because writing down all data by hand while switching between flow variables, adjusting the pressure and etc, increases the risk of the error. There are quite a lot of steps to do for each measurement, especially if switching between different mafs or comparing data with cylinder head measurements, not just with orifice plates.

black-top, 2025-02-13:
Ok, I understand what You are saying. In that case, offset calibration is correct, but we should adjust the curve to match flow with different size orifices or generate a new curve from scratch.

DeeEmm, 2025-02-13:
Yes. I think that if we use a method like the one I outlined, then we can match the curve to the bench. The question is whether this method will give accurate results.

it is obviously bette with a range of calibrated orifices, but I do not think that this is achievable for the average user.

It should also follow that the MAF sensor should give the same results for every bench without the curve needing to be adjusted, just the initial offset calibration. This is also something that we need to validate.

black-top, 2025-02-16:
I did some more testing today. My flow bench pipe is 100mm id and I use original MAFs housings, and I made adapters with gradually changing diameters to match MAF housings, but as You have mentioned that housing diameter might be the problem, I did a simple experiment.  I set flow bench at 28" pressure with a 100cfm orifice and just kept reducing the MAF housing diameter setting until flow was matching 100CFM.  After that, I tested CD at 10" at pressure. 
Results are with reduced maf size diameter to match 100cfm orifice, the CD flow also match expected flow in 1-2CFM range. 
So the MAF curve is matching with these settings.
 I did same test with two MAF mounted one after the other (067 and 5WK), with two mafs the correct 067 readings was at 70mm diameter setting (close to 72mm 5wk diameter) and did same test with only 067 maf in the system, at this time I get correct readings with 80mm housing diameter even if original housing is 82mm and the pipe is 100mm. 
In all tests, the calibration flow offset was set to 0.