DIY-Flow-Bench/DIY-Flow-Bench #37: BME280 Sensor fault causes blank display.

930MAF, 2020-08-04:
If  a BME 280 is selected in the config.h for Baro, Temp or humidity sensor it will not boot up the display, it will stay blank.  but if you fire up the serial monitor it will return The sensor did not respond. Please check wiring. 
Using SPARKFUN_BME280

**Arduino (please complete the following information):**
Firmware Version: Release V1 Beta Release #13 · Build 20080401
Hardware Type Mega 2560 DFROBOT display

**Additional context**
Add any other context about the problem here.

DeeEmm, 2020-08-05:
I think that this is a conflict with I2C addresses between the display and BME unit. (they both default to 0x77)

On the BME unit I have, there are some solder pads on the underside that need to be changed to change the address. There are three pads at the opposite end of the hole marked 'ADR'

https://cdn.sparkfun.com/r/600-600/assets/learn_tutorials/4/1/9/bottomside.png

The trace to the left pad (Pad 'I') needs to be cut
The trace to the left pad (Pad 'O') needs to be soldered

I'm changing the code at the moment but have some compilation errors I'm working through.

DeeEmm, 2020-08-06:
I'm still seeing the issue after resolving the compilation errors. I think it may be a conflict between the Sparkfun and TCmenu libraries. I'm going to reach out to the TCmenu dev and see if he has any ideas.

DeeEmm, 2020-08-06:
Okay, after a bit of digging I've found the reason for the issue. On line 138 there is a freeze statement

`while(1); //Freeze`

This was in the original BME280 example. It's not a very good way of error handling as it just locks up the system.

I've commented it out for now but need to work out why the BME is failing to initiate.

DeeEmm, 2020-08-06:
Fixed in Build 20080603

DeeEmm, 2020-08-07:
Going to mark this as closed.

closed 2020-08-07 by DeeEmm