DIY-Flow-Bench/DIY-Flow-Bench #60: Support for digital gauges

DeeEmm, 2021-07-29:
Add support for standard digital gauge protocol to be able to record valve lift as suggested by Edvinas Juodviršis.

- Support standard serial gauge protocol.
- Lift value recorded and stored with flow data 
- Lift value displayed on dashboard

DeeEmm, 2021-07-29:
Some information on serial gauge protocol - https://sites.google.com/site/marthalprojects/home/arduino/arduino-reads-digital-caliper

DeeEmm, 2021-07-29:
Additional Gauge protocol information

https://hackaday.io/project/511-digital-dial-indicator-cnc-surface-probe

https://forum.arduino.cc/t/garant-digital-dial-indicator-encoder-digimatic-rs232-without-data-cable/543582

DeeEmm, 2021-09-22:
Received ebay el-cheapo digital dial indicator that has serial data capabilities.

Not planning to take a look at this just yet as I need to finalise the shield design and testing but it's here and ready to go when I do.

[image]

[image]

[image]

[image]

baker27, 2021-09-23:
Do you need code for reading the dial indicator?

DeeEmm, 2021-09-23:
If you have some or want to write some, that'd be great.

I was going to adapt the examples above

KirikauKiwi, 2024-11-13:
@DeeEmm do you have the [serial cable](https://vi.aliexpress.com/item/1005006288776709.html) to plug in to your digital dial indicator?

It also looks like there are [USB versions](https://vi.aliexpress.com/item/1005002989551967.html) of the DDI too, the port on the DDI is USB and not a proprietary style like on the serial version.

I wonder if the data output is the same for the DDI's measuring 0.01mm and 0.001mm as the USB version has a 0.01 or 0.001 selector switch on the box in the middle of the USB cable.

DeeEmm, 2024-11-13:
I don't recall. I suspect not. I seem to recall that I found the pin outs and intended to solder direct. But that looks like a bargain and worth it for the cable alone.

The USB version looks like it's using a UART Serial converter.

IIRC the intent was to use Serial 2 to connect the gauge, so the first style is more appropriate. 

Gauge[Serial] -> ESP32[serial 2]

The USB version would need to be converted back to straight serial to connect to serial 2 so you would need to add a USB host controller into the pipe...

Gauge[Serial] -> UART Slave -> UART Master -> ESP32[serial 2]

DeeEmm, 2024-11-13:
> I wonder if the data output is the same for the DDI's measuring 0.01mm and 0.001mm as the USB version has a 0.01 or 0.001 selector switch on the box in the middle of the USB cable.

Not sure. I seem to recall that they follow a common standard, so suspect that they output would be the same. This is definitely true if they are using a UART as they are simply converting the data stream

DeeEmm, 2024-11-13:
Bought a couple for 'research' purposes :laughing:

stefan63, 2024-11-17:
Me thinking loud again.........
If we are using 1.8 deg (200 steps/rev) stepper motors driving a lead screw with 2mm pitch will gwt us (1 mm) / 800 = 
0,00125 mm  / revolution. Knowing this will make the messurement with a digital gauge unnecessary!
https://www.circuitist.com/how-to-calculate-steps-per-mm-lead-screw-and-gt2-timing-belt/

DeeEmm, 2024-11-17:
> Me thinking loud again......... If we are using 1.8 deg (200 steps/rev) stepper motors driving a lead screw with 2mm pitch will get us (1 mm) / 800 = 0,00125 mm movement/step. Knowing this will make the messurement with a digital gauge unnecessary! https://www.circuitist.com/how-to-calculate-steps-per-mm-lead-screw-and-gt2-timing-belt/

Indeed. This is the exact idea. Also with automated valve adjustment and automated flow control it will be possible to map the entire valve lift range automatically.

stefan63, 2024-11-17:
Maybe one of those could work, almost plug&play!
https://www.aliexpress.com/item/1005007160376705.html?

stefan63, 2024-11-17:
> Something like the second one would work, not too sure of the drive ratio
> of the first item.
>
> Best option would be lead screw with pitch of 1mm, but 2mm would probably
> work fine.
>
> For an overall 18mm of valve travel, measured at 1.5mm increments, the 2mm
> pitch would have a resolution of 135 steps per period compared to the 180
> steps of the 1mm pitch. The 1mm also has a torque advantage and I suspect
> could operate most valves with standard springs fitted.

There is also .9 degree steppers available ;)