DIY-Flow-Bench/DIY-Flow-Bench #61: OTA Update

DeeEmm, 2021-08-05:
Add support for OTA firmware update.

Using SPIFFS / ESP partitioning it is possible to perform an 'over the air' firmware update allowing users to update firmware via the web GUI.

This massively simplifies the process of installing firmware updates making this more accessible to the end users.

OTA does however require some amount of free partition space, which will impact storage capacity for datalog files.

One solution is to add support for and SD card and move all datalog storage there.

DeeEmm, 2021-08-05:
Added request for SD card support #62

stefan63, 2024-10-29:
OTA   Github => ESP32
https://medium.com/@adityabangde/esp32-firmware-updates-from-github-a-simple-ota-solution-173a95f4a97b

Best regards/Stefan

DeeEmm, 2024-10-29:
Thanks Stefan.

I am trailing the  ESPAsyncWebServer version as this is the Webserver library that we use. Although I did take a look at a few others.

One caveat that I did not know is that once the ESP has been updated Over-The-Air, it will no longer update using serial, which I think is a bit of a barrier at this point in time. This is related to the way that the partition schema works.  Essentially the OTA update is alternately stored on the OTA1 and OTA2 partitions. Whereas the USB update is stored to the Main partition which is only a fallback in case OTA1/2 are corrupted or cleared.

For OTA to work successfully the update process needs to solely use OTA. This is not an issue per-se, but does require commitment that once the switch is made it will define how all future updates are managed.

What I am currently working on is to extend the current file upload functionality so that the user can manually upload the firmware binary via the GUI. This means that we can package up the Binary and index.html files into each Github release so that the update path for anyone with a working installation is simply a case of uploading the firmware.bin and index.html files via the browser. No need to crack out a computer compile and connect it up to the flow bench.

Ultimately we could extend this to use the full  OTA functionality, in which case the update process would fetch the files from a remotely hosted server triggered by an update function in the GUI. This then negates the need for users to download anything and allow them to check to see if there is an update available from the UI.

But I think for the time being a simplified disconnected update using the DIYB GUI is a good compromise.

stefan63, 2024-10-29:
I agree and ESP32 download tool is already available an quite easy to use.
https://www.aeq-web.com/esp32-flash-tool-exported-program-upload-bin-hex-file/?lang=en

DeeEmm, 2024-10-29:
The binary created by VSCode compiler is available within the project tree at .pio/build/release/firmware.bin

I've merged commits from the OTA branch as the GUI firmware update is now working. #185 

For time being I will close this out as complete but may revisit it if we decide to implement OTA proper

closed 2024-10-29 by DeeEmm