DIY-Flow-Bench/DIY-Flow-Bench #101: Message: 'Flow limit reached'

DeeEmm, 2021-09-23:
Message: 'Flow limit reached'

Description: 'Test object flow exceeds Vac source capability'

Need to display to user when measured flow approaches maximum flow capability of vac source.

Also need to include config setting to set threshold
Also need to be able to calibrate this value i.e. record flow with no test piece

_Originally posted by @DeeEmm in https://github.com/DeeEmm/DIY-Flow-Bench/issues/77#issuecomment-925510153_

KirikauKiwi, 2024-12-12:
Something similar to this would be useful to show that the MAF sensor has gone over range, currently the MAF tile displays 0.0 when the over range occurs.

DeeEmm, 2024-12-12:
haha, I noted earlier that there is already a TODO comment in the MAF code to remind me to do this

It should ideally trigger a system message that is displayed on the GUI where it currently reports the uptime

The only issues is that the the system messaging is yet to be started.