DIY-Flow-Bench/DIY-Flow-Bench #157: LiftGraph.png file has blank data

DeeEmm, 2023-12-17:
Yes it saves liftdata.json and it show values correctly in text format when I open it with notepad, but image LiftGraph.png show 12kb size and 800x750 pix, but when the file is opened it is empty, there is no image, the opened image shows as a transparent rectangle. A simple solution for now is just a screenshot, but probably there is a bug in code, because same problem on phone and pc when image is saved. 
Another small detail that would be convenient is the possibility to change not only the lift interval but also the maximum lift and maximum flow values on the graph in the configuration. Now the graph looks out of proportion with the maximum 1000cfm when measuring for example up to 350cfm.

_Originally posted by @black-top in https://github.com/DeeEmm/DIY-Flow-Bench/discussions/155#discussioncomment-7875544_

DeeEmm, 2024-11-25:
Just testing this and appears to be working fine.

[image]

@black-top are you still seeing this issue?

Does anyone else see the same problem?

black-top, 2024-11-25:
Yes, I get a blank image. If I open an image with a different program, I still see the same view. Same on an Android phone.
  
[image]
[image]

DeeEmm, 2024-11-25:
Can I ask what browser you are using? Looks to be Chrome?

I'm using Chrome / Firefox / Opera here wth no issues but I run a MAC so do not have a way to test Internet Explorer. 

The conversion from the SVG image displayed in the UI to downloadable image happens entirely in the browser. The process takes the SVG image, converts it and presents it the browser as a download. This is done in javascript at runtime.

I suspect that the issue is therefore be browser related. The big question is how / why.

One easy thing to test is to change the file format. I've just committed a change to the DEV branch that changes the file from PNG to JPG. Give that a whirl and see if it is any different.

You will need to compile / upload as well as use the index.html in the data folder.

Let me know if that makes a difference

black-top, 2024-11-30:
Don't have any time to do testing right now, so tested  only different browsers with same firmware in esp32. It looks like the problem is with Firefox, because at the flow bench win7 pc with Firefox and on an Android phone with Firefox, the problem is the same.  With Opera browser, image saving is working.

stefan63, 2024-11-30:
Firefox is picky my experience to!
Chrome is restrictive but at least it give you an option ;)

DeeEmm, 2024-12-01:
Thanks @black-top. Will take a look at the requirements for the image saving in FF.

I use Chrome.

DeeEmm, 2024-12-10:
I have replicated this issue locally in FF.

Changing file download type from PNG to JPG did not resolve the issue.

Seems as if the canvas is not being saved properly. (SVG is painted onto HTML canvas and then saved as an image). There are some references to issues with FF and fixes discussed, but it looks like those fixes are already present in the code.

Currently we have no CSS data associated with the canvas as none is needed, but there is at least some commentary online that it is important to integrate the CSS. Perhaps this needs to be present for this work in FF.

Will try and test again.

DeeEmm, 2025-01-09:
Just re-tested this to work on a fix and this issue now appears to be resolved.

This may possibly be Firefox related - perhaps a recent update. My FF version is V 133.0.3

closed 2025-01-09 by DeeEmm