DIY-Flow-Bench/DIY-Flow-Bench #162: Compilation error: 'tiny' does not name a type

stefan63, 2024-08-05:
**Software Version**
- RELEASE (e.g. V.2.0-RC.6) 
- BUILD_NUMBER (e.g. Build 23041301)

**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
Trying to upload but Arduino IDE keep complaining with.
"Compilation error: 'tiny' does not name a type   🙁
Seams to be an issue with sensors.cpp how di I solve this?"
......................#ifdef BME_IS_ENABLED
#define TINY_BME280_I2C
#include "TinyBME280.h"
tiny::BME280 _BMESensor;
TwoWire I2CBME = TwoWire(0);
#endif............

"Arduino IDE Version: 2.3.2
Date: 2024-02-20T10:04:35.814Z
CLI Version: 0.35.3"

**Steps to reproduce the bug**
I simply try to upload the sketch and no complains about missing librarys.

**ESP32 (please complete the following information):**
 - Hardware Wemos D1 R32]
 - ESP Firmware ?

**Additional context**
Would be nice to know wich librarys and library versions that was used during a successful upload :D

DeeEmm, 2024-08-06:
Hi Stefan.

I suspect that it might be that the TinyBME280 library is not registered in the Arduino IDE so it cannot find the TinyBME280.h file.

My recommendation is to consider swapping to use the VSCode IDE as there are environment files provided in the project that automagically set up library dependencies. This has to be done manually on the Arduino IDE.

/DM

DeeEmm, 2024-08-06:
For info on installing with the VSCode IDE please refer to the wiki documentation

https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Installation

Let me know how you get on.

/DM

stefan63, 2024-08-18:
Sorted out for now by commenting out the currently non existing BME280 in my test setup.

closed 2024-08-18 by stefan63