DIY-Flow-Bench/DIY-Flow-Bench #163: Toggle switch to allow for dual use of Diff press sensor Pdiff / Pitot

stefan63, 2024-08-21:
Just a thought, if we could implement a switch "software or hardware controlled" to select betwen Pdiff and Pitot we might be able to save some  $  and a sensor?

stefan63, 2024-11-03:
Thinking loud is not always the best haha...
Maybe just close this suggestion since I realize that this need to involve a solenoid valve to be able to switch negative pressure port location from tube to ambient pressure and therefore no savings to be made!
Kind regards/Stefan

DeeEmm, 2024-11-04:
On some aftermarket systems they use a pneumatic manifold to switch tubing configuration. These are commonly used in pneumatic logic and are available at industrial suppliers. This would take care of the physical connection.

The manifolds also can have electrical switches to determine the position. 

Switching the sensor source based on an input or even a soft key on the GUI is definitely possible.