DIY-Flow-Bench/DIY-Flow-Bench #164: [BUG] Baro pressure incorrectly displayed in UI

DeeEmm, 2024-09-03:
**Software Version**
- RELEASE V2.0-RC.6
- BUILD_NUMBER 23041301

**Describe the bug**
Baro pressure value is / 10

maher-nakesh, 2024-09-03:
try this
https://t.ly/zy17A
select the desired compiler in the installer

DeeEmm, 2024-09-03:
Code review in progress.

DeeEmm, 2024-09-04:
OK. so I've fixed this issue, it was sending the incorrect value to the UI

I will not release the fix just yet as there is a small amount of refactoring happening that I am still in the middle of.

Here's the fix for anyone who wants to manually implement it...

in webserver.cpp on line 1042 change...

`  dataJson["BARO"] = sensorVal.BaroKPA;`

to

`  dataJson["BARO"] = sensorVal.BaroHPA;`

This fix will appear from build 24090301 onwards.

closed 2024-09-04 by DeeEmm