DIY-Flow-Bench/DIY-Flow-Bench #180: [FEATURE REQUEST] Add MODE to CFM value smoothing

DeeEmm, 2024-10-24:
Include 'MODE' for averaging function for flow value smoothing

Suggested by @black-top

DeeEmm, 2024-10-28:
I've added a data filter drop down to the configuration tab in latest commit

So far all that is functional is Rolling Median smoothing, but it is written to include Cyclical Average and Mode, but the maths for those needs to be written.

DeeEmm, 2024-10-29:
Discussion relating to Mode filtering in flow calibration thread 

https://github.com/DeeEmm/DIY-Flow-Bench/issues/179#issuecomment-2443394451

DeeEmm, 2024-10-29:
I'm going to assign this target to the V3 release, however it will likely be possible to squeeze additional filters in between now and then as the main core filter management code is now in place.

Potential to include filters to truncate and then use standard deviation methods...

Truncated Median
Truncated Average
Truncated Mode

We can also consider removing larger deviants from the data set...

change > x cfm

In theory if we add a number of (different) filters then hopefully each end user will find one that is suitable for their setup.

DeeEmm, 2024-11-04:
From discussion in issue https://github.com/DeeEmm/DIY-Flow-Bench/issues/178#issuecomment-2453430005

Additional flow value manipulation using truncation and rounding...

Changes to javascript.js to truncate displayed values...

```
var decimalPlacesFlow = 0;
var decimalPlaces = 1;
```
```
if (key === 'FLOW' || key === 'AFLOW' || key === 'MFLOW' || key === 'FDIFF'){
  document.getElementById(key).innerHTML = myObj[key].toFixed(decimalPlacesFlow); 
} else {
  document.getElementById(key).innerHTML = myObj[key].toFixed(decimalPlaces); 
}
```

> Okay. so after some experimenting it looks like I can use templating variables in the javascript file which means we can pull these values out into the configuration.
> 
> I have also been playing around with the round function to allow us to round to nearest 0.5cfm
> 
> ```
> round(x * 2.0) / 2.0;
> ```
> 
> I will package up this up within the configuration settings tab as a new group of settings under the filters
> 
> Round to nearest integer 
> Round to nearest 0.5  
> 0/1/2 Decimal places for flow variables
> 0/1/2 Decimal places for everything else

DeeEmm, 2024-11-04:
@black-top from https://github.com/DeeEmm/DIY-Flow-Bench/issues/178#issuecomment-2453379664

> I repeated the same test in a few days, and the results looks good. CD flows in 1-2CFM with same calibrations. For cylinder head flow, there are the same problems that we were discussing about filtering. The value fluctuates in the 1–5 CFM range and sometimes in the 5–10 CFM range (with cycle average filter). Maybe turbulence from the cylinder head influences that, but my MAF is far from the cylinder head and 4 or 5 90deg bends, so it should be quite laminar. I was trying to write down minimum values, because it looks like it repeats most accurately. With these results, I think it will be a good option to have at settings to select simple number rounding to a whole number, because the system will probably never be that stable to see less than 1CFM with MAF setup. Also, the"Average Minimum" filter may be worth trying.

After pondering this last night. I think that this needs further investigation. We should easily be able to get down to sub CFM measurements. We see fractional values in the MAF table and also interpolate the MAF data. If we use simulated input, the readings are very stable, which leads me to think that the fluctuations are external, and potentially hardware related. I think the pulsing that you see when you close off the inlet at 28" is indicative of this. It would be really good to try and isolate the cause.

It would be good to try and pin down the exact cause of the fluctuations. Whilst software filters are an 'easy' solution, they are not taking causality into account. We are only addressing the symptom of the problem and not the problem itself.

Do you have an oscilloscope?. The first step is to determine whether the fluctuations are external to the board. You can monitor the input voltage of the MAF wth a scope to determine this.

Assuming that the fluctuations are external, we then need to determine if they are from the MAF, or as a result of system design, which ideally requires measuring the air flow with a secondary device or monitoring system pressure.

If it is a physical phenomenon, I would expect it to be easily measurable.

If it is not a physical phenomenon, then the solution would be electronic filtering in the form of RC filter(s). There is already a rudimentary RC filter in the design. But the values can easily be tweaked. I think that this would be a more appropriate solution and more robust.

DeeEmm, 2024-11-04:
@Black-top I've added the changes to truncate / round the flow values in the GUI to the configuration tab. Let me know how this works for you. You will need to upload both the project and the index.html file

DeeEmm, 2024-11-04:
Screenshots...

[image]

[image]

black-top, 2024-11-04:
I have been trying to solve those issues since the beginning. The latest iteration has properly shielded cable to MAF, and since rs4 MAF needs 12V and 5V for reference, I made a small pcb based on a speeduino power supply, so a 5V power supply with all filtering and stabilization is connected a few centimeters away from the MAF connector. The 5V is as stable as possible, don't see any variation. I was probing signals with scope and don't see any noise, unless easp32 is connected to the pc, but I connect it only for firmware update, so it is not connected all the time. The MAF signal at the connector, compared to the signal at the PCB is undisturbed. All grounds are connected to the same place, so there are no ground loops. I see some signal pulsating directly from MAF, there is also some small variation in depression. My guess is that air flow is pulsing, and that might be the cause.  I think I need to add a manual valve near the pump, for "controlled leak", so the pump will run stable. 

I lost track a little bit. I don't need to do code modifications, it is already in the master?

black-top, 2024-11-04:
Well, close loop control will be great, but it requires a lot of work, and probably quite complex strategies. Probably one PWM output for vfd and a second pwm output for DBW. Or maybe at first just DBW pwm output, but pump control leave manual, because two control systems might fight each other. Or maybe a simple strategy for vfd just ramp up pump speed open loop and PID only for DBW control.

DeeEmm, 2024-11-04:
A very basic implementation might not be too bad. Just controlling the bleed valve, with manual vac motor control.  No bells and whistles, just a very basic control.

I'm thinking that if it adds value to the initial release and helps us to produce a more stable / accurate measurements, then it will actually help with validation, which I see as the most important issue at present.

black-top, 2024-11-04:
Yes, even basic control setup will be a big step forward and probably will solve some issues with stability.

DeeEmm, 2024-11-04:
I will take a look after I've finished with what I'm currently working on.

black-top, 2024-11-05:
I tested the rounding functions, and it looks like everything is working as intended, didn't find any issues.

KirikauKiwi, 2024-11-05:
@black-top great to read that probing with a scope the DIYFB signals appear noise free and that the MAF and pressure sensors are showing signs that are indicating actual air flow pulsation, awesome stuff that the sensors are picking this up.
Do you have a water manometer that can measure 28" it would be a good test to see if the water level is also pulsing up and down too, double verification.
Do the pulsations exist all the time or do they increase when the level of vacuum increases.
Then you can be certain about where they are coming from and then look at physical methods of trying to control the pulsations.

I would be curious to know if the pulsations are 50 or 60Hz or multiples of. If so, this could be due to the motor being AC driven, the pulsations could also be coming from blades on the impeller inside the housing. You may have to look at coming up with some sort of pulsation damper for your system.
Are you are using a regenerative style of blower/airpump?

black-top, 2024-11-06:
@KirikauKiwi
 I have somewhere my old water manometer from the old flow bench setup, but the pump exceeds 100"h2o at low lifts, so I need to be careful with vfd to not suck water into the system. I also have manometers for carb/itb synchronization, so I think I will connect one of these for reference. Pump is driven by VFD, so I think it is probably the resonance of the system, but not just 50/60hz from main.  I don't have much spare time, so I can't play too much with it, but I will do more electronics testing, maybe FFT on signals, to see frequencies if close loop bleed valve control still don't give a better stability. I think it needs some air just to keep running stable.  Yes, both pumps s are regenerative style of blower/air pump.

KirikauKiwi, 2024-11-07:
Great idea with the FFT on the signals, it could help identify if it is indeed air pulsations in the pipes. 
One idea I had today whilst out driving if it is air pulsations is to try either an accumulator plumbed in to your system, or the likes of a quarter wave or hermholtz side branch resonator to act as a fluid damper.

DeeEmm, 2024-11-11:
Based on the philosophy of 'if it's currently in the GUI it has to work' I've moved this into the RC9 stuff.

DeeEmm, 2025-01-06:
Stumbled across the following whilst looking for data on issue #307 

Savitzky–Golay filter - https://en.wikipedia.org/wiki/Savitzky-Golay_filter 

Seems like it might be suitable for flow smoothing. 

It requires equally spaced data and can work on a live data stream.

DeeEmm, 2025-01-13:
I've added some basic filters for Average, median and mode.

Each filter can be tweaked / tuned using the following macro variables in system.h

```
// MAF Data Filters
#define ALPHA_MEDIAN 0.01f
#define ALPHA_AVERAGE 0.65f
#define ALPHA_MEAN 0.1f
```

Will close this  out for the time being, we can revisit this later if these do not meet our needs or need to be modded

closed 2025-01-13 by DeeEmm