DIY-Flow-Bench/DIY-Flow-Bench #182: Temp conversion "RANKINE"

KirikauKiwi, 2024-10-28:
Whilst looking through the code for #179 I noticed this in the calculations.cpp Line 180 The RANKINE offset should be +491.67 looks like a copy and paste with the KELVIN offset.
Not sure if it is actually used in the code anywhere, but thought best to point it out in case it is.

```
// Then convert DegC to required value and return
  switch (unitsOut) 	{

    case DEGF:
      convertedTemp = (inputTempDegC * 1.8) + 32;
      return convertedTemp;
      break;

    case RANKINE:
      convertedTemp = (inputTempDegC + 273.15 ) * 1.8;
      return convertedTemp;
      break;

    case KELVIN:
      convertedTemp = inputTempDegC + 273.15;
      return convertedTemp;
      break;
```

DeeEmm, 2024-10-28:
This is correct.

°R equals K multiplied by, 9 divided 5

9/5 = 1.8

Hence conversion to Kelvin and then from Kelvin to Rankine

Could also have used 

°R = °C * 1.8 + 491.67

Much of a muchness really

DeeEmm, 2024-10-28:
BTW.

In an older iteration of the software Rankine was used in the conversion from Mass flow to volumetric flow. But those calcs now use SI units so currently not used anywhere.

DeeEmm, 2024-10-28:
Sorry was a bit hasty in closing this as it's blocked responses,

KirikauKiwi, 2024-10-28:
Yes I see, may bad in reading the code in relation to Kelvin, cheers for clarifying.

DeeEmm, 2024-10-28:
All good. It's nice to have someone to check my work. 

Your feedback / advice / input is welcomed and valued

closed 2024-10-29 by DeeEmm