DIY-Flow-Bench/DIY-Flow-Bench #187: [BUG]

stefan63, 2024-10-31:
**Software Version**
- RELEASE (e.g. V.2.0-RC.8) 
- BUILD_NUMBER 24103101

**I want to use a XGZP6899A010KPDPN 

And made this config.**

* CONFIGURE REFERENCE PRESSURE SENSOR
* If you want to modify the code to include additional reference pressure sensors
* You will need to add your volts to kPa algorithm in the function sensors->getPRef()
*
* Recommended sensor is the MPXV7007DP or XGZP6899A007KPDPN 
***/

#define PREF_IS_ENABLED                                     // Comment to disable reference pressure related code

#define FIXED_REF_PRESS_VALUE 1                             // Fixed pressure value in Pascals

// Set signal source (Uncomment One line only)
// #define PREF_SRC_PIN
#define PREF_SRC_ADC

// Set sensor type (Uncomment One line only)
// #define PREF_SENSOR_TYPE_LINEAR_ANALOG 
//#define PREF_SENSOR_TYPE_MPXV7007        
// #define PREF_SENSOR_TYPE_MPXV7025
// #define PREF_SENSOR_TYPE_XGZP6899A007KPDPN        
#define PREF_SENSOR_TYPE_XGZP6899A010KPDPN        

#define PREF_MV_TRIMPOT 0.0                               // Millivolt offset
#define PREF_ANALOG_SCALE 1.0                             // Scaling factor used for raw analog value
#define PREF_ADC_CHANNEL 1      

**Then I get this :( when I compile (build) in PlatformIO is there a bug in censors.ccp?** 

ESP32/DIY-Flow-Bench/sensors.cpp: In member function 'double Sensors::getPRefValue()':
ESP32/DIY-Flow-Bench/sensors.cpp:651:37: error: expected ';' before 'if'
   sensorVal = sensorVolts * 5 - 12.5
                                     ^
                                     ;
ESP32/DIY-Flow-Bench/sensors.cpp:660:2:
  if (sensorVal > 0) {
  ~~
ESP32/DIY-Flow-Bench/sensors.cpp:662:4: error: 'else' without a previous 'if'
  } else {

stefan63, 2024-10-31:
Compiles fine if I change to #define PREF_SENSOR_TYPE_MPXV7007 or MPXV7025
 But nider of the XGZP6899A007KPDPN    or   XGZP6899A010KPDPN   works.

DeeEmm, 2024-10-31:
Sorry. 

Yes the line is missing the terminator character.

in sensors.cpp, around line 651 change...

```
sensorVal = sensorVolts * 5 - 12.5
```

to...

```
sensorVal = sensorVolts * 5 - 12.5;
```

The code will then compile.

I believe the maths is correct as I would have pulled this from the datasheet when I added it.

DeeEmm, 2024-11-01:
Fix for this pushed in build 24100101 so I'm going to close this as is now resolved.

closed 2024-11-01 by DeeEmm

stefan63, 2024-11-01:
I just checked 24100101 and I guess you overlooked the 007 sensor (sorry for being pain in the a...).

	#elif defined PREF_SENSOR_TYPE_XGZP6899A007KPDPN

		// Linear response. Range = 0.5 ~ 4.5 = -7 ~ 7kPa
		sensorVal = sensorVolts * 3.5 - 8.75

	#elif defined PREF_SENSOR_TYPE_XGZP6899A010KPDPN

		// Linear response. Range = 0.5 ~ 4.5 = -10 ~ 10kPa
		sensorVal = sensorVolts * 5 - 12.5;
		
		Also shouldnt it be? I guess both might be working but I did:  sensorVal = (sensorVolts * 5 - 12.5);

DeeEmm, 2024-11-02:
Haha. that's typical lol.

Thanks. I'll fix it up.