DIY-Flow-Bench/DIY-Flow-Bench #205: Post-Compile Configuration

DeeEmm, 2024-11-08:
It's a fair bit of work to get to a dropdown for each setting. First we need to get all of the settings out into dedicated files and then we can look to create GUI elements to manage each setting. Ultimately down the track something like this would be the goal but for the time being just the ability to change settings post compile time will allow the OTA firmware to function correctly.

It's quite interesting changing the settings from pre-compile to post-compile. It presents a number of technical challenges. 

We need to change the following files to be loaded after the ECU has booted

- pins.h  - pins definitions + I/O assignments
- configuration.h - configuration settings > begin() & initialise() vars setup
- maf.cpp - MAF data creation
- language.h language file 

Changing to a JSON format and reading the variables in from a file is not too hard as we already use the arduinoJSON library which makes relatively easy work of this, most of the settings are easy enough to translate apart from the large data array that is used for the MAF file, which I have yet to do. There is some creative coding needed to ensure that we can typecast correctly from char* arrays to whatever variable type we require to get the settings from JSON. This typecasting was previously taken care of automagically by the #define precompile directive and was transparent. So there is an amount of work to do there. Also char* / string management in C++ is woeful. Give me PHP any day lol.

There are also some additional challenges depending where we need to replace compile time directives within the code. Previously we #include the files at the top of each class or file as required. This meant that the values from those files were available right away to use. However when we have some #ifdef statements in the header section of the class definitions, we cannot use regular conditional logic to replace them, as we cannot use if-then-else logic in the header section of the file. So some creative solutions are required.

We also need to bypass pretty much all of the code until the I/O and configuration files are present. So we need to be able to block out large portions of the code so that it skips directly to the webserver to allow the relevant files to be uploaded. The big challenge there is dealing with unassigned I/O and variables (because we've essentially removed most vars from the code) and so we have to bypass the code that relies on them whilst still allowing it to function afterwards. The existing code flow is not ideal for this as we did all of the I/O and configuration stuff first, whereas we are now trying to do it last.

Today has been working through many errors line by line to get the code to compile without having the configuration and I/O assigned.

Very challenging lol.

_Originally posted by @DeeEmm in https://github.com/DeeEmm/DIY-Flow-Bench/discussions/189#discussioncomment-11174761_

DeeEmm, 2024-11-08:
Parking this for the time being.

Changes are stashed on WIP branch.

Look to implement non-breaking changes from WIP into main code over time so that eventual move to post-compile configuration handling has less impact.

Also need to look at better controlling initialisation sequence. Sequence needs more granularity so that breaking aspects like I/O and Variable assignment can be more easily worked around.

Need to bring ALL initialisation into main startup.

DeeEmm, 2024-11-15:
Stumbled across an interesting stand alone binary uploader...

https://github.com/marcelstoer/nodemcu-pyflasher

[image]

This could be very useful coupled with the ability to upload binaries via the GUI. It would mean that no one needs to use VSCode to get up and running. Just download this program, flash the binary in the first instance and then all updates and setup is done via the GUI.

I did download it and test it. It seemed to happily uploaded the binary, but there is unfortunately an error on boot. It seems that this error means that there is no program present.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x63206f74
invalid header: 0x63206f74
invalid header: 0x63206f74
invalid header: 0x63206f74
invalid header: 0x63206f74
invalid header: 0x63206f74
invalid header: 0x63206f74
```

Hopefully the source could be updated to work. (I suspect that this might actually be a partition issue.) anyone game enough to try?

stefan63, 2024-11-15:
I have used this ESP32 Flash tool for some of my previous projects and it works fine:
I sugest you try it before doing any unnessesary changes :D
https://www.aeq-web.com/esp32-flash-tool-exported-program-upload-bin-hex-file/?lang=en

DeeEmm, 2024-11-15:
Thanks Stefan.

Will give them a try.

DeeEmm, 2024-11-27:
Been having a play with the firmware uploader, and have a working version.

[image]

I've stripped out the chaff to leave just what is necessary.

ATM I'm looking at how we can spit out a combined binary as this is required for the firmware loader. This includes bootloader.bin + partitions.bin along with the firmware.bin in a single file.

Currently platformio does not support this out-of-the-box, but others have integrated the ability to do this.

With a single binary, any firmware uploader can be used, there are even browser based loaders such as this Adafruit version

https://adafruit.github.io/Adafruit_WebSerial_ESPTool/

and an expressif version

https://espressif.github.io/esptool-js/.

It should actually be noted that these browser based uploaders do support uploading separate bootloader / partition / firmware binaries, it is just a case of putting the correct memory addresses in. For example using the expressif tool works...

[image]

(Using this method it is also possible upload the spiffs.bin file as well.)

The underlying code that supports the browser based upload is available for anyone to use - esptool-js, which means that we can easily deploy it on the main website. [ESP-tool Documentation](https://espressif.github.io/esptool-js/docs/index.html). We could also modify it to link the source code directly to the current release to pull in the appropriate files.

We can also do the same with the Firmware loader app. Then installing code is simply a case of visiting the URL or using the app. There would be no need to manually handle files or type in register addresses, we can include all of that boilerplate stuff within the app / browser loader so that it is transparent to the user

Of course this is only needed for the initial upload. Once the device has firmware installed, we then move to the GUI style update. In this case it should be possible to do a similar thing where we can either link to, or fetch the remote firmware file.

If this can be implemented, then the steep learning curve of having to use VSCode or the Arduino IDE to flash the board will be eliminated, which I think will be a gamechanger.

There's still a bunch of work to do on this, but we now have a basic proof of concept.

DeeEmm, 2024-11-27:
So managed to sort out the creation of a merged binary. 

Also expanded on this to populate a 'release' folder with the merged binary, firmware binary and GUI file...

[image]

The merged binary is created by the merge-firmware.py file which is called as an extra script in platformio at compile time.

So now for each commit and pull request binaries are automatically created. This means that it should be very straightforwards for anyone wanting to upgrade. Unless you are not running the default MCU. That will however be addressed when I sort out the rest of the post-compile configuration stuff.

DeeEmm, 2024-11-28:
Updated the wiki to include instructions for uploading firmware using a firmware uploader. Currently the web based version is working and the app just needs a bit of tweaking.

https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Installation

I'm going to pull this back into the V2 release. I think that it will just make a more rounded product and hopefully open it up to people who are not so tech savvy.

I know it is a bit more work, but not too much.

DeeEmm, 2024-11-28:
Ahh yes. the Firmware uploader code is now uploaded to the repo - https://github.com/DeeEmm/DIYFB-firmware-flasher

At the moment the Mac App works from the VSCode environment but stalls when run standalone.  

I have no way of testing on Windows. If anyone is keen to have a go. 

I know that the app is not a requirement for the V2 release, but would be good to get both executables working

KirikauKiwi, 2024-11-28:
> Updated the wiki to include instructions for uploading firmware using a firmware uploader. Currently the web based version is working and the app just needs a bit of tweaking. [...]

This is awesome!!!!!
Would be good to add to the Web flash section about which browser to use, e.g. Google Chrome and Microsoft Edge [version 89 or later](https://developer.mozilla.org/en-US/docs/Web/API/Serial#browser_compatibility) browsers

DeeEmm, 2024-11-29:
Will add that in. There' still a lot of tidying to do in the WIKI.

Current status of this is that the mods to get the config stuff moved over to configuration.json are done but not 100% tested. This is due to the MAF stuff still being in process. Currently I've created a MAF.json file and have read the MAF data into a JSON object. 

```

    "sensor_type" : "ACDELCO_92281162",
    "output_type" : "voltage",
    "maf_units" : "KG_H",
    "maf_scaling" : 0.1,
    "maf_diameter" : 94,
    "maf_lookup_table" : {
        "8" : 28,
        "15" : 34,
        "23" : 43,
        "31" : 52,
        "38" : 64,
        "46" : 78,
...
        "477" : 15091,
        "485" : 15954,
        "492" : 16852,
        "500" : 17787 
    }
}
```

The JSON object does allow some iteration, but this is very basic compared to using an array or vector as we did previously. I've been scouring the ArduinoJSON documentation to see how the JSON object can be accessed in a similar manner to the existing code, but it very much looks like what I need to do is create a vector from the JSON object (actually vector of vectors as it is a 2D array) and then just use the existing code as-is.

So next step is to try to iterate over the ArduinoJSON object and create the necessary vector. (as I do not see an available method to create it directly).

So, we are very close, but still a little way to go.

Will try and get this done over the weekend. I have some other commitments but keen to get this refactor completed before I get too out of step with the DEV branch.

DeeEmm, 2024-12-01:
Had a bit of a nightmare with this, Somehow managed to lose all changes by closing / reopening VSCode, so spent most of yesterday recovering the lost work. Fortunately VS history had a copy of most changes.

Was definitely something screwy going on with my VS as I previously has an issue with a locked PR. Still unsure what that issues is so need to be more vigilant with committing progress. (hadn't committed for four days - lesson learned).

Still got one small compile error that I need to fix.

```
/Users/mick/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/datahandler.cpp.o:(.literal._ZN11DataHandler13parsePinsDataEN11ArduinoJson8V6215PB218StaticJsonDocumentILj1024EEE+0x10): undefined reference to `pins'
```

I'm out for the rest of the day so that will have to wait until tonight.

DeeEmm, 2024-12-02:
Compile error fixed, just working through validation and tidying.

Have updated the WIKI for the configuration settings... https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Configuration

Will work through MAF settings next

DeeEmm, 2024-12-10:
OK. So I think this is pretty much done.

The change had a massive impact as pretty much every aspect of the code relied on data that was present in the precompile macros. This means that the code changes are pretty substantial.

Making the change broke a lot of functionality, so there has been an amount of time spent on fixing things and in some cases finding different ways of doing things. Other changes were made in the name of code improvement, wherever possible I have replaced if-else logic with switch / select, which is both faster and easier to read.

In reviewing the code I've also found other errors which I have addressed as they have arisen. These are recorded in the changelog.

Some caveats.

- The 5v pin is unable to be used to measure board voltage. It appears that this pin is related to WiFi, so whilst Wifi is in use the pin cannot be used to measure analog voltages. This is actually a bit weird as prior to the change from pins.h to pins.json the compiler never complained. I suspect that this could be as a result of the issue not being presented to the compiler error checking due to being masked by the precompile macro. 
- Currently the 5v is set to fixed value. I will of course address this in future PCB versions. It should also be possible to solder a jumper to an unused pin. I'm not sure how much of an error running a fixed 5v value will introduce. I've not seen any variation to the 5v bus in testing here, but other environmental influences may need to be considered. Of course YMMV.

- Memory requirements have changed a lot and the ARDUINO_LOOP_STACK_SIZE has been increased substantially to accommodate the memory consumed by copying the JSON file data into memory. Along with this the allocation for JSON object sizes has also been increased. The consideration here is that the JSON files are to a degree liable to change size based on various factors. We can allow for this where we can predict size changes, such as for the configuration.json file but other files such as the MAF.json and language.json files are able to be modified by the end user, which could result in stack overflows / stack canary watchdog errors if the changes exceed the memory allocation. Not an immediate issue, but one to bear in mind if you see such errors.

- I have created pins.json and maf.json files, however these have not been validated. I know that there will be errors in the pins files. I will work though those next. The format of the files has changed and so will not look familiar. Unfortunately it is not possible the comment JSON files as this results in compilation errors, so the files need to be modified in conjunction with the documentation, which I also need to check and update.
- The WEMOS pins file is fully working.

- File names are dynamically detected. This is to allow the auto versioning to increment file names. These can be uploaded as-is, there is no requirement to rename the files. The code searches for 'index' / 'pins' / 'maf' strings in the filenames. 

- From this version onwards it is now possible to upgrade firmware by using the 'upgrade' button in the GUI. This allows you to upload the 'update_binary' from the releases folder. This uses your existing pins / maf / config data.
- Where updates to pins / maf / config files are required as part of an update we will need to look at modifying them programatically.
- Next stages are to look at version / update detection and possibly even upgrading from the remotely hosted binaries. Lots of possibilities, but will leave those for V3. The foundation is now in place, we can build on it later.

Working branch is here - https://github.com/DeeEmm/DIY-Flow-Bench/tree/205-post-compile-configuration

I still have some tidying up to do and also want to make sure that this is tested before merging into DEV.

DeeEmm, 2024-12-10:
Ahh yes amongst the changes I changed the way that the capture works. Previously after every submission the GUI refreshed, which was a bit of a PITA. So I changed the operation to submit the form data by XHR. The only caveat here is that in getting rid of the refresh the graph does not update (it is created at page load). So to view your changes you need to refresh your browser.

I will change this behaviour, but it is a fair bit of code to move the graph data from template variables to ajax / event data, so will not do this until a later version.

At some point we will rewrite the data graph and add in the additional info to the GUI. We can look at changing the behaviour at that stage.

KirikauKiwi, 2024-12-10:
Great work under the bonnet, to the end user it won't look very different!!!!
Three things I have noticed are.
- Pitot GUI display not updating, via the API the sensor is working as expected.
- MAF Flow GUI display, with the Bosch MAF when the voltage is below 1V the display shows a very long number for some of the look up table values and displays correctly for others, it appears to alternate when doing a voltage sweep. Working as expects at voltages above 1V.
- There appears to but a latency with the GUI display updating, most noticeable with the Environmental data.

MAF Data error
[image]

Version and hardware information
[image]

Latency of GUI display updating
https://github.com/user-attachments/assets/757381ad-4f65-415a-aebf-db797994c482

KirikauKiwi, 2024-12-10:
Further information for the MAF GUI,

It appears that with the 20mV steps in the MAF lookup table that when the voltage is below 1V the large number is being display when the MAF voltage is between the 20mV steps.
```
        "86" : 94, 
        "88" : 83, 
```
[image]

[image]

[image]

DeeEmm, 2024-12-10:
I had also previously noted the out of range value for the flow. I had added fabs() to each value generated by ADC and also changed the code for the validation to prevent values going negative, which I assumed were the cause. In local tests this appeared to eliminate the out of range value, but that is obviously not the case.

I also noticed some latency in the GUI, which seemed to settle after a short time. I am unsure of the reason, possibly as a result of additional processing requirements ?? 

Will take another look at the OOR value and optimising the code

DeeEmm, 2024-12-10:
BTW your GUI version is very different from the current version - 2412100100. I don't think that this could be the case for the latency or OOR value, but nothing should be overlooked.

KirikauKiwi, 2024-12-10:
Thanks for the heads up, I initial flashed this 205 Build over top of the the DEV build on my ESP32.
I have cleared the flash on the ESP32 and then ran the did an upload and monitor which showed on the serial monitor as up and running, I couldn't access DIYFB local.
Went back and run the GULP commands and the upload and monitor which has worked. However the GUI version numbers is out of whack.

[image]

KirikauKiwi, 2024-12-10:
The pitot ∆ display is working as expected other than when it is zero'd.
After zeroing the Pitot Display the ∆ has a a small area of voltage between 2.47 - 2.5V where the ∆ is displayed as a negative above to 2.5V crossing point the ∆ is displayed correctly and it is also displayed correctly as a positive value from 0 - 2.47 V.

DeeEmm, 2024-12-10:
The @@version@@ is the template placeholder that is replaced during the gulp build.

This is present in the source file (/src/index.html)  but gets replaced during the gulp tasks when the /build/index.html is created and is carried into the /releases/V2.0-RC8_2412100100_index.html file, which is essentially just the /build/index.html file renamed and copied into the releases folder

You can see this in the gulp.js file...

`
	var result = data.replace("@@version@@", new_build_number);`

The overall process is that the GUI source files reside in the src folder

```
index.html
style.css
javascript.js
```

These are the three main files for the GUI that get combined and minfied by GULP

Additionally there is the preload.html file, which is the basis for the HTML code presented in the boot loop. This actually resides in the language.json file as a minified version but the unmodified souce version is in the /src folder.

When GULP is run (combine) the index.html, style.css and javascript.js files are combined into a single file that is stored in the /build folder.

Previously this was the index.html file that we used in the GUI. But now we copy this to the releases folder and rename it with the current version info. A small change tot eh build process but one that allows us to easily keep track of versions. In addition to the @@version@@ variable displayed in the info modal, we also have the filename as a point of reference.

It's a bit of a change for those of us used to grabbing the index.html file from the data folder, but moving forwards the documentation will direct everyone to the release folder.

DeeEmm, 2024-12-10:
> The pitot ∆ display is working as expected other than when it is zero'd. After zeroing the Pitot Display, the ∆ has a a small area of voltage between 2.47 - 2.5V where the ∆ is displayed as a negative. Above the 2.5V crossing point the ∆ is displayed correctly, it is also displayed correctly as a positive value from 0 - 2.47 V.

I want to look at all of the pressure sensors in the context of zeroing / calibration values. We added zeroing to the pitot and differential, but we really need to make trimpot and calibration functions much easier to use.

I feel that there is a bit of work to so there

DeeEmm, 2024-12-10:
As this is working with a few residual issues not directly related to the post-compile-config I've merged PR into main and will close out this issue

We can flag those incidental issues up individually under new issue reports

The zeroing / calibration discussed above is probably best taken care of as part of #236

closed 2024-12-10 by DeeEmm