DIY-Flow-Bench/DIY-Flow-Bench #207: [BUG]Decimal places on the GUI

KirikauKiwi, 2024-11-08:
24110704"
#define RELEASE "V.2.0-RC.8"
**Software Version**
- RELEASE V.2.0-RC.8
- BUILD_NUMBER Build 24110704

**Describe the bug**
When changing the Flow decimal accuracy from "tenths" to "hundredths" on the web GUI configuration page after clicking save the change is made, however all the other displayed units are rounded to their nearest whole number.

It also appears that the General Decimal accuracy drop down has disappeared for the GUI

**Expected behavior**
When changing the Flow accuracy that the other displayed numbers are not changed

**Missing General Decimal Accuracy**
[image]

**Flow Accuracy set to Hundredths**
[image]

DeeEmm, 2024-11-08:
Hmnnn something strange happening here. I note that the drop down is missing in the GUI. 

Likely related. Suspect an issue in the javascript. Will take a look.

Most likely a search and destroy victim

[image]

DeeEmm, 2024-11-08:
Was two issues.

- Missing JSON statements preventing decimal places from being saved to config file.
- Missing ' from HTML code for dropdown, preventing value from being sent in POST. (always sent zero)

Test the updated code and let me know if it works for you

KirikauKiwi, 2024-11-08:
Yes all working for the decimals points 👍 

One other thing I just noticed is when on the Dashboard page of the GUI and I click refresh in Safari it takes me to the Configuration page, is this the expected behaviour? or should it refresh to the page you were viewing?

DeeEmm, 2024-11-08:
Depending on where you have been you might find that the URL in the address bar is set to /?view=config. This is intentional and is usually a result of the the redirect created when you save data whilst viewing the configuration tab.

In these cases when you save the configuration, it redirects you back to the configuration tab after the page refreshes,  which is the default behaviour.

If you then navigate to another tab the URL is not changed, as it does not 'navigate', but instead changes the display using javascript. So if at some point you refresh the page it will send you to the 'config' tab.

We can negate this behaviour by always redirecting to the main page after saving data.

KirikauKiwi, 2024-11-08:
I think that would be a useful small change as I would think in most cases that after saving a change you would want to go to the main page.

DeeEmm, 2024-11-09:
An easy fix. Consider it done.

As this issue is fixed, I'll close this out.

closed 2024-11-09 by DeeEmm