DIY-Flow-Bench/DIY-Flow-Bench #208: [FEATURE REQUEST]SCFM Conversion/Display/Logging

KirikauKiwi, 2024-11-08:
With a few of us getting close to having the DIYFB hardware and Flow benches/Test rigs up and running, would it be useful to implement SCFM conversions in the DIYFB firmware as this would give us a way to normalise our reading since we are not all in the same room on the same day.

Then, when we start testing our systems will are able to have flow results that will be directly comparable because of the standard set of conditions.

Having a poke around on the web this is the formula the would do it.
SCFM = ACFM × (P_act / P_std) × (T_std / T_act) × (1 / (1 - (RH / 100)))

Where the standard condition are;
P_std = Standard pressure (usually 101.325 kPa or 14.7 psi or 29.92 inHg)
T_std = Standard temperature (usually 293.15 K or 20°C or 68°F)
Dry Air 0%RH

DeeEmm, 2024-11-08:
Yes. Absolutely.

I will try and find the conversation that we previously had about standardisation, it is worth a read.

I cannot quote the exact standard number from memory, will have to look it up IIRC it was a DIN standard. 

But, the most appropriate standard (and there are many) is the standard that relates to automotive power testing. i.e the standard that governs how car manufacturers test their engines / vehicles to be able to provide a SAE/DIN power rating in KW.

Using the 'dyno' test standard makes a lot of sense when you consider that (ideally) the flow testing will then be carried out exactly the same conditions as the dyno testing.

Of course It should also be possible to swap out the reference standard so that people can use a difference reference standard should they wish. 

Easy enough to achieve.

I will put something together.

DeeEmm, 2024-11-09:
[image]

OK so I've hashed in the basic code, now I just need to do the maths part and test.

I've included several reference standards that specify Temp / Baro / RH. Unfortunately NIST do not specify RH so cannot be included (I will remove this entry)

All of these standards are either directly relevant, or generic.

DeeEmm, 2024-11-09:
I've corrected my previous reply above. it is actually ISO1585 that is the Automotive engine testing standard. SAE J1394 is the standard for non-metallic tubing, so somewhat unrelated.

DeeEmm, 2024-11-09:
> Having a poke around on the web this is the formula that would do it.
> SCFM = ACFM × (P_act / P_std) × (T_std / T_act) × (1 / (1 - (RH / 100)))

What's the source for this?

Seems a little too simplistic.

DeeEmm, 2024-11-09:
Okay. I've re-jigged the calcs for air density and broken them out into a dedicated method (calculateAirDensity). This method is used by both mass flow to volumetric flow calcs, as well as transposing between current and reference standard conditions.

I've based calcs on the following - https://neutrium.net/general-engineering/conversion-of-standard-volumetric-flow-rates-of-gas/

[image]

Specifically as we can calculate both the current air density and the air density at the reference standard.

The following reading may also be of interest https://www.buildingcalculators.com/standard-cfm-and-actual-cfm-convert 

I've pushed the changes to the following branch - https://github.com/DeeEmm/DIY-Flow-Bench/tree/SCFM-Conversion-%23208 but not keen to create a PR until we are happy that this is correct.

I also note a comment I made within the mass to volumetric conversion relating to how we add the reference pressure to the baro for the purpose of the conversion. Whilst seperate to this conversion, it is worthwhile getting a sanity check on this

calculations->convertFlow()
```
  // TODO validate reference pressure adjustment - do we add it or subtract it? Should be baro pressure less vac amount
  double refPressurePascals = sensorVal.BaroPA - this->convertPressure(sensorVal.PRefKPA, PASCALS) ;
```

DeeEmm, 2024-11-09:
BTW the dropdown is not working. At present it is just using default values for ISO 1585

DeeEmm, 2024-11-09:
It is interesting to see that I am getting an approximate 10cfm difference between my current reading and the standard

```
Local Conditions - 23.4 | 101.79 | 45.5%
ISO 1585 Std     -   25 | 100    | 0%
```

https://github.com/user-attachments/assets/0e205fb1-80a6-426c-ae7f-9686a7c5c339

DeeEmm, 2024-11-09:
OK. Dropdown now also working.

If anyone is happy to test this and give some feedback on the SCFM.

KirikauKiwi, 2024-11-09:
Yes it is interesting, and in deed and shows exactly why when trying to compare one flow bench to another it is like comparing 🍏  to 🍊.
At least with Standard conditions applied we will be able to compare 🍏 to 🍏 and 🍊 to 🍊 

In your example of a 10CFM difference this is mostly influenced by the Humidity difference as it will have the biggest scaling factor.
I see at your location it is 23.4 | 101.79 | 45.5%, I currently have 20.7 | 101.80 | 59.6%, also interesting to note as I have Temp/Humidity monitoring in several rooms at home, all temps are within 1 °C of each other, but there is a 7% difference in RH

KirikauKiwi, 2024-11-09:
Running the SCFM version now, looks bloody good, Nice work 👍 

Only so far comment is that on the dashboard display it would be great to include the SCFM depression inches and the Standards number.
That way we can easily see what version of SCFM we are using at a glance.

Nice to see ISA in there too.

I am liking the look of ISO5011 as it has a good real world values for the Temp,  %RH and is based at STD sea level.
i.e. the Temp and %RH are probably a good real world median.

Will take a while to verify the formulas used in the calculations but noting is looking out of sorts currently.

❤️ this update

DeeEmm, 2024-11-09:
The depression is not part of the reference standard. As such we are free to nominate a depression reference for collective results. Obviously 28"/H2O is the obvious depression to choose, but not all of us have systems that will achieve 28" (myself included). For comparing results I propose that we measure across a range of depressions so that it can cater for all benches.

I agree that ISO5011 is the closest to real world conditions, but do like that ISO1585 specifically relates to engine test conditions. I suppose that the actual reference standard chosen is more or less arbitrary and the actual temp/baro/RH is immaterial. Once we have a result that conforms to one reference standard it is easy to convert to any other standard.

My take on ISO 1585 compared to ISO 5011 is that 1585 follows the other International standards that use 0%RH as the baseline. Obviously it is almost impossible (perhaps actually impossible) to test engines at 0%RH and so the 0%RH figure only serves as a comparison baseline. 

ISO 5011 on the other hand, uses a mid range RH value which is easily achievable in test conditions. I suspect that the actual RH can directly impact the efficiency of air filters (ISO5011 is an air filter test standard) and that testing at 0% yields far more favourable results. 

I agree that ISO 5011 will give results that most closely match real world conditions. 

Happy to change the default to 5011

DeeEmm, 2024-11-09:
Default changed to ISO5011 and reference added to tile

[image]

KirikauKiwi, 2024-11-09:
I wasn't referring to fixing the SFCM tests to 28 inH2O, only to display what the inH2O has been set to in the configuration page.

Maybe what I am trying to convey is, would it also be useful to be able to view the "Standard adjusted flow" too.

DeeEmm, 2024-11-10:
> I wasn't referring to fixing the SFCM tests to 28 inH2O, only to display what the inH2O has been set to in the configuration page.
> 
> Maybe what I am trying to convey is, would it also be useful to be able to view the "Standard adjusted flow" too.

There is no way to specifically offset the depression for SCFM. The depression setting in the configuration is only for the adjusted flow value, it does not change or translate the SCFM value. 

SCFM is displayed at whatever the current active depression is. So the actual depression is already being displayed on the 'Depression' Tile

All calibrations (and therefore bench comparisons) must be done at the actual depression that the calibration orifice was originally tested at.

So if the orifice was tested at 10in/H2O, that is what the standard flow musts also be measured at.

Hope this makes sense.

KirikauKiwi, 2024-11-10:
Yes that makes sense.
Fact checking here. so if we have a SCFM value at what ever the measured depression is say 10 inH2O as you refer to above. I thought the SCFM @10inH2O would be able to be converted to SCFM @ ?? inH20 with the pressure ratio calculation as it is done with ACFM.
Secondly would't you be able to do the SCFM conversion on the ACFM adjusted flow, so it would make sense that the math should work either way, or am I over looking something here.

DeeEmm, 2024-11-10:
> The case for Adjusted Standardised flow
> 
> I accept that some people will want to do this. But it must be applied to the Adjusted flow value and not the other way around.

Further to this, it would also need to be obvious to the user, what adjustments are applied to the adjusted flow figure...

We currently have...

[image]

We would need to change that to...

[image]

But the change is not obvious and is easily overlooked / misunderstood.

90% of blokes do not bother to read instructions / the WIKI so they would miss this sublety and end up with a measurement result that was not what they expected, plus be completely oblivious to it

It is a similar story with adding an additional tile

[image]

We would end up with 5 different flow tiles in that position, which for some  may be information overload when all they want is a CFM reading

-----------------------------------------------

So I am happy to add it to the adjusted flow, but unsure of the best way to proceed.

Perhaps it is a setting in the configuration and the end user selects one or the other when they initially set their bench up

DeeEmm, 2024-11-11:
Checkbox in configuration added to enable / disable standardised flow for adjusted flow value

Changes in master

KirikauKiwi, 2024-11-11:
Thanks for adding it in.

I agree with you that each bench should only be used within its actual physical performance capabilities and that calculated data is usually not as good as measured data so is best to avoid if possible, you could argue that the adjusted ACFM should be hidden away too.
The main reason to add it in for me was that it is at least keeping the standard conditions calculations across the board and in line with what can be done with the ACFM, I am interested to see how repeatable testing will be comparing ACFM to SCFM over a period of time, as in my eyes ACFM measurements are essential used like a comparator, so my test today won't be the same as my test tomorrow if the ambient conditions have changed. So if working away on a project for a few months using ACFM readings it will be hard to get any form of repeatability due to the different conditions each time the measurement are made, only the changes you make on the day will be able to be seen i.e did you improve the flow or not.

It will be interesting overall to see how good the accuracy and precision will be of the different DIYFB builds over time.

PS, Thanks for you work in this, in creating a great software and hardware tool for the community.

DeeEmm, 2024-11-11:
Thanks.

I think it is a useful feature, and for some will likely be the default mode, especially those with a shop-vac and floating depression style setup. I mean it doesn't get any simpler than a MAF and shop-vac. If that kind of system can spit out repeatable and somewhat accurate flow numbers then the software truly becomes a useful tool.

I too am very keen to see how the figures compare from bench to bench as this will be the real test. I want to get some of these additional requests added in so that we can sew up the V2 stuff. Your pitot request is next on the list.

The final change will be adding in some form of 'system report' for comparing performance data. I've got some ideas on that but will start another thread (if there is not one already)

closed 2024-11-11 by DeeEmm