DIY-Flow-Bench/DIY-Flow-Bench #211: [BUG]Flow Diff view on the DIYFB GUI

KirikauKiwi, 2024-11-08:
**Software Version**
- RELEASE V.2.0-RC.8
- Build 24110802
- 

**Describe the bug**
Should the three labels at the lower area of the "Flow Diff" match the first three labels from the configuration page Calibration Data, currently they do not.

It appears that all three of the Calibration Data offsets gets applied to the "Flow Rate" and "Adjusted Flow" as expected and only two of them appear to be applied to the "Flow Diff" the total numeric value is correct, but when you switch between the three options for "Flow Diff" the value only changes twice, I would expect it to change three time if there is data in all three of the calibration data.

Would it be useful to add the applied offset value to the text label that is displayed at the lower part of the "Flow Diff"
[image]

[image]

KirikauKiwi, 2024-11-08:
LOL, got stuck and couldn't getting my head around it, but the penny dropped finally lol.

KirikauKiwi, 2024-11-08:
I think I got caught as it looks like the "User Target" has not yet been full setup, is this correct?

DeeEmm, 2024-11-09:
Flow targets were working, but I can see that now it is not.

Normally you click on the 'Unit' part of the tile to cycle through 'User target' / 'Leak Test' / 'Baseline'. 

Will take a look

DeeEmm, 2024-11-09:
I can see the event triggering in the console and being received in the serial monitor...

[image]

[image]

but I can also see an error...

[image]

I note that this is only happening for negative depression values so suspect related to that.

Will continue later this eve..

KirikauKiwi, 2024-11-09:
I think I see the same as you on the screen.
What I think is happening is that on the dashboard page, when "Flow Diff, User Target (cfm)" is displayed where is it picking up its data value from? 

Because on the configuration page there in no field labeled "User Target", I only have "Baseline" and "Offset". 
Should there be a "User Target" field on the configuration page?

[image]

DeeEmm, 2024-11-09:
Ahhh okay. 

To set the user target you need to click on the value in the tile...

https://github.com/user-attachments/assets/91107995-2d66-434f-a5fb-fa3927905bfc

I added this to the dashboard rather than the config as it is something that will likely get changed often. So this way it is quick to edit.

KirikauKiwi, 2024-11-09:
It would be good I think to add the value of the offset to the view if possible on the dashboard page.

E.g. Currently its displays "Baseline (cfm)" I am suggesting to display  "Baseline (Value)(cfm)"

KirikauKiwi, 2024-11-09:
> Ahhh okay.
> 
> To set the user target you need to click on the value in the tile...
> 
>  Screen.Recording.2024-11-09.at.8.51.58.pm.mov 
> 
> I added this to the dashboard rather than the config as it is something that will likely get changed often. So this way it is quick to edit.

Ahhhh so that is what I was missing, got it now 👍

KirikauKiwi, 2024-11-09:
😂 I can confirm that it is all working as it should!!!!!! Now that I know what to do!!!!

Layer 1, User issue!!!!!!!!

closed 2024-11-11 by DeeEmm