DIY-Flow-Bench/DIY-Flow-Bench #212: [BUG]Configuration page, INFO display.

KirikauKiwi, 2024-11-09:
**Software Version**
- RELEASE V.2.0-RC.8 
- BUILD_NUMBER Build 24110802

More of a refinement than a BUG.
When viewing the INFO page, I am unable to scroll the page. I cannot see the last Pressure sensor.

Can you group the pressure sensors so they are displayed next to each other too.

[image]

DeeEmm, 2024-11-09:
I've reduced the font size a little to hrlp stop the overflow. Let me know how you get on

KirikauKiwi, 2024-11-09:
It has improved it a little. The text is still partially of the screen.
https://github.com/user-attachments/assets/f44f31cc-f237-4d8b-9636-003e2b5a9a79

Personally I preferred the previous text size as it appears to have changed the Dashboard too.

DeeEmm, 2024-11-11:
On the current code I've changed the test size back to previous but added a height component to the modal dialog. Let me know if that changes the issue.

You can also try to manually change the CSS using web developer in chrome and finding a height that works for your display

DeeEmm, 2024-11-11:
OK added scroll to the modals when content overflows. 

Should be able to scroll the info modal upwards now.

Changes in master

Let me know how you get on

KirikauKiwi, 2024-11-11:
BUILD_NUMBER "24111107"
RELEASE "V.2.0-RC.8"

It looks like the modal changes have affected the main GUI page and not the INFO pop up page.
https://github.com/user-attachments/assets/6adb2504-d453-4b62-b457-01a8a8a912f7

I liked how the main page had a sticking point so it aligned with the top of the browser window, this has gone now.
https://github.com/user-attachments/assets/6c286a17-f875-4a93-b537-b066e2f23dae

DeeEmm, 2024-11-22:
Might rewind a bit here then.

The overall aim of the GUI is to not need to scroll. It is designed for a touchscreen / tablet.

So everything that needs to be viewed, should be visible within the view-frame without scrolling. Sometimes I forget this when I'm sitting at my 27" monitor 😆 

Adding the padding in the footer has created some room for one item but caused a couple of other problems, so not really a workable solution.

I think the best solution here is to revert the changes to the footer and then look at improving the layout of the info modal, perhaps even change it to a tabbed modal to get more realestate.

DeeEmm, 2024-11-22:
There was a suggestion / discussion (somewhere) to include a page for I/O and include mimics to show the state of inputs and outputs. That seems like a logical place to move the sensor stuff. So perhaps creation of the 'diagnostics' page / modal / tab will be the end goal here and kill two birds with one keyboard.

DeeEmm, 2024-11-22:
Ok I've pencilled this in for V3, but it should be possible to get basic tabbed structure in place before then, so that we can move the sensor info.

This will reduce size of info modal and make sensor info readible.

DeeEmm, 2024-11-22:
Got a bit of an interim fix...

[image]

Changed it to a 2 column layout.  Let me know if that works for you.

DeeEmm, 2024-11-22:
Just tidied up the alignment a little...

[image]

KirikauKiwi, 2024-11-23:
> Might rewind a bit here then.
>
> The overall aim of the GUI is to not need to scroll. It is designed for a touchscreen / tablet.
>
> So everything that needs to be viewed, should be visible within the [...]

Yes this works great. 
Totally agree with you about the rewind and the GUI design/functionality being used primarily for tablet 👍

DeeEmm, 2024-11-23:
Unfortunately still need to scroll on the configuration page, but as that is not accessed too much. is a small compromise and probably less annoying than have to remember which one of the ten tabs had the setting you needed on it. 😆

KirikauKiwi, 2024-11-23:
Minor detail, I am still seeing the scroll bar in the modal window. I notice in your screen shot that it is not there.
[image]

DeeEmm, 2024-12-11:
Closing as completed

closed 2024-12-11 by DeeEmm