DIY-Flow-Bench/DIY-Flow-Bench #213: [FEATURE REQUEST]GUI Pitot display changes

KirikauKiwi, 2024-11-09:
These are suggested changes for the "Pitot" area of the DIYFB GUI.

Suggested Changes
- Heading of "Pitot" change to "Air Speed" or "Air Velocity"
- Main displayed value is currently "∆Pitot" change to "Speed Value" e.g. display the calculated speed V = sqrt [2 * {pt – ps} / r ]
- "inH2O" to "ft/s" or "m/s" or other preferred unit of velocity
- ADD "Pitot ∆ (Value) inH2O" as this I think will be useful to have on the display next to the velocity unit being displayed.

[image]

Note: units have to be kg/m^3^ and Pa or slug/ft^3^ and lbs/ft^2^

DeeEmm, 2024-11-16:
I've closed out #213 as I feel that it can be rolled up within this change. Whatever solution is implemented on Pitot can be utilised on pDiff / pRef

> **Is your feature request related to a problem? Please describe.** Differential sensors zero point may drift depending on system design / construction / ambient conditions
> 
> **Describe the solution you'd like** Add a zero button to differential sensors to allow ambient condition to be zero'd
> 
> This applies to Pitot, Diff pressure and also Ref pressure as they all use MPX7007DP sensors

DeeEmm, 2024-11-16:
Also closing out #159 - we'll keep all of the pitot stuff in this issue

> Will be nice to have feature for pito input calibration for air speed probing so it will display air speed in m/s or ft/s instead of differential pressure if pito input is unused with maf or orifice style flow bench.
> 
> My small pump starts to lose pressure after approximately 150-160cfm @ 28in H2o, so corrected and uncorrected flow numbers start to diverge, so I probably will change to my big pump that was too powerful for my electrical installation, but now I have VFD, so it should be ok.
> 
> I still need to fix few leaks and do calibrations for flow bench.
> 
> _Originally posted by @black-top in [#155 (reply in thread)](https://github.com/DeeEmm/DIY-Flow-Bench/discussions/155#discussioncomment-7918628)_

DeeEmm, 2024-11-16:
First blush...

[image]

I will eventually add the GUI descriptors to the language overrides so that people can rename things like pitot / depression / etc to their liking.

DeeEmm, 2024-11-21:
So Pitot functionality is working (to a basic fashion).

Both Pitot and PDiff values can be zeroed down by clicking on the value in the tile.

I don't have a sensor or pot hooked up to test with so the values are just floating on my test bench...

https://github.com/user-attachments/assets/53818823-430f-40f0-81d6-1f4f4c3d4c5f

One thing that need to be determined is where the delta is measured between.

Hooking up the pitot across the sensor should return the delta without needing to calculate this in the code, but as I do not have a pitot sensor to test with, someone else may need to validate this for me.

There are a bunch of changes in this commit including the changes to loading the pin definitions and expansion of translation info. 

Once everyone is hjppy I will merge the changes into the main branch

black-top, 2024-11-21:
I have a pitot tube. I will order MPXV7007DP for pitot and test it when I get it.

DeeEmm, 2024-11-21:
Created #249 - Pitot improvements to capture additional functions / improvements in pitot functionality for V3

KirikauKiwi, 2024-11-23:
- I am able to zero the differential sensor.
- I cannot zero the Pitot value, the reading ∆ ?? inH2O is not getting displayed.
- the main pitot value being displayed appears still to be the pressure value and not the calculated velocity.
- possibly add a coma between from this m/sec ∆??inH2O to m/sec, ∆??inH20

[image]

KirikauKiwi, 2024-11-23:
Would also be good to distinguish in someway that you have zero'd the display, as with the Differential tile, the display looks the same if it is zero'd or not.

DeeEmm, 2024-11-23:
The pitot behaviour is a bit strange for me to test. As the value is floating and not stable, it tends to jump around a bit.

A colour or intensity change might be appropriate to indicate if it has been zeroed.

At the moment clicking the value zeroes it, but there is no way to 'un-zero' the value.

So perhaps...

- Click value to zero. [value changes colour indicating it has been zeroed]
- Click value to un-zero. [value changes colour  indicating it has been un-zeroed]

This saves adding yet another control to the GUI and follows the existing control philosophy of clicking / touching the value to manipulate it.

Obvious choice is that the colour change could be red = unset / green = zeroed, but could also use an intensity change such as grey = unset / black = zeroed

[image]

Red / green is obviously much easier to differentiate between, however we might consider reserving red for any error reporting that we might want to add in the future.

So perhaps pink or a reddish orange.

Pink (kinda looks like a muted red)

[image]

Which is very easy to differentiate from red...

[image]

Personally I'm a fan of subtlety, and would use grey / back, however this is open to error of interpretation so a more distinct change will help people know whether the value has been zeroed without them needing to 'think'. Plus a black colour does not distinguish it from any other value state, so is easily overlooked.

Light grey...

[image]

And of course black..

[image]

We could of course make the colours user definable, but that's definitely a down the track thing.

I will make the changes and we can hone it after.

DeeEmm, 2024-11-24:
Actually discovered an error. Pitot was saving to wrong calibration variable 😨

DeeEmm, 2024-11-24:
OK so above changes are implemented but I've not created a PR yet as I want to do some more testing.

Will finish off testing once real-life work is out of the way. 😄

DeeEmm, 2024-11-24:
Ok, changes now in DEV. including cookies to retain selected tiles across page loads and reporting sensor voltage to API for pDiff and pRef sensors to aid with setting trim pots

24112402 - #213 Pitot / differential zeroing
                - #254 Report sensor voltage to API
                - Cookie to remember tile status

DeeEmm, 2024-11-24:
[image]

DeeEmm, 2024-11-24:
@KirikauKiwi  comments moved from #249

> Pitot Display
> 
> * I cannot get the Pitot to zero, the number is changing colours, however the numerals stay the same.

The numbers are zeroing here but I don't have a pitot to test with, so flying blind a little.

Are you using Build 24112402?

> * The ∆ pressure is only displaying either 0, 1 or 2, you are not able to see the units/ones, or the decimal values.
> * Based on the above I suggest to put the m/sec above the ∆ ??? inH20, keeping them centre aligned, I think this will be less cluttered and give more space for the ∆ to be displayed.

That's the only issue with adding additional chars in there, real estate is very limited. Adding an additional line will increase the overall viewport height as it will also add a line to all of the other tiles. 

I'll revert the change. I don't think that there is a need for an additional commer or delimeter as the ∆ symbol itself is the delimiter between the data.

> * The colour changing number are great to identify what can be interacted with and zeroed, like the green, not so much a fan of the pink/salmon colour it disappears in to the background for me, lack of contrast I think.

See comment above ^^^

KirikauKiwi, 2024-11-25:
Yes on Build 24112402, apologies forgot that detail!!!!!
https://github.com/user-attachments/assets/b2f4adf6-eb4a-4e1d-ab64-f404902517db

DeeEmm, 2024-11-25:
Ok think I might have found the issue...

When zeroing the current value is saved as the calibration offset...

Calibration::setPitotCalOffset()
```
  if (calVal.pitot_cal_offset == 0) {
    // update config var
    calVal.pitot_cal_offset = sensorVal.PitotKPA;
    _message.debugPrintf("Calibration::setPitotDeltaOffset %d kPa\n", calVal.pitot_cal_offset);
  } else {
    // update config var
    calVal.pitot_cal_offset = 0.0;
    _message.debugPrintf("Calibration::resetPitotDeltaOffset %d \n", 0.0);
  }

  saveCalibrationData();   
```

This value is then used to offset the value assigned in the main task...

TASKgetSensorData( void * parameter )
```
        #ifdef PITOT_IS_ENABLED
        sensorVal.PitotKPA = _sensors.getPitotValue() - calVal.pitot_cal_offset;
        sensorVal.PitotH2O = _calculations.convertPressure(sensorVal.PitotKPA, INH2O) ;
        sensorVal.PitotVelocity = _sensors.getPitotVelocity();
        #endif
```

So we are only zeroing the differential

But, then when we calculate the velocity, it is grabbing the data directly from the sensor rather than the global struct...

```
	pitotPressure = this->getPitotValue();

	// get air density
	airDensity = _calculations.calculateAirDensity(sensorVal.TempDegC, sensorVal.BaroKPA, sensorVal.RelH);
```

It should be...

```
	pitotPressure = sensorVal.PitotKPA;

```

KirikauKiwi, 2024-11-25:
The zeroing is working now.

I think it might be useful to display something where the Velocity is displayed for when the ∆P is negative. I know we shouldn't probably display a negative velocity, well I guess you can, as it means it is in the opposite direction to what is expected.
e.g. Your Pitot tube is connected up incorrectly, P_total and P_static have been connected to the wrong ports and/or something else could be causing a problem.

Secondly if the output for the Pitot sensor was for some reason reading below its cross over point when both ports are open to atmosphere, this won't get displayed currently (negative velocity), so wouldn't alert you to calibrate/offset for pitot MV_TRIMPOT.

https://github.com/user-attachments/assets/2ed255a7-6a7d-4893-a012-909ca90c5f58

DeeEmm, 2024-11-25:
At the moment the velocity value's sign is removed using fabs(). So if there is negative velocity it's just displayed as a positive number.  It was also the same with the differential but I removed fabs() from that after your previous comments. It's trivial to remove it from both but...

I guess the determining factor here is how / where the user zeroes down the value. If they zero the value and then move the probe to a lower pressure area, it will show a negative value.

But I think that we need to consider if this is actually valid use case. The only real way for the value to go negative is for the user to zero down with the probe in the port. If it is zeroed at ambient then this should never happen. 

Swapped probe leads should not be an issue. The calcs only look at the differential, not individual port values. In this case the fabs() function actually solves the issue of swapped leads and 'negative' velocity by dropping the sign. So the fact that the probe reads 'backwards' is not an issue if we leave the fabs() function in place. The probe will read correctly no matter which way around the leads are connected. Ultimately a lot more 'fool-proof'.

>if the output for the Pitot sensor was for some reason reading below its cross over point 

Define 'some reason'. 

If it is an error we are trying to detect with this change, then we should probably check for the specific error and then flag the error to the user.  

The planned 'manometer' or graphic display will assist in helping the user to better visualise what is going on with the pitot values, including crossing the zero line. 

As with most processes, the procedure itself forms a large part of the correct operation of the device / machine / process. The philosophy of trying to catch deviations from the correct operational procedures can have a large adverse effect on ease of use. Consider it in the same way as calibration. If it's not done correctly, then the bench will be inaccurate. So we have to trust that the user will follow the correct procedure. This is why the Wiki is so important.

KirikauKiwi, 2024-11-25:
Don’t think I explained that very well, 
In the video you can see when the delta P goes negative the velocity reads zero.

KirikauKiwi, 2024-11-25:
I hadn't taken a look at the formula in the sensors.cpp for the pitot velocity before now.
```
	// Convert Pitot pressure to velocity (m3/min)
	// airVelocity = sqrt(2 * (pitotPressure - sensorVal.PRefKPA) / airDensity );
	airVelocity = sqrt(2 * (pitotPressure) / airDensity );

	// Lets make sure we have a valid value to return - check it is above minimum threshold 
	// TODO - 'borrowing' min_bench_pressure
	double pitotComp = fabs(airVelocity);
	if (pitotComp > config.min_bench_pressure) { 
		return airVelocity;
	} else { 
		return 0.0001; // return small non zero value to prevent divide by zero errors (will be truncated to zero in display)
	}	
```
Yes the fabs will work for the air velocity.  The airVelocity won't return a value for the sqrt of a negative ∆Pressure.

The velocity formula will need to take the absolute value of ∆Pressure.
```
airVelocity = sqrt(2 * abs(pitotPressure) / airDensity );
```

DeeEmm, 2024-11-25:
> Don’t think I explained that very well,
> In the video you can see when the delta P goes negative the velocity reads zero.

Yes understood

The solution is to revert the change where I allow the differential to go negative

BTW. abs() only works for integers. Needs to be fabs() for floats / doubles. You will note that this is already present on line 978

```
double pitotComp = fabs(airVelocity);
```

We only apply this to the result, not the calculation itself

DeeEmm, 2024-11-25:
Should now only display positive values.

When we add the ability to view pitot pressure we will include ability to view =/-ve pressure values.

This is not necessary for velocity for the reasons outlined previously.

Will add item to Wiki update to outline procedure for pitot zeroing / calibration

KirikauKiwi, 2024-12-01:
I have made a pull request #272 with a few suggested changes that should close this out.

DeeEmm, 2025-01-13:
As #272 is now closed I am going to close this oout

closed 2025-01-13 by DeeEmm