DIY-Flow-Bench/DIY-Flow-Bench #216: [BUG] MAF data sudenly does not show in UI

stefan63, 2024-11-10:
**Software Version**
Version:V.2.0-RC.8
Build:24110901

 Wemos D1 R32
After that the bench has been running for some time (random) the MAF data dissapears and only shows 0.0.
The only way I can get it back is to delete  config.json, liftdata.json AND cal.json and then power cycle the board.
The files above is then recreating themselfs and the board and UI works normal again.
Next time it hapens I will try deleting one file at the time to pinpoint the trouble maker ;)

DeeEmm, 2024-11-11:
Did you manage to replicate this?

stefan63, 2024-11-11:
Yes, will try to upload latest version since I'm still on build:24110901.
I attatch the files, removing the config file kind of solved the problem but when I then change the maf housing size it shows that  X1000 value again. Will report back if problem still valid after update.
[cal.json](https://github.com/user-attachments/files/17702565/cal.json)
[config.json](https://github.com/user-attachments/files/17702566/config.json)
[liftdata.json](https://github.com/user-attachments/files/17702567/liftdata.json)

stefan63, 2024-11-11:
Realized that I had to change the file type to attach html ;)
[index.html.txt](https://github.com/user-attachments/files/17702630/index.html.txt)

DeeEmm, 2024-11-11:
I cannot see anything untoward in the file.

Would definitely recommend updating to the current version in Master (including index.hml file). This contains all recent changes.

stefan63, 2024-11-11:
rev 06 works much better, will try you MAF look up table A.S.A.P

DeeEmm, 2024-11-22:
Is this now resolved with current version in DEV?

DeeEmm, 2025-01-11:
I'm going to close this issue as MAF Code rewritten