DIY-Flow-Bench/DIY-Flow-Bench #217: Scaled MAF housing returns out of range value

DeeEmm, 2024-11-11:
I am starting to work out the flow of the code!!!! it is a bit like trying to read a foreign language at times.

With the most current build 24111102, I am still getting wacky numbers on the GUI when I enter a value in the MAF Housing diameter on the config page.

They appear at least in the ball park and correct order of magnitude when I divide the result by one million., see below.
```
// Calculate the 'velocity' for the original pipe area
		mafVelocity = flowRateKGH  / PI * pow(status.mafDiameter / 2, 2);

		// scale the result with the new pipe area and convert back to mass flow
		transposedflowRateKGH = mafVelocity * PI * pow(config.maf_housing_diameter / 2, 2) / 1000000;

		return transposedflowRateKGH;
```
This is what I get on the GUI,
[image]

Again without the divide by one million.
```
		// Calculate the 'velocity' for the original pipe area
		mafVelocity = flowRateKGH  / PI * pow(status.mafDiameter / 2, 2);

		// scale the result with the new pipe area and convert back to mass flow
		transposedflowRateKGH = mafVelocity * PI * pow(config.maf_housing_diameter / 2, 2);

		return transposedflowRateKGH;
```
This is what I get on the GUI,
[image]

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/discussions/174#discussioncomment-11208752_

DeeEmm, 2024-11-11:
Confirm that using different MAF diameter results in abnormal / out of range value

[image]

Will investigate

DeeEmm, 2024-11-11:
Is this related to #203

KirikauKiwi, 2024-11-11:
> Is this related to #203

No, IIRC that was down to and index.html problem, I had an aged index.html, it was a little tricky to get it to update back then, but have had no other problems like that since.

DeeEmm, 2024-11-11:
Cool. kinda lost track as the two have similar symptoms.

KirikauKiwi, 2024-11-11:
If I calculate the values with pen and paper I get values that seem sensible and expected. When I did the divide by one million it puts the displayed values in the correct order of magnitude but I don't think the values are correct.

IIRC when I did the divide, it displayed about 700-800 KGH, when the manually calculated value was about 280KGH.

Good luck hunting the gremlins.

DeeEmm, 2024-11-11:
Need to track down the source that I used. I note that I did not include the URL.

I suspect that it is a scaling error - similar to that you originally suggested with mm / metres, but tied to the mass flow units

DeeEmm, 2024-11-11:
I cant find the source but the conversion formula is pretty well documented.

https://phys.libretexts.org/Bookshelves/College_Physics/College_Physics_1e_(OpenStax)/12%3A_Fluid_Dynamics_and_Its_Biological_and_Medical_Applications/12.01%3A_Flow_Rate_and_Its_Relation_to_Velocity

https://jackwestin.com/resources/mcat-content/fluids/continuity-equation

However the actual way I am using the formula is a bit bodgy. The formula is essentially for volumetric flow. but I am electing to ignore that on the basis that it is essentially just a ratio-metric relationship and that for the instant in time where we convert the value, there cannot be any changes in volume based on environmental changes to temp, baro and RH. So we can safely ignore them and apply the same ratio directly to the Mass Flow value

What we end up with is a simple scaling formula, This is more evident now that I have broken the area calcs out of each formula

```
// Calculate original MAF area
oldMafArea = (PI * pow((status.mafDiameter / 2), 2)) / 1000000;
// Calculate new MAF area
newMafArea = (PI * pow((config.maf_housing_diameter / 2), 2)) / 1000000;

// Calculate the 'velocity' for the original pipe area
mafVelocity = flowRateKGH / oldMafArea;
// scale the result with the new pipe area and convert back to mass flow
transposedflowRateKGH = mafVelocity * newMafArea;

```

I noted your comments relating to m3, which is the SI for Q in the formula Q=VA, so converting the area A into m2 rather than mm2 is the logical change. This is the divide by 1,000,000 aspect. Doing this generates reasonable results and the large values are now gone.

So the maths now appears to be sound, but I think that we still need to validate my bodgy formula. Have I oversimplified the conversion? 

I would like to validate what we are seeing in the GUI using a third party method such as an online calculator.

DeeEmm, 2024-11-11:
Changes are on the 'Out-of-range-value-#217' branch

https://github.com/DeeEmm/DIY-Flow-Bench/tree/Out-of-range-value-%23217

DeeEmm, 2024-11-11:
I am getting very close to the GUI value using the following calculators

https://www.sensorsone.com/volumetric-flow-rate-and-diameter-to-flow-velocity-calculator/#volume-flow

and

https://www.sensorsone.com/flow-velocity-and-diameter-to-volume-flow-rate-calculator/

But there is a discrepancy of about 4cfm based on a change from the 82mm MAF to 100mm. However I see the same 4cfm discrepancy testing at 50 cfm as I do at 250cfm, So I suspect that the discrepancy is not as a result of an issue with the calculation but some other variable.

To give some perspective -  240cfm@82mm increases to 360cfm@100mm.

I'm going to create a PR and bring this into the main code. I am mostly satisfied that the calcs work but would like to do further real world testing.

KirikauKiwi, 2024-11-11:
I entered what I had typed unto verify the formulas and passed it through ChatGPT to help with the Markdown formatting and it came up with this below, using 82mm for the MAF mousing and 100mm for the New housing.

**ChatGPT output**
Ah, got it! You want the formulas and explanation **without LaTeX**, but still in a readable format. Below is the step-by-step solution, written in plain HTML for easy understanding.

### Problem Breakdown:

You have two cases for the MAF sensor: one in the original housing and one in a different-sized housing. The objective is to calculate the changes in mass flow rate due to the different cross-sectional areas of the housings, while the velocity of the air remains the same.

---

### Step 1: Case 1 — Solve for Velocity (V1)

The formula for mass flow in **Case 1** (original housing) is:

**m₁ = ρ * A₁ * V₁**

Where:
- **m₁** is the mass flow rate (kg/s),
- **ρ** is the air density (kg/m³) (same for both cases),
- **A₁** is the cross-sectional area of the original housing (m²),
- **V₁** is the air velocity (m/s) in the original housing.

We want to solve for the **velocity** (V₁). To do this, simply rearrange the formula:

**V₁ = m₁ / A₁**

Where:
- **m₁** is the mass flow rate (kg/s) in the original housing.

---

### Step 2: Case 2 — Calculate Mass Flow (m₂) in the New Housing

In **Case 2**, the MAF sensor is in a new housing with a different cross-sectional area **A₂**, but the velocity **V₁** remains the same (because the sensor reads the same air velocity).

The formula for mass flow in **Case 2** is:

**m₂ = ρ * A₂ * V₁**

Since we know **V₁** from Case 1, we can substitute it into the equation:

**m₂ = A₂ * (m₁ / A₁)**

Where:
- **m₂** is the new mass flow rate in the new housing,
- **A₂** is the cross-sectional area of the new housing,
- **A₁** is the cross-sectional area of the original housing,
- **m₁** is the mass flow rate in the original housing.

Rearranging this:

**m₂ = (A₂ / A₁) * m₁**

This equation shows that the mass flow rate **m₂** in the new housing is proportional to the ratio of the areas **A₂ / A₁** and the original mass flow rate **m₁**.

---

### Conclusion:

So, the formula for mass flow in the new housing becomes:

**m₂ = (A₂ / A₁) * m₁**

Where:
- **m₁** is the mass flow rate in the original housing,
- **A₁** is the cross-sectional area of the original housing,
- **A₂** is the cross-sectional area of the new housing.

---

### Example:

Let's say:
- The original mass flow rate **m₁ = 363.5 kg/h**,
- The area of the original housing **A₁ = 82 mm²**,
- The area of the new housing **A₂ = 100 mm²**.

#### Convert Mass Flow Rate to kg/s:
Since we are working in **kg/s**, we first need to convert **m₁** from kg/h to kg/s:

**m₁ = 363.5 kg/h ÷ 3600 = 0.101 kg/s**

#### Calculate Velocity (V₁) in Original Housing:
Now, we can solve for **V₁**:

**V₁ = m₁ / A₁ = 0.101 / (82 x 10⁻⁶) = 1231.7 m/s**

#### Calculate New Mass Flow Rate (m₂) in the New Housing:
Now, use the formula for **m₂**:

**m₂ = (A₂ / A₁) * m₁**

Substitute the values:
- **A₁ = 82 mm² = 82 × 10⁻⁶ m²**
- **A₂ = 100 mm² = 100 × 10⁻⁶ m²**

So:

**m₂ = (100 × 10⁻⁶ / 82 × 10⁻⁶) * 0.101 = 0.123 kg/s**

#### Convert m₂ back to kg/h:
**m₂ = 0.123 * 3600 = 442.8 kg/h**

Thus, in this example, the mass flow rate **m₂** in the new housing would be **442.8 kg/h**, which is higher than the original mass flow rate **m₁ = 363.5 kg/h**, because the area **A₂** is larger than **A₁**.

---

### HTML Representation (Without LaTeX):

```html
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>MAF Sensor Mass Flow Calculation</title>
</head>
<body>
    <h1>MAF Sensor Mass Flow Rate Calculation</h1>

    <p>We have two cases for the MAF sensor:</p>

    <h2>Case 1: Original Housing</h2>
    <p>The mass flow rate in the original housing is:</p>
    <pre><code>m₁ = ρ * A₁ * V₁</code></pre>
    <p>To solve for velocity (V₁), we rearrange the formula:</p>
    <pre><code>V₁ = m₁ / A₁</code></pre>

    <h2>Case 2: New Housing</h2>
    <p>In the new housing, the mass flow rate is:</p>
    <pre><code>m₂ = ρ * A₂ * V₁</code></pre>
    <p>Substitute V₁ from Case 1:</p>
    <pre><code>m₂ = A₂ * (m₁ / A₁)</code></pre>
    <p>Final formula:</p>
    <pre><code>m₂ = (A₂ / A₁) * m₁</code></pre>

    <h3>Example Calculation</h3>
    <p>Given:</p>
    <ul>
        <li>m₁ = 363.5 kg/h</li>
        <li>A₁ = 82 mm²</li>
        <li>A₂ = 100 mm²</li>
    </ul>

    <p>Step 1: Convert m₁ to kg/s:</p>
    <pre><code>m₁ = 363.5 / 3600 = 0.101 kg/s</code></pre>

    <p>Step 2: Solve for V₁:</p>
    <pre><code>V₁ = m₁ / A₁ = 0.101 / (82 × 10⁻⁶) = 1231.7 m/s</code></pre>

    <p>Step 3: Calculate m₂:</p>
    <pre><code>m₂ = (100 × 10⁻⁶ / 82 × 10⁻⁶) * 0.101 = 0.123 kg/s</code></pre>

    <p>Step 4: Convert m₂ to kg/h:</p>
    <pre><code>m₂ = 0.123 * 3600 = 442.8 kg/h</code></pre>

    <h3>Conclusion:</h3>
    <p>The new mass flow rate in the larger housing is 442.8 kg/h.</p>
</body>
</html>
```

---

### Summary:
- **In the new housing**, with the same velocity, the mass flow rate changes because the area of the housing has changed.
- The mass flow rate in the new housing **m₂** can be calculated by multiplying the original mass flow rate **m₁** by the ratio of the areas **A₂ / A₁**.

Let me know if you'd like further clarification or more examples!

KirikauKiwi, 2024-11-11:
So it looks like it can all be done in one calculation!!!!!

**m₂ = (A₂ / A₁) * m₁**

DeeEmm, 2024-11-11:
Essentially the equation that GPT is using 

```
m₁ = ρ * A₁ * V₁
```

Is the same that I used, where Q = volumetric flow and mass is ignored. i.e. flow but actual denomination is unimportant

```
Q = V * A
```

The development of the equation uses the continuity equation which states Q1 = Q2 which can be further expressed as

```
V1 * A1 = V2 * A2
```

So when we transpose for V2 we get

```
V2 = (V1 * A1) / A2
```

Which essentially gives the same  result as GPT where it scales the value based on the relationship between A1 - A2

DeeEmm, 2024-11-11:
I'm going to close this out as working.

I will also tidy up the code and include the GPT version as a comment

```
// scale the result with the new pipe area and convert back to mass flow
transposedflowRateKGH = (flowRateKGH / oldMafArea) * newMafArea;

// Chat-GPT version		
// transposedflowRateKGH = flowRateKGH * (newMafArea / oldMafArea);
```

closed 2024-11-11 by DeeEmm