DIY-Flow-Bench/DIY-Flow-Bench #219: [FEATURE REQUEST] Support for RC-Servo operated valves

stefan63, 2024-11-11:
I realize that stepper motor is in the todo list, but I stll want to lift the posibilty to use RC-servos to controll valves.

DeeEmm, 2024-11-11:
Is this PWM output?

stefan63, 2024-11-11:
Yes and there are loads of code hints for esp32 as well as Arduino in general.
https://en.wikipedia.org/wiki/Servo_control

DeeEmm, 2024-11-11:
PWM is pretty trivial to implement but it will require additions electronics as I suspect that the servomotor will exceed the capabilities of standard output current capacity (nominal 500mA). This is easy enough to do using a mosFET. The new PCB also now has a prototype area where additional electronics like this can be added which includes direct access to molex pins.

I suspect that we will also use the same philosophy for PWM based IAT bypass valve to fine control the bleed pressure

KirikauKiwi, 2024-11-11:
The PWM implementation will also be useful for motor control as discussed in #186 Good for DIYFB V3