DIY-Flow-Bench/DIY-Flow-Bench #224: [BUG]Calibration offset value error

KirikauKiwi, 2024-11-11:
**Software Version**
BUILD_NUMBER "24111107"
RELEASE "V.2.0-RC.8"

I am seeing an incorrect value being generated for the calibration offset, I am using the default 14.4 CFM and 10 inH2O from the calibration orifice settings no other offsets applied.

**My pre calibration flow and pressure**
[image]

**My post calibration flow and pressure**
[image]
[image]

Is anybody else seeing similar?

KirikauKiwi, 2024-11-11:
Secondly with the leak test offset the PRef "Depression" value should also be zero'd. Using the pictures below, to then get a 28 inH2O change in "Depression" after the leak test I would have to increase my depression from -8 to -36, if it was zero'd with the leak test calibration I can then easily get to my testing "Depression".

**Pre leak test calibration**
[image]

**Post leak test calibration**
[image]

DeeEmm, 2024-11-11:
> I am seeing an incorrect value being generated for the calibration offset, I am using the default 14.4 CFM and 10 inH2O from the calibration orifice settings no other offsets applied.

This seems to be the same issue that I found on your PR #214. Are you still using your modified code?

> Secondly with the leak test offset, the PRef "Depression" value should also be zero'd.
> Using the pictures below, to then get a 28 inH2O change in "Depression" after the leak test I would have to increase my depression from -8 to -36, if it was zero'd with the leak test calibration I can then easily get to my testing "Depression" with out having to think.

Same thing applies here. This looks like it could be related to the issue I saw with your PR.

Additionally there is also an issue with terminology. At some point (not sure when) it was requested to change the description from 'pRef' to 'depression'. The issue with that change is that we have now created this 'double negative' issue.

'depression' by definition is 'vacuum'.

Originally the description was Ref Pressure (based on pRef) i.e. pressure.

So we have changed from a -ve value meaning a negative pressure i.e. vacuum, to a -ve value meaning a negative vacuum i.e. pressure - a double negative

All of the underlying code is based on +/- kPa values (pRef), and as with the PR #214 request, there are a lot of underlying code changes that have to happen to be able to flip the sign on the GUI. 

The entire code philosophy is now based on SI units (kPa). So we do all calculations based on +/-kPa, DegC, hPa, etc. not depression in inches of water and cubic feet. We just convert the end units for the GUI to give us CFM and inH20. Originally the code was written using imperial maths - as it was based on feet and inches, and TBH it was a nasty mess and hard to follow. Thank heavens for metrification lol.

I would advise in the first case to make sure that you are running the standard unmodified code as the symptoms appear to be in line with the issues that I saw in your PR. Then if you are still seeing an issue we can take a deeper look.

This issue actually has some further value as I also note that a few of the other issues being reported are as a result of using an out of date index.html so I think that I need to add versioning for the generated index.html file.

KirikauKiwi, 2024-11-12:
I not using the formula I suggested in PR #214, this is with the DIYFB STD formula for Flow calibration and leak offsets.
```
// Apply Flow calibration and leak offsets
sensorVal.FlowCFM = sensorVal.FlowCFMraw  - calVal.leak_cal_baseline - calVal.leak_cal_offset - calVal.flow_offset;
```
I did at first have problems with the index.html, I will ask about the index.html in another post to keep them separate.
The formula is working in line with the expected results so I am certain it has reverted back to the DIYFB STD.

Yes understand what you say about the negative values and the depression heading, being a double negative!!!!
Personally I would prefer the  GUI "Title" to say something like Pressure 1 and Pressure 2 as I think this will help keep the code less confusing when reverse flow is implemented etc as the code for the tasks PRef and PDiff are used for, invert PRef will be doing the job of PDiff and vice versa, possible changing of the labelling of the sensors on the PCB too.
- I think currently the "Title" should match what is on the PCB currently.
Need to consider that in V3 the additional Temp sensor needed for reverse flow and its naming too, possibly a new naming convention of example T1, P1 and T2, P2. Temp and Pressure at location 1 & location 2.

Yeah, in agreement about SI units, I think they must of been thought up by someone with a bit of common sense 👍

KirikauKiwi, 2024-11-12:
Thinking forward a little too in relation to reverse flow. What we see currently as a leak is a gain in CFM, in reverse flow the leak will be a loss in CFM and in theory should be the same absolute value., but in reality could be slightly different due to the reverse flow testing will create a positive change for the reverse flow "increased Baro Pressure", where as the corrections for the normal flow direction see a "decreased Baro Pressure"
**From Calculations.cpp**
```
  // TODO validate reference pressure adjustment - do we add it or subtract it? Should be baro pressure less vac amount
  double refPressurePascals = sensorVal.BaroPA - this->convertPressure(sensorVal.PRefKPA, PASCALS) ;
```

DeeEmm, 2024-11-12:
A lot of valid commentary here. I'm breaking this out into a seperate issue - #230 - Standard Terminology

KirikauKiwi, 2024-11-12:
OK hands up 🙌, I had shot myself in the foot, only partially though. I had not rolled back all the formulas.
Which I have now done, so need a fact check as something is still happening with my calibration offset.

***First fact check please***
This is what is have reverted back to.
**DIY-Flo-Bench.ino**
```
        // Apply Flow calibration and leak offsets
        sensorVal.FlowCFM = sensorVal.FlowCFMraw  - calVal.leak_cal_baseline - calVal.leak_cal_offset  - calVal.flow_offset;

        // Create Flow differential values
        switch (sensorVal.FDiffType) {

        case USERTARGET:
          sensorVal.FDiff = sensorVal.FlowCFM - calVal.user_offset;
          strcpy(sensorVal.FDiffTypeDesc, "User Target (cfm)");
          break;

        case BASELINE:
          sensorVal.FDiff = sensorVal.FlowCFMraw - calVal.flow_offset - calVal.leak_cal_baseline;
          strcpy(sensorVal.FDiffTypeDesc, "Baseline (cfm)");
          break;
        
        case BASELINE_LEAK :
          sensorVal.FDiff = sensorVal.FlowCFMraw - calVal.flow_offset - calVal.leak_cal_baseline - calVal.leak_cal_offset;
          strcpy(sensorVal.FDiffTypeDesc, "Offset (cfm)");
          break;
                
        default:
          break;
        }
```
The leak calibration offset is working correctly, however when I do a calibration offset is is picking up a value higher than expected and it is also a negative, so with it being subtracted from the value we get the double negative and it is adding it to the raw value. This is with no other offsets applied.

**Flow Calibration**
https://github.com/user-attachments/assets/9c422c9a-1106-46f5-9aad-9ffffc37c76c

KirikauKiwi, 2024-11-12:
Now that my calibrations are being saved and applying the offsets correctly I have noticed something else that I would like to ask about too.

Without any calibration or offsets applied I run my theoretical bench and have a leak of 8.3 ACFM and a MAF Flow reading of 17.5 KGH both on the GUI, I do the automatic leak test on DIYFB and it now the respective flow figures are 0.00 ACFM and 0.00 KGH as would be expected, I then start my testing and increase the flow so the ACFM reading increases to say 2.5 ACFM now when I go to see the MAF Flow it is displaying 22.9 KGH.

At my weather conditions this is what I get converting it manually.
2.5 CFM of air under the given conditions (1030 hPa, 22°C, RH = 55.5%) is approximately 4.93 kg/h.

So it looks like the KGH is not being converted when either a Calibration or Leak offset is applied, or the GUI is displaying the actually RAW MAF sensor KGH

@DeeEmm can you confirm this and does it have any affect on the other calculation in DIYFB whilst a calibration and/or any other offset is applied, be it Leak, or User Target.

KirikauKiwi, 2024-11-12:
Regarding my PR #232 , I am running the master code and I am still having problems with the Calibration Flow Test, I have added a message to the PR, that hopefully better explains what is happening when I try to do the Calibration Flow test.

DeeEmm, 2024-11-13:
So what are you seeing in the GUI when you calibrate the flow?

- Disable minimum flow / pressure thresholds to display all flow values
- Reset Flow offset - note the uncorrected ACFM value
- Perform flow calibration - note value of Flow offset + corrected ACFM value

KirikauKiwi, 2024-11-13:
So with all offsets and calibrations set to zero and the minimums set to zero.

- The uncorrected ACFM value is 2.1 CFM
- Calibration flow offset -12 and the corrected ACFM is 14.4 CFM

This is with the default calibration orifice CFM of 14.4

All explained above and in the updated message in the PR #232

KirikauKiwi, 2024-11-13:
Please accept my apologies for wasting your time, I shut everything down did a clean, build and upload and then rebooted the ESP32. the Default DIYFB is working and I can no longer repeat what I was had stated above.

Sorry again for the wild goose chase 🤦‍♂️

KirikauKiwi, 2024-11-13:
I was starting to go a little crazy 😜, I can now walk away with the problem unknowingly and unwittingly resolved with a bit of egg on my face.

closed 2024-11-13 by KirikauKiwi

DeeEmm, 2024-11-13:
All good. No shame at all in raising an issue where you find one. Please don't stop. I can appreciate that these kind of issues can become massively frustrating.

The way I see it is that it is much better to raise an issue that ends up to be a non-issue, than not raise an issue that ends up significant.

I'm hoping that the versioning for the index.html file will go a long way towards helping diagnosis of problems such as these. We only have a handful of users at the moment and versioning has already been a contributing factor to several issues. So adding versioning is hopefully a good first level sanity check.

KirikauKiwi, 2024-11-13:
Cheers, Yeah there was a bit of flawed thinking going on too.