DIY-Flow-Bench/DIY-Flow-Bench #226: Stepper Driver Testing

DeeEmm, 2024-11-11:
The last remaining outstanding test on the PCB is the stepper drivers.

Need to undertake a test to validate the stepper drivers.

Test does not have to be related to DIYFB project. It can simply consist of arbitrary code to test PCB functionality

DeeEmm, 2024-11-17:
OK so spent this evening testing steppers.

Have unfortunately found the first issue with the PCB. Not entirely sure how, but the direction pin for the flow control stepper 
(FLOW_VALVE_DIR_PIN) is assigned to a pin that can only only used for an input (35).

It also seems that the pin mapping differs from the physical layout as there are a few discrepancies, so I suspect that this has happened in the PCB design stage.

A workaround is to jumper a link from the spare output (4), which just happens to be the pin next to it. 

I have amended the pin map to suit.

closed 2024-11-17 by DeeEmm

stefan63, 2024-11-24:
Found this regarding pin35
https://www.esp32.com/viewtopic.php?t=26008

DeeEmm, 2024-11-24:
That's interesting. Will have to check was versions the boards I have are.