DIY-Flow-Bench/DIY-Flow-Bench #228: [FEATURE REQUEST] index.html versioning

DeeEmm, 2024-11-12:
Add versioning to index.html file

Versioning to follow and utilise existing build numbers

Version number to be updated as part of GULP task.

DeeEmm, 2024-11-14:
OK. So I have the index.html versioning working.

This works as a gulp task and replaces a template marker within the HTML with the version number when the file is processed by gulp (compressed / minified / etc). 

The resultant 'compiled' index.html file will have the version number 'hardcoded' within it. The only time that this gets overwritten is when the file is recompiled again from source by running the gulp task

The version number itself is pulled from an external file - in this case version.json. I elected to use JSON as it fitted with other aspects of the project and meant that we could use the existing gulp processes to pull in the JSON data without having to use an additional libraries. 

Using JSON meant having to partially revisit the post compile stuff that I worked on as part of #205, which also meant that I could pull in the code from that branch saving a bunch of work, whilst also addressing some of the shortfalls that I previously noted.

What this now means is that moving other config files from pre-compile to upload-able JSON files like mafData and configuration.h is now possible. I have actually already moved pin definitions over as part of the current changes as well a getting rid of the external language file. The plan with the language file is to be able to upload an override file, otherwise it just pulls the default language values from structs.

The one thing that really needs to happen next is to implement a similar strategy for the main C++ compiler. This way all versioning can then be managed from within the version.json file

I have a little tidying to do and then I will publish the code.

-------------------------------

There are a bunch of follow on changes that can now be implemented...

- Remote callback to version.json to check if DIYFB is up to date (both firmware and GUI)
- Automated versioning for main DIYFB code
- Move configuration.h into configuration.json #195
- Move MAFdata.h into MAFdata.json #108
- Create languge.json override

closed 2024-11-14 by DeeEmm

DeeEmm, 2024-11-14:
OK. So before moving on to other stuff, I need to update the documentation to support these changes, as well as create new pin files for the other board variants.

To be able to compile this code there are also some changes to the platformio environment, so it is important for this to be captured.

I have also spent some time pondering if I should just complete the remaining changes to move MAF and Config to post compile. TBH the hard part was getting the PINs to work, which has now been resolved, so the changes to MAF and config are relatively trivial.

What this does is it then allows the GUI firmware update to work properly as the compiled binary will be board agnostic. Existing pindefs and configs will be retained across updates. So the provision of the complied binary and index file is all that will be needed. This is a natural progression of the index.html versioning and further narrows down the possibility of mismatched updates.

I know that this might seem like a bit of a detour from the current project path, but I suspect that what I spend in time implementing it I will save tenfold in support.

KirikauKiwi, 2024-11-14:
I have setup a DEV branch on my Git and have had a look at the new stuff,

I have created the pinout.json for the M5 Stack Core2 and also the other board I am working with, the ESPRESSIF ESP32S3-DEV-KIT-C, and  currently having a little error when I try to upload the board specific .json

This is from the serial monitor after I attempt to upload the pins.json
```
Waiting...
[ 58594][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/spiffs/ESPRESSIF_ESP32_S3_DEV_KITC.json) failed
[222777][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/spiffs/ESPRESSIF_ESP32_S3_DEV_KITC.json) failed
```

Looks like I jumped the gun, got ahead of the ball, but had seen the post 8 hours ago and after dinner tonight got in to it and have come back here to see your latest post about updating the documentation and the new pin files.

Should I wait for what you mention above to be done, or would you like me to do a Pull Request so you get the small bit of work I have done on the json pin file etc.

DeeEmm, 2024-11-14:
Sorry I have pins here, just haven't pushed the changes yet. Only just got back into it. 

```
{
    "BOARD_TYPE" : "M5STACK-CORE2",
    "VAC_SPEED_PIN" : -1, 
    "VAC_BLEED_VALVE_PIN" : -1,
    "VAC_BANK_1_PIN" : 19,
    "VAC_BANK_2_PIN" :-1, 
    "VAC_BANK_3_PIN" :-1,
    "AVO_STEP_PIN" :  -1,
    "AVO_DIR_PIN" : -1,
    "FLOW_VALVE_STEP_PIN" : -1,
    "FLOW_VALVE_DIR_PIN" : -1,
    "VCC_3V3_PIN" : -1,
    "VCC_5V_PIN" : 35,
    "SPEED_SENSOR_PIN" :-1,
    "SWIRL_ENCODER_PIN_A" : -1,
    "SWIRL_ENCODER_PIN_B" : -1,
    "ORIFICE_BCD_BIT1_PIN" : -1,
    "ORIFICE_BCD_BIT2_PIN" : -1,
    "ORIFICE_BCD_BIT3_PIN" : -1,
    "MAF_PIN" : -1,
    "REF_PRESSURE_PIN" : -1,
    "DIFF_PRESSURE_PIN" : -1,
    "PITOT_PIN" : -1,
    "TEMPERATURE_PIN" : -1,
    "REF_BARO_PIN" : -1,
    "HUMIDITY_PIN" : -1,
    "SERIAL0_TX_PIN" :1,
    "SERIAL0_RX_PIN" :3,
    "SERIAL2_TX_PIN" :-1,
    "SERIAL2_RX_PIN" :-1,
    "SDA_PIN" : 32,
    "SCL_PIN" : 33,
    "SD_CS_PIN" : 4,
    "SD_MOSI_PIN" : 23,
    "SD_MISO_PIN" : 38,             
    "SD_SCK_PIN" : 18,
    "WEMOS_SPARE_PIN_1": -1
}

```

Not 100% on the error. But I think it might be an error in your platformIO.ini as I had a similar error here.

Take a look at any build flags where  you are defining  overrides and check that there are no spaces

```
	-D LAST_BUILD_TIME=$UNIX_TIME

        # not

	-D LAST_BUILD_TIME = $UNIX_TIME

```

I'm guessing that you are using a custom platformIo.ini. I've actually tidied it up and refactored it a little, so you might need to either pull the changes in to your local version or add any mods you've done into the new version. The changes were part of the requirements for updating the BUILD_NUMBER / VERSION variables in the main code.

You will note that there is a python script called user-actions.py called at compile time. This pulls in the version data from version.json and adds the variables to the build environment so that they replace the variables at compile time.

We check that this has happened at the bottom of system.h which is the file that system configuration precompile directives now reside in. For example...

```
#ifndef MAJOR_VERSION 
#error MAJOR_VERSION UNDEFINED
#endif
...
```
Will throw a compiler error if the variables are not present in the environment.
[... 97 lines cut ...]
                if (Serial.available() > 0) {
                    status.serialData = Serial.read();
                    _api.ParseMessage(status.serialData);
                }
            }                            
        }

        if (SPIFFS.exists("/pins.json") && SPIFFS.exists("/index.html")) doLoop = false;

        vTaskDelay( 1 );
    
    } while (doLoop = true);

    tempServer->reset();

}
```

Once the files are received, it should then exit the boot loop.

I've left the serial API calls in the loop as it can be useful for debugging, however some of the commands will cause the ECU to crash as they are not initialised at this point in the code.

I have found that sometimes when uploading the files it seems to get stuck in the loop, even after the files are uploaded but rebooting the ECU then allows it to boot up normally. It is trivial to force a reboot at this point but I wanted to try and avoid that as then the WiFi needs to be reconnected, which if the end user is doing manually can be a bit frustrating.

Hopefully that gives you a bit of a head start on the new code flow

DeeEmm, 2024-11-15:
^^^ a lot of this stuff is what I'm now trying to capture in the documentation. 😄

KirikauKiwi, 2024-11-15:
Impressed with all the code changes/ refactoring that you have in this update.

DeeEmm, 2024-11-15:
Thanks 😄 

Ahh yes, forgot to mention, you will need to rename your pins file to pins.json, otherwise it will not get picked up. Ideally I would like to prefix or suffix the JSON files and then use a wildcard or regex search to ignore the rest of the filename. This way the files would not need to be renamed. But for the time being it needs to be pins.json when it is uploaded.

Hmnnn maybe we could just change the extension to 'pins'. Might be easier to decode. 🤔

KirikauKiwi, 2024-11-15:
Yes, renaming the file to pins.json, seems obvious now when that it states the file names on the upload GUI,
I did however learn that the error that I posted above was due to the file name being too long, SPIFFS has a max file name of 32 characters.

Yes it would great to be able to use a logical and obvious name for the file, I have always been a fan of it does what it says on the can!

One other question is in side the .json file the "BOARD_TYPE" do we have to use, the "#ifdef" name or the '#define BOARD_TYPE' name. 
```
#ifdef ESPRESSIF_ESP32_S3_DEV_KITC
#define BOARD_TYPE              "esp32-s3-devkitc-1"
```

I have tried with both now after renaming the file to pins.json and I am now getting similar to what I had with the undefined pins, which -1 fixed, it appears to be back with this update.
```
Defaulting to AP Mode 
Creating WiFi Access Point:  DIYFB  
Access Point IP address: 192.168.4.1 
Multicast: http://diyfb.local 
[   438][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   447][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   453][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   459][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   465][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   471][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   478][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[   484][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
```

This is my pins.json, they are set the same as they were before the changing to DEV.
```
{
    "BOARD_TYPE" : "esp32-s3-devkitc-1",
    "VAC_SPEED_PIN" : -1, 
    "VAC_BLEED_VALVE_PIN" : -1,
    "VAC_BANK_1_PIN" : 19,
    "VAC_BANK_2_PIN" : -1, 
    "VAC_BANK_3_PIN" : -1,
    "AVO_STEP_PIN" : -1,
    "AVO_DIR_PIN" : -1,
    "FLOW_VALVE_STEP_PIN" : -1,
    "FLOW_VALVE_DIR_PIN" : -1,
    "VCC_3V3_PIN" : -1,
    "VCC_5V_PIN" : 1,
    "SPEED_SENSOR_PIN" : -1,
    "SWIRL_ENCODER_PIN_A" : -1,
    "SWIRL_ENCODER_PIN_B" : -1,
    "ORIFICE_BCD_BIT1_PIN" : -1,
    "ORIFICE_BCD_BIT2_PIN" : -1,
    "ORIFICE_BCD_BIT3_PIN" : -1,
    "MAF_PIN" : -1,
    "REF_PRESSURE_PIN" : -1,
    "DIFF_PRESSURE_PIN" : -1,
    "PITOT_PIN" : -1,
    "TEMPERATURE_PIN" : -1,
    "REF_BARO_PIN" : -1,
    "HUMIDITY_PIN" : -1,
    "SERIAL0_TX_PIN" : 43,
    "SERIAL0_RX_PIN" : 44,
    "SERIAL2_TX_PIN" : -1,
    "SERIAL2_RX_PIN" : -1,
    "SDA_PIN" : 8,
    "SCL_PIN" : 9,
    "SD_CS_PIN" : 4,
    "SD_MOSI_PIN" : 23,
    "SD_MISO_PIN" : 38,             
    "SD_SCK_PIN" : 18
}
```
Where do the PIN numbers from the pins.json file end up next to set the pins, with this new method?

DeeEmm, 2024-11-15:
The pins are now assigned in hardware...
```
Hardware::assignIO (JsonObject pinData)
```

We initially pull the JSON data into a struct, the same as we did before with other variables. Using extern structs allows us to easily pass data between methods without having to create convoluted methods to pass by value or reference. We can make entire variable collections available within methods by simply including the reference to the original struct.

```
  extern struct Pins pins;
  extern struct DeviceStatus status;

  status.boardType = pinData["BOARD_TYPE"].as<String>();

  // Store input pin values in struct
  pins.VCC_3V3_PIN = pinData["VCC_3V3_PIN"].as<int>();
  pins.VCC_5V_PIN = pinData["VCC_5V_PIN"].as<int>();
  pins.SPEED_SENSOR_PIN = pinData["SPEED_SENSOR_PIN"].as<int>();
...etc
```

Then further down we assign the I/O...

```
 pinMode(pins.VCC_3V3_PIN, INPUT);   
  pinMode(pins.VCC_5V_PIN, INPUT);   
  pinMode(pins.SPEED_SENSOR_PIN, INPUT);   
  pinMode(pins.ORIFICE_BCD_BIT1_PIN, INPUT);   
  pinMode(pins.ORIFICE_BCD_BIT2_PIN, INPUT);   
```

DeeEmm, 2024-11-15:
I've included the board type in the pins file as it seems the most logical place.

```
status.boardType = pinData["BOARD_TYPE"].as<String>();
```

JSON also does not support comments, so it provides a way to identify what type of board the file is for.

DeeEmm, 2024-11-15:
So doing the same for the configuration and MAF files is mostly just a case of typing out all of the import statements.

Although I till need to work out how to process arrays in JSON for the MAF data

DeeEmm, 2024-11-15:
@KirikauKiwi I note that the BOARD_NAME you are using is "esp32-s3-devkitc-1". This is (at this stage) just a textual descriptor for user ID purposes. It's not actually used anywhere in the code other than to display in the GUI

Would 'M5STACK-CORE2' be more appropriate and easier to recognise?

KirikauKiwi, 2024-11-15:
They are two different boards, the long winded name is an ESPRESSIF dev board, that is it full name 😂, Since the pins are noe defined externally the PlatformIO board type no longer is need I assume, So I will most likely use a more user friendly name. I had the board sitting around and I am using to stay more in line with the Main DIYFB build currently to better assist in testing etc.

Having some interesting times with the pins with how they are now defined.
So the main problem I am experiencing is that three of the pins that are showing up as invalid are the pin definitions for the BME, So Temp, Baro and RH%>

Do you actually need to define a single pin to an I2C device (unless it has a physical interrupt) as you should only need to define the two pins for the I2C bus, Logically it doesn't make sense to me define a pin to a device that actually needs two pins which are defined already as the I2C bus pins, and since you can have 128 devices on the bus you would run out of pins pretty quickly if they needed to be assigned for each device.
The instance of the device is initialised with the I2CDevice.begin() is it not, so shouldn't need to be assigned to a single pinout.???

DeeEmm, 2024-11-15:
The single pin definitions are unused if the I2C is used. In that case the maf / baro / etc are connected either via the ADS or come from the BME.

If for some reason you did not want to use the BME or ADS, then you could connect directly to the ESP32 via the single pins that are defined.

So it's an either-or situation. You use one or the other

TBH early on we did not really have a standardised setup, so people were using all kinds of sensors connected directly to the ESP. Over time we've found that the best solution by far is to use the BME and ADS connected via I2C as they have far superior resolution when compared to the 12bit resolution of the ESP's inputs. 

So it's kind of a legacy situation retained to cater for those who might like to do things a little differently 🤪 

~The actual pin initialisation is the same as previously, even using the same values, so it should not act any differently.~

I now remember that you had to test with #ifdefs to determined if the pins were used.

I guess we need a way to replicate this behaviour now that the we have moved to post compile I/O handling

It would be easy enough to test for the -1 value in each case, but that's a whole lot of if then elses.

I will have a ponder to see if I can come up with something a little more elegant.

DeeEmm, 2024-11-15:
So maybe a key=>value array where 'key' is a uint but has a constant definition in the constants.h file

i.e. 

```
int 'VCC_3V3_PIN' = 1;
```

Then the array can be iterated in a for loop using the count to index through the key values.

Each value is then tested. If it has a value greater than or equal to zero it gets assigned, if not it gets ignored.

DeeEmm, 2024-11-15:
Or we can just iterate thought it using the pointer.

KirikauKiwi, 2024-11-15:
I think the SDA/SCL pins configs will be fine, it is more the pin configs that are being set for the following.
```
"MAF_PIN" : -1,
"REF_PRESSURE_PIN" : -1,
"DIFF_PRESSURE_PIN" : -1,
"PITOT_PIN" : -1,
"TEMPERATURE_PIN" : -1,
"REF_BARO_PIN" : -1,
 "HUMIDITY_PIN" : -1,
```
Through the process of elimination the four instances of the invalid pins I have narrowed it down to,
```
"PITOT_PIN" : -1,
"TEMPERATURE_PIN" : -1,
"REF_BARO_PIN" : -1,
 "HUMIDITY_PIN" : -1
```
Overall it appears to be the pinouts that are being defined for I2C devices. that are being problematic.

DeeEmm, 2024-11-15:
Have you tried 

> Overall it appears to be the pinouts that are being defined for I2C devices. that are being problematic.

The easy test is just comment those pin initialisations in hardware.cpp

```
 // pins.REF_PRESSURE_PIN = pinData["PREF_SRC_PIN"].as<int>();
  // pins.DIFF_PRESSURE_PIN = pinData["PDIFF_SRC_IS_PIN"].as<int>();
  // pins.PITOT_PIN = pinData["PITOT_SRC_IS_PIN"].as<int>();
  // pins.TEMPERATURE_PIN = pinData["TEMPERATURE_PIN"].as<int>();
  // pins.HUMIDITY_PIN = pinData["HUMIDITY_PIN"].as<int>();
  // pins.REF_BARO_PIN = pinData["REF_BARO_PIN"].as<int>();
```

DeeEmm, 2024-11-15:
oops i mean the initalisation (it's late here)

```
pinMode(pins.MAF_PIN, INPUT);   
  pinMode(pins.REF_PRESSURE_PIN, INPUT);   
  pinMode(pins.DIFF_PRESSURE_PIN, INPUT);   
  pinMode(pins.PITOT_PIN, INPUT);   
  pinMode(pins.TEMPERATURE_PIN, INPUT);   
  pinMode(pins.HUMIDITY_PIN, INPUT);   
  
```

DeeEmm, 2024-11-16:
OK so in latest change to DEV branch I've added a simple check to the pin assignment. If you get a chance to test, let me know if it addresses your pin configuration issues

KirikauKiwi, 2024-11-16:
Quick update, still trying to get to the bottom of this!!!

Hands up another dumb moment whilst burning the midnight oil in the small hours, SO obvious with some sleep 🤦‍♂️ about either using analog or I2C sensors for the Environmental sensors. I got my wires crossed about this because I wasn't thinking correctly about the way the problem I am having is presenting it self!!!!!

So my problems I am looking in to are either related to;
- ~The ESP32 S3 boards have additional pins, up to 45 available compared to the 39 available with the ESP32 boards. Which isn't a problem any longer as the Arduino framework was updated not long after the S3 variant boards came out and the boards are supported in PlatformIO.~
- The pinouts I have defined for my the DEV Kit C board I am using, any pin can be defined as a GPIO due to the way its matrix works,  different to the ESP32 boards. There are a few dedicated pins if you want to us USB +D and-D etc, but these can still be used GPIO pins if not require for the specific feature. 
- Layer 1, user problems (Get some sleep!!!!)

I have fixed the layer 1 problem, and update to the this DIYFB version/build,
```
"MAJOR_VERSION" : "V2",
"MINOR_VERSION" : "0",
"BUILD_NUMBER" : "24111601",
"GUI_BUILD_NUMBER" : "24111401",
"RELEASE" : "V2.0.RC.8",
```
**I am all back up and working** with the ESP32 S3 DevKitC-1 board, while researching the problem I was having with the invalid pins, I found some interesting information about setting the pins manually via pinMode, the interaction with Serial.begin, Wire.begin and SPI.begin and board default pin configurations which I will put in a pull request.

KirikauKiwi, 2024-11-16:
One thing that I wasn't sure of initially was when doing the upload and monitor there is an noticeably longer time period passes before you see any output on the serial monitor for the DEV compared to the current master, when it is building the file system etc.
Is it possible to pop a message up before this, as the first couple of time I was thinking that something had stalled.

Now that the process gets past the pin hurdle I was having, the other information after that point showing the status of the file uploading on the serial monitor I found really useful.

Then once the two files are loaded is it possible to print to the serial monitor to "Rest the board" or is this something you can automate with a serial RTS command, then Followed by a "Connect your computer to the DIYFB WiFi" upon the reboot. I think this will help guide other end users through the process of getting DIYFB up and running/connected.

DeeEmm, 2024-11-17:
Great news. 👍 

> * The pinouts I have defined for my the DEV Kit C board I am using, any pin can be defined as a GPIO due to the way its matrix works,  different to the ESP32 boards. There are a few dedicated pins [...]

This is actually true of the ESP32 as well, each pin has multiple potential assignments as can be seen from the WEMOS schematic

[image]

However, there are some functions that are tied to specific pins, such as the ADC's and comms such as SCL, SDA, MISO, MOSI etc as the rely on hardware componentry that is tied to those physical pins.

In a way this is a similar issue that you face as the M5Stack has additional componentry attached to physical pins (screen etc), which I suspect means that those pins may be hardcoded within the M5 library and not available to reassign.

> One thing that I wasn't sure of initially was when doing the upload and monitor there is an noticeably longer time period passes before you see any output on the serial monitor for the DEV compared to the current master, when it is building the file system etc.

I'm not seeing a delay in the serial monitor here. Whilst you cannot see the API call echoed to the screen, I'm sending '~' to the serial monitor straight after the first load...

https://github.com/user-attachments/assets/27126e05-c849-4ecd-bfb0-d613d95a94b4

The major change that could be causing a small delay is that now we are scanning the filesystem two additional times to look for wildcard file names for PINS_ and MAF_ files. However, we print to the serial monitor before this point

[image]

There is one other change that might be causing this...

Previously we included the pins definitions as part of the precompile. So they were already available within the code as hardcoded constants. This meant we knew what pins were assigned for Serial 1, so we could begin the serial with the pins assigned...

```
#if defined SERIAL0_ENABLED
Serial.begin(SERIAL0_BAUD, SERIAL_8N1 , SERIAL0_RX_PIN, SERIAL0_TX_PIN);
```
Now that we have moved the pin definition to post compile, the pins are not available until after we have read the PINS_.json file. However, in order to be able to report to the serial monitor. We can still issue a serial begin WITHOUT defining the pins. This will fall back to the default pins for the board. i.e. the USB connection.

This way we are able to start reporting to the serial monitor much earlier in the code. we now do this at the top of DataHandler.begin it is now one of the first things we do...

```
void DataHandler::begin() {

    extern struct ConfigSettings config;
    extern struct Language language;
    extern struct DeviceStatus status;

    Hardware _hardware;
    Pins pins;
    Messages _message;
    Comms _comms;

    StaticJsonDocument<1024> pinData;
    // StaticJsonDocument<1024> mafData;

    // Need to set up the data environment...

    // Start serial comms
    this->beginSerial(); 

    _message.serialPrintf("\r\nDIY Flow Bench");                                         
    // _message.serialPrintf("\nDIYFB Version: %s \nBuild: %s \nGUI: %s \n", RELEASE, BUILD_NUMBER, GUI_BUILD_NUMBER);                                         
    _message.serialPrintf("\nDIYFB Version: %s \nBuild: %s \n", RELEASE, BUILD_NUMBER);                                         

```

I wonder if that is somehow causing a delay in the M5 library??

DeeEmm, 2024-11-17:
Sorry missed this part...

> Then once the two files are uploaded is it possible to print to the serial monitor to "Rest the board" or is this something you can automate with a serial RTS command, then Followed by a "Connect your computer to the WiFI called DIYFB and then type diyfb.local in to your web browser " upon the reboot.Trying to think how else to guide the user through the process. As most people won't be familiar with browsing to a local add

It is trivial to do a reboot, however once the files are loaded it should automatically redirect to the main page.  I'm trying to avoid forcing a reboot as it disconnects the user from WiFi, and depending on how you have set up your WiFi connection, this can be a PITA (i.e. no auto re-connect + manually typing credentials every time). The main consideration is that it is also possible to get to the upload page after you have booted and are running. This is the case if you delete an existing file to update it, for example as part of an update.

There is a balance between how much info to include in the serial monitor and how much info to keep in the WIKI. On the one hand if you are viewing info via serial monitor and compiling code, we can assume a minimum level of competence. But on the other this may be the first experience for some users. I think that adding a link to the WIKI might be the best bang for buck here and then improving the step-by-step instructions

The serial commentary is predominantly to give a basic idea of where the code execution is up to, so that if it fails we can ask,  '_what was the last thing printed to the serial monitor before it crashed'_  this will then give a really good indication of what the issue is, or at least in what process the code was running when it fell over. For the most part there is a comment at the beginning an end of each process.

To a degree there is an assumption that to get to the point where you are connecting via serial monitor, then either you know what you are doing, or you're reading the documentation. 

The 'Hand Holding', really starts in the GUI after the user has connected by Wifi with the 'Please upload the following files". This will detect which files are missing and ask for them, This could be easily expanded with additional prompts on what to do for each file.

For example we could add something like this for the pins file...

>Upload the PINS.JSON file for your board. This can be found in the PINS folder of the project.
>If you do not see a file for your board, please see the WIKI for instructions on how to create one.

We can then link directly to the relevant section.

The use case for the _'complete novice'_ is that the default configuration and hardware is used. This requires no configuration and apart from flashing the firmware is about as simple as it gets. I think that using a third party firmware loader, or even forking an existing loader is the way to go there. The loader can even be expanded to use the API to grab the Wifi credentials and give them to the user without having to open up or view a serial monitor.

The other option is to provide pre-flashed ECUs and completed shields. I.e. plug and play.

DeeEmm, 2024-11-17:
[image]

KirikauKiwi, 2024-11-17:
Currently working with the ESP32_S3 board which is not an M5 Stack product, so none of their libraries are being used currently.

Once the board is all setup and running the serial monitor is fine, I was referring to having a board with an empty flash and the first upload and monitor, it seems to take a while to build the SPIFFS I think.
Everything works, I wanted to report the observation as the pause had me starting to thing something had hung or stalled.

https://github.com/user-attachments/assets/36fc78e5-8259-4f9f-8743-2b948f3dc4c3

After the pause which is about 30 seconds at a guess this is output, and after then any resets, the monitor is fine.

[image]