DIY-Flow-Bench/DIY-Flow-Bench #231: [BUG] Wrong flow readings after an update to V2.0-RC.8, build:24111106.

black-top, 2024-11-12:
**Software Version**
- RELEASE (V2.0-RC.8,) 
- BUILD_NUMBER (build:24111106)

**Describe the bug**
Wrong flow readings after an update to V2.0-RC.8, build:24111106. 

More at https://github.com/DeeEmm/DIY-Flow-Bench/issues/220

> @DeeEmm Scaling looks like it is working ok, but the flow numbers are now are completely wrong. With the latest master. Same setup, same cylinder head, all lift curve now read about 10-12CFM, before at 13mm was about 300CFM. BTW the readings now are very stable; they do not fluctuate at all, but they are not even close to true. CD flow is also wrong now, after calibration now I get -3, before my value was +4, but after calibrating, the flow numbers are still wrong. All settings are the same as before. I tried to change settings to see what could be causing it, but don't see any significant effect from any changes at the settings.
> 
> Maybe something with MAF housing calculations is causing it now. I use original MAF housing, so my value always was 0.

black-top, 2024-11-12:
Today's flow readings are about 20 CFM, but it looks like they are stuck, all adjusted, non-adjusted do not change more than a few CFM. I can't find a setting that can cause it, probably something with MAF housing changes in the code.

stefan63, 2024-11-12:
Uploaded Build 24111107 and it worked with  Audi Bosch .......67   but still not with Range Rover ......10 only shows strange low numbers on flow with tha cpp file.

KirikauKiwi, 2024-11-12:
@black-top the MAF Housing calculations are ignored if the value is either 0 or is the same as the internal diameter of the original MAF housing as noted is in the mafData cpp file.
So for the Bosch sensor if 0 or 82 is entered in the GUI config page the calculations are ignored, if any other value is entered the calculations will be processed.

DeeEmm, 2024-11-12:
Is the issue related to the scaling? Can we confirm this.

Also if there was a version that works, can you let me know the build number so that I can do a code compare.

I suspect that an old change may have been pulled in

black-top, 2024-11-13:
Don't know the exact build number, but the master that was working OK was downloaded on November 1.

black-top, 2024-11-13:
I don't see any significant effect on flow readings by changing all settings, especially new ones. So I dont know what is causing it, I just uploaded the (build:24111106) to test the scaling function.

DeeEmm, 2024-11-13:
Ok just so that I understand.

1. The issue is that the flow readings are showing 10-12CFM, but were previously showing 300cfm which is what you would expect from the head being measured.

>I don't see any significant effect on flow readings by changing any settings

2. What settings are you changing and what change are you expecting to see

3. Also, Is this with a scaled MAF housing?

black-top, 2024-11-13:
1: The first day when I tested the new code, flow numbers were stuck at about 10-12CFM. The next day I did some more testing, and the flow was stuck at about 20CFM. At full lift, the head was flowing before the update, about 300CFM, now it just stuck at same number at all lifts and changes only about 1CFM.  The CD flow calibration also does not work anymore, after new flow calibrations, readings are still wrong.  Today I didn't done any more testing. 

2: I was just changing all the settings to see if any setting gave effect to the wrong readings to find the problem source. 

3 I use the original MAF housing, my scaling was always  0.

black-top, 2024-11-13:
So I uploaded the ( build:24111106) to the spare esp32 board and tested it now, looks like it's working OK. I will try later to do more testing and to rewrite the code to the old esp32 board and test what it is doing.

DeeEmm, 2024-11-13:
> So I uploaded the ( build:24111106) to the spare esp32 board and tested it now, looks like it's working OK. I will try later to do more testing and to rewrite the code to the old esp32 board and test what it is doing.

That's great news. happy to hear that it is working again.

Let me know how you get on

closed 2024-11-22 by DeeEmm