DIY-Flow-Bench/DIY-Flow-Bench #233: BME68X Pressure sensor integration

DeeEmm, 2024-11-12:
BME68X incorporated in to the code please.

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/discussions/189#discussioncomment-11230280_

DeeEmm, 2024-11-18:
Started to take a quick peek at this. 

First hurdle is always trying to find a good library.

Adafruit is most common, but I will not use their libraries in a project like this as it requires an additional framework. (bloatware)

DFRobot Library is also very bloaty

I previously tried using the Bosch Sensortechnic library for the BME280 but it had a lot of issues, so I am apprehensive about using it again

The author of the TinyBME library that is used currently, does not have an equivalent BME680 library, which is a shame as it is nice and cut down.

The Zanduino library looks fairly lightweight 
https://github.com/Zanduino/BME680/tree/master but is written for Arduino so may need porting

Similar story for Seeed, although this still needs trimming down as well 
https://github.com/Seeed-Studio/Seeed_Arduino_BME68x/tree/master

I've forked the Zanduino library and will see if I can get it working as it is the most lightweight. 

https://github.com/DeeEmm/BME680

DeeEmm, 2024-11-18:
OK, also found the following library which specifically stated it works with the Wemos D1
https://github.com/malokhvii-eduard/arduino-bme280/tree/master

KirikauKiwi, 2024-11-19:
I have my branch of the Pitot/BME68X setup, after uploading my pins.json and index.html, then rebooting. The serial monitor is hanging near the end of the "INPUT"
```
DIY Flow Bench
DIYFB Version: V2.0.RC.8 
Build: 24111801 
For help please visit the WIKI:
https://github.com/DeeEmm/DIY-Flow-Bench/wiki
Initialising File System 
Loading Configuration 
Loading Lift Data 
Loading Calibration Data 
PINS file Found: /PINS_ESP32_S3_DEV_KITC.json
Loading Pins Data 
Parsing Pins Data 
Defaulting to AP Mode 
Creating WiFi Access Point:  DIYFB  
Access Point IP address: 192.168.4.1 
Multicast: http://diyfb.local 
Initialising I/O 
Input MAF_PIN: 0
Input REF_PRESSURE_PIN: 0
Input DIFF_PRESSURE_PIN: 0
Input PITOT_PIN: 0
Input SERIAL0_RX_PIN: 44
Input SDA_PIN: 8
Input SCL_PIN: 9
Input SD_MISO_PIN: 0
Input SD_SCK_
```
The SD card is not defined, and my pins.json for my ESP32 S3 DEVKITC board has the pins for the SD card as -1.

When performing a rest on the board the serial monitor hangs at the same place each time, other than the length of the last lines output differs by one or two characters. e.g. Input SD_SCK_, Input SD_SCK_PIN:, Input SD_SCK_PIN: 0

DeeEmm, 2024-11-19:
I eventually pulled the BME280 class files directly into the project so work on them locally, as updating the remote repo quickly became a PITA.

I managed to get the code to compile and loaded, but I do not get any output. I note that even if I manually send hardcoded values direct from classes 'getSensorData' method, I do not see those values in DIYFB. Kinda weird. I suspect that is it some kind of scope issue but haven't been able to find out what's happening.

I then tried to get the example working directly on the ESP32 without the DIYFB code, which should be pretty straightforwards, but was having a lot of dependency issues. 

At which point I parked it as it was taking the branch I was working on too far off topic. So I've merged this changes back into Dev so that I can now create a dedicated branch for this issue.

The branch contains all of the changes that I made to integrate the 680 into the code, you will just need to adjust the configuration.h file to enable the BME680 and set temp / baro / relh to BME680.

Rather than uncommenting the library call, I have found that it is easier to copy the DeeEmm_BME680.cpp/h files into the main project directory and work on them locally.

DeeEmm, 2024-11-19:
BTW. I updated the changes that I've made to the library which are mostly just code tidying so that it is readable (I mean seriously who minifies source code lol)

I was in the process of commenting out the API stuff to simplify the code, but did not finish, so this might (probably has) broken the code. 

Just uncomment the commented out SPI stuff which is easy enough to spot. 

Code is very similar to BME280. Each measurement requires getting values from a bunch of registers - these are defined at top of main file. Those register values are then 'decoded' to generate final values. Not particularly tricky but lots of number crunching so difficult to follow.

We can assume that the number crunching part is okay, it's just how the generated values are accessed from code external to the library.

It should be... 

- Create class instance
- Initialise Class / sensor data
- interrogate Class method to obtain values

It seems to be failing on the last part, but could also be any other part. Kinda hard to determine as there are no error messages.

Best bet might be to try and get the example working and then go from there, but I was having trouble setting up my environment to get the example to compile. I had a bunch of issues with project paths. (a local environment issue not necessarily a project issue).

All of those changes are pushed to the library, so feel free to hack on those to get it working, or if it works for you let me know how 😆

KirikauKiwi, 2024-11-20:
With this DEV branch, 
```
"MAJOR_VERSION" : "V2",
"MINOR_VERSION" : "0",
"BUILD_NUMBER" : "24111901",
"GUI_BUILD_NUMBER" : "24111702",
"RELEASE" : "V2.0.RC.8",
```

<details><summary>I am getting compile errors with the BME680 library.</summary>
```
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:82:1: error: extra qualification 'BME680_Class::' on member 'BME680_Class' [-fpermissive]
 BME680_Class::BME680_Class() {
 ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:82:1: error: 'BME680_Class::BME680_Class()' cannot be overloaded with 'BME680_Class::BME680_Class()'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:101:3: note: previous declaration 'BME680_Class::BME680_Class()'
   BME680_Class();                           // Class constructor (unused)
   ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:88:1: error: extra qualification 'BME680_Class::' on member 'BME680_Class' [-fpermissive]
 BME680_Class::~BME680_Class() {
 ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:88:1: error: 'BME680_Class::~BME680_Class()' cannot be overloaded with 'BME680_Class::~BME680_Class()'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:102:3: note: previous declaration 'BME680_Class::~BME680_Class()'
   ~BME680_Class();                          // Class destructor (unused)
   ^
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:97:6: error: extra qualification 'BME680_Class::' on member 'begin' [-fpermissive]
 bool BME680_Class::begin() {
      ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:97:6: error: 'bool BME680_Class::begin()' cannot be overloaded with 'bool BME680_Class::begin()'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:103:11: note: previous declaration 'bool BME680_Class::begin()'
   bool    begin();                          // Start using I2C Communications
           ^~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:106:6: error: extra qualification 'BME680_Class::' on member 'begin' [-fpermissive]
 bool BME680_Class::begin(const uint32_t i2cSpeed) {
      ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:106:6: error: 'bool BME680_Class::begin(uint32_t)' cannot be overloaded with 'bool BME680_Class::begin(uint32_t)'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:104:11: note: previous declaration 'bool BME680_Class::begin(uint32_t)'
   bool    begin(const uint32_t i2cSpeed);   // I2C with a non-default speed
           ^~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:116:6: error: extra qualification 'BME680_Class::' on member 'begin' [-fpermissive]
 bool BME680_Class::begin(const uint32_t i2cSpeed, const uint8_t i2cAddress) {
      ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:116:6: error: 'bool BME680_Class::begin(uint32_t, uint8_t)' cannot be overloaded with 'bool BME680_Class::begin(uint32_t, uint8_t)'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:106:11: note: previous declaration 'bool BME680_Class::begin(uint32_t, uint8_t)'
   bool    begin(const uint32_t i2cSpeed, const uint8_t i2cAddress);  // Set speed and I2C Addr.
           ^~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:142:6: error: extra qualification 'BME680_Class::' on member 'begin' [-fpermissive]
 bool BME680_Class::begin(const uint8_t chipSelect) {
      ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:142:6: error: 'bool BME680_Class::begin(uint8_t)' cannot be overloaded with 'bool BME680_Class::begin(uint8_t)'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:105:11: note: previous declaration 'bool BME680_Class::begin(uint8_t)'
   bool    begin(const uint8_t chipSelect);  // Start using either I2C or HW-SPI
           ^~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:160:6: error: extra qualification 'BME680_Class::' on member 'begin' [-fpermissive]
 bool BME680_Class::begin(const uint8_t chipSelect, const uint8_t mosi, const uint8_t miso,
      ^~~~~~~~~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:160:6: error: 'bool BME680_Class::begin(uint8_t, uint8_t, uint8_t, uint8_t)' cannot be overloaded with 'bool BME680_Class::begin(uint8_t, uint8_t, uint8_t, uint8_t)'
In file included from libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:8:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:107:11: note: previous declaration 'bool BME680_Class::begin(uint8_t, uint8_t, uint8_t, uint8_t)'
   bool    begin(const uint8_t chipSelect, const uint8_t mosi, const uint8_t miso, const uint8_t sck);
           ^~~~~
libdeps/esp32-s3/DeeEmm BME680/src/DDeeEmm_BME680.cpp:182:6: error: extra qualification 'BME680_Class::' on member 'commonInitialization' [-fpermissive]
 bool BME680_Class::commonInitialization() {
      ^~~~~~~~~~~~
[... 351 lines cut ...]
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.h:147:34: note:   template argument deduction/substitution failed:
[... cut ...]
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.cpp: At global scope:
libdeps/esp32-s3/DeeEmm BME680/src/DeeEmm_BME680.cpp:539:1: error: expected unqualified-id at end of input
 }
 ^
Compiling .pio/build/esp32-s3/libc5c/ESPmDNS/ESPmDNS.cpp.o
*** [.pio/build/esp32-s3/lib242/DeeEmm BME680/DDeeEmm_BME680.cpp.o] Error 1
*** [.pio/build/esp32-s3/lib242/DeeEmm BME680/DeeEmm_BME680.cpp.o] Error 1
========================================= [FAILED] Took 32.11 seconds =========================================
```
</details>

DeeEmm, 2024-11-20:
Yes I think this is as a result of the changes I made to the library whilst testing. 

I was trying to get the example in the library to compile and run stand-alone.

DeeEmm, 2024-11-20:
The DIYFB code part is actually done. It's just the library part that needs to work. I suspect that its not too far away from working, just need to finish what I started. 😆

DeeEmm, 2024-11-20:
OK. some progress...

[image]

I have the example project compiling and running. It is however reporting nonsense. But it is a step in the right direction.

Had to re-jig the project layout, add a platformIO.ini file and some other stuff so that it would compile as a stand alone project uner VScode / plstormIO, so not too sure if it will now work as a library. Figured that I would get the example working first and then sort out the library aspect after.

DeeEmm, 2024-11-20:
Ok so sorted the project paths out. Library can now be included in code and also built as a standalone project using the default I2CDemo

Now to get it working

KirikauKiwi, 2024-11-23:
I have had a brief read up about the BME68X sensor and look through the example and the library you have created.
Amongst the reading there was a mention of calculations having to be performed on the raw values from the sensor to be able to get the Temp, Pressure, Humidity and Gas resistance.
I could not find anything in the example or your library that looked like the calculations that are in the [Bosch API](https://github.com/boschsensortec/BME68x_SensorAPI/blob/master/bme68x.c), the calculations section is from line 831.

I can report the example code and your library work with my BME688 sensor, with exactly the same numbers being output???? The displayed values stayed fixed didn't change as I would expect.
```
Starting I2CDemo example program for BME680
- Initializing BME680 sensor
- Setting 16x oversampling for all sensors
- Setting IIR filter to a value of 4 samples
- Setting gas measurement to 320°C for 150ms

Loop Temp°C Humid% Press hPa   Alt m Air mΩ
==== ====== ====== ========= ======= ======
[  1777][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
   1 -80.00 84.000   5600.00-17044.257800.00
   2 -80.00 84.000   5600.00-17044.257800.00
   3 -80.00 84.000   5600.00-17044.257800.00
   4 -80.00 84.000   5600.00-17044.257800.00
```

DeeEmm, 2024-11-23:
Yes this is the same output that I see.

The library I used was taken from the Adafruit library, which was taken from the Bosch library.

The calcs are the same...

Line 842 in the Bosch library...
```
    var1 = ((int32_t)temp_adc >> 3) - ((int32_t)dev->calib.par_t1 << 1);
    var2 = (var1 * (int32_t)dev->calib.par_t2) >> 11;
    var3 = ((var1 >> 1) * (var1 >> 1)) >> 12;
    var3 = ((var3) * ((int32_t)dev->calib.par_t3 << 4)) >> 14;
    dev->calib.t_fine = (int32_t)(var2 + var3);
    calc_temp = (int16_t)(((dev->calib.t_fine * 5) + 128) >> 8);

    /*lint -restore */
    return calc_temp;
```

Line 395 in DeeEmm_BME680.cpp
```
  //*******************************//
  // First compute the temperature //
  //*******************************//
  var1         = ((int32_t)adc_temp >> 3) - ((int32_t)_T1 << 1);  // Perform calibration/adjustment
  var2         = (var1 * (int32_t)_T2) >> 11;                     // of Temperature values according
  var3         = ((var1 >> 1) * (var1 >> 1)) >> 12;               // to formula defined by Bosch
  var3         = ((var3) * ((int32_t)_T3 << 4)) >> 14;
  _tfine       = (int32_t)(var2 + var3);
  _Temperature = (int16_t)(((_tfine * 5) + 128) >> 8);
```

Lots of bit shifting and register interrogation, it is hard to follow. 

Back when I was integrating the BME280, I pulled the BME280 Bosch library apart to try and condense it, but it was hard going and eventually I found the tiny_BME library so left that as-was.

I suspect that the basic code in the library is okay, and that the issue lay in how the resultant data is handled and communicated.

TBH I haven't looked at it since I posted above.

KirikauKiwi, 2024-11-23:
Ah, yes I had overlooked it!!!!

No rush as there are better things to be working on other than adding the option of having another type of environmental sensor available.

stefan63, 2025-01-28:
Another contender?

[image]

DeeEmm, 2025-01-29:
Interesting combination. Two temperature and humidity sensors on one BOB. I wonder what the thinking behind that is?

stefan63, 2025-01-29:
The reason i guess is that is actually is a ATH20+BME280.
Look at the shape of the chip covers.
https://community.home-assistant.io/t/aht20-bmp280-combined-sensor/366411