DIY-Flow-Bench/DIY-Flow-Bench #234: Basic version of PWM close loop pressure control.

DeeEmm, 2024-11-12:
The basic version of PWM close loop pressure control.

_Originally posted by @black-top in https://github.com/DeeEmm/DIY-Flow-Bench/discussions/189#discussioncomment-11223082_

DeeEmm, 2024-11-14:
> @DeeEmm mentioned about available power from the shield

VAC_BLEED_VALVE_PIN is already assigned for PWM or analog control of a 'bleed device'. It can be used for either

VAC_SPEED_PIN provides an analog output that can be connected to a VFD.

The ESP32 only has those two ADC converters, so depending on what speed control philosophy is used, the VAC_SPEED_PIN may need to drive either a VFD or a PWM style ICV / DBW device.

The overall current limit for the shield is determined by the LM2596 Buck PSU's which should provide up to 3amps with no heat sink. So we have 3v@3A plus 5v@3A. For 12v the limit is whatever sized PSU you are using to power the shield. Consider that any PWM controlled DBW or IAVC valve will require 12v as they will invariably be an automotive device so the limiting factor there is the 12v PSU.

There is a 50mA current limit on the ESP outputs. There is also a voltage disparity as well, the ESP is a 3.3v device, so it cannot drive a 5v/10v/12v device. So some form of mosFET needs to be used. Something like the IRLB8721PB which has a low enough gate voltage that it can be driven by the 3.3v output. This will allow the ESP32's ADC to drive PWM based devices at whatever current / voltage is required up to 30v/60A.

To drive a VFD, again some form of transistor will be needed. This time a mosFET is not needed as this is a low current application, so something like the trusty old 2N2222 should work just fine. Full voltage needs to be limited to 10v as the VFD requires a 0-10v reference.

For switching 12v device via the ULN2003 there is a 0.5A limit per channel, so an interposing transistorised relay is recommended. (vac motors / etc)

The stepper drivers are direct from the 12V bus. If you use an A4988 they will supply up to 3A, which will happily drive a NEMA 32 which is massive overkill for this application.

As far as using hobby servos are concerned, my advice is don't. They are not robust enough. This is essentially a semi-industrial process / machine. So component selection is important to ensure reliability and accuracy. Pay the extra few dollars for something reliable.

The choice of using something like a NEMA 14 for the valve lift is so that it can drive a lead screw that operates on the valve stem. This has the advantage of a massive force multiplication. from a typical 18Ncm NEMA 14, driving a 1mm pitch you will get 120N of force. That's a 666% increase. I appreciate that the spring is changed, but when you have 28" of pressure pushing against the valve, that's a lot of force.

Of course, there are many ways to skin a cat.

KirikauKiwi, 2024-11-14:
Thanks, this is really helpful in explaining the overall scope and the roadmap for you DIYFB project.