DIY-Flow-Bench/DIY-Flow-Bench #236: [FEATURE REQUEST]V3, Data Display accessed from Configuration GUI page

KirikauKiwi, 2024-11-13:
Would it be useful to have a data display page that is accessible like the "Info" page is.

The idea would be to display the live sensor voltages and the sensor readings with their native units and possibly the calculated and converted values too.

I think this could be useful while trouble shooting any possible anomalies you may experience with the DIYFB like a possible faulty pressure sensor etc.

Sort of a diagnostics display.

DeeEmm, 2024-11-13:
Could also add the I/O status to this as well.

DeeEmm, 2024-12-10:
Also need to integrate #274 Trimpot Tuning into this

DeeEmm, 2024-12-10:
Additional items

- Zeroing of differential sensor voltages
- Trimpot zeroing
- General calibration / sensor reporting

DeeEmm, 2024-12-13:
[image]

Have a very basic modal working. 

You can view it by enabling in webmaster tools. Look for datModal in the source and then set display: block

DeeEmm, 2024-12-18:
An update on this issue.

The changes to add additional modal dialogs for displaying additional data fields resulted in GUI latency issues.

The latency is not only as a result of this change, but rather that this change was the proverbial straw that broke the camels back. The actual issue is memory related and largely as a result of the changes made as part of #205 - Post Compile Configuration where data previously resident in pre-compile macros was moved into JSON files and stored in SPIFFS.

The issue there was that to load the data into run-time memory it required that it was first loaded into a JSON object before it could be assigned to run-time variables. The net result was that the data occupied twice the memory, which then caused various memory issues.

The solution to this has been to move the post compile data from JSON files into NVM. The method from which NVM 'preferences' can be moved into runtime vars is much less memory intensive.

So at the moment I am moving all settings and preferences into the NVM memory space, which will free up memory on the stack to be used for things like a data display page and also allow for an easier user experience as there will be no JSON files to manage

Added to this I have also rewritten the basic sitemap and navigational relationship between functions. Essentially now everything gets it's own dedicated HTML page and javascript header. This keeps individual page load times and sizes to an absolute minimum, which also helps to manage memory requirements.

There's still a lot of core code changes to undertake before the GUI can be fixed up. Hoping to get those done before xmas.

DeeEmm, 2025-01-05:
[image]

Mimic page now added as pat of the conversion to use NVM for system settings.

Code still being tested but currently an issue with the MAF conversion to be resolved.

Going to close this item out as complete. We can add to / edit content of mimic page as required.

closed 2025-01-05 by DeeEmm