DIY-Flow-Bench/DIY-Flow-Bench #238: WIKI Updates

DeeEmm, 2024-11-16:
With the recent changes and upcoming RC9 release there are (and are yet to be) a lot of changes to the functionality of the code.

Whilst I try to make sure that all such changes are captured int he WIKI to support code changes, there is a definite opportunity to updated the WIKI along side existing changes.

If there is anything that you want to see, or feel could be improved on, discuss below.

DeeEmm, 2024-11-17:
From #228 Linking WIKI to GUI for targetted help...

>... in the GUI after the user has connected by Wifi with the 'Please upload the following files". This will detect which files are missing and ask for them, This could be easily expanded with additional prompts on what to do for each file.

> For example we could add something like this for the pins file...

```
Upload the PINS.JSON file for your board. This can be found in the PINS folder of the project.
If you do not see a file for your board, please see the WIKI for instructions on how to create one.
```

>We can then link directly to the relevant section.

KirikauKiwi, 2024-11-17:
Would it be useful as many starting with flow testing/porting etc may not be aware of the ball park figures of CFM to theoretical HP they may need from a flow bench, which may help them with bench design, motor selection, MAF selection. I know most of this information is out there already.

~Should I choose the RS4 MAF or the ACDELCO etc. using the very ball park figure of 0.5 CFM per HP the ACDELCO MAF would be good for about 590HP per cylinder as 277CFM @ 28inH2O vac is about 296 CFM at 1 atm and the ACDELCO should give better resolution compared to the BOSCH RS4 0-4V up to the limit of ACDELCO~ probably don't need to bore the end user with that, TLDR is that the ACDELCO is good for about 590HP per cylinder, more than enough for most I would think. @DeeEmm did I get that about right as it was question I had about MAF selection?

DeeEmm, 2024-11-17:
Yes sounds about right. 

There are MAF data files for a couple of ACDELCO MAFs. but only one has validated data: the ACDELCO_92281162. This is basically the LS2 MAF used by GM and Holden all most LS2 based engines. Commonly available and easy to source.

The big question about MAF selection is the transfer data / data file. Very few have accurate published data. So often the choice is driven by available data rather then ideal MAF size.

At the top of the MAF Data files there is some info on flow rates and whether the data has been validated

I think it would be good to add a table into the WIKI that includes flow rates and potential horsepower / Kw ratings based on the airflow figures along with the validation / test status of the data.

DeeEmm, 2024-11-17:
BTW, all of the MAF data that we currently have has come from online sources. 

For clarity:

**Validated** data is data where the provenance has been validated. I.e. known to work, either from being officially published OEM data, or used as part of something like the Megasquirt project. It is not necessarily validated by the DIYFB project, but equally is not just randomly untested published data. 

**Tested** data is where the data has been tested on a DIYFB bench and the results verified

DeeEmm, 2024-11-17:
The table can also link direct to the MAF files in the Repo.

Can also have the same kind of data for pins files - a table that outlines available types etc - can also link to direct to file in repo

DeeEmm, 2024-11-24:
/cib

DeeEmm, 2024-11-24:
/cib

KirikauKiwi, 2024-11-25:
Would it be worth mentioning in the WIKI. 
To recommend using a USB isolator when connecting your computer to the DIYFB shield, when the shield is being powered by its own PSU.

DeeEmm, 2024-11-25:
OK MAF table added to WIKI...

https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Component-selection

DeeEmm, 2024-11-25:
> Would it be useful as many starting with flow testing/porting etc may not be aware of the ball park figures of CFM to theoretical HP they may need from a flow bench, which may help them with bench design, motor selection, MAF selection. I know most of this information is out there already.

It is hard to know how much info to include. The project will support pretty much any bench, so there is no ideal one size fits all, there are also vast differences in opinions on what a bench should / shouldn't be or include. for example the eternal 28" reference argument versus floating depression style benches. I feel that steering people too much may ultimately be bad for the project.

Flow figures differ vastly between heads, depending on type and valve size. A two valve wedge shaped head will flow a lot different to a multivalve pent roof head. A 30mm valve will flow differently to a 40mm valve. Porting shop A is different porting shop B. So HP targets are not something that can be easily generalised. I think that this side of things can be left to the user to find out.

I think some details that point to what the users need to consider without steering them towards any particular solution is a good idea.

DeeEmm, 2024-11-25:
#213 - Outline procedure for pitot zeroing / calibration along with differential sensor zeroing / calibration

DeeEmm, 2024-11-25:
Further comments from above thread...

> When you build your bench, make provision for fitting extra motors, then just blank off the extra holes. Its almost zero extra work, but one day you may be very glad you did this.

> Yes I definitely want to make sure it's open for that. I played with the numbers some it looks like 5 of those motors are kinda the magic number to be able to flow up to 350cfm area at 28"(44").

Elsewhere there are comments that outline the drivers for motor selection

> Most of us are limited by the available power source to some maximum total flat out amperage, and we are (or should be) looking for the most efficient motors to get the highest possible CFM flow from whatever available power we have.
> If there is then a choice between four motor "A" or six much smaller "B" motors, and both have similar efficiency, we might logically choose the lower cost option.
> The trick is to find motors that have their efficiency peak around 44 inches, and the magic number for judging motors is CFM per amp at 44 inches.
> The worst may be about 8CFM per amp, the very best about 18CFM per amp, and most "average" motors maybe around 12 CFM per amp.
> That will give you some idea of how to judge the numbers from the available data.

DeeEmm, 2024-11-25:
Haha, didn't even notice this initially...

They have a dedicated page for flow bench motors...

> The vacuum motors on this page are suggested for your use with the "FlowBench" plans from [www.flowbenchtech.com](http://www.flowbenchtech.com/)

https://centralvacuummotor.com/flowbench.htm#gsc.tab=0

They list two motors

- LLH9563
- LH2093

I think that this data along with the selection process above gives a great starting point for vac motor selection.

I will wrap this up and add it to the Wiki

DeeEmm, 2024-11-26:
OK. Done - https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Vacuum-Motor-Selection

DeeEmm, 2024-11-28:
Added some additional 'getting started / installation' pages to the WIKI. On-boarding is now a lot easier to understand. Still need to flesh out the differential bench stuff and create some connection diagrams along with updating the configuration pages

I've added the details for the firmware flasher installation pathway as currently the online expressif tool can be used. This will make it easier for the non-tech savvy to get it up and running provided they are using the default / recommended setup.

https://github.com/DeeEmm/DIY-Flow-Bench/wiki/Installation

Would really love to get the post-compile stuff sorted out for V2 as well. I know that this is a bunch of extra work but I think that for the V2 release it would round the release out nicely and be a good starting point for what happens next.

DeeEmm, 2024-12-12:
From #279 - update procedure for initial Buck PSU voltage set up and trim pot settings

DeeEmm, 2025-01-16:
WIKI Updated.

DeeEmm, 2025-01-30:
Need to update WIKI and add links in mimic for new items

DeeEmm, 2025-02-04:
Need to update MAF table with new sensors

DeeEmm, 2025-02-08:
Updated WIKI broke sensor selection out into separate section with dedicated pages for MAF / Pressure / Environmental sensors

closed 2025-02-08 by DeeEmm