DIY-Flow-Bench/DIY-Flow-Bench #239: Update API calls

DeeEmm, 2024-11-17:
Fitting in with the ethos of "if its currently part of the project it needs to work", the API needs a quick review and update once the RC9 stuff has been finalised.

There's not too much to do, just a quick review and update.

If there are any additional API calls that would be of use, list them below and we can potentially include them as part of the update

DeeEmm, 2024-11-17:
I will add that at this point in time the API HTTP access is a long way behind the API serial access. This will likely remain this way until V3 where we can address this as a seperate item. 

I will create a seperate item for that

KirikauKiwi, 2024-11-18:
For the pressure sensors on the ADS1115, 
- PRef add the Gauge value kPa
- add PDiff with ADC Value, Voltage and Gauge value kPa
- add Pitot with ADC Value, Voltage and Gauge value kPa

IIRC for the MAF sensor, these 3 reading from the ADS1115 are already in the serial API.

Would it be useful to display the PWM values and the DAC values too, they could be useful for trouble shooting when people are adding a MOSFET or OPAMP to drive their peripherals.

DeeEmm, 2024-11-24:
> * PRef add the Gauge value kPa

This is already available as part of Enum1 [E]

> * add PDiff with ADC Value, Voltage and Gauge value kPa

Added... all ADC voltage values under and raw values under [a]

[A] - ADC Voltage values
[a] - ADC Raw values

This is good from the perspective that we have now free'd up the secondary API function for each sensor (lower case values), which was previously used as voltage for most sensors. We can now customise the secondary value for something else.

> * add Pitot with ADC Value, Voltage and Gauge value kPa

Added... 

[P] - Pitot value (in/H20) - Note main API values report same units as GUI.  

[e] - enum2 - Pitot kPa values + Swirl value - kPa value available as part of Enums

So we have main enum [E] which reports general values and [e] which reports secondary values.

> IIRC for the MAF sensor, these 3 reading from the ADS1115 are already in the serial API.

Yes. For MAF there is the Raw MAF value [M], MAF voltage [m] and flow in kg/h [f], which is a little out of line from other sensor values.

But now we are reporting all voltages and RAW ADC values under the ADC API call, we can re-jig things a little

So main values are reported as such...

[M] - MAF Flow in KG/h 
[D] - Differential in in/H2O
[T] - Temp in Deg C
[B] - Baro in hPa
[R] - Reference pressure in in/H20
[H] - Humidity in %
[P] - Pitot value in in/H2O

Nice and easy to remember

However, at present [S] is currently used for WIFI SSID and [s] for SPIFFS file list, So I've moved those to [W] and [/] respectively

We can now use secondary values for something different

[m] - MAF secondary function
[p] - Pitot secondary function
etc..

This is probably most useful for calibration / setup process - for example like tuning or setting up the trimpots as we were discussing in #254 

I vote that for the time being we reserve these secondary functions in lieu of utilising them for initial setup (whatever that might look like).

ATM I'm trying to get the workflow automation working for auto branch creation. I'll upload the changes once I've got that figure out.

DeeEmm, 2024-11-24:
ahh yes - summary of current API functions...

```
  DIYFB API Commands
  ==============================
  API Response Format
  Command : Value : Checksum
  ==============================
  3 : 3.3V Voltage Value
  5 : 5V Voltage Value
  A : ADC Voltage Values
  a : ADC Raw Values
  B : Barometric Pressure
  C : Flow Offset Calibration
  D : Differential Pressure value
  E : Enum1
  F : Flow Value in CFM
  f : Flow Value in KG/H
  H : Humidity Value
  I : IP Address
  J : JSON Status Data
  j : JSON Configuration Data
  K : MAF Data Key Value
  k : MAF Data Lookup Value
  L : Leak Test Calibration
  l : Leak Test
  M : MAF RAW ADC Value
  m : MAF Voltage
  N : Hostname
  o : Active Orifice
  P : Pitot Value
  Q : MAF Data Max Value
  q : MAF Data Key Max Value
  R : Reference Pressure Value
  S : Swirl value
  T : Temperature in Celcius
  t : Temperature in Fahrenheit
  U : Uptime in hhhh.mm
  V : Version
  v : Valve lift data in JSON format
  W : WiFi SSID
  X : xTask memory usage   
  ? : Help
  / : SPIFFS File List
  ~ : Restart ESP
  $ : Reset WiFi
  @ : Stream Status
  ! : Debug Mode
  < : Last Error
  ============================== 
```

DeeEmm, 2024-11-25:
OK, looks like the workflow actions are now working. 

Adding an assignee automagically creates a branch.

Will create a PR for the API changes

DeeEmm, 2024-11-25:
Ignore the most recent activity - Just testing the workflow out.

I've added the additional API functionality for setup to issue #242 and rolled it up with the WIKI changes for new users.

Makes more sense to look at that process holistically - documentation + GUI + Serial + etc

I'll close this out for the time being. 

We can always add more suggestions + reopen later.

KirikauKiwi, 2024-11-25:
Should the new Serial API be working with Build 24112402, or has it not been rolled out in a release yet?
As the new commands I have tried are not reporting anything back.

DeeEmm, 2024-11-25:
Yes it should be.

Take a look at what is reported when you send '?'

Should be the same as above

DeeEmm, 2024-11-25:
Ooops no sorry

24112501 onwards

DeeEmm, 2024-11-25:
It's in the PR linked above 

[image]

DeeEmm, 2024-11-29:
After looking at #266 I just remembered that we are able to interrogate the API whilst in the boot loop, however some of the API functions are not initialised at that point.

Need to mask those items that result in incorrect data / kernel panics when in boot loop.

Easy enough to do with the **status.doBootLoop** variable

DeeEmm, 2024-12-10:
Boot loop API items now masked (build 2412060044)

closed 2024-12-10 by DeeEmm