DIY-Flow-Bench/DIY-Flow-Bench #242: [FEATURE REQUEST] WIKI

stefan63, 2024-11-19:
For a newbee on flow benches maybee a section of the WIKI could help to straighten some ?? in regards of flow bench types :)
Ilustration would be helpfull.
What differs those types and how do they work, where to place the sensors/hoses?
1. MAF based , "the new black"
2. Pressure drop
3. Orifice
4. Venturi

I know how to solder and take messurements (and some coding) but the worings of the flow bench types make my head spin haha...

Kind regards/Stefan

DeeEmm, 2024-11-19:
Yes we definitely need to expand the existing WIKI info to include the differential style / pressure style bench now that we have it in the code. This is currently missing.

I think it might also be good to leverage @danielerpl's experience with using the PTS plans to build a bench as this is equally applicable to other differential style benches.

DeeEmm, 2024-11-25:
I'm going to roll this action up with some additional items to improve the on-boarding process for new users.

#239 - Utilise API to assist with initial setup

danielerpl, 2024-11-25:
> Yes we definitely need to expand the existing WIKI info to include the differential style / pressure style bench now that we have it in the code. This is currently missing.
>
> I think it might also be [...]

For sure! if you need any info I am here to help everyone

DeeEmm, 2025-01-16:
I've now updated the WIKI to include info on the settings, configuration and pins pages. Clicking on the info icon on any tiles on those pages opens up the relevant section of the WIKI in a new window... 

[image]

(maybe we need a 'dark mode' for the GUI 😄 )

[image]