DIY-Flow-Bench/DIY-Flow-Bench #244: [FEATURE REQUEST]Suggested simplification of "Resolution and Accuracy" in GUI Configuration.

KirikauKiwi, 2024-11-19:
I am suggesting a small change to the Resolution and Accuracy in the GUI Configuration page.

**From this**
```
<br>
                    <h3>Resolution and Accuracy</h3>
                    <div class="config-row tooltip">
                        <label class="config-label">Flow Value Rounding:</label>
                        <span>
                            <select name='ROUNDING_TYPE' class='config-select'>
                                <option value='NONE' selected>None</option>
                                <option value='INTEGER'>Whole number</option>
                                <option value='HALF'>Half value </option>
                            </select>
                        </span>
                        <span class="tooltiptext">Rounding used in flow calculations</span>
                    </div>
                    <div class="config-row tooltip">
                        <label class="config-label">Flow Decimal Accuracy:</label>
                        <span>
                            <select name='FLOW_DECIMAL_LENGTH' class='config-select'>
                                <option value='0'>None</option>
                                <option value='1' selected>Tenths</option>
                                <option value='2'>Hundredths </option>
                            </select>
                        </span>
                        <span class="tooltiptext">Number of decimal places used in UI for flow values</span>
                    </div>
                    <div class="config-row tooltip">
                        <label class="config-label">General Decimal Accuracy:</label>
                        <span>
                            <select name='GEN_DECIMAL_LENGTH' class='config-select'>
                                <option value='0'>None</option>
                                <option value='1'>Tenths</option>
                                <option value='2' selected>Hundredths </option>
                            </select>
                        </span>
                        <span class="tooltiptext">Number of decimal places used in UI</span>
                    </div>
                    <br>
```

**To this**
```
<br>
                    <h3>Resolution and Accuracy</h3>
                    <div class="config-row tooltip">
                        <label class="config-label">Flow Decimal Accuracy:</label>
                        <span>
                            <select name='FLOW_DECIMAL_LENGTH' class='config-select'>
                                <option value='0'>1.    Whole</option>
				<option value=‘0.’5>0.5 Half</option>
                                <option value='1' >0.1  Tenths</option>
                                <option value='2'>0.001 Hundredths </option>
                            </select>
                        </span>
                        <span class="tooltiptext">Number of decimal places used in UI for flow values</span>
                    </div>
                    <div class="config-row tooltip">
                        <label class="config-label">General Decimal Accuracy:</label>
                        <span>
                            <select name='GEN_DECIMAL_LENGTH' class='config-select'>
                                <option value='0'>1     Whole</option>
				<option value='0.5'>0.5 Half</option>
                                <option value='1'>0.1   Tenths</option>
                                <option value='2' selected>0.01 Hundredths </option>
                            </select>
                        </span>
                        <span class="tooltiptext">Number of decimal places used in UI</span>
                    </div>
                    <br>
```

It feels more intuitive to me. 
I am assuming the Rounding and Decimal places on the GUI is only affecting the displayed values, and not changing the data used in the calculations etc.

DeeEmm, 2024-11-20:
Ahh yes. Agreed. Makes more sense.

DeeEmm, 2024-11-20:
It's funny. Had to look at the source for your post as the notification that I received showed that you had posted the HTML code.

Was scratching my head trying to work out how you managed to get the CSS from the project to work here in the forums.

😆🤣😆

KirikauKiwi, 2024-11-20:
I did first try and post the HTML as I was not sure if it was possible to or not!!!, so I edited the comment and replaced the html with photos instead!!!!

DeeEmm, 2024-11-22:
So changing the descriptions is trivial and done. However combining the decimal places with rounding is not so trivial.

We pass the decimal places directly to the GUI javascript via AJAX.

So, for FLOW_DECIMAL_LENGTH dropdown selection values...

- No DP = 0 
- 1 DP (0.1) = 1
- 2 DP (0.01) = 2

We pass this value directly to the GUI...

```
 document.getElementById(key).innerHTML = myObj[key].toFixed(~FLOW_DECIMAL_LENGTH~);  
```

So it is not possible to add an additional value for 0.5, even though this makes sense, as rounding and DP are different mathematical functions.

So the rounding and decimal places functions need to remain seperate for the time being as they apply to two different parts of the code.

I have changed the language as suggested for the DP drop-downs, and also changed the terminology of the rounding to further differentiate it from the DP to remove any ambiguity.

[image]

[image]

We can of course revisit this down the track but given the amount of work still to do for RC9 I'm going to close this out for now.

closed 2024-11-22 by DeeEmm

KirikauKiwi, 2024-11-23:
It amazes my that Human logic is often apparently simple, but at times still difficult in the binary world. I guess that is good for us Humans vs the Robots.

KirikauKiwi, 2024-11-23:
The decimal point shift and the rounding are only happening at the GUI level, is that correct?

If so would a more approbate Heading for this section possibly be?
**Display Resolution**

DeeEmm, 2024-11-23:
Yes only at the GUI level.  Rounding at the calc level just introduces error. 

The original intent is only to make it easier to read (for a human). i.e. to help prevent the rapid changing of fractional values in the GUI.

This effect does not impact data collection or API values as the data snapshot grabs all sensor data within a microsecond window. As such it is impervious to such change.