DIY-Flow-Bench/DIY-Flow-Bench #245: DEV branch, The serial monitor is hanging near the end of the "INPUT"

KirikauKiwi, 2024-11-20:
The serial monitor is hanging near the end of the "INPUT" with the latest DEV branch. This is happening with my ESP32 S3 board setup. 
```
DIY Flow Bench
DIYFB Version: V2.0.RC.8 
Build: 24111901 
For help please visit the WIKI:
https://github.com/DeeEmm/DIY-Flow-Bench/wiki
Initialising File System 
Loading Configuration 
Loading Lift Data 
Loading Calibration Data 
PINS file Found: /PINS_ESP32_S3_DEV_KITC.json
Loading Pins Data 
Parsing Pins Data 
Defaulting to AP Mode 
Creating WiFi Access Point:  DIYFB  
Access Point IP address: 192.168.4.1 
Multicast: http://diyfb.local 
Initialising I/O 
Input MAF_PIN: 0
Input REF_PRESSURE_PIN: 0
Input DIFF_PRESSURE_PIN: 0
Input PITOT_PIN: 0
Input SERIAL0_RX_PIN: 44
Input SDA_PIN: 8
Input SCL_PIN: 9
Input SD_MISO_PIN: 0
Input SD_SCK_P
```
The SD card is not defined, and my pins.json for my ESP32 S3 DEVKITC board has the pins for the SD card as -1.

When performing a rest on the board the serial monitor hangs at the same place each time, other than the length of the last lines output differs by one or two characters. e.g. Input SD_SCK_, Input SD_SCK_PIN:, Input SD_SCK_PIN: 0

I have also tried with the pins defined as my boards default SPI pins, with the same results in the serial monitor.
```
"SD_CS_PIN" : 10,
"SD_MOSI_PIN" : 11,
"SD_MISO_PIN" : 13,             
"SD_SCK_PIN" : 12,

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/issues/233#issuecomment-2486878468_

DeeEmm, 2024-11-20:
The issue will be the line after

This look to be the 'WEMOS_SPARE_PIN_1'

This is not present in the M5Stack pins.json file, which is I suspect why it is failing. This line need to be present.

KirikauKiwi, 2024-11-20:
I am currently working on the other board and not the M5 Stack.

This is my pins.json, the memos spare pin is present
```
{
    "BOARD_TYPE" : "esp32-s3-devkitc-1",
    "VAC_SPEED_PIN" : -1, 
    "VAC_BLEED_VALVE_PIN" : -1,
    "VAC_BANK_1_PIN" : -1,
    "VAC_BANK_2_PIN" :-1, 
    "VAC_BANK_3_PIN" :-1,
    "AVO_STEP_PIN" :  -1,
    "AVO_DIR_PIN" : -1,
    "FLOW_VALVE_STEP_PIN" : -1,
    "FLOW_VALVE_DIR_PIN" : -1,
    "VCC_3V3_PIN" : -1,
    "VCC_5V_PIN" : -1,
    "SPEED_SENSOR_PIN" : -1,
    "SWIRL_ENCODER_PIN_A" : -1,
    "SWIRL_ENCODER_PIN_B" : -1,
    "ORIFICE_BCD_BIT1_PIN" : -1,
    "ORIFICE_BCD_BIT2_PIN" : -1,
    "ORIFICE_BCD_BIT3_PIN" : -1,
    "MAF_PIN" : -1,
    "REF_PRESSURE_PIN" : -1,
    "DIFF_PRESSURE_PIN" : -1,
    "PITOT_PIN" : -1,
    "TEMPERATURE_PIN" : -1,
    "REF_BARO_PIN" : -1,
    "HUMIDITY_PIN" : -1,
    "SERIAL0_TX_PIN" :43,
    "SERIAL0_RX_PIN" :44,
    "SERIAL2_TX_PIN" :-1,
    "SERIAL2_RX_PIN" :-1,
    "SDA_PIN" : 8,
    "SCL_PIN" : 9,
    "SD_CS_PIN" : -1,
    "SD_MOSI_PIN" : -1,
    "SD_MISO_PIN" : -1,             
    "SD_SCK_PIN" : -1,
    "WEMOS_SPARE_PIN_1": -1
}
```

KirikauKiwi, 2024-11-20:
Digging a little further it would appear to my eye that something is happening before "Initialising I/O" because the first pin that appears in the serial monitor is the MAF_PIN it displays the pin is set to 0, however in my pins.json it is set to -1.
The pinMode setting in the hardware.cpp is >=0, so I shouldn't be seeing the pin set, This is true from the MAF_PIN onwards in my pins.json, the pins prior to this don't appear in the serial monitor as expected as they are -1.

Does this indicate that the values from the pins.json file are not setting the int value correctly in the struct Pins part of the struct.h

From the hardware.cpp 
```
if (pins.MAF_PIN >= 0 ) {
    _message.serialPrintf("Input MAF_PIN: %d\n", pins.MAF_PIN );
    pinMode(pins.MAF_PIN, INPUT); 
```
```
Initialising I/O 
Input MAF_PIN: 0
Input REF_PRESSURE_PIN: 0
Input DIFF_PRESSURE_PIN: 0
Input PITOT_PIN: 0
Input SERIAL0_RX_PIN: 44
Input SDA_PIN: 8
Input SCL_PIN: 9
Input SD_MISO_PIN: 0
Input SD_SCK_P
```
Comparing that to my pins.json
```
    "MAF_PIN" : -1,
    "REF_PRESSURE_PIN" : -1,
    "DIFF_PRESSURE_PIN" : -1,
    "PITOT_PIN" : -1,
    "TEMPERATURE_PIN" : -1,
    "REF_BARO_PIN" : -1,
    "HUMIDITY_PIN" : -1,
    "SERIAL0_TX_PIN" :43,
    "SERIAL0_RX_PIN" :44,
    "SERIAL2_TX_PIN" :-1,
    "SERIAL2_RX_PIN" :-1,
    "SDA_PIN" : 8,
    "SCL_PIN" : 9,
    "SD_CS_PIN" : -1,
    "SD_MOSI_PIN" : -1,
    "SD_MISO_PIN" : -1,             
    "SD_SCK_PIN" : -1,
    "WEMOS_SPARE_PIN_1": -1
```

stefan63, 2024-11-20:
During my initial tests I dissabled "comented out" all sensors just to try
to get the board show a UI, this failed, I then enabled MAF sensor (without
having one hooked up) and then the board compiled correctly. Looks like the
MAF is mandatory even if not used....

DeeEmm, 2024-11-21:
> I am currently working on the other board and not the M5 Stack.

What board is this?? Generally there should be no need to change the the platformIO.ini - The esp32dev board / environment should work for ALL ESP32's (even the M5Stack) as they all use the same processor. 

The BOBs that the processor is mounted to, along with any additional peripherals attached are largely irrelevent. Pin clashes for peripherals should be able to be programmed around with the pin-defs.

Try compiling using the default 'esp32dev' environment

DeeEmm, 2024-11-21:
> During my initial tests I dissabled "comented out" all sensors just to try to get the board show a UI, this failed, I then enabled MAF sensor (without having one hooked up) and then the board compiled correctly. Looks like the MAF is mandatory even if not used.... 

It *should* work as there is a use case for differential benches where the MAF is not used, but that is a relatively new change so it is possible that the code will fail if the MAF is disabled. 

At some point in V3 I will integrate the post compile configuration (configuration.json). At this point I will review each use case and make sure that disabling the MAF works as intended

DeeEmm, 2024-11-21:
> first pin that appears in the serial monitor is the MAF_PIN it displays the pin is set to 0, however in my pins.json it is set to -1.

If it is set to -1 then it should not display. It should be skipped over. Essentially only pins that are actively assigned should be reported to the serial monitor.

I did previously note a strange behaviour relating to the -1 value and suspected it was as a result of an unsigned int, however the pins struct declares all pins as ints, which are signed by default.

A point to note there is that pinmode actually accepts an unsigned int for the pin value, so sending it a '-1' is essentially setting it out of range. I note that with my board, this causes an error, hence the reason for adding the logic to test for the '-1' value

It should not be possible for pins with a value of '-1' to pass the test, it dos not make logical sense

```
  // Set Inputs
  if (pins.VCC_3V3_PIN >= 0 ) {
    _message.serialPrintf("Input VCC_3V3_PIN: %d\n", pins.VCC_3V3_PIN );
    pinMode(pins.VCC_3V3_PIN, INPUT);   
  }
  if (pins.VCC_5V_PIN >= 0 ) {
    _message.serialPrintf("Input VCC_5V_PIN: %d\n", pins.VCC_5V_PIN );  
    pinMode(pins.VCC_5V_PIN, INPUT);
  }  
```

So the fact that it is able to print to the serial monitor is I think a clue. 

The value bing printed is '0', so I would deduce that the value of the pins.struct was '0' to begin with.

By default the pins.structs are uninitialised. They only receive a value when the pins.json file is read. I suspect that the issue may lay there.

DeeEmm, 2024-11-21:
also unfortunately '0' is a valid pin number

DeeEmm, 2024-11-21:
Ok so did a quick test..

```
  // TEST - save pins data to file
  String output;
  serializeJson(pinsFileData, output);
  this->writeJSONFile(output,"/pintest.json", CONFIG_JSON_SIZE);
```

Added to bottom of load pinsData method to immediately bounce results back to a file.

Results of test...

```
{
  "BOARD_TYPE": "WEMOS_D1_R32",
  "VAC_SPEED_PIN": 25,
  "VAC_BLEED_VALVE_PIN": 26,
  "VAC_BANK_1_PIN": 13,
  "VAC_BANK_2_PIN": 12,
  "VAC_BANK_3_PIN": 14,
  "AVO_STEP_PIN": 32,
  "AVO_DIR_PIN": 33,
  "FLOW_VALVE_STEP_PIN": 27,
  "FLOW_VALVE_DIR_PIN": 4,
  "VCC_3V3_PIN": -1,
  "VCC_5V_PIN": 2,
  "SPEED_SENSOR_PIN": 15,
  "SWIRL_ENCODER_PIN_A": -1,
  "SWIRL_ENCODER_PIN_B": -1,
  "ORIFICE_BCD_BIT1_PIN": 34,
  "ORIFICE_BCD_BIT2_PIN": 36,
  "ORIFICE_BCD_BIT3_PIN": 39,
  "MAF_PIN": -1,
  "REF_PRESSURE_PIN": -1,
  "DIFF_PRESSURE_PIN": -1,
  "PITOT_PIN": -1,
  "TEMPERATURE_PIN": -1,
  "REF_BARO_PIN": -1,
  "HUMIDITY_PIN": -1,
  "SERIAL0_TX_PIN": -1,
  "SERIAL0_RX_PIN": -1,
  "SERIAL2_TX_PIN": 16,
  "SERIAL2_RX_PIN": 17,
  "SDA_PIN": -1,
  "SCL_PIN": -1,
  "SD_CS_PIN": 5
}
```

So the '-1' is getting picked up.

Next test is to dump content of pins.struct

DeeEmm, 2024-11-21:
OK,

```  
serializeJsonPretty(pinsFileData, Serial);
```
Added to bottom of load pinsData 

Results...

```
Parsing Pins Data 
{
  "BOARD_TYPE": "WEMOS_D1_R32",
  "VAC_SPEED_PIN": 25,
  "VAC_BLEED_VALVE_PIN": 26,
  "VAC_BANK_1_PIN": 13,
  "VAC_BANK_2_PIN": 12,
  "VAC_BANK_3_PIN": 14,
  "AVO_STEP_PIN": 32,
  "AVO_DIR_PIN": 33,
  "FLOW_VALVE_STEP_PIN": 27,
  "FLOW_VALVE_DIR_PIN": 4,
  "VCC_3V3_PIN": -1,
  "VCC_5V_PIN": 2,
  "SPEED_SENSOR_PIN": 15,
  "SWIRL_ENCODER_PIN_A": -1,
  "SWIRL_ENCODER_PIN_B": -1,
  "ORIFICE_BCD_BIT1_PIN": 34,
  "ORIFICE_BCD_BIT2_PIN": 36,
  "ORIFICE_BCD_BIT3_PIN": 39,
  "MAF_PIN": -1,
  "REF_PRESSURE_PIN": -1,
  "DIFF_PRESSURE_PIN": -1,
  "PITOT_PIN": -1,
  "TEMPERATURE_PIN": -1,
  "REF_BARO_PIN": -1,
  "HUMIDITY_PIN": -1,
  "SERIAL0_TX_PIN": -1,
  "SERIAL0_RX_PIN": -1,
  "SERIAL2_TX_PIN": 16,
  "SERIAL2_RX_PIN": 17,
  "SDA_PIN": -1,
  "SCL_PIN": -1,
  "SD_CS_PIN": 5
}
```

Again '-1' value is getting picked up.

So a bit odd really. pins.struct is definitely '-1.'

This only leaves the comparison logic itself...

```
  if (pins.VCC_3V3_PIN >= 0 ) {
    _message.serialPrintf("Input VCC_3V3_PIN: %d\n", pins.VCC_3V3_PIN );
    pinMode(pins.VCC_3V3_PIN, INPUT);   
  }
```

Need to find a different way to test for same

DeeEmm, 2024-11-21:
Tried a bunch of stuff, nothing works so far

Flipping the logic does not work

```
  if (!(pins.VCC_3V3_PIN < 0) ) {
    _message.serialPrintf("Input VCC_3V3_PIN: %d\n", pins.VCC_3V3_PIN );
    pinMode(pins.VCC_3V3_PIN, INPUT);   
  }
```

Still lets '-1' through and prints it as '0'

I can only assume that the '0' is a limitation of the output formatting (behaving as an unsigned int) even though %d should  print as '-1'

Very bizarre behaviour.

DeeEmm, 2024-11-21:
OK. Nothing seems to work as expected. Have done a bunch  of tests.

```
 if (!(pins.VCC_3V3_PIN == -1) ) {
```

```
 if (pins.VCC_3V3_PIN != -1)  {
```

```
 if (!(pins.VCC_3V3_PIN == 99) ) {
```

Unsure what is going on there. Truly bizarre.

Note that whilst I have used the VCC_3V3_PIN in the above examples,  but the very first if statement actually works in many cases (VCC_3V3_PIN), all others afterwards do not.

Again another clue perhaps?

DeeEmm, 2024-11-21:
```
 if (pins.VCC_5V_PIN ) {
```

Works. This is because the value '0' is interpreted as false. So '-1' is seen as 0 and therefore false. (all non zero value are considered true)

I think that the issue is that the pins.struct is somehow seen an ambiguous and so does not behave as expected, this is despite it being initialised as int's. I tried hard to typecast it to try and force the desired behaviour, but nothing seems to work. Similar story if the pins.struct is initialised with all '-1' values, which I had hoped would remove the ambiguity.

I think that the best solution is to dump the pins.struct in favour of discrete variables and drop the '-1' nomenclature. The solution above is not usable simply as '0' is a valid GPIO value on most boards, but it will do as a quick and dirty fix for the interim.

Definitely needs more work

KirikauKiwi, 2024-11-21:
I am cannot recall what the changes were in respect to pin setting, between this Build: 24111901, being the current DEV build and the previous one, the previous build didn't appear to exhibit this behaviour.

I am having oddities that it appears to work for some pins and not others regarding the serial print. I added  a serial print of the prior to the pin set to show the value.
```
  _message.serialPrintf("pins.ORIFICE_BCD_BIT3_PIN value: %d\n", pins.MAF_PIN);
  if (pins.ORIFICE_BCD_BIT3_PIN >= 0 ) {   
    _message.serialPrintf("Input ORIFICE_BCD_BIT3_PIN: %d\n", pins.ORIFICE_BCD_BIT3_PIN );
    pinMode(pins.ORIFICE_BCD_BIT3_PIN, INPUT);   
  }
    _message.serialPrintf("pins.MAF_PIN value: %d\n", pins.MAF_PIN);
  if (pins.MAF_PIN >= 0) {
    _message.serialPrintf("Input MAF_PIN: %d\n", pins.MAF_PIN);
    pinMode(pins.MAF_PIN, INPUT);
```

This is what was output
```
Initialising I/O 
pins.ORIFICE_BCD_BIT3_PIN value: 0
pins.MAF_PIN value: 0
Input MAF_PIN: 0
Input REF_PRESSURE_PIN: 0
Input DIFF_PRESSURE_PIN: 0
Input PITOT_PIN: 0
Input SERIAL0_RX_PIN: 44
Input SDA_PIN: 8
Input SCL_PIN: 9
Input SD_MISO_PIN: 0
Input SD_SCK_PIN: 0
Input WEMOS_SPARE_PIN_1: 0
Output SD_MOSI_PIN: 0
Output SE
```

Strange to me that the serial print works for the orifice, i.e does not serial print as you would expect from the code, but the MAF pin does print.
```
Initialising I/O 
Input MAF_PIN: 0
Input REF_PRESSURE_PIN: 0
Input DIFF_PRESSURE_PIN: 0
Input PITOT_PIN: 0
Input SERIAL0_RX_PIN: 44
Input SDA_PIN: 8
Input SCL_PIN: 9
Input SD_MISO_PIN: 0
Input SD_SCK_P
```

If the pin is saying it is 0 from this, before the pins are being set by the pinMode, this would point to the variable not either being initialised and/or their values being set correctly?
```
 _message.serialPrintf("pins.ORIFICE_BCD_BIT3_PIN value: %d\n", pins.MAF_PIN);
```

KirikauKiwi, 2024-11-21:
Here is a comparison from the values set in the code to my pins.json for the input pins only.

int values
```
pins.VCC_3V3_PIN value: -1                                    
pins.VCC_5V_PIN value: -1
pins.SPEED_SENSOR_PIN value: -1
pins.ORIFICE_BCD_BIT1_PIN value: -1
pins.ORIFICE_BCD_BIT2_PIN value: -1
pins.ORIFICE_BCD_BIT3_PIN value: -1
pins.MAF_PIN value: 0
pins.REF_PRESSURE_PIN value: 0
pins.DIFF_PRESSURE_PIN value: 0
pins.PITOT_PIN value: 0
pins.TEMPERATURE_PIN value: -1
pins.HUMIDITY_PIN value: -1
pins.REF_BARO_PIN: value -1
pins.SERIAL0_RX_PIN value: 44
pins.SDA_PIN value: 8
pins.SCL_PIN value: 9
pins.SD_CS_PIN value: -1
pins.SD_MISO_PIN value: 0
pins.SD_SCK_PIN value: 0
pins.SD_SCK_PIN value: 0
pins.WEMOS_SPARE_PIN_1 value: 0
```

The respective pins.json values
```
    "VCC_3V3_PIN" : -1,
    "VCC_5V_PIN" : -1,
    "VAC_SPEED_PIN" : -1,
    "ORIFICE_BCD_BIT1_PIN" : -1,
    "ORIFICE_BCD_BIT2_PIN" : -1,
    "ORIFICE_BCD_BIT3_PIN" : -1,
    "MAF_PIN" : -1,
    "REF_PRESSURE_PIN" : -1,
    "DIFF_PRESSURE_PIN" : -1,
    "PITOT_PIN" : -1,
    "TEMPERATURE_PIN" : -1,
    "HUMIDITY_PIN" : -1,
    "REF_BARO_PIN" : -1,
    "SERIAL0_RX_PIN" :44,
    "SDA_PIN" : 8,
    "SCL_PIN" : 9,
    "SD_CS_PIN" : -1,
    "SD_MISO_PIN" : -1,             
    "SD_SCK_PIN" : -1,
    "WEMOS_SPARE_PIN_1" : -1
```

Odd that some values are being set from the pins.json and others are not.

KirikauKiwi, 2024-11-21:
Found a reason for the MAF, REF_PRESS, DIFF_PRESS and PITOT_PIN the data handler.cpp is setting the pins from the SRC, so somewhere there needs to be a way to if/else the pins due to the configuration.json having "MAF_SRC_IS_PIN" : false, "MAF_SRC_IS_ADC" : true, or do we only need to change pinData["MAF_SRC_IS_PIN"] to pinData["MAF_PIN"] as this matches what is in the pins.json

```
pins.MAF_PIN = pinData["MAF_SRC_IS_PIN"];
pins.REF_PRESSURE_PIN = pinData["PREF_SRC_PIN"];
pins.DIFF_PRESSURE_PIN = pinData["PDIFF_SRC_IS_PIN"];
pins.PITOT_PIN = pinData["PITOT_SRC_IS_PIN"];
```

DeeEmm, 2024-11-21:
Do you mean configuration.h? the config.json file currently only saves values related to the GUI settings.

There is actually a terminology change required here to allow configuration.h to be moved into configuration.json. The existing config.json will likely be renamed to GUI_config or something similar to allow that to happen.

Can definitely guide the code flow using the existing  #defines 

```
#define MAF_SRC_IS_PIN
```

But this will eventually change when the configuration is moved from pre-compile to post-compile.

(likely another can of worms lol)

I think that the SPI stuff can likely be left out as I think that it degrades to defaults when not specified.

KirikauKiwi, 2024-11-21:
The data handler.cpp is trying to set the pins as if they are using analogue sensors,
```
pins.MAF_PIN = pinData["MAF_SRC_IS_PIN"];
pins.REF_PRESSURE_PIN = pinData["PREF_SRC_PIN"];
pins.DIFF_PRESSURE_PIN = pinData["PDIFF_SRC_IS_PIN"];
pins.PITOT_PIN = pinData["PITOT_SRC_IS_PIN"];
```
If it was to use the I2C sensor it would be,
```
pins.MAF_PIN = pinData["MAF_SRC_IS_ADC"];
pins.REF_PRESSURE_PIN = pinData["PREF_SRC_ADC"];
pins.DIFF_PRESSURE_PIN = pinData["PDIFF_SRC_IS_ADC"];
pins.PITOT_PIN = pinData["PITOT_SRC_IS_ADC"];
```
Currently neither MAF_SRC_IS_PIN or MAF_SRC_IS_ADC is in the pins.json. The json has MAF_PIN

DeeEmm, 2024-11-21:
quick and easy fix

```
 }
  if (pins.ORIFICE_BCD_BIT3_PIN > 0 ) {
    pinMode(pins.ORIFICE_BCD_BIT3_PIN, INPUT);   
    _message.serialPrintf("Input ORIFICE_BCD_BIT3_PIN: %d\n", pins.ORIFICE_BCD_BIT3_PIN );
    pinMode(pins.ORIFICE_BCD_BIT3_PIN, INPUT);   
  }
  #ifdef MAF_SRC_IS_PIN
  if ((pins.MAF_PIN > 0)  ) {
    _message.serialPrintf("Input MAF_PIN: %d\n", pins.MAF_PIN );
    pinMode(pins.MAF_PIN, INPUT);   
  }
  #endif
  #ifdef PREF_SRC_PIN
  if (pins.REF_PRESSURE_PIN > 0 ) {
    _message.serialPrintf("Input REF_PRESSURE_PIN: %d\n", pins.REF_PRESSURE_PIN );
    pinMode(pins.REF_PRESSURE_PIN, INPUT);   
  }
  #endif
  #ifdef PDIFF_SRC_IS_PIN
  if (pins.DIFF_PRESSURE_PIN > 0 ) {
    _message.serialPrintf("Input DIFF_PRESSURE_PIN: %d\n", pins.DIFF_PRESSURE_PIN );
    pinMode(pins.DIFF_PRESSURE_PIN, INPUT);   
  }
  #endif
  #ifdef PITOT_SRC_IS_PIN
  if (pins.PITOT_PIN > 0 ) {
    _message.serialPrintf("Input PITOT_PIN: %d\n", pins.PITOT_PIN );
    pinMode(pins.PITOT_PIN, INPUT);   
  }
  #endif
  #ifdef TEMP_SENSOR_TYPE_LINEAR_ANALOG
  if (pins.TEMPERATURE_PIN > 0 ) {
    _message.serialPrintf("Input TEMPERATURE_PIN: %d\n", pins.TEMPERATURE_PIN );
    pinMode(pins.TEMPERATURE_PIN, INPUT);   
  }
  #endif
  #ifdef RELH_SENSOR_TYPE_LINEAR_ANALOG
  if (pins.HUMIDITY_PIN > 0 ) {
    _message.serialPrintf("Input HUMIDITY_PIN: %d\n", pins.HUMIDITY_PIN );
    pinMode(pins.HUMIDITY_PIN, INPUT);   
  }
  #endif
  #ifdef BARO_SENSOR_TYPE_LINEAR_ANALOG
  if (pins.REF_BARO_PIN > 0 ) {
    _message.serialPrintf("Input REF_BARO_PIN: %d\n", pins.REF_BARO_PIN );
    pinMode(pins.REF_BARO_PIN, INPUT);     
  }
  #endif
  if (pins.SERIAL0_RX_PIN > 0 ) {
    _message.serialPrintf("Input SERIAL0_RX_PIN: %d\n", pins.SERIAL0_RX_PIN );
    pinMode(pins.SERIAL0_RX_PIN, INPUT);   
  }
  if (pins.SERIAL2_RX_PIN > 0 ) {
    _message.serialPrintf("Input SERIAL2_RX_PIN: %d\n", pins.SERIAL2_RX_PIN );
    pinMode(pins.SERIAL2_RX_PIN, INPUT);   
  }
```

KirikauKiwi, 2024-11-21:
Currently I have changed the datahandler.cpp to this,
```
pins.MAF_PIN = pinData["MAF_PIN"];
pins.REF_PRESSURE_PIN = pinData["PREF_PIN"];
pins.DIFF_PRESSURE_PIN = pinData["PDIFF_PIN"];
pins.PITOT_PIN = pinData["PITOT_PIN"];
```

DeeEmm, 2024-11-21:
Just mask them using same #ifdefs

At the point when we move the precompile stuff we will end up with something like...

```
"configuration.tempSensorType" :  1,
```

Where the numerical integer represents the existing temp sensor options

Then in constants we define...

```
int TEMP_SENSOR_NOT_USED = 0
int TEMP_SENSOR_TYPE_FIXED_VALUE = 1
int TEMP_SENSOR_TYPE_LINEAR_ANALOG = 2
int TEMP_SENSOR_TYPE_BME280 = 3
int TEMP_SENSOR_TYPE_BME680 = 4
int TEMP_SENSOR_TYPE_SIMPLE_TEMP_DHT11 = 5
```

Which allows us to use switch case logic to process sensor types...

```
switch (configuration.tempSensorType) {

case (TEMP_SENSOR_NOT_USED):

break;

case(TEMP_SENSOR_TYPE_FIXED_VALUE):

break;

etc...
```

We can then also use logic in the initialiseIO method...

if (configuration.tempSensorType == TEMP_SENSOR_TYPE_LINEAR_ANALOG) {
...do initialisation stuff
}
```

DeeEmm, 2024-11-21:
> Currently I have changed the datahandler.cpp to this,
> 
> ```
> pins.MAF_PIN = pinData["MAF_PIN"];
> pins.REF_PRESSURE_PIN = pinData["PREF_PIN"];
> pins.DIFF_PRESSURE_PIN = pinData["PDIFF_PIN"];
> pins.PITOT_PIN = pinData["PITOT_PIN"];
> ```

What part? method?

KirikauKiwi, 2024-11-21:
In the parsePinsData, this
```
/***********************************************************
* @name ParsePinsData
* @brief Updates pins struct from passed JSON data
* @param pinsData JSON document containing pins data
***/
void DataHandler::parsePinsData(StaticJsonDocument<1024> pinData) {

  extern struct CalibrationData calVal;
  extern struct Pins pins;
  extern struct DeviceStatus status;

  Messages _message;

  _message.serialPrintf("Parsing Pins Data \n");    

  status.boardType = pinData["BOARD_TYPE"].as<String>();

  // Store input pin values in struct
  pins.VCC_3V3_PIN = pinData["VCC_3V3_PIN"];
  pins.VCC_5V_PIN = pinData["VCC_5V_PIN"];
  pins.SPEED_SENSOR_PIN = pinData["SPEED_SENSOR_PIN"];
  pins.ORIFICE_BCD_BIT1_PIN = pinData["ORIFICE_BCD_BIT1_PIN"];
  pins.ORIFICE_BCD_BIT2_PIN = pinData["ORIFICE_BCD_BIT2_PIN"];
  pins.ORIFICE_BCD_BIT3_PIN = pinData["ORIFICE_BCD_BIT3_PIN"];
  pins.MAF_PIN = pinData["MAF_PIN"];
  pins.REF_PRESSURE_PIN = pinData["PREF_PIN"];
  pins.DIFF_PRESSURE_PIN = pinData["PDIFF_PIN"];
  pins.PITOT_PIN = pinData["PITOT_PIN"];
  pins.TEMPERATURE_PIN = pinData["TEMPERATURE_PIN"];
  pins.HUMIDITY_PIN = pinData["HUMIDITY_PIN"];
  pins.REF_BARO_PIN = pinData["REF_BARO_PIN"];
  pins.SERIAL0_RX_PIN = pinData["SERIAL0_RX_PIN"];
  pins.SERIAL2_RX_PIN = pinData["SERIAL2_RX_PIN"];
  pins.SDA_PIN = pinData["SDA_PIN"];
  pins.SCL_PIN = pinData["SCL_PIN"];
  pins.SD_CS_PIN = pinData["SD_CS_PIN"];
  pins.SD_MISO_PIN = pinData["SD_MISO_PIN"];
  pins.SD_SCK_PIN = pinData["SD_SCK_PIN"];
  pins.WEMOS_SPARE_PIN_1 = pinData["WEMOS_SPARE_PIN_1"];

  // Store output pin values in struct
  pins.VAC_BANK_1_PIN = pinData["VAC_BANK_1_PIN"];
  pins.VAC_BANK_2_PIN = pinData["VAC_BANK_2_PIN"];
  pins.VAC_BANK_3_PIN = pinData["VAC_BANK_3_PIN"];
  pins.VAC_SPEED_PIN = pinData["VAC_SPEED_PIN"];
  pins.VAC_BLEED_VALVE_PIN = pinData["VAC_BLEED_VALVE_PIN"];
  pins.AVO_STEP_PIN = pinData["AVO_STEP_PIN"];
  pins.AVO_DIR_PIN = pinData["AVO_DIR_PIN"];
  pins.FLOW_VALVE_STEP_PIN = pinData["FLOW_VALVE_STEP_PIN"];
  pins.FLOW_VALVE_DIR_PIN = pinData["FLOW_VALVE_DIR_PIN"];
  pins.SD_MOSI_PIN = pinData["SD_MOSI_PIN"];
  pins.SERIAL0_TX_PIN = pinData["SERIAL0_TX_PIN"];
  pins.SERIAL2_TX_PIN = pinData["SERIAL2_TX_PIN"];
}
```

Here is my modified serial output, you can see the correct values are coming in from the data handler.cpp for the MAF, REF PRESS, DIFF PRESSURE and PITOT.

I am still having problems with the SPI PINS

KirikauKiwi, 2024-11-21:
Currently is is trying to parse this,
```
pins.MAF_PIN = pinData["MAF_SRC_IS_PIN"];
pins.REF_PRESSURE_PIN = pinData["PREF_SRC_PIN"];
pins.DIFF_PRESSURE_PIN = pinData["PDIFF_SRC_IS_PIN"];
pins.PITOT_PIN = pinData["PITOT_SRC_IS_PIN"];
```
the SRC's are not part of the pins.json file

DeeEmm, 2024-11-21:
Haha an unfortunate mistake

The pins.json file is correct, the code parsing it is not

```
    "MAF_PIN" : 0,
    "REF_PRESSURE_PIN" : 0,
    "DIFF_PRESSURE_PIN" : 0,
    "PITOT_PIN" : 0,
    "TEMPERATURE_PIN" : 0,
    "REF_BARO_PIN" : 0,
    "HUMIDITY_PIN" : 0,
```

So it should be 

```
pins.MAF_PIN = pinData["MAF_PIN"];
pins.REF_PRESSURE_PIN = pinData["REF_PRESSURE_PIN"];
pins.DIFF_PRESSURE_PIN = pinData["DIFF_PRESSURE_PIN"];
pins.PITOT_PIN = pinData["PITOT_PIN"];
```

DeeEmm, 2024-11-21:
(There is some disparity in the variable naming convention)

DeeEmm, 2024-11-21:
Here's what I have in configure pins

```

/***********************************************************
 * @brief Configure pins
 * @note Conditional configuration based on board type and hardware
 * @details Initiates I/O from pin configuration in struct and reports to serial monitor
 * @details Messaging identifies I/O in event initialisation crash
 *
 ***/
void Hardware::initaliseIO () {

  Messages _message;

  extern struct Pins pins;
  extern struct DeviceStatus status;

  _message.serialPrintf("Initialising I/O \n");   

  // Set Inputs
  if (pins.VCC_3V3_PIN > 0 ) {
    _message.serialPrintf("Input VCC_3V3_PIN: %d\n", pins.VCC_3V3_PIN );
    pinMode(pins.VCC_3V3_PIN, INPUT);   
  }
  if (pins.VCC_5V_PIN ) {
    _message.serialPrintf("Input VCC_5V_PIN: %d\n", pins.VCC_5V_PIN );  
    pinMode(pins.VCC_5V_PIN, INPUT);
  }  
  if (pins.SPEED_SENSOR_PIN > 0 ) {
    _message.serialPrintf("Input SPEED_SENSOR_PIN: %d\n", pins.SPEED_SENSOR_PIN );
    pinMode(pins.SPEED_SENSOR_PIN, INPUT);   
  }
  if (pins.ORIFICE_BCD_BIT1_PIN > 0 ) {
    _message.serialPrintf("Input ORIFICE_BCD_BIT1_PIN: %d\n", pins.ORIFICE_BCD_BIT1_PIN );
    pinMode(pins.ORIFICE_BCD_BIT1_PIN, INPUT);   
  }
  if (pins.ORIFICE_BCD_BIT2_PIN > 0 ) {
    _message.serialPrintf("Input ORIFICE_BCD_BIT2_PIN: %d\n", pins.ORIFICE_BCD_BIT2_PIN );
    pinMode(pins.ORIFICE_BCD_BIT2_PIN, INPUT);   
  }
  if (pins.ORIFICE_BCD_BIT3_PIN > 0 ) {
    pinMode(pins.ORIFICE_BCD_BIT3_PIN, INPUT);   
    _message.serialPrintf("Input ORIFICE_BCD_BIT3_PIN: %d\n", pins.ORIFICE_BCD_BIT3_PIN );
    pinMode(pins.ORIFICE_BCD_BIT3_PIN, INPUT);   
  }
  #ifdef MAF_SRC_IS_PIN
  if ((pins.MAF_PIN > 0)  ) {
    _message.serialPrintf("Input MAF_PIN: %d\n", pins.MAF_PIN );
    pinMode(pins.MAF_PIN, INPUT);   
  }
  #endif
  #ifdef PREF_SRC_PIN
  if (pins.REF_PRESSURE_PIN > 0 ) {
    _message.serialPrintf("Input REF_PRESSURE_PIN: %d\n", pins.REF_PRESSURE_PIN );
    pinMode(pins.REF_PRESSURE_PIN, INPUT);   
  }
  #endif
  #ifdef PDIFF_SRC_IS_PIN
  if (pins.DIFF_PRESSURE_PIN > 0 ) {
    _message.serialPrintf("Input DIFF_PRESSURE_PIN: %d\n", pins.DIFF_PRESSURE_PIN );
    pinMode(pins.DIFF_PRESSURE_PIN, INPUT);   
  }
  #endif
  #ifdef PITOT_SRC_IS_PIN
  if (pins.PITOT_PIN > 0 ) {
    _message.serialPrintf("Input PITOT_PIN: %d\n", pins.PITOT_PIN );
    pinMode(pins.PITOT_PIN, INPUT);   
  }
  #endif
[... 82 lines cut ...]
    pinMode(pins.FLOW_VALVE_STEP_PIN, OUTPUT);
  }
  if (pins.FLOW_VALVE_DIR_PIN > 0 ) {
    _message.serialPrintf("Output FLOW_VALVE_DIR_PIN: %d\n", pins.FLOW_VALVE_DIR_PIN );
    pinMode(pins.FLOW_VALVE_DIR_PIN, OUTPUT);
  }
  #ifdef SD_CARD_IS_ENABLED
  if (pins.SD_MOSI_PIN > 0 ) {
    _message.serialPrintf("Output SD_MOSI_PIN: %d\n", pins.SD_MOSI_PIN );
    pinMode(pins.SD_MOSI_PIN, OUTPUT);
  }
  #endif
  // if (pins.SERIAL0_TX_PIN > 0 ) {
  //   _message.serialPrintf("Output SERIAL0_TX_PIN: %d\n", pins.SERIAL0_TX_PIN );
  //   pinMode(pins.SERIAL0_TX_PIN, OUTPUT);
  // }
  // if (pins.SERIAL2_TX_PIN > 0 ) {
  //   _message.serialPrintf("Output SERIAL2_TX_PIN: %d\n", pins.SERIAL2_TX_PIN );
  //   pinMode(pins.SERIAL2_TX_PIN, OUTPUT);
  // }

  _message.debugPrintf("I/O Initialised");

}
```

DeeEmm, 2024-11-21:
Test the MAf / baro / MISO / etc using precompile macros (#ifdefs)

Comment out the serial stuff.

Now working

I will look to tidy it up when I move the precompile stuff to post compile.

Note that I'm also testing for pin value to be > 0

Using '-1' and testing for values to be > -1 means that when the structs default to a 0 value, we see that as a false positive. So it is much better to use an arbitrary positive value such as 99 for the test.

At any rate, we never actually assign anything to pin 99, it's just a test

DeeEmm, 2024-11-21:
SORRY: let me correct that

testing for < 99

DeeEmm, 2024-11-21:
And in the structs we initialise everything to 99

DeeEmm, 2024-11-21:
```

/***********************************************************
 * Pin Data
 ***/
struct Pins {
  int VAC_SPEED_PIN = 99;
  int VAC_BLEED_VALVE_PIN = 99;
  int VAC_BANK_1_PIN = 99;
  int VAC_BANK_2_PIN = 99; 
  int VAC_BANK_3_PIN = 99;
  int AVO_STEP_PIN = 99;
  int AVO_DIR_PIN = 99;
  int FLOW_VALVE_STEP_PIN = 99;
  int FLOW_VALVE_DIR_PIN = 99;
  int VCC_3V3_PIN = 99;
  int VCC_5V_PIN = 99;
  int SPEED_SENSOR_PIN = 99;
  int SWIRL_ENCODER_PIN_A = 99;
  int SWIRL_ENCODER_PIN_B = 99;
  int ORIFICE_BCD_BIT1_PIN = 99;
  int ORIFICE_BCD_BIT2_PIN = 99;
  int ORIFICE_BCD_BIT3_PIN = 99;
  int MAF_PIN = 99;
  int REF_PRESSURE_PIN = 99;
  int DIFF_PRESSURE_PIN = 99;
  int PITOT_PIN = 99;
  int TEMPERATURE_PIN = 99;
  int REF_BARO_PIN = 99;
  int HUMIDITY_PIN = 99;
  int SERIAL0_TX_PIN = 99;
  int SERIAL0_RX_PIN = 99;
  int SERIAL2_TX_PIN = 99;
  int SERIAL2_RX_PIN = 99;
  int SDA_PIN = 99;
  int SCL_PIN = 99;
  int SD_CS_PIN = 99;
  int SD_MOSI_PIN = 99;
  int SD_MISO_PIN = 99;             
  int SD_SCK_PIN = 99;
  int WEMOS_SPARE_PIN_1 = 99;
};
```

DeeEmm, 2024-11-21:
So now it is only the pins enabled in pins.config that are being initialised and reported to the serial monitor.

There are no more serial hangs

[image]

DeeEmm, 2024-11-21:
This means that the pins file effective works as an override.

If you only want to set up a MAF sensor, then you only really need to define the one line...

```
{
"MAF_PIN" : 23,
}
```

The remaining items will automatically be ignored as they are initialised to 99

closed 2024-11-21 by DeeEmm

KirikauKiwi, 2024-11-21:
Nice work, I have updated the relevant files and now get past the Initialising IO part of on the serial monitor, I am now encountering a problem that the I2C devices are not being found. It is showing the correct pins for my I2C.
```
Multicast: http://diyfb.local 
Initialising I/O 
Input SDA_PIN: 8
Input SCL_PIN: 9
=== SPIFFS File system info === 
Total space:      1.26 MB 
Total space used: 60.54 KB 
Initialising Hardware 
Scanning for I2C devices...
Found 0 device(s). 
Initialising ADS1115 
[128700][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
ADS1115 Connection failed
```