DIY-Flow-Bench/DIY-Flow-Bench #249: Pitot Improvements

DeeEmm, 2024-11-21:
Additional functionality for pitot probes.

Original discussion - #247 

- [ ] Ability to add an offset to velocity calc to allow for single ended probe (prob also good idea to tune differential probe as well)
- [ ] Ability to select static pressure source (Baro / pRef / pDiff / Pitot)
- [ ] Ability to change measurement units
- [ ] Ability to change between velocity and pressure

KirikauKiwi, 2024-11-22:
> Additional functionality for pitot probes.
> 
> Original discussion - #247
> 
> * [ ]  Ability to add an offset to velocity calc to allow for single ended probe (good idea to tune differential probe as well)
> * [ ]  Ability to select static pressure source (Baro / pRef / pDiff / Pitot)
> * [ ]  Ability to change measurement units
> * [ ]  Ability to change between velocity and pressure

See my comments here https://github.com/DeeEmm/DIY-Flow-Bench/discussions/247#discussioncomment-11346849

* [ ]  Ability to add an offset to velocity calc to allow for single ended probe (good idea to tune differential probe as well)
Why would this be needed? based on my [comments](https://github.com/DeeEmm/DIY-Flow-Bench/discussions/247#discussioncomment-11346849) above.

* [ ]  Ability to select static pressure source (Baro / pRef / pDiff / Pitot)
Why would this be needed? other than if you were eliminating a pressure sensor from your shield and using PRef and PDiff to calculate ∆P and having to manually reconfiguring their connections when wanting to measure in cylinder head port ∆P or velocity

* [ ]  Ability to change measurement units
Yes this is useful

* [ ]  Ability to change between velocity and pressure
Yes this is useful

DeeEmm, 2024-11-22:
There is already a zero function for the pitot differential method, this zeroes down the pressure differential to zero the gauge in the GUI.

The velocity offset would allow the delta calculation to be corrected due to the larger differential between the pressure port and static port. i.e. it applies to the velocity, not the pressure. For a regular pitot probe this would give you the option to calibrate the probe against a reference source. For a single ended probe (read here also 'single ended pressure sensor') it would allow you to correct the flow value error to to compensate for a lower static pressure.

The idea is that the static pressure side of the equation can be taken from any reference, even atmospheric. However once you move the static reference away from the locality of the probe, the velocity calculation will return an incorrect result as the static pressure would be higher / lower than it should be. However if you understand what that offset is and can quantify it, then you can correct the result. It's just maths.

Consider this. In an ideal scenario the static pressure of the port would be stable, it is defined by the depression and port characteristics. Assuming it is stable, it would be no different from any other point of reference that was stable. i.e. Pref, pDiff or Baro. It's just like connecting two ends of a manometer to two different locations, the relationship between those two points is still tied to the fact that the calculation is a differential pressure calculation.

The ability / requirement to change the static reference source depends on a few factors. For a single ended probe on a MAF bench the most accurate reference would be pRef. For a single ended probe on a differential bench it would be pDiff. You also might elect to use Baro, which would replicate probing with a regular manometer, a technique used by some. In this case you would not be looking at velocity, but just pressure, hence the ability to change between velocity and pressure.

Of course the pitot connected to a differential sensor would be the most accurate / best choice, but I don't think that we should exclude other options on the basis that they are inferior. For some it might be the only option available. 

Think of it in the same terms as the Adjusted flow figures. They are not 'real' measurements, but still very useful none the less. Or like a toolbox. Not everyone will use every tool.

KirikauKiwi, 2024-11-24:
Pitot Display
- I cannot get the Pitot to zero, the number is changing colours, however the numerals stay the same.
- The ∆ pressure is only displaying either 0, 1 or 2, you are not able to see the units/ones, or the decimal values.
- Based on the above I suggest to put the m/sec above the ∆ ??? inH20, keeping them centre aligned, I think this will be less cluttered and give more space for the ∆ to be displayed.
- The colour changing number are great to identify what can be interacted with and zeroed, like the green, not so much a fan of the pink/salmon colour it looks disappears in to the background for my, lack of contrast I think.