DIY-Flow-Bench/DIY-Flow-Bench #266: GUI hangs after uploading index / pins file on initial upload

DeeEmm, 2024-11-28:
GUI should automatically redirect to main GUI page after missing files have been uploaded.

Suspect stuck in boot loop and presence of files does not break loop

Check detection of file presence

DeeEmm, 2024-11-28:
Yes the WiFi hang can often be recovered manually.

I believe the hang on initial setup is that the code gets stuck in the boot loop in datahander.cpp

```
 do {
    // capture program pointer in loop and wait for files to be uploaded

        // Process API comms
        if (settings.api_enabled) {        
            if (millis() > status.apiPollTimer) {
                if (Serial.available() > 0) {
                    status.serialData = Serial.read();
                    _api.ParseMessage(status.serialData);
                }
            }                            
        }

        // This is the escape function. When all files are present and loaded we can leave the loop
        if ((status.pinsLoaded == true) && (status.mafLoaded == true) && (SPIFFS.exists("/index.html")) status.doBootLoop = false; 

        // if (status.pinsFilename.isEmpty();

        vTaskDelay( 1 );
    
    } while (status.doBootLoop = true);
```

It should sit there until the missing files have been uploaded. This is to prevent things like pins initialisation until the pin definitions have been loaded.

The following code tests for the files.  

```
        if ((status.pinsLoaded == true) && (status.mafLoaded == true) && (SPIFFS.exists("/index.html")) status.doBootLoop = false; 
```

**status.doBootLoop** is reset in the **fileUpload** method...

```
    if (_data.checkUserFile(PINSFILE)){
      _data.loadPinsData();
      status.pinsLoaded = true;
    } 
    if (_data.checkUserFile(MAFFILE)) {
      // _data.loadMafData();
      status.mafLoaded = true;  
    }
```

It should work, but does not

DeeEmm, 2024-11-28:
The behaviour is that instead of redirecting to the index page once the files have been uploaded, it stays on the file upload page which is served as part of the boot loop process.

KirikauKiwi, 2024-11-28:
This appears to have it working, however an Anysync error now appears on the serial monitor and the diyfb.local appears as a blank white page.

- } while (status.doBootLoop = true); The single = was causing an infinite loop I believe
- status.doBootLoop = false; was missing {}
```
    do {
    // capture program pointer in loop and wait for files to be uploaded

        // Process API comms
        if (config.api_enabled) {        
            if (millis() > status.apiPollTimer) {
                if (Serial.available() > 0) {
                    status.serialData = Serial.read();
                    _api.ParseMessage(status.serialData);
                }
            }                            
        }

        // if (status.pinsFilename.isEmpty();
        if ((status.pinsLoaded == true) && (SPIFFS.exists("/index.html"))) {status.doBootLoop = false;
        }
        vTaskDelay( 1 );
    
    } while (status.doBootLoop == true);

    tempServer->reset();

}
```

Serial monitor
```
Loading Pins Data 
Parsing Pins Data 
Initialising I/O 
Input VCC_5V_PIN: 2
Input SPEED_SENSOR_PIN: 15
Input ORIFICE_BCD_BIT1_PIN: 34
Input ORIFICE_BCD_BIT2_PIN: 36
Input ORIFICE_BCD_BIT3_PIN: 39
Input SDA_PIN: 21
Input SCL_PIN: 22
Output VAC_BANK_1_PIN: 13
Output VAC_BANK_2_PIN: 12
Output VAC_BANK_3_PIN: 14
Output VAC_SPEED_PIN: 25
Output VAC_BLEED_VALVE_PIN: 26
Output AVO_STEP_PIN: 32
Output AVO_DIR_PIN: 33
Output FLOW_VALVE_STEP_PIN: 27
Output FLOW_VALVE_DIR_PIN: 4
=== SPIFFS File system info === 
Total space:      1.26 MB 
Total space used: 81.87 KB 
Initialising Hardware 
Scanning for I2C devices...
Found address: 72 (0x48)
Found address: 118 (0x76)
Found 2 device(s). 
Initialising ADS1115 
ADS1115 Initialised
Hardware Initialised 
Initialising Sensors 
Initialising BME280 
BME280 Initialised
Sensors Initialised 
[ 65621][E][AsyncTCP.cpp:1393] begin(): bind error: -8
Server Running 
```

DeeEmm, 2024-11-28:
> * } while (status.doBootLoop = true); The single = was causing an infinite loop I believe

Good catch

> This appears to have it working, however an Anysync error now appears on the serial monitor and the diyfb.local appears as a blank white page (related to the Anysync error???)

Yes asynctcp is required for the webserver

KirikauKiwi, 2024-11-29:
If I manually rest the board at this point it springs back to life, looks like the Anysync error might be related to a port already in use.

DeeEmm, 2024-11-29:
I suspect that it might be that we are spinning up a temporary webserver in datahandler.cpp to be able to load the file upload page without the Asynctcp code as it is only needed for the asynchronous events. (pushing the JSON data to the page to update the GUI)

So when we come to create the actual webserver in webserver.cpp there is an issue.

The original intent was to create a temp webserver for the boot loop and then destroy it. then create the actual webserver instance when we continue with the initialisation of the actual webserver

Need to follow the flow of the code through to see if the Asyntcp instance is being created after we leave the boot loop

KirikauKiwi, 2024-11-30:
Stopping the web server before the rest has solved the issue, see PR #270

DeeEmm, 2024-12-01:
Great work. Will test this end and report back.

DeeEmm, 2024-12-10:
Confirmed resolved

closed 2024-12-10 by DeeEmm