DIY-Flow-Bench/DIY-Flow-Bench #268: [BUG]GUI Configuration Minimum Bench Pressure Dead band issue

KirikauKiwi, 2024-11-28:
Version: V2.0.RC.8
Build: 24112601
GUI: 24112502

When the "Minimum Bench Pressure" is set to 1 in the GUI configuration, it is causing a dead band across all pressure sensors of around +/- 4 inH2O which is 1 kPa. 

The expected behaviour would be a dead band of +/- 1 inH2O

[image]

The 1 that is coming from the Minimum Bench Pressure is 1 kPa not 1 inH2O. The pressure formulas are in kPA so the Minimum Bench Pressure value will need to be converted in to kPa to correct this.
I didn't do a pull request for this as I don't know if it would prefer to do the conversion elsewhere in the code first to allow for easier unit conversions later, or add the conversion factor to the formula below.
```
// Lets make sure we have a valid value to return - check it is above minimum threshold
	double pDiffComp = fabs(sensorVal);
	if (pDiffComp > config.min_bench_pressure) { 
		return sensorVal;
	} else { 
		return 0.0001; // return small non zero value to prevent divide by zero errors (will be truncated to zero in display)
```

DeeEmm, 2024-11-28:
Yes, the code works in SA units (kPa). Which includes the threshold. Originally we did not have any units associated with this, so it did not really matter. The intent is that the code just reports a zero value to the GUI when it is not running.

For 'correctness' and as we have now added units to this value, we should really update it.

I will take a look to see where is most appropriate to make the change as it is used in a few places.

DeeEmm, 2024-12-10:
I've considered this as part of the #205 update and note that there is also some impact on initial set up. 

To a degree this ties in with #274 'TrimPot tuning' and #236 'config data display', specifically when initially setting up and testing.

I have made improvements to the API and it is now possible to view unfiltered data there, but it would be useful to have a single page of all values. This way we can keep the GUI clean (filtered) and display unfiltered values on a configuration page.

It should be noted that the 'deadband' is intentional on the GUI. The intent is just to ensure that when the bench is off it displays zero. It can however be tuned, hence the min flow value. Ideally this should be renamed as it is now just an arbitrary threshold used in several places.

DeeEmm, 2025-01-10:
Changed comparison to use INH2O.
Increased resolution of min bench pressure / min flow values

closed 2025-01-10 by DeeEmm