DIY-Flow-Bench/DIY-Flow-Bench #273: [FEATURE REQUEST]Does the GUI need to be reloaded when pressing the "Start Bench" or "Stop Bench"

KirikauKiwi, 2024-12-01:
I would like to know if it is necessary to reload the GUI when either of the "Start Bench" or "Stop Bench" buttons are pressed in the GUI?
It would feel smoother to the user if the page didn't reload. 

*For me when the page reloads it goes from what I am viewing.*

[image]

*To this after the page reload*

[image]

DeeEmm, 2024-12-01:
It should not be necessary. It may be redirecting within the event defined in webserver.cpp

KirikauKiwi, 2024-12-01:
Is it not the, window.location.href = '/' in index.html code snippet below that is doing the page reload? It is the same for the Stop Bench button.
```
  document.getElementById('on-button').addEventListener('click', function(){
    console.log('Bench On');
    xhr.open('GET', '/api/bench/on');
    xhr.onload = function() {
      if (xhr.status === 200) window.location.href = '/';
    };
    xhr.send();
```

DeeEmm, 2024-12-01:
Yes, that line is basically saying that if the event receives a http 200 response (load okay), then redirect the page.

This looks like it is in the javascript file,  I was originally referring to the event definitions in webserver.cpp which is the other end of this transmission...

```
  // API request handlers [JSON confirmation response]
  server->on("/api/bench/on", HTTP_GET, [](AsyncWebServerRequest *request){
      Messages _message;
      Hardware _hardware;
      _message.Handler(language.LANG_BENCH_RUNNING);
      _message.debugPrintf("Bench On \n");
      _hardware.benchOn(); 
      request->send(200, "text/html", "{\"bench\":\"on\"}"); 
      });
```

However looking at that now I can see that part does not redirect (some do)

It should be enough to just comment out the onload() function...

```
  document.getElementById('on-button').addEventListener('click', function(){
    console.log('Bench On');
    xhr.open('GET', '/api/bench/on');
    // xhr.onload = function() {
    //   if (xhr.status === 200) window.location.href = '/';
    // };
    xhr.send();
```

DeeEmm, 2024-12-10:
Fixed in Build 24120302

closed 2024-12-10 by DeeEmm