DIY-Flow-Bench/DIY-Flow-Bench #288: [BUG]40-60mV Offset with voltages from the ADS1115

KirikauKiwi, 2024-12-11:
Add this so it can be looked in to with a bit more detail.

I have notice when I connect. my MXVP7007DP sensor that they are showing a higher than expected voltage output when open to the atmosphere, I expect 2.5V and have about 2.6V give or take a few mV across the three pressure sensors. So sitting idle in the picture below you can see my Pitot is sitting at about 1" H2O.

I also noticed that with my 5V supply at 5V I was not able to get to the last couple of lookup points to display on the GUI for the MAF sensor using a 10 kOhm pot, adjusting the supply voltage up to ≈ 5.1V they are then accessible.

In the screen shot below the Pref and MAF are connect to Potentiometers and the Pitot is connected to a pressure sensor,

[image]

First referenced by @blacktop https://github.com/DeeEmm/DIY-Flow-Bench/discussions/147#discussioncomment-11508087

DeeEmm, 2024-12-11:
I've noted a similar general behaviour regarding the top end of the scale. The range seems to be shifted downwards and it is not possible to see FSD even with the pot wound right up.

I suspect that this might be somewhere in the code, however I have not found where. Initially I suspected the transfer function, but I believe that this is in line with the data sheet, although I could just be making the same error transposing the data sheet transfer function.

If the offset solves the issue then it could be a valid solution but it would obviously be better to get an understanding of why this behaviour is happening.

blacktop, 2024-12-11:
Different @black-top

KirikauKiwi, 2024-12-11:
I managed to get a bit of testing done this afternoon with the ads1115, doing a comparison between the DIYFB raw values and voltages. No changes were made to the hardware. I put together a basic bit of code to test the ADS1115 outside of the DIYFB code base.

- Library used for the was the Adafruit ADS1x15 library
- The same RAW ADC value to voltage conversion formula was used other than it being hard coded.

I tested at the voltages voltages below, adjustments to the voltage were made with a 10 turn, 10kOhm potentiometer, all voltages tested to the nearest hundredth uV, e.g. 5.000763 V
-2.50V being the zero crossing point for the MXVP sensor 
- 4.98V being the max lookup value for the Bosch MAF sensor
- 5.00V being the max voltage for all sensors.

The voltages tested were compared to each other with their RAW ADC values and compared to the ideal raw RAW ADC value would be for each of the voltages.
- 2.5V, 13333
- 4.98V, 26560
- 5.00V, 26667

DIYFB RAW ADC values
- 2.50015V, 13654, this is 321 bits above the ideal
- 4.979705V, 27195 this is 635 bits above the ideal
- 5.000763V, 27310 this is 643 bits above the ideal
with the ADS1115 LSB of 187.5 uV this is 60 - 120mV above the ideal voltages.

Testing code RAW ADC values
- 2.500066V, 13314, this is  19 bits below the ideal
- 4.978354V, 26518, this is 42 bits below the ideal
- 5.000319V, 26635, this is 32 bits below the ideal
with the ADS1115 LSB of 187.5 uV this is 3.563 - 7.875mV below the ideal voltages.

What has not been tested.
- Swapping the library used in DIYFB for the ADS1115
- ADS1115 configuration either in DIYFB or the library used in DIYFB.
- Other possibilities not yet considered??? Suggestions please.

I suspect that this points to the library used in DIYFB for the ADS1115 somehow converting the input voltages to a higher RAW ADC value than expected

The ADS1115 code used for testing.
```
#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_ADS1X15.h>

Adafruit_ADS1115 ads1115;

void setup(void)
{
  Serial.begin(9600);
  Serial.println("Hello!");
  
  Serial.println("Getting single-ended readings from AIN0..3");
  Serial.println("ADC Range: +/- 6.144V (1 bit = 187.5uV)");
  ads1115.begin();
}

void loop(void)
{
  int16_t adc0, adc1, adc2, adc3;
  double vadc0, vadc1, vadc2, vadc3;

  adc0 = ads1115.readADC_SingleEnded(0);
  adc1 = ads1115.readADC_SingleEnded(1);
  adc2 = ads1115.readADC_SingleEnded(2);
  adc3 = ads1115.readADC_SingleEnded(3);
  vadc0 = ads1115.readADC_SingleEnded(0) * 6.144 / 32727;
  vadc1 = ads1115.readADC_SingleEnded(1) * 6.144 / 32727;
  vadc2 = ads1115.readADC_SingleEnded(2) * 6.144 / 32727;
  vadc3 = ads1115.readADC_SingleEnded(3) * 6.144 / 32727;
  Serial.print("AIN0: "); Serial.print(adc0); Serial.print(" "); Serial.println(vadc0, 6);
  Serial.print("AIN1: "); Serial.print(adc1); Serial.print(" "); Serial.println(vadc1, 6);
  Serial.print("AIN2: "); Serial.print(adc2); Serial.print(" "); Serial.println(vadc2, 6);
  Serial.print("AIN3: "); Serial.print(adc3); Serial.print(" "); Serial.println(vadc3, 6);
  Serial.println(" ");
  
  delay(1000);
}
```

DeeEmm, 2024-12-11:
It's funny. I had pondered last night whether the issue was ADC related. It seemed like the most likely suspect.

It should be considered that it could also be hardware as well. It is worth probing VCC at the ADC and at the sensor to see if there is any difference as this would produce a similar effect. Testing my board here shows 5v at the buck, 5v at VCC pin on the ADS1115 and 5v at VCC on the pressure sensors. I don't expect it would be any different for you, but worth eliminating.

I've had mixed success with Adafruit and Sparkfun libraries. I seem to recall replacing the BME library due to issues.

If we can pin point the issue to the library, then it would give us a direction to move in. 

I wonder if the wider community have had similar issues with the Adafruit library

DeeEmm, 2024-12-11:
Actually we are not using the adafruit library

DeeEmm, 2024-12-11:
Should be able to swap in following library call...

```
https://github.com/pprasse/ADS1115-Lite.git
```

DeeEmm, 2024-12-11:
This is interesting... https://forum.arduino.cc/t/reading-ads1115-in-continuous-mode/363701/6

DeeEmm, 2024-12-11:
just going though the ADS1115 setup and we are currently set to 128 samples per second, which is massive overkill

in Hardware::begin
```
adc.setSampleRate(ADS1115_REG_CONFIG_DR_128SPS); // Set ADC Sample Rate - 128 SPS, or every 7.8ms  (default)
// adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS); // Set ADC Sample Rate - 8 SPS, or every 7.8ms  (default)
    
```

We can safely change this to 8 samples per second.

DeeEmm, 2024-12-12:
OK. Think I might have it.

in Hardware::getADCVolts where we calculate the voltage from the raw ADC value

```
    case ADS1115:
      volts = rawADCval * config.ADC_GAIN / config.ADC_RANGE;
    break;
``

We are using structs to store the ADC gain and range. Substitute these with the actual values...

```
      volts = rawADCval * 6.144 / 32767;
```

This is what I got originally...

```
A:2.376049:3.536241:0.536149:0.536149
a:12976:19312:2928:2928
```

And this is what I get with the change made...

```
A:2.430074:3.621111:0.546017:0.552017
a:12976:19312:2912:2944
```

Note that there is a difference in the calculated voltage, but the Raw value is the same.

Unsure why this is. I have quickly experimented with using different datatypes, for example changing int to int_16 as this is one of the differences with your test code, but it does not change the outcome. 

Can you make the change above and re-test

KirikauKiwi, 2024-12-12:
I knew you were not a fan of the Adafruit libraries (often bloated), but it was the easiest one for me to grab with out having to do much to test with.

Many people may have not noticed the difference depending on what they were measuring, I have seen many experienced builders using spirt and laser levels that are clearly out of whack, but they still believe it to be true.

> It should be considered that it could also be hardware as well. It is worth probing VCC at the ADC and at the sensor to see if there is any difference as this would produce a similar effect. Testing [...]

I will test the hardware voltages further tomorrow, my initial check have shown the externally measured voltages to be slightly higher than what the voltages are shown within the DIYFB API, e.g when looking at the MAF sensor with it 4.98V lookup point, the API was at about 5.03V and my scope was saying about 5.1V, but the scope voltage depends on wether I am looking at the bottom, top or a mean value, I couldn't find a measurement point that aligned.

This is with the pot adjusted to give a reading of 2.500xxxV on the MAF Sensor pin of the ADS1115.

[image]

This is my 5V supply line, bumped up over 5V to allow the FS reading of the MAF lookup values.

[image]

Tis is with the Pot adjusted to give the MAF FS reading of 1805.5KGH, 4.98V on the look up table.

[image]

DeeEmm, 2024-12-12:
What scope are you using? Looks a lot nicer than my Hantek 😄 (I'm too cheap to buy a proper scope). I did have a nice scope that I left back in the UK when I moved here some 20+ years ago (too big and heavy), just wasn't doing much electronics at the time.

> my initial check have shown the externally measured voltages to be slightly higher than what the voltages are shown within the DIYFB API, e.g when looking at the MAF sensor with it 4.98V lookup [...]

This raises am interesting question about initial setup. Originally the intent is to use the trimpot settings to being any deviation back in range, however this depends on where you are viewing the voltage. 

From the ADCs perspective, it will be accurate if VCC at the ADC is in spec 
From the Pressure sensors perspective, it will be accurate if VCC at the sensor is in spec
From the MCUs perspective, it views I2C data over the bus, so should not be impacted by voltage deviations, however if we utilise the internal ADC of the MCU to measure the VCC_5V_PIN, then MCU will be accurate if VCC at the MCU is in spec.

My measurements above are only a quick look with a DMM. I will set the scope up and take a harder look at this, but we may have a fundamental problem. 

Based on the above we need to add a trimpot for the ADC itself. OR we set the 5V bus so that VCC at the ADC is correct and then trim the values for each sensor + MCU.

I think that the latter is the better solution.

- MCU does not care if 5v supply is nigher as it has on board regulator, it just needs to be trimmed so that the VCC_5V_PIN reading is accurate
- Likewise pressure sensors do not care what VCC is, but it needs to be identified and included within the transfer function as the pressure sensor is ratiometric across the acceptable VSS range. So the trim for each pressure sensor needs to be the difference between the measured value (VCC_5V_PIN) and VCC at each sensor.

So what we end up with is a commissioning procedure that looks like this...

- Set the 5v buck so that you get 5v at the VCC pin of the ADS1115 
- Set the VCC_5V_TRIMPOT value so that the MCU reports 5v
- Set PREF_MV_TRIMPOT to correct voltage at VCC pin on pRef sensor
- Set PDIFF_MV_TRIMPOT to correct voltage at VCC pin on pDiff sensor
- Set PITOT_MV_TRIMPOT to correct voltage at VCC pin on pitot sensor

Furthermore if we look at a typical transfer function we can se that the voltage compoinent is taken from

KirikauKiwi, 2024-12-12:
Test comparing,  volts = rawADCval * config.ADC_GAIN / config.ADC_RANGE; VS volts = rawADCval * 6.144 / 32767;

DIYFB RAW ADC values with config.ADC_GAIN / config.ADC_RANGE

- 2.500015V, 13654, this is 321 bits above the ideal
- 4.980071V, 27196 this is 636 bits above the ideal
- 5.039949V, 27542 this is 662 bits above the ideal

DIYFB RAW ADC values with 6.144 / 32767

- 2.500951V, 13336, this is 3 bits above the ideal
- 4.980527V, 26560 this is 0 bits above the ideal
- 5.033216V, 26844 this is 36 bits above the ideal

DeeEmm, 2024-12-12:
All of this is interesting as it also relates to #279 - Add probe points on PCB

So in addition to 0v / 3.3v and 5v we also have:

- VCC on ADS1115
- VCC pin on pRef sensor
- VCC pin on pDiff sensor
- VCC pin on pitot sensor

DeeEmm, 2024-12-12:
> Test comparing, volts = rawADCval * config.ADC_GAIN / config.ADC_RANGE; VS volts = rawADCval * 6.144 / 32767;
>
> DIYFB RAW ADC values with config.ADC_GAIN / config.ADC_RANGE
>
> * 2.500015V, 13654, this [...]

Looks like that's it 👍

KirikauKiwi, 2024-12-12:
> What scope are you using? Looks a lot nicer than my Hantek 😄 (I'm too cheap to buy a proper scope)

It was an early Christmas present to myself, it is the baby of the picoscope range [picoscope 2204A](https://www.picotech.com/oscilloscope/2000/picoscope-2000-overview) Black Friday delivered to my door for a shade under £90, regular price £115

DeeEmm, 2024-12-12:
Oooh very nice. Might be time for an upgrade. Looks heaps better than the Hantek which is pretty basic...

[image]

KirikauKiwi, 2024-12-12:
The Picoscope [software](https://www.picotech.com/downloads) has a demo mode. It has been a while since I have used a scope I am pretty impressed with it and yeah the software seems slick, very glad that I went with a USB style scope of the like of the Rigol's etc.
Would love one of their automotive scopes but don't do that stuff anymore since moving from NZ.

DeeEmm, 2024-12-12:
> The Picoscope [software](https://www.picotech.com/downloads) has a demo mode.

Cool. Will have a play

KirikauKiwi, 2024-12-12:
> Looks like that's it 👍

It is a weird one, on paper it should be doing the same math as the values being pulled in to the formula as they are the same as in those in the hardcode. 🤷‍♂️

Main thing at least it was not too difficult to resolve!!!!!!

KirikauKiwi, 2024-12-12:
<details><summary>Did a bit of a search on the possible reasons, I am don't know enough about C++, to me the maths should work either way.</summary>
Got it! Since rawADCval is defined as an int, you'll need to ensure the division is done in floating-point arithmetic to avoid any precision issues. Here’s how you can modify the expressions to ensure accurate calculations:
```
cpp
double volts1 = rawADCval * ADC_GAIN / static_cast<double>(ADC_RANGE);
double volts2 = rawADCval * 6.144 / 32767.0;
```
By casting ADC_RANGE to double in volts1, and using 32767.0 in volts2, you ensure that the division is performed with floating-point precision. This way, both calculations should yield the same result.
So, your final code should look something like this:
```
cpp
int rawADCval; // Ensure this is properly initialized
double ADC_GAIN = 6.144;
int ADC_RANGE = 32767;

double volts1 = rawADCval * ADC_GAIN / static_cast<double>(ADC_RANGE);
double volts2 = rawADCval * 6.144 / 32767.0;
```
Both volts1 and volts2 should now give the same voltage values. If they still differ, there might be other factors at play, such as how rawADCval is obtained or any compiler-specific optimizations. 🛠️🔍
</details>

DeeEmm, 2024-12-12:
> > Looks like that's it 👍
> 
> It is a weird one, on paper it should be doing the same math as the values being pulled in to the formula as they are the same as those in the hardcode. 🤷‍♂️
> 
> Main thing at least it was not too difficult to resolve!!!!!!

Yes, could have been infinitely more complex 😆 

I'll close this out now. Changes are now in DEV

DeeEmm, 2024-12-12:
> you'll need to ensure the division is done in floating-point arithmetic to avoid any precision issues.

Ahh, yes makes sense. I wonder if we have any other instances of the same?

Would be worth auditing to see if we have the same issue elsewhere

DeeEmm, 2024-12-12:
Similar potential issue...

```
		case LINEAR_ANALOG : {
			long refPressRaw = analogRead(pins.REF_PRESSURE_PIN);
			sensorVolts = refPressRaw * (_hardware.get3v3SupplyVolts() / 4095.0);
			break;
		}
```

refPressRaw is a long

closed 2024-12-12 by DeeEmm

DeeEmm, 2024-12-12:
OK updated for all LINEAR_ANALOG cases. Changes are on the current branch I'm working on so they will be brought in on my next PR.

```
		case LINEAR_ANALOG: {
			long mafRaw = analogRead(pins.MAF_PIN);
			sensorVolts = static_cast<double>(mafRaw) * (_hardware.get3v3SupplyVolts() / 4095.00F);
			break;
		}
```

DeeEmm, 2024-12-12:
Also found in SensorData struct...

```
  long BaroPA = 0;
```

Have changed to double. This is used in Air density calcs which is obviously part of mass - volumetric conversion

DeeEmm, 2024-12-12:
Same in 5v calcs...

```
    long rawVoltageValue = analogRead(pins.VCC_5V_PIN);  
    vcc5vSupplyVolts = (2 * static_cast<double>(rawVoltageValue) * 0.805860805860806F) ;
    return vcc5vSupplyVolts + config.VCC_5V_TRIMPOT;
```

KirikauKiwi, 2024-12-12:
Having a double check of everything and the raw adc values and the voltages are spot on how ever the GUI for the MAF is not displaying the correct lookup for 4.98V which should be 1805.5 KGH.

[image]

DeeEmm, 2024-12-12:
> Having a double check of everything and the raw adc values and the voltages are spot on, however on the GUI, the MAF value is not displaying the correct lookup for 4.98V which should be 1805.5 KGH.

What MAF are you using?

DeeEmm, 2024-12-12:
ahh ok I assume the Bosch unit.based on the value

4.97 is not seen as equal or greater than the max value, so the actual value is interpolated.

It uses linear interpolation to scale the value between the current value, and the next lowest value.

```
		} else if (Key > refValue && rowNum > 0) { // The value is somewhere between this and the previous key value so let's use linear interpolation to calculate the actual value: 

			u_int KeyPrev = status.mafLookupTable[rowNum - 1][0]; // Key value for the previous row (x1)
			u_int ValPrev = status.mafLookupTable[rowNum - 1][1]; // Flow value for the previous row (y1)

			// Linear interpolation y = y1 + (x-x1)((y2-y1)/(x2-x1)) where x1+y1 are coord1 and x2_y2 are coord2
			lookupValue = ValPrev + (refValue - KeyPrev)*((Val-ValPrev)/(Key-KeyPrev));
			sensorVal.MafLookup = lookupValue;
			break;   

		} else if (rowNum == status.mafDataTableRows && refValue > Key) { //we're at the largest value, this must be it
			lookupValue = Val; // lets use the associated lookup value
			sensorVal.MafLookup = Val;
			// TODO status message MAX FLOW
			break;
		}
```

However it looks as if the comparison is just returning the previous value.

I can also see that there are similar issues as above where we are mixing doubles and ints. I will sit down later and take a look at changing those. Need to do a spot of DIY before the wife comes home 😆

KirikauKiwi, 2024-12-12:
Well spotted, yes I have the Bosch MAF sensor json currently loaded.
I am only 3 bits shy, 562.5 uV is not bad for manually adjusting a potentiometer 😄 

It only updates to 1805.5 KGH once I get above 5.03V.

[image]

black-top, 2024-12-12:
Those software issues might be a significant contributor to the fluctuating values and inaccuracies, but I think there are hardware issues too. Measuring voltage directly on the ADC board A0 (MAF input) I see 1.06V, if I disconnect MAF signal wire from the main board and measure MAF signal I see 1.00V, and on the ADC board A0 0.06V, so somewhere at the board that additional 0.06V is induced. I tested at different ground points at PCB for the oscilloscope, but don't see differences, so probably not a ground loop. I also soldered 10uf tantalum capacitor directly on ADC power pins. 
Maybe ADC traces go near power traces and get induced voltage.
[image]

KirikauKiwi, 2024-12-12:
I have made some measurements with my setup as I am on a prototype/bread board with wire jumpers.

- The first two voltages (top to bottom) from the API connected as a working system e.g. MAF voltage is connected to A0, I adjusted the pot to get as close to 1.06V as I could. (Screen shot 1)
- The third and fourth voltages from the API are measuring the MAF output voltage not connected to A0, with the scope displaying the MAF voltage. (first photo)
- The fifth and sixth voltages from the API are with the MAF disconnnected from A0 the scope measuring A0. (Screen shot 2)

Screen shot 1.

[image]

Screen Shot 2.

[image]

black-top, 2024-12-12:
@KirikauKiwi  Does  "MAF_MV_TRIMPOT" work for you correctly?  From yesterday testing, I don't see much change at flow measurements with  0 or -0.18  "MAF_MV_TRIMPOT".

DeeEmm, 2024-12-12:
> if I disconnect MAF signal wire from the main board and measure MAF signal I see 1.00V, and on the ADC board A0 0.06V, so somewhere at the board that additional 0.06V is induced. 

You are measuring the floating ADC input of the ADS1115 so you are not really seeing the voltage of the PCB, just the inside of the ADC1115.  You cannot really measure the ADC in this manner.

The pressure sensor connects direct to the ADC via a 10k current limiting resistor and tied to ground by C18 a 330nf capacitor which acts as a filter so there is no possibility that the line is getting pulled up. The voltage you see is the ADCS1115, which may not necessarily be an issue.

Did you apply the changes discussed above? They address the 0.06v issue, which is related to an issue in the ADC code

> I tested at different ground points at PCB for the oscilloscope, but don't see differences, so probably not a ground loop. I also soldered 10uf tantalum capacitor directly on ADC power pins. Maybe ADC traces go near power traces and get induced voltage. 

Induced voltages should not be an issue with a DC only system. This is normally more of an issue with AC based systems or signals, especially high frequency signals where the issue presents itself as noise.

KirikauKiwi, 2024-12-12:
> @KirikauKiwi Does "MAF_MV_TRIMPOT" work for you correctly? From yesterday testing, I don't see much change at flow measurements with 0 or -0.18 "MAF_MV_TRIMPOT".

Yes they are working for me.

Are you deleting the existing configuration.json file via the Configuration GUI page and then uploading the new configuration.json file?

black-top, 2024-12-12:
Yes, yesterday I uploaded latest dev with all .json files,  CFM values are almost identical without trim pot adjustment, the results of yesterday measurements:

https://github.com/DeeEmm/DIY-Flow-Bench/discussions/147#discussioncomment-11534482

black-top, 2024-12-12:
@DeeEmm  I don't see the release folder at the latest DEV for upload files, or do I need to update differently this time?

DeeEmm, 2024-12-12:
If you go to the main page and click the dropdown,  you will see a list of current branches. 

[image]

You can also navigate directly - https://github.com/DeeEmm/DIY-Flow-Bench/tree/DEV

black-top, 2024-12-13:
@DeeEmm  Uploaded latest dev, will do testing soon.
This is the problem I was mentioning before.
Pref and Pitot after upload config are the same, even though they are set differently at config. 
[image]

DeeEmm, 2024-12-13:
ok will take a look. likely a cut and paste error on my part

DeeEmm, 2024-12-13:
As suspected, a cut and paste error.

It is fixed in DEV in commit [ce47e0a](https://github.com/DeeEmm/DIY-Flow-Bench/commit/ce47e0a949f000e787c5786b7ba444d3069d4c64)