DIY-Flow-Bench/DIY-Flow-Bench #292: [BUG]preload.html, Temp Webserver hangs with json file uploads

KirikauKiwi, 2024-12-12:
Since the major changes post 205 Build, I have notice that when uploading the four json files via the preload.html page that the webpage is often not refreshing. The serial monitor shows the file as parsed.

Reloading the page in browser often get it to update the page, sometimes not. My experience is that it is more problematic with the MAF.json file compared to the other three json files.
When attempting to upload the MAF.json the web url is often stuck at diyfb.local/api????/upload, apologies as lost my copy of it from my clipboard.

This is similar to #266 and the boot loop.

I have tried increasing the delay of vTaskDelay from 1 to 1000, it doesn't appear to help as it did with increasing the delay after tempServer ->end

DeeEmm, 2024-12-12:
Take a look at your serial monitor as you upload the file in the browser. I suspect that you will see a panic related to  asyncTCP.

The file appears to upload okay, but it is crashing just after file upload for som reason.

I think that this is the reason that the redirect does not get actioned as the browser connection is closed before it is sent.

I've not got to the bottom of why this is happening yet. I suspect that it is memory related, just haven't figured out how as increasing the loop task size does not fix the issue

KirikauKiwi, 2024-12-12:
Looking at the boot loop would it be worth declaring the objects outside of the loop along with the web server initialisation as repeated creation of the objects and web server initialisation within the loop could possibly cause memory management issues or excessive resource usage.
```
void DataHandler::bootLoop()
{
    extern struct DeviceStatus status;
    extern struct BenchSettings settings;
    extern struct SensorData sensorVal;

    Hardware _hardware;
    Calculations _calculations;
    Messages _message;
    Webserver _webserver;
    API _api;
    DataHandler _data;

    bool shouldReboot = false;

    // if the weberver is already running skip sever initialisation
    if (!status.webserverIsRunning) {

      _message.serialPrintf("Spinning up temporary web server\n");
      _message.serialPrintf("View browser to upload missing files\n");

      tempServer = new AsyncWebServer(80);
      // tempServerEvents = new AsyncEventSource("/events");

      // Upload request handler
      tempServer->on("/api/file/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
          Messages _message;
          _message.debugPrintf("/api/file/upload \n");
          // request->send(200);
          },
          _webserver.fileUpload); 

      // Index page request handler
      tempServer->on("/", HTTP_ANY, [](AsyncWebServerRequest *request){
          Language language;
          Webserver _webserver;
              request->send_P(200, "text/html", language.LANG_INDEX_HTML, _webserver.processLandingPageTemplate); 
          });

      tempServer->onFileUpload(_webserver.fileUpload);
      // tempServer->addHandler(tempServerEvents);
      tempServer->begin();

      _message.serialPrintf("Waiting...\n");
    
    }

    do {
    // capture program pointer in loop and wait for files to be uploaded

        // Process API comms
        if (settings.api_enabled) {        
            if (millis() > status.apiPollTimer) {
                if (Serial.available() > 0) {
                    status.serialData = Serial.read();
                    _api.ParseMessage(status.serialData);
                }
            }                            
        }

        // This is the escape function. When all files are present and loaded we can leave the loop
        if ((status.configLoaded == true) && (status.pinsLoaded == true) && (status.mafLoaded == true) && (status.GUIexists == true) ){
          status.doBootLoop = false; 
          break;
        } 
        // if (status.pinsFilename.isEmpty();

        vTaskDelay( 1 );
    
    } while (status.doBootLoop == true);

    // if the weberver is already running skip sever reset
    if (!status.webserverIsRunning) {
      tempServer->end();  // Stops the server and releases the port
      delay(1000);  // 1000ms delay to ensure the port is released
      tempServer->reset();
    }
}
```

DeeEmm, 2024-12-13:
There are no declarations within the loop itself.

```
    do {
    // capture program pointer in loop and wait for files to be uploaded

        // Process API comms
        if (settings.api_enabled) {        
            if (millis() > status.apiPollTimer) {
                if (Serial.available() > 0) {
                    status.serialData = Serial.read();
                    _api.ParseMessage(status.serialData);
                }
            }                            
        }

        // This is the escape function. When all files are present and loaded we can leave the loop
        if ((status.configLoaded == true) && (status.pinsLoaded == true) && (status.mafLoaded == true) && (status.GUIexists == true) ){
          status.doBootLoop = false; 
          break;
        } 
        // if (status.pinsFilename.isEmpty();

        vTaskDelay( 1 );
    
    } while (status.doBootLoop == true);

```

The other declarations are masked by the webserverIsRunning variable

```
    // if the weberver is already running skip sever initialisation
    if (!status.webserverIsRunning) {
        ...
    }
```

This way if we are already running and then delete a file, we do not recreate the webserver instance, we simply use the one that is already running.

It might be worth considering a reload after the final file is uploaded to clear down the memory.

Did you check to see whether you also see the asyncTCP panic?

DeeEmm, 2024-12-13:
``` this is what I am seeing in the serial monitor when uploading files

--- Terminal on /dev/cu.usbserial-144430 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
CONFIG file Found: 
Loading Configuration 
Loading Configuration 
PINS file Found: /PINS_WEMOS_D1_R32.json
Loading Pins Data 
Parsing Pins Data 
Loading Pins Data 
Parsing Pins Data 
MAF file Found: /MAF_DELPHI_AF10118.json
Loading MAF Data 

***ERROR*** A stack overflow in task async_tcp has been detected.

Backtrace:0x40083a85:0x3ffdfbe00x4008bb9d:0x3ffdfc00 0x4008f071:0x3ffdfc20 0x4008d85f:0x3ffdfca0 0x4008bcac:0x3ffdfcd0 0x4008bc5c:0x0000007c  |<-CORRUPTED

  #0  0x40083a85:0x3ffdfbe0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008bb9d:0x3ffdfc00 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x4008f071:0x3ffdfc20 in vApplicationStackOverflowHook at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:394
  #3  0x4008d85f:0x3ffdfca0 in vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3504
  #4  0x4008bcac:0x3ffdfcd0 in _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
  #5  0x4008bc5c:0x0000007c in _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: 0000000000000000

Rebooting...

```

DeeEmm, 2024-12-13:
Very limited info on this error

```
***ERROR*** A stack overflow in task async_tcp has been detected.
```

KirikauKiwi, 2024-12-13:
> Did you check to see whether you also see the asyncTCP panic? 

Yes getting a very similar asyncTCP panic to the one you posted above. There is not a lot of information available when I have been looking in to it.

The panic happens most often with the MAF.json, and most of the time for me with the other three json files. At random time some of the smaller json files will load/parse as expected.

The best I have found so far has been this,
The corruption marker in the backtrace, particularly the 0x0000007c |<-CORRUPTED, indicates that the stack memory has been overwritten or corrupted. This often occurs due to:
- Stack Overflow: The task's stack size is too small for the operations it needs to perform, leading to overwriting memory regions.
- Memory Corruption: Another part of the program might be writing into the task's stack space inadvertently, causing corruption.

KirikauKiwi, 2024-12-13:
I noticed you have added in some code to check for the stack over flows via relevant hooks, has this highlighted anything useful?

DeeEmm, 2024-12-13:
Not yet, I've not been able to spend much time on this so far.

The issue is definitely memory related, the overflow indicates that memory is being written past defined registers, which kind of implies that the memory assigned to asyncTCP is too small. But I cannot find anywhere that refers to how to manage the memory for the asyncTCP task. Perhaps this is a wild red herring chase.

I did find one issue that sounds like it might be related but not had a chance to digest it yet

https://github.com/me-no-dev/ESPAsyncWebServer/issues/1381

DeeEmm, 2024-12-13:
ok, found the issue...

in Webserver::fileUpload we check to see what file was uploaded and then do two things.

1 - We load the file if appropriate
2 - We set a variable to say that the code is loaded. 

This is how we break out of the boot loop.

(commented out code below)

```
  if (final)  {
    _message.debugPrintf("Upload Complete: %s, %u bytes\n", filename.c_str(), file_size);
    request->_tempFile.close();

    // if (_data.checkUserFile(CONFIGFILE)){
    //   _data.loadConfiguration();
    //   status.configLoaded = true;
    //   redirectURL = "/";
    //   // redirectURL = "/?view=upload";
    // } 
    // if (_data.checkUserFile(PINSFILE)){
    //   _data.loadPinsData();
    //   status.pinsLoaded = true;
    //   redirectURL = "/";
    //   // redirectURL = "/?view=upload";
    // } 
    // if (_data.checkUserFile(MAFFILE)) {
    //   _data.loadMAFData();
    //   status.mafLoaded = true;  
    //   redirectURL = "/";
    //   // redirectURL = "/?view=upload";
    // }
    // if (_data.checkUserFile(INDEXFILE)) {
    //   status.GUIexists = true;  
    //   redirectURL = "/";
    // }

    request->redirect(redirectURL);
  }
}
```

Obviously executing blocking / delaying code at this point in the async process is casing the issue. So I suspect that this needs to be moved outside of Webserver::fileUpload

I did play with this before but ran into some issues

Will take another look at how to achieve this without blocking the async process

DeeEmm, 2024-12-13:
Ok. I've managed to isolate this to a single line.

Removing the redirectURL assignment seems to have 'fixed' the issue.

There is now a little oddness in that the browser does not redirect to the bootloop, but this in itself is not an issue.

For example. We delete the MAF file. The browser stays on the file manager dialog, but we then decide to close the file managed dialog and carry on using DIYFB normally.

The MAF data is already in memory and will remain there until the MCU is rebooted.

At reboot the boot sequence detect that the MAF file is missing and sends code execution into the boot loop.

DeeEmm, 2024-12-13:
we now have this...

```

  if (final)  {
    _message.debugPrintf("Upload Complete: %s, %u bytes\n", filename.c_str(), file_size);
    request->_tempFile.close();

    if (_data.checkUserFile(CONFIGFILE)){
      _data.loadConfiguration();
      status.configLoaded = true;
    } 
    if (_data.checkUserFile(PINSFILE)){
      _data.loadPinsData();
      status.pinsLoaded = true;
    } 
    if (_data.checkUserFile(MAFFILE)) {
      _data.loadMAFData();
      status.mafLoaded = true;  
    }
    if (_data.checkUserFile(INDEXFILE)) {
      status.GUIexists = true;  
    }

    request->redirect(redirectURL);
  }

```

DeeEmm, 2025-01-09:
Going to close this out as file upload replaced by NVM data

closed 2025-01-09 by DeeEmm