DIY-Flow-Bench/DIY-Flow-Bench #293: [BUG]Latcency

KirikauKiwi, 2024-12-12:
Version: V2.0-RC8
Build: 2412120056

Adding this here as struggling to note where we discussed this yesterday.

Since the updates to DEV branch post 205-post-compile-configuration, there is a noticeable latency in the GUI updating, most obvious with the BME sensor data, the data from the pressure sensors appears to  updating more frequently.

Whilst working on #288 I have noticed the latency while using the API too, at times taking more than one minute for a response from the API.

KirikauKiwi, 2024-12-12:
A little more information. The latency on the GUI is still present, but when switching the the relay for Start Bench and Stop Bench this is working as normal with no noticeable difference to the previous builds.

DeeEmm, 2024-12-12:
The on / off commands are sent direct to the ESP via HTTP request sent by XHR.

GUI -> javascript event listener -> Button event -> XHR request -> http://.../API/bench/on -> AsyncWebServerRequest

This works with no latency

Similarly the Server Side Events going in the opposite direction also work fine

DataHandler::buildSSEJsonData -> JSON SSE -> browser -> addEventListener('JSON_DATA') -> GUI update

There is a difference in refresh rates between environmental and sensor data, simply as the environmental data does not change as fast.

```
int ADC_SCAN_DELAY_MS = 250;
int BME280_SCAN_DELAY_MS =  1000;
```

I'm not seeing any lag or latency to the refresh values here, they appear to be working at normal rate, This is echoed by bench on/off response.

What I am seeing a delay in is the operation of the API. I'm not seeing 60 second delays but still maybe 20 seconds in some cases. Still way higher than it should be. I am also seeing some increase in page load times and initial GUI population

The latency has increased since the post-compile stuff was done. This would be related to the additional stack memory used. which has basically doubled from what it was before.

It should be noted that API calls are made in the loop, whereas all other actions are made asynchronously.

Sensor updates are done via xtasks and semaphore handoff - each sensor task (ADC / Enviro) are carried out on its own dedicated core.

Outgoing server tasks (SSE) is asynchronous

Same thing with GUI browser tasks (XHE)

There might be a possibility that upgrading the arduinoJSON7 can improve the situation. This is because it uses dynamic memory allocation for JSON documents, whereas at the moment we need to manually define them. This means tha we need to cater for the 'worst case scenario' i.e. the largest file which in our case is a 27k MAF file. It also may increase the issue as it takes up more memory than V6.

We can also spend some time going through code optimisation to see if we can make gains there although this is usually done last thing

KirikauKiwi, 2024-12-12:
I am experiencing very odd behaviour with Chrome or Safari, very slow to update and at times when you reload the page the tiles remain blank for a longtime before being populated

Here is a screen shot of two Safari Browser windows and one from Chrome, the Safari browser window that has the same values as the Chrome browser I opened as a private window, the chrome window is not an incognito window.

All three Tiles update when the voltages are changed, but MAF difference remains.

[image]

DeeEmm, 2024-12-12:
> I am experiencing very odd behaviour with Chrome or Safari, very slow to update and at times when you reload the page the tiles remain blank for a longtime before being populated
> 

Yes this is similar to what I mentioned above,  increase in page load times and initial GUI population.

> 
> All three Tiles update when the voltages are changed, but MAF difference remains.
> 

Sorry I don't understand what you are saying. Can you please explain what the 'MAF difference' is.

KirikauKiwi, 2024-12-13:
Sure no problem to clarify.

I have two browser windows open simultaneously with diyfb.local as the url.
The MAF values being displayed are different in the two browser windows?? how can this be if they are using the same HTML to access the same "FLOW" data from the ESP32???

PRef, PDiff and Pitot sensors follow each other as expected in both browser windows. The MAF values will both change if I adjust the input voltage but the Values displayed are different???

KirikauKiwi, 2024-12-13:
Ignore that LOL.

It is obviously because the MAF tiles are not set the same LOL!!!!!!! Another one of those Dumb moments!!!!!!

DeeEmm, 2024-12-13:
Haha. Screen blindness. Cannot see the wood for the trees. Happens to the best of us. 😆

DeeEmm, 2024-12-13:
So browser latency might be related to the javascript.

The pipe between the MCU and the Browser is fine

It is only the GUI that is sluggish, not the data.

I have parked some stuff that I was working on by commenting it out (GUI serial monitor). Might be worth stripping that out of the code and seeing if that makes a difference (it's right at the bottom)

```

// Serial monitor modal
// https://developer.chrome.com/docs/capabilities/serial
// var port, textEncoder, writableStreamClosed, writer;
// async function connectSerial() {

//     try {
      
//         // Prompt user to select any serial port.
//         const port = await navigator.serial.requestPort();
//         // const port = await navigator.serial.getPort()
//         await port.open({ baudRate: 115200 });

//         textEncoder = new TextEncoderStream();
//         writableStreamClosed = textEncoder.readable.pipeTo(port.writable);

//         writer = textEncoder.writable.getWriter();
//         listenToPort();
//     } catch {
//         alert("Serial Connection Failed");
//     }
// }
// async function sendCharacterNumber() {
//     document.getElementById("lineToSend").value = String.fromCharCode(document.getElementById("lineToSend").value);
// }
// async function sendSerialLine() {
//     dataToSend = document.getElementById("lineToSend").value;
//     if (document.getElementById("addLine").checked == true) dataToSend = dataToSend + "\r\n";
//     if (document.getElementById("echoOn").checked == true) appendToTerminal("> " + dataToSend);
//     await writer.write(dataToSend);
// }
// async function listenToPort() {
//     const textDecoder = new TextDecoderStream();
//     const readableStreamClosed = port.readable.pipeTo(textDecoder.writable);
//     const reader = textDecoder.readable.getReader();
//     // Listen to data coming from the serial device.
//     while (true) {
//         const { value, done } = await reader.read();
//         if (done) {
//             // Allow the serial port to be closed later.
//             reader.releaseLock();
//             break;
//         }
//         // value is a string.
//         appendToTerminal(value);
//     }
// }
// const serialResultsDiv = document.getElementById("serialResults");
// async function appendToTerminal(newStuff) {
//     serialResultsDiv.innerHTML += newStuff;
//     if (serialResultsDiv.innerHTML.length > 3000) serialResultsDiv.innerHTML = serialResultsDiv.innerHTML.slice(serialResultsDiv.innerHTML.length - 3000);

//     //scroll down to bottom of div
//     serialResultsDiv.scrollTop = serialResultsDiv.scrollHeight;
// }
// document.getElementById("lineToSend").addEventListener("keyup", async function (event) {
//     if (event.keyCode === 13) {
//         sendSerialLine();
//     }
// })
// document.getElementById("baud").value = (localStorage.baud == undefined ? 115200 : localStorage.baud);
// document.getElementById("addLine").checked = (localStorage.addLine == "false" ? false : true);
// document.getElementById("echoOn").checked = (localStorage.echoOn == "false" ? false : true);

// // CODELAB: Add feature detection here.
// const notSupported = document.getElementById('notSupported');
// notSupported.classList.toggle('hidden', 'serial' in navigator);
```

There are also some extra modal windows - for serial monitor and the data modal along with accompanying javascript. This may also have an impact.

I really want to get the data modal up and running as it gives us somewhere to report load times and latency

DeeEmm, 2024-12-13:
We could also consider splitting everything up into seperate files and having only a minimal file for the main GUI, however this does complicate things considerably, especially for the end user needing to manage multiple files. I think that this is a last resort.

I am confident that code optimisation can resolve the issue

DeeEmm, 2024-12-13:
Fixed it for you 😉

DeeEmm, 2024-12-13:
Although javascript is just about to get larger due to additional configuration / user params.

I have previously considered embedding the HTML within the core code. We already do this with the Boot Loop if there is no index.html file. We serve up a minified version of the page that is stored within one of the language structs and only contains the file upload dialog. It is in the language struct so that people can translate it. Similarly anyone can translate the entire index.html file although we do break out individual titles and headings into the language file to prevent having to do that

In essence we already store the file in SPIFFS and send it to the browser on page load. There's really no difference where that data comes from. It takes up the same amount of room on the device and still needs to be sent to the browser on each page load.

Again it's another large change and I'm not sure that the benefits outweigh the work needed to do it.

Comparatively speaking the  HTML / Javascript is nothing compared to something like Facebook. The issue will no doubt just be crappy coding 😆

KirikauKiwi, 2024-12-13:
> I have parked some stuff that I was working on by commenting it out (GUI serial monitor). Might be worth stripping that out of the code and seeing if that makes a difference (it's right at the bottom)

I have deleted that from the src/javascript.js file and cleared the flash on my esp32 and uploaded to test. Latency is still present with no perceivable change.

DeeEmm, 2024-12-13:
Not sure how familiar you are with the Chrome 'webserver tools' extension, kind of valuable for web stuff.

It has some great features for measuring page load times etc. You can view the time taken for the first SSE event to come through...

Go to the 'Network' tab and click on 'events' after page load, this will show you to SSE data along with timestamps.

You an also view the raw data at http://diyfb.local/events

[image]

KirikauKiwi, 2024-12-13:
> It is only the GUI that is sluggish, not the data.

If that was so, would the serial API response not be affected??? As I am seeing the sluggishness with the API too in CoolTerm and serial monitor in VSCode.

DeeEmm, 2024-12-13:
> If that was so, would the serial API response not be affected??? As I am seeing the sluggishness with the API too in CoolTerm and serial monitor in VSCode.

I was just talking from GUI perspective, but yes you are right, so likely not javascript related.

DeeEmm, 2024-12-13:
This points to the main code being the root cause (common source)

DeeEmm, 2024-12-13:
Page load times - most of which is waiting for server response. On top of this we have the SSE delay, which again is likely mostly server delay, so for me all up I am seeing 800ms + 2207ms so just over 3 secs all up. 

Not overly long, but definitely a lot longer than it was originally.

[image]

KirikauKiwi, 2024-12-13:
I am less familiar with java, html web based code than I am with C++ and I am still not very literate at that either.
Safari with the develop tools enabled has similar too I think. I do like how Chrome has the web serial too 👍 

You can record a timeline of the page load, events etc.

https://github.com/user-attachments/assets/6efd9826-430c-441b-ba9f-9a63fc1adfd0

DeeEmm, 2024-12-13:
Ahhh Safari.

The only time I use it is to check browser compatibility and then not so often nowdays. It's a toy compared to Chrome for webdev stuff.

Web Developer tools is the defacto standard really. Safari dev tools are not bad, but definitely prefer Chrome.

KirikauKiwi, 2024-12-13:
Looking at the event stream in Chrome as you suggested I am typically seeing 2-4s and not as frequently, but still regularly 5-11s delays.

KirikauKiwi, 2024-12-13:
What changes were made to how the I2C data bus is setup/used with the 205-post-compile-configuration changes.

I am not sure if I have set it up correctly on my picoscope, but I was curious to look at the I2C bus as the scope can decode it.

The I2C bus with the current DEV build, the data appears to be not very good i.e it looks incomplete, non existent, compared to the DEV build prior to the 205-post-compile-configuration updates.
It also looks a bit inconsistent in the previous build too, i.e the data is intermittent but it looks complete as it has acknowledgements and data bytes.

I2C Bus DEV build pre 205-post-compile-configuration.

[image]

I2C Bus DEV build post 205-post-compile-configuration.

[image]

DeeEmm, 2024-12-13:
Will have a look. I think it was only the address decoding was changed from hex to decimal as JSON cannot store hex.

Wouldn't have thought that would be the issue

KirikauKiwi, 2024-12-13:
Here are a couple of wave form captures that you should be able to open in the picoscope software. Remove the .txt from the file name before opening, I had to add it so I could upload the files here.

I have a better trigger point in these files and also scoping both I2C devices and changed my scoping points to be at the pins of the ESP32, previous screen capture where taken between the BME and ADS1115.

DEV build post 205-post-compile-upload.
[20241213 DIYFB DEV build pre 205-post-compile-configuration-0002.psdata.txt](https://github.com/user-attachments/files/18126182/20241213.DIYFB.DEV.build.pre.205-post-compile-configuration-0002.psdata.txt)

DEV build pre 205-post-compile-upload.
[20241213 DIYFB DEV build post 205-post-compile-configuration-0003.psdata.txt](https://github.com/user-attachments/files/18126184/20241213.DIYFB.DEV.build.post.205-post-compile-configuration-0003.psdata.txt)

KirikauKiwi, 2024-12-13:
> Will have a look. I think it was only the address decoding was changed from hex to decimal as JSON cannot store hex.
> 
> Wouldn't have thought that would be the issue

It maybe a red hearing, given I have never measured an I2C bus before, but there was a noticeable difference in the decoding of the bus by the picoscope software that I thought best to flag it as a possibility

KirikauKiwi, 2024-12-13:
Further testing of the I2C bus
If I remove the I2C bus connection from the ESP32 and check just the ESP32 I2C pins I am only getting messages about the ADS1115 at address 0x48 nothing coming from the ESP32 for the BME280 at address 0x76.
Once I connect the ADS1115 and BME back on to the bus the data from the ADS1115 is now appears.

It appears I now have the settings correct on the scope as READ and WRITES are now displayed, previously I could only see READS.
No devices connected to the ESP32 I2C bus, current DEV build.

[image]

ADS115 and BME connected to the ESP32 I2C bus, current DEV build.

[image]

No devices connected to the ESP32 I2C bus, previous DEV build you get the ADS1115 Failed message.

[image]

ADS115 and BME connected to the ESP32 I2C bus, previous DEV build.

[image]

[image]

KirikauKiwi, 2024-12-13:
Another observation is that the previous build on my was PSU 12.49V, 0.06A, 75W, the current DEV build is 12.49V, 0.09A, 1.12W so something is churning away.

DeeEmm, 2024-12-14:
> Another observation is that the previous build on my was PSU 12.49V, 0.06A, 75W, the current DEV build is 12.49V, 0.09A, 1.12W so something is churning away.

I'm seeing around 100mA here, but my PSU is pretty agricultural and only has 0.1A accuracy

At the moment I'm trying to resolve the file upload issue. I think that all of these are related to the same thing - memory allocation / memory management. This is exactly what I'm seeing with the file upload - stack and canary errors

The additional memory consumed by moving user data post compilation means that it takes up a massive amount of memory. Previously all of that data was handled by precompile macros, which essentially excluded anything that was not relevant and then compiled what was left it into its smallest form. Now we are dealing with with uncompressed text/json documents that we have to load into memory twice (once for the file in SPIFFS and again in operational memory on the stack) and so our memory usage has increased by about 300%

This will either be a memory leak, bad memory management or a combination of both. My money is on the latter.

Regarding I2C. It generally either works or it does't. I'm not sure that disconnecting a device is a valid line of enquiry. How are you initialising the bus with only one device without changing the code?, it should fail with an error on initialisation. If you are simply disconnecting the bus whilst the system is running, this is a failure mode that is not considered or tested, so essentially it is uncontrolled behaviour. It does not surprise me that it is giving you unexpected results.

It is worth considering that there is some prioritisation to the bus messaging. The ADC works on a much faster poll time than the BME.

Can I ask what symptoms are you seeing that indicate there is an issue with BME reads? 

I would advise setting a test in the code to validate those observations.

For example if you suspect that there are dropped device reads, or that the device read hangs, construct a test to validate that, and then look for causality.

DeeEmm, 2024-12-14:
However. I will say that way back in my first reply I mentioned that this was a memory issue, and I think that it would be more efficient to pool together with that goal in mind.

I spent some time last night looking for memory and task reporting from the FreeRTOS codebase, however few of those fucntions are available to platformio. This is mainly because the arduino framework that platformio uses works from a compiled binary version of FreeRTOS and at teh time of compilation those additional functions were not enabled.

So it is not possible to enable them at runtime, the only way to get access to those tool is to recompile FreeRTOS from scratch.

Obviously this is a large undertaking an one outside of the scope of this project, so the quest continues to find othe suitable ways of testing  what is going on.

DeeEmm, 2024-12-14:
Dang. Looks like somehow I've fried the serial port on the ESP as I'm only seeing the bluetooth port in PIO. A reboot hasn't resolved it. Guess that's karma for being short in my reply. (sorry)

I think that what happened was that another device on the USB hub drew power back through the ESP, overloading the serial chip. I noted that when I switched my PSU off, the other USB device on the hub (Traktor mixer) rebooted. 

Really need to get a powered USB hub

So entirely self-inflicted and avoidable, but also something that I've done many times before with no ill-effects. (Good job I bought a stack of Wemos boards.)

I added an API function to view heap data (x). so we now have xTask stack data (X) and Heap data (x)

Decided to do some tests...

Stack and heap with with only the index page loaded (86k)

> x:Free Heap=151684 / Max Allocated Heap=94196 
> X:Stack Free Memory EnviroTask=9204 / SensorTask=9204 

Then with congifuration.json uploaded (approx 2k filesize)

> x:Free Heap=151364 / Max Allocated Heap=86004 
> X:Stack Free Memory EnviroTask=9204 / SensorTask=9204 

Next with pins file uploaded (approx 900 bytes filesize)

> x:Free Heap=138232 / Max Allocated Heap=83956 
> X:Stack Free Memory EnviroTask=672 / SensorTask=808 

Note that at this point, it left the bootloop despite still requiring MAF file. Need to look at why this happened

Next I deleted the index file (86k)

> x:Free Heap=135696 / Max Allocated Heap=73716 
> X:Stack Free Memory EnviroTask=656 / SensorTask=808 

And then re-uploaded the index file (86k)

> UploadStart: /V2.0-RC8_2412140027_index.html 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=0 len=177 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=177 len=1436 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=1613 len=612 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=2225 len=1436 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=3661 len=612 
> Writing file: '/V2.0-RC8_2412140027_index.html' index=4273 len=624 

But then the serial seems to hang with no upload complete response from serial monitor

So I try try again but then get a panic...

```
UploadStart: /V2.0-RC8_2412140027_index.html 
Writing file: '/V2.0-RC8_2412140027_index.html' index=0 len=192 
Writing file: '/V2.0-RC8_2412140027_index.html' index=192 len=1436 
Writing file: '/V2.0-RC8_2412140027_index.html' index=1628 len=612 
Writing file: '/V2.0-RC8_2412140027_index.html' index=2240 len=1436 
Writing file: '/V2.0-RC8_2412140027_index.html' index=3676 len=612 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40112b7c  PS      : 0x00060030  A0      : 0x8010d994  A1      : 0x3ffb1ff0  
A2      : 0x3ffe9548  A3      : 0x00000d30  A4      : 0x3ffe9560  A5      : 0x3ffc7a20  
A6      : 0x63203a6e  A7      : 0x00000000  A8      : 0x00004620  A9      : 0x3f408c4c  
A10     : 0x00000001  A11     : 0x3ffe9548  A12     : 0x3ffe9560  A13     : 0x00000b38  
A14     : 0x3ffe8a70  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400895c4  LEND    : 0x400895da  LCOUNT  : 0xffffffff  

Backtrace:0x40112b79:0x3ffb1ff00x4010d991:0x3ffb2030 0x401644de:0x3ffb2050 0x4010b1e4:0x3ffb2070 

  #0  0x40112b79:0x3ffb1ff0 in tcp_output at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_out.c:1305 (discriminator 4)
  #1  0x4010d991:0x3ffb2030 in tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:999
  #2  0x401644de:0x3ffb2050 in _tcp_recved_api(tcpip_api_call_data*) at libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:444
  #3  0x4010b1e4:0x3ffb2070 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:172
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154
```

The MCU reboots but then goes to waiting for maf.json file. So it appears that the file transfer above completed despite giving an error

so I upload the VDO MAF file (approx 2k)

And then something interesting happens

> x:Free Heap=151128 / Max Allocated Heap=86004 
> X:Stack Free Memory EnviroTask=9204 / SensorTask=9204 

The stack watermark shows free memory of 9204 for both tasks (cores)

Obviously this is not right. 

Will keep digging

KirikauKiwi, 2024-12-14:
Initial I looked at the I2C bus as part of a diagnostic work flow looking at the chain of inputs. Based on what I was observing and what I am able to physically check.

Obserservations.
- GUI is responsive to the Bench Start/Stop ✅
- GUI is reloading as expected e.g. browser refresh ✅
- GUI incoming data updating appears sluggish (initial observation and the reason behind this Bug report) 
- No physical hardware changes ✅
- Change to the code base.❓
- Additional files to upload. ❓
- Serial API responses a deathly slow ❌

I then began testing the physical.
- Checking voltage outputs from the pressure sensor, and trim pots. ✅
- Now armed with the tool to decode the I2C bus ( I know!!! entering the danger zone) it showed some noticeable differences between the two build versions. ❌ hence reporting it.  (I did note that it maybe a red herring)
- Checking the code base ❓Difficult as I don't know what was changed and lack of knowledge with C++, so a slow process for me

Summary,
1. It appears that once the code is running that the GUI was working as expected other than the incoming data.
2. It may be possible that the I2C bus is affected by the memory issue. Otherwise what else would make it appear different between the builds.
3. The API response is the slowest of them all and at a guess runs in a different process which is also being affected somehow.
4. The convergence of the fault finding of #293 and #292 would indicate that it is most likely a Memory/Processing issue.

> Can I ask what symptoms are you seeing that indicate there is an issue with BME reads?

- One of the observed differences I noticed in the GUI data updates is that the BME data appears not to be updated at all for me, the ADS1115 Data was being updated albeit slowly, I am aware of the 500ms and 1000ms so I would expect to see the behaviour of the BME updates to be about twice as slow as those of the ADS updates in the GUI. It was much longer than double.
- Upon checking the decoding of the I2C bus and comparing between the two builds there was no signs of any data read or writes for the BME on the bus for the current DEV build.
- With the ESP32 being the Master on the I2C bus it would indicated that something is happening inside the ESP32 preventing the BME being the slave device providing its data to the ESP32 as the master. 
- This aligns with my observations.

Is it likely something wrong with the I2C bus, no as it is working as expected with the previous DEV build, it does indicated that there is something different that is doing something to the I2C!!!

Take a look at column F in this [CSV](https://github.com/user-attachments/files/18134968/20241213.DIYFB.DEV.build.pre.205-post-compile-configuration-0004.csv) (I2C address) , it is an export of the decode from the I2C bus of the previous DEV build. you can see regular Read/Writes for the BME and more frequent Read/Writes for the ADS as expected. The read/writes are not present for the BME in the second [CSV](https://github.com/user-attachments/files/18134970/20241214.DIYFB.build.post.205-post-compile-configuration-0005.csv) which is from the current DEV build.

I am suspecting that once the memory issue is resolved/minimised that the full functionality of the I2C will return.

PS

> Guess that's karma for being short in my reply. (sorry)

I didn't find you being short in your reply, I have no problem with being questioned/challenged with this sort of thing, Trouble shooting can be a royal PITA. It is too easy to construct a test that contains bias that then confirms your observation and then conclude the wrong thing because of this. 
It is always good to have a second set of eyes, ears and thought processes to fact check!!!

You are right in challenge it too, given the "DUMB" moments that occur for me.

PSS

>  I added an API function to view heap data (x). so we now have xTask stack data (X) and Heap data (x)

This looks promising 👍

KirikauKiwi, 2024-12-14:
> Regarding I2C. It generally either works or it does't.

>  I'm not sure that disconnecting a device is a valid line of enquiry. How are you initialising the bus with only one device without changing the code?

I didn't, this was was an observed difference between the the two DEV builds.

>  it should fail with an error on initialisation. If you are simply disconnecting the bus whilst the system is running, this is a failure mode that is not considered or tested, so essentially it is uncontrolled behaviour. It does not surprise me that it is giving you unexpected results.

The expected results would be in both instance for the I2C bus to stop working based on my observation of the previous builds and how the the software works, the current DEV build didn't exhibit this same behaviour as the I2C bus is still working when I removed the bus from the ESP32.

- By removing the I2C bus on the previous build you get the ADS Failure, Wire read error on the serial monitor and DIYFB stops working, which is visible in this [screen shot above](https://private-user-images.githubusercontent.com/80394127/395628288-8998ca62-8292-4306-96f1-374d082fb22d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxNjU3MTIsIm5iZiI6MTczNDE2NTQxMiwicGF0aCI6Ii84MDM5NDEyNy8zOTU2MjgyODgtODk5OGNhNjItODI5Mi00MzA2LTk2ZjEtMzc0ZDA4MmZiMjJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE0VDA4MzY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4NGE5NTM5NTEzOTFmY2EyODQyMjE4ZDA0NGY3YTQ0MDJlZTcxNzNjZjAyNTRkM2VmZDY0ZjU0NjRlMzRkMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Rvw0IjEUdGGuItthHQoQdvFHddCtabqn8mRgNjyGIno)

When I remove the I2C bus on the current DEV build the ADS failure, Wire read error does not appear on the serial monitor and the ESP32 appears to still be running, which is visible on the above [screen shot](https://private-user-images.githubusercontent.com/80394127/395621545-54563568-5211-419c-bdda-487d797f4c73.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxNjU3MTIsIm5iZiI6MTczNDE2NTQxMiwicGF0aCI6Ii84MDM5NDEyNy8zOTU2MjE1NDUtNTQ1NjM1NjgtNTIxMS00MTljLWJkZGEtNDg3ZDc5N2Y0YzczLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE0VDA4MzY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkYWIyMzg3OGZjZGYxYmUyY2FkNTMwMDkzMWI1ZDkzZTBjOTE5ZWEwYTkzZjVmNWQ5ZGY2YTc3NjQ2N2RhZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.BG88tNjiU1OizqMt_jtdrAhcseiYy-q5Xu_blw05U9I) you can see that the ESP32 is sending out data still that the decode is showing it is for the ADS1115, there is no decode showing anything on the I2C bus for the BME at address 77.

DeeEmm, 2024-12-14:
I think that it is possibly more of a symptom than a cause, which is likely the same issue that I'm seeing with the file upload. 

My problem is that C++ is also not my main language. My background is using memory safe languages where you do not have to worry about stuff like managing the heap and stack. Or more specifically bad coding practices do not impact code execution in the same way (stack overflow / canary watchdogs etc). So it is relatively new territory for me too, as well as a steep learning curve. 

C++ is very powerful in that the ability to have more granular control over how memory is managed makes for more powerful potential, but also means that there are a myriad more ways to screw it up 😆  

I'm beginning to think that a rethink of how me manage user data might be in order. The 'solution' using flat files is proving to be too memory hungry, especially for the larger files such as the MAF and Index files.

It is possible to move index file data within the main code. I think that this will reduce code size as the resultant data will be compiled with the core. It also removes one step from the end user - having to upload / manage the index file.

It would also then pay to have seperate HTML pages for settings and the dataGraph rather than using a monolithic file structure as we currently do. This would reduce page size and javascript requirements. Essentially reducing the GUI page down to a minimum footprint.

We can also move the configuration and PINS off into EEPROM. Not sure why I didn't think of this initially. This completely removes the overhead of not only managing the files but also converting them on system load / config update. (This IS the way)

I think the trickiest one is the MAF file. 

When we look at the index file (86k) we are not actually doing anything with that data. We are essentially grabbing it from SPIFFS and sending it to the browser. The browser then runs the javascript code. So there's minimal overheads for MCU. The Main code is basically just spitting the SSE data out to the browser. It is the browser that processes the task of updating the GUI, not the MCU.

The MAF on the other hand is a different kettle of fish. Not only are we loading it into working memory, we are then interrogating it multiple times a second to determine the flow rate. This consumes an amount of processing power which is directly related to the size of the data array. In the case of the Delphi MAF file we are talking about 1024 data points. That's actually double the size of the EEPROM. (which excludes it as a possibility) 😦 

I did consider trying to revert back to using transfer functions instead of lookup tables. But this requires that the data is accurately converted. Perhaps this is an ideal task for AI to undertake. If we can feed it 1024 data points and get an accurate transfer function back in return, then we only need to include the transfer function within the code, which again can be part of the core code.

Maybe that's a dream? But it should be achievable. It should be possible to accurately represent the MAF curve with a transfer function. This then moves the hardcore memory intensive stuff offline.

<details><summary>ChatGPT response to: How do i calculate the transfer function from a key:value dataset</summary>

DM
ChatGPT
To calculate a transfer function from a key:value dataset, you'll generally need to follow these main steps:

Understand the Data: Ensure that your data format is appropriate for creating a transfer function. Typically, you would have input-output data from a system, where one variable represents the input (independent variable, (x)), and the other represents the output (dependent variable, (y)).

Choose a Model Form: Determine what type of model you want to fit your data. Common forms are linear or nonlinear models, and for transfer functions in control systems, you might use forms like:

A first-order transfer function: [ G(s) = \frac{K}{\tau s + 1} ]
A second-order transfer function: [ G(s) = \frac{K \cdot \omega_n^2}{s^2 + 2\zeta\omega_n s + \omega_n^2} ] Here, (K) is the system gain, ( \tau ) is the time constant, (\omega_n) is the natural frequency, and (\zeta) is the damping ratio.
Fit the Model to Data:

Gather your dataset in usable form (e.g., as arrays or lists).
Use least-squares regression or other fitting techniques to determine the parameters of your model that best fit your dataset. This can often be done using software tools or programming languages (like Python, MATLAB, R, etc.).
Calculate the Transfer Function:
If your dataset has already been fitted to a model form (e.g., polynomial or nonlinear), you can express the output in terms of the input through the transfer function:

For a linear regression model:
If you have (y = mx + b), you could represent it in transfer function form based on the system dynamics you've established.
Validate the Model: After obtaining parameters for the transfer function, it's essential to validate the model against the dataset:

Compare the predicted outputs using your transfer function with the actual outputs from your dataset.
Use metrics such as the coefficient of determination (R²), residual plots, or cross-validation to ensure the model fits well.
Example Using Python
Here's an example of how to fit a simple first-order transfer function to a dataset using Python:

python
Copy
import numpy as np
import matplotlib.pyplot as plt
from scipy import signal
from scipy.optimize import curve_fit

[... cut ...]
t_out, y_out = signal.step(system)

# Plotting the results
plt.plot(t_out, y_out, label='Model Response')
plt.scatter(x_data, y_data, color='red', label='Data')
plt.xlabel('Time')
plt.ylabel('Output')
plt.title('System Response and Fitted Model')
plt.legend()
plt.show()
Conclusion
Understand and prepare your dataset.
Determine the appropriate model type for your transfer function.
Use regression or other fitting techniques to estimate the model parameters.
Validate the model against your data.

<details>

DeeEmm, 2024-12-14:
Maybe we could move MAF into EEPROM if we limit the number of data points.

If we reserve half of the 512 available data points for configuration and half for MAF. We end up with MAF resolution of 20mv which is what we generally see with the Bosch / Siemens data. Some MAF data files have an even lower resolution. We also interpolate data. So resolution is somewhat higher than this.

With that said. With the current method of linear interpolation that is used, we actually have a small error - because the data is not linear. 

Using a transfer function, we can correctly calculate any intermediate value along the curve. However we are reliant on the accuracy of the transfer function.

All of which comes back to system accuracy.

DeeEmm, 2024-12-14:
Also on MAF data...

Maybe we need to rethink about how we store and use that data.. At the moment we have a variety of different data schemas for the MAF files. They can be any length and consist of any kind of scaling. The original intent was to build in flexibility.

Perhaps it would be better to define a fixed length array of data and then translate all MAF data to fit. 

Again this moves some of the number crunching offline and we simplify the code operation.

KirikauKiwi, 2024-12-14:
> I think that it is possibly more of a symptom than a cause, which is likely the same issue that I'm seeing with the file upload

I would agree with you now we are more aware of the problem overall.

I did attempt a linear regression on the Bosch data in Excel earlier in the week when helping @black-top with trying to get the MV_TRIMPOT value for him to input. I got close but not close enough.

In theory most MAF sensors must follow a similar curve due to them all being an orifice that is flowing air.
So the actual physical Flow curve would closely follow the same equation used to calculate the orifice flow. The electronics in the MAF sensor is then only really providing a voltage output that has to mapped to the curve. We do not need to worry about the electronics inside other than knowing what voltage it outputs for a given flow.
I would expect the only differences between MAF sensors to be the voltage output for a given flow, the overall curve/transfer function will be the same.
Well in theory this is what I would expect in practice humans are involved so it may not be as such. Might have ago at graphing the MAF data from DIYFB to see.

I assume this is why the pressure sensor work without much processing power required since they have a transfer function that can easily be processed in the ESP32.

Another option could possible be to involve a second MCU, I have noticed M5 Stack does this with many of their peripherals, as an example the closest they have in their line up to the ADS115 uses an STM MCU to process the analogue inputs and then provides an I2C output to be used by the ESP32.
This is the particular device I am talking about [Pb.Hub](https://shop.m5stack.com/products/i-o-hub-1-to-6-expansion-unit-stm32f0) this also allows for it to be used as an expansion device to for PWM etc.
So essentially off boarding the MAF number crunching if we cannot work out a transfer function for them.

DeeEmm, 2024-12-14:
> I did attempt a linear regression on the Bosch data in Excel earlier in the week when helping @black-top with trying to get the MV_TRIMPOT value for him to input. I got close but not close enough.

I used excel early on but found the same limitations.

We took data measured from a superflow bench and tried to create a MAF function from the data.

[image]

Then used this to create an excel app to generate key:value data using a three point method - basically same as you would for a temperature sensor. You punch in three values along the curve and it generates the data table for you

[image]

IIRC this is using data that is derived as an exponent via a logarithmic function.

What I now know is that mathematically we should be using the least-squares method, which is far more appropriate. 

I guess what I'm thinking with where things currently stand is that the change made to move the user definable data into post-compile is basically not working as intended. I certainly seem to be digging a hole with every change causing additional issues. which is not where i wanted this to be. So I think that it needs a second bite to try and resolve some of these issues.

We know that the basic code is valid, apart from perhaps some minor improvements to improve accuracy. But the way that we are managing the user data is not the most appropriate. I cannot even upload a file without seeing a panic.

I knew that the move the post compile stuff was going to be a big change. I guess it's just still not finished.

I think that moving the PINS and Config into EEPROM is a no brainer, but suspect that this alone is not going to solve the issue. 

I'll take a look at this along with moving the index file on board.

Are you able to use your chatGPT skills to see if you can get it to generate some code to do a least-squares translation of a MAF data file?

DeeEmm, 2024-12-14:
> Another option could possible be to involve a second MCU

I would like to try and avoid this as it introduces unnecessary complexity which will likely be too difficult for the average end user.

The ESP can be made to work, we are nowhere near its capacity. It's just poor coding. Plain and simple.

DeeEmm, 2024-12-14:
> Are you able to use your chatGPT skills to see if you can get it to generate some code to do a least-squares translation of a MAF data file?

Use the Bosch sensor as Black-Top can possibly validate the results. I only have a test-bench set up at present.

black-top, 2024-12-14:
I didn't notice this topic, you are doing a lot of work. This is probably somehow related, because after uploading yesterday's dev, I noticed that is much more difficult to connect to DIYFWB. It does not connect from the first try, after that, it disappears from wifi network list, I need to restart esp32, and it takes 4-5 tries to connect. Another issue that Opera browser do not load DIYFWB page it says "this site can't be reached, it took too long to respond",  but with Firefox it works ok. I need to take another laptop to check the serial monitor, because I can't get serial drivers to work with that old pc dedicated for flow bench, so I can't give any details right now about what is happening.

DeeEmm, 2024-12-14:
Yes there has been some issues with latency but at this stage I do not know the cause. I'm working to try and find a fix to the immediate issue, and then will look to try and improve the latency.

It might be worth rolling back to a previous version for the time being.

You can see a list of previous commits on [this page](https://github.com/DeeEmm/DIY-Flow-Bench/commits/DEV/)

The 'Merge Pull Requests' are the major changes. 

If you manage to roll back to a version where these issues are not present, let me know what commit that is, as this might give some idea on the cause of the issue.

KirikauKiwi, 2024-12-14:
Going back over and testing with the previous commits Dec 10, cf2238a8 works well but is still a bit sluggish with the data updates for the BME but looks good for the ADS data, there are four json files to upload with this commit.
Going one back one further pull request to Dec 9, 1a1c532 this appears to have very consistent data updates for both the BME and ADS, there are two json files to upload with this commit.

KirikauKiwi, 2024-12-14:
I uploaded each commit between the Merge Pull Requests 1a1c532 and cf2238a, the issue is not present on 1a1c532 and it is appears at the first commit after this, which is 84509ce.

@black-top in platformio if you have the DEV showing in the bottom left corner, you should be able to roll back to this commit with the commands git stash, this parks anything you have currently, then use the command git checkout 1a1c532.
You should then see 1a1c532 in the bottom left corner of platformio, erase the flash on the ESP32 and upload.

KirikauKiwi, 2024-12-14:
I have created an updated version of 1a1c532 that contains the fixes for the Pitot Tile Display changes and corrections to the API commands that had been lost with the roll back.

@DeeEmm I wasn't certain of the best way to do a pull request for this so it is sitting on my git [here](https://github.com/KirikauKiwi/DIY-Flow-Bench/tree/DEV-Stable)

DeeEmm, 2024-12-15:
Hmnnn, that is some considerable time ago. A LOT of work has been done since then. (I've been coding more-or-less full time most days) Rather than junk everything that has been done, it would be better to identify the cause.

A better option would be to update master with 1a1c532 and leave DEV as-is.

Whilst the intent is for DEV to always compile and work, there will be times where it does not. After all it is the bleeding-edge of what we are doing. As long as master is stable then we are all good.

I think that we can do this by checking out PR 277 and then creating a new PR to master

```
gh pr checkout 277
```

But before we do that, can I ask did you check to see whether all commits to DEV work, or only your own changes?

https://github.com/DeeEmm/DIY-Flow-Bench/commits/DEV/ESP32/DIY-Flow-Bench

---------------------------

## Work done since 1a1c532

Commits on Dec 14, 2024
[2412140026 - Basic Data Modal Added (WIP)](https://github.com/DeeEmm/DIY-Flow-Bench/commit/1cac7197d971e5a4498af2ecda07d7a6c7cb49ff)

Commits on Dec 13, 2024
[2412130038 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/82e92cd1ea29d464df0b94bbec85fccd7aab113e) https://github.com/DeeEmm/DIY-Flow-Bench/issues/288 [Sensor status info incorrect](https://github.com/DeeEmm/DIY-Flow-Bench/commit/82e92cd1ea29d464df0b94bbec85fccd7aab113e)

yesterday
[2412130037 - Changed pre:user_actions](https://github.com/DeeEmm/DIY-Flow-Bench/commit/1068fe6fc88224b8168b01ce5add3e7222cc7be6)

yesterday
[2412130025 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/55314ac32c0f81056e6b86cfb4e95b6a6576adcc) https://github.com/DeeEmm/DIY-Flow-Bench/issues/292 [Changed Webserver::fileUpload 'final' code](https://github.com/DeeEmm/DIY-Flow-Bench/commit/55314ac32c0f81056e6b86cfb4e95b6a6576adcc)

2 days ago
[2412130015 - Fixed Issue with pre-compile tasks deleting release](https://github.com/DeeEmm/DIY-Flow-Bench/commit/961702454d0820fe6d060533f954018c3e57058f)

Commits on Dec 12, 2024
[Merge branch 'DEV' into 236-feature-requestv3-data-display-accessed-from-configuration-gui-page](https://github.com/DeeEmm/DIY-Flow-Bench/commit/1195e3e8917584839963b7ac27b4b6af4f990658)

3 days ago
[Update platformIO](https://github.com/DeeEmm/DIY-Flow-Bench/commit/efa4afb93f9ed3ccd4e01661fc810610c3a3254e)

3 days ago
[2412120051 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/c4d553a6a258abf27e70f6e9d00f04ab6484648a) https://github.com/DeeEmm/DIY-Flow-Bench/issues/288 [Updated Analog calcs to cast to double](https://github.com/DeeEmm/DIY-Flow-Bench/commit/c4d553a6a258abf27e70f6e9d00f04ab6484648a)

3 days ago
[2412120002 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/c88c312ca423f5b712c8f5d0262295ece85fa3ad) https://github.com/DeeEmm/DIY-Flow-Bench/issues/288 [fixed 60mv sensor deviation](https://github.com/DeeEmm/DIY-Flow-Bench/commit/c88c312ca423f5b712c8f5d0262295ece85fa3ad)

3 days ago
Commits on Dec 11, 2024
[2412110117 - Refactored MAF JSON code handling](https://github.com/DeeEmm/DIY-Flow-Bench/commit/e36c9889a9cd651a4e2ee5bb7bf9796b59f1fcf1)

4 days ago
[2412110117 - Refactored MAF JSON code handling](https://github.com/DeeEmm/DIY-Flow-Bench/commit/fef75a16d0fe7830dd806dac448bafc8ac62f34c)

4 days ago
[2412110103 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/7f15b02b71627d5544835529be4e0bb444386607) https://github.com/DeeEmm/DIY-Flow-Bench/issues/285 [increase template size in DataHandler::loadJSONFile](https://github.com/DeeEmm/DIY-Flow-Bench/commit/7f15b02b71627d5544835529be4e0bb444386607)

4 days ago
[2412110068 - Moved MAF initalisation into DataHandler](https://github.com/DeeEmm/DIY-Flow-Bench/commit/2a4e6ff16b197be538ae6555839423f4dec2a64d)

4 days ago
[Merge branch 'DEV' into 205-post-compile-configuration](https://github.com/DeeEmm/DIY-Flow-Bench/commit/100e72ec957617b5b3b7b53443ce7a1acc9e036f)

4 days ago
Verified
[Create binary](https://github.com/DeeEmm/DIY-Flow-Bench/commit/98efcf596b34d64e0b33ec408ba3d600f7bbfc80)

4 days ago
[2412100102 - Updated PINS files](https://github.com/DeeEmm/DIY-Flow-Bench/commit/92c83ed6d9f4369d0fb53d547298787d261fa5a7)

4 days ago
[2412100102 - Updated PINS files](https://github.com/DeeEmm/DIY-Flow-Bench/commit/133b89ec9090d1d27e52a80ef8ba73ea7f6b39e4)

4 days ago
Commits on Dec 10, 2024
[Fixes for Pitot GUI](https://github.com/DeeEmm/DIY-Flow-Bench/commit/20f523de989f489c5ff38e070f99907a3c652def)

4 days ago
[New versioning and .bins created with upload](https://github.com/DeeEmm/DIY-Flow-Bench/commit/7549afe1f3e44a027c9dcb531c8bbc8e75eb4865)

5 days ago
[2412100089 - Align GUI build # with Code build #](https://github.com/DeeEmm/DIY-Flow-Bench/commit/f9bec10835959d69a55d3068a0abb91becfe606c)

5 days ago
[2412100087 - Moved voltage trim outside of validation test](https://github.com/DeeEmm/DIY-Flow-Bench/commit/b24825600c81e479d5666857e5533148aebdb9e8)

5 days ago
Commits on Dec 9, 2024
[2412090117 - Fixed GUI info for MAF sensor type](https://github.com/DeeEmm/DIY-Flow-Bench/commit/9a96b049ceb784686f219def99ab344d90734424)

5 days ago
[2412090109 - Fixed sensor_src type](https://github.com/DeeEmm/DIY-Flow-Bench/commit/5c799585593587ae72e8f87917dfdb61439d17b9)

5 days ago
[2412090001 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/84509ceee4d07ed6c400d050868e7c8235049b61) https://github.com/DeeEmm/DIY-Flow-Bench/pull/277 [Temperature correction for ISO 5011](https://github.com/DeeEmm/DIY-Flow-Bench/commit/84509ceee4d07ed6c400d050868e7c8235049b61)

last week
[2412090001 -](https://github.com/DeeEmm/DIY-Flow-Bench/commit/a92196da343685d4612e49e018bfa47fcfc3081d) https://github.com/DeeEmm/DIY-Flow-Bench/pull/277 [Temperature correction for ISO 5011]

DeeEmm, 2024-12-15:
This is the compare between the last working 'good' version and the one where you noted the issue appears.

https://github.com/DeeEmm/DIY-Flow-Bench/compare/1a1c532..84509ce

Not surprisingly, this looks to be the change where we moved from precompile macros to post-compile files.

Heres the diffs from the changelog...

```
2412090065      - Fixed XHR form submission

2412080010      - GUI fixed version increment in GUI info
2412080005      - Test build target for pre user actions
2412060066      - Reformat JSON Data in API
2412060044      - Updated Serial API responses (inc boot loop)
2412060001      - Update boot loop file prompts to link to Github
2412050201      - Convert MAF JSON to vector of vectors
                - Change key>val longs to u_ints
2412050090      - Recalculated JSON memory allocation
                - Increased LOOP_STACK_SIZE
                - Added MAF / config JSON file dumps to API
2412050001      - Update build number format [2 to 4 digit]
                - Auto build number update
24120401        - Utilise api/file/upload path for all uploads
                - Remove redirect from valve lift capture (change to XHR)
                - Updated gulp to auto version GUI html file
24120302        - Remove redirect from bench on/off
                - Utilise boot loop for runtime file handling
24120301        - Update index.html
                - Change I2C addressing to decimal
24120205        - Update MAF.json handling
                - Address issue with loading configuration.json
24120204        - Update GUI file load management
24120203        - Update sensor type handling
24120202        - Update sensor enable handling
24120201        - MAF data file json format
24120102        - Fixed compile error
24120101        - Tidied up status definitions in sensors.begin
24113002        - Removed precompile macros assoc/w configuration.h
24113001        - #205 Post compile configuration refactor
24112901        - Pitot Tile not displaying correct value for ∆P
                - Correction in calibaration.cpp for calData pdiff and pitot.
                - API Corrections
                - Added merge-firmware.py
                - Capitalised DOCS + CHANGELOG directories
                - Change datagraph image download to JPG
```

It's interesting to compare changes.

I think the previous comments above are on the right track to address these issues (moving files into PROGMEM and settings into EEPROM)

Last night I did manage to move the web files into PROGMEM and it instantly resolved many of the issues I was seeing.  the uC stopped panicking and the GUI was able to load. However there were still residual issues. My next step is to split the index file up into it's constituent parts (index / datagraph / settings) and split the javascript file up so that each page has a smaller dedicated javascript file. The goal is obviously to make everything as small as possible and only call what is immediately needed.

Apart from some navigation changes where you will see a page refresh when changing between the tabs there will no discernible difference in the GUI. 

I want to follow this change to it's conclusion, as it **_should_** work, but will concede that if it doesn't we may then need to revert back to the setup we had pre-compile. If this happens then we may beed to review how custom hardware is supported.

In this case (and I know some of you will not like this) we may need to consider running a fixed hardware design. This would make much of the offending code superflous and also removes the requirement to do any post compile changes at all. A lot of consideration and testing went into determining the recommended hardware configuration and for good reason. This is not to say that the code cannot be used on other hardware, just that it is not supported out-of-the-box.

Think of this a bit like Apple versus Micro$oft. Windows has so many issues, as it has to cater for an almost infinite number of possible hardware permutations, it is actually impossible to manage effectively and proactively. There is a constant (reactive) bugfix cycle where issues are addressed as they arise. And arise they do. 

Apple on the other hand only uses a handful of different hardware configurations. So MacOS is very easy to manage and test, which why is just works. Bugs of the kind that you see on M$ products are extremely rare on Apple products. The update cycle is much slower and more considered.

Comparatively, and to round out my analogy. Windows has had 2 updates every month this year, the last one had 124 bugfixes in it. MacOS on the other hand has had just two updates this entire year. Common hardware is the reason why.

The current issues are as a result of the change to post compile configuration. This change was to make it easy for anyone who did not want to run the recommended hardware to perform firmware updates. 

The update pathway for the recommended hardware has always been that it just works out of the box, with no changes required. So we could just flip this issue and push it back on to those who want to run custom hardware, such that they need to manage their own update workflows and hardware integration. It would certainly make development much more straightforwards. And faster. And easier to support.

I am confident that we can resolve the current issues. For me it is much better to cater for a variety of configurations, and so this has always been the preference, but if this impacts the progress of the project then we may need to have a rethink. Of course, none of this stops anyone tweaking the code to work on whatever hardware they want to use. Its not that the code will not work on other MCU's or with other sensors, it's just that the update pathway is not trivial with precompile directives that need to be managed.

KirikauKiwi, 2024-12-15:
Looking back over my notes from going through the various commits,  I had written down that the Serial API was working well in some of the commits after the issue appeared this was not my primary goal so I wasn't always checking this.

I have just now reloaded cf2238a to look at it in a more granular way.
Observations are.
- BME data updating about every second as expected.
- Serial API working well.
- ADS1115 data not visible on the GUI.
- the four .json file upload without any errors on the serial monitor.

It appears we maybe looking at more than one issue that has crept in. I will have a look now jumping forward through the major revisions to try and identify the commit where the serial monitor or BME data becomes unstable or both.

DeeEmm, 2024-12-15:
Just noted another thing.

I am experiencing much more reliable activity accessing the GUI via the IP address. I am specifically noticing this with the larger javascript file which is now being served up as a seperate file. using the diyfb local address it will often hang.

So I think that this must relate to the mDNS component which seems a bit flaky.

I am generally using the IP address. Not sure if you are. But might be worth trying this to see if it resolves or improves any of the other issues.

I suspect that this may also extend to POST / GET data as well. i.e. file uploads and possibly the XHE data.

This might go a long way towards explaining why I never seems to experience the same magnitude of issues as were being reported.

KirikauKiwi, 2024-12-15:
100E72EC
- 4x json uploaded ✅
- BME ✅
- ADS Data not viewable on GUI
- Serail API ✅

2594213
- 4x json uploaded ✅
- BME Sluggish
- ADS Normal with a few stumbles
- Serail API ✅

ef84b77
- 4x json upload, asyncTCP crash with MAF upload, other three upload normally.
- BME Sluggish
- ADS Sluggish
- Serail API Sluggish

> I am generally using the IP address. Not sure if you are. But might be worth trying this to see if it resolves or improves any of the other issues.

99% of the time I am using diyfb.local, I have only used the IP address when looking at issues with the temp server and file upload issues.

KirikauKiwi, 2024-12-15:
It looks like there are two separate issues at.
- 84509ce
- ef84b77

KirikauKiwi, 2024-12-15:
> A better option would be to update master with https://github.com/DeeEmm/DIY-Flow-Bench/commit/1a1c532dd2ae8500fccaa27e3298e4dd950f0e1d and leave DEV as-is.

Yes that seems to be sensible 👍

DeeEmm, 2024-12-15:
The mDNS might be a local network issue.

Will compare these later. 

https://github.com/DeeEmm/DIY-Flow-Bench/commit/1a1c532dd2ae8500fccaa27e3298e4dd950f0e1d merged into main.

DeeEmm, 2024-12-16:
Configuration settings are now stored in and read from NVM (non-volatile memory)

The index page loads and SSE is set up so values are updating from MCU.

Data and settings pages still need to be sorted out

I'm getting between 0.2-0.4 page load times for the index page, including initialisation of data.

Still lots more to do, but definitely a step in the right direction.

Need to fix up the data and settings pages as well as create a form / forms the configuration settings.

If anyone wants to test and see what kind of page load times they are seeing and note any latency, would be appreciated.

Bear in mind that this is not fully functioning code at the moment. It's just the index page and associated files to test file load and data transfer.

Code is in DEV.

DeeEmm, 2024-12-16:
PR #302 - [https://github.com/DeeEmm/DIY-Flow-Bench/pull/302](https://github.com/DeeEmm/DIY-Flow-Bench/pull/302)

KirikauKiwi, 2024-12-16:
I have tested: #302 
- preload HTML, the File upload button works in Chrome, couldn't get it to work in Safari.
- in Chrome file uploads are working but the page doesn't refresh correctly, and when going to upload the next file it is pulling up the file browser at the last directory you uploaded from, e.g. upload pins.json, go to upload the next json and the file browser will be at the pins directory it should be one level up.
- BME and ADS data updating a 1000ms intervals, a very occasional stumble but still updating almost every 1000ms.
- BME data looks correct.
- ADS data is showing at the MAX voltage of the ADS of 6.144V across all four of its analogue inputs.

[image]

DeeEmm, 2024-12-16:
Yeah actual data / functionality needs to be validated, it's more just to test latency on index page. The other two pages are static so should not be an issue. Neither are tested yet.

[image]

Just tested Safari and also see same error. Not much info on the error other than it appears to be fairly common with services such as Netflix. Unfortunately no one specifies what the actual cause of it is, only a bunch of pre-canned solutions such as restart / clear cache / change DNS / etc. So no way to test.

I'm going to ignore Safari for the time being. It's most likely a cache issue.

KirikauKiwi, 2024-12-16:
With Safari. I didn't get any error message, it was behaving like the button press wasn't doing anything, no response at all.
I tested Safari with a Private Window too to check for caches etc, it was still like the button press was not working.

DeeEmm, 2024-12-16:
Ahh okay.

Yes the settings and datagraph pages have not been coded yet, they just consist of the results of a quick cut and paste to divide up the code, so some aspects may work, but others may not.

I'm using a dedicated javscript file for each page to keep individual page loads smaller. It is more overall memory space consumed , but smaller individual page resources. The javascript for the data and settings pages has not been sorted yet.

DeeEmm, 2024-12-18:
Moved calibration settings into the NVM.

Next will do same for preferences and PINs

Once this is done I will move on to the GUI pages for data / config / etc. There are some new pages to add once the core code is written

I've had a stable GUI main page running for over 38 hours with no issues. Seems as if performance is back to normal with these changes.

[image]

Need to have a bit of a think about how we handle the PINS data in the NVM. Would like to try and capture I/O assignment issues and then redirect user to PINs page. 

Also need to work out how we manage pre-loading configurations. At the moment I'm thinking that by default ALL pins are disabled. User can then choose a pre-programmed schema from a drop down. This works similar to choosing a PINS file, where all pin assignments are then set. User then has the option of editing individual pins to customise the configuration. Or if the user is starting from scratch they can just set individual pins as required.

Once pins data is saved if there are any I/O assignment conflicts we catch the error and send the user back to the pins configuration page to fix it

DeeEmm, 2024-12-18:
(changes are on branch for #236 )

DeeEmm, 2024-12-21:
Quick update.

Slowly making my way through the changes to the code but been a bit distracted by xmas stuff.

Did get a quick chance this evening to take a look at creating some python code to generate a transfer function that represents the MAF data. Results look very promising...

[image]

I've ran a few of the mafData files that we have and the results are the same in each case.

So I am quietly confident that we can replace the mafData files with transfer functions. Just need  to finish the code off and test how it works in practice.

[7 more replies]

closed 2025-01-05 by DeeEmm