DIY-Flow-Bench/DIY-Flow-Bench #296: [FEATURE REQUEST]MAF Flow KGH needs a Standardised tile SKGH @ ISO5011.

KirikauKiwi, 2024-12-13:
Since working through the [Calculated Orifice Flows @ ISO 5011 Std. Conditions](https://github.com/DeeEmm/DIY-Flow-Bench/discussions/147#discussioncomment-11523594) and having just seen #294 it became apparent that DIYFB should also have a Standardise Mass Flow too.

The number of tile views have been increasing and may become a bit unwieldy, would it be worth considering for version 3, along with unit selection that it maybe possible to reduce the apparent number of tile views by first selecting your desired views from the configuration page and/or via a metric/imperial/SI units selector.

DeeEmm, 2024-12-13:
The number of tiles is always the same (6), but the choice may be overwhelming to some.

I think that some people will get confused about the differences between SCFM / ACFM and corrected flows. 

Perhaps a different 'mode' might be more appropriate where those additional options are hidden by default or a checkbox list of what ones you want to be able to access via the GUI.

Ultimately I would love a drag-and-drop style interface for the GUI where you can simply rearrange it to suit whatever your needs are. I have taken a brief look into this. it's not massively difficult, but also not trivial. 

TBH as the GUI file is not tied to the main code, anyone could create whatever GUI they wanted for this. Maybe one day we'll get a GUI / Webdev on board who can take my basic HTML / javascript GUI and make something a lot more slick leveraging the existing datastreams and API. This is in part why it was written this way.

DeeEmm, 2025-01-23:
OK just had a chance to take a look at this.

The MAF sensor reports mass per time period not volume, so is uninfluenced by temperature, pressure and humidity. 

If the temperature, pressure or humidity changes, causing the density of the charge to differ i.e. the number of molecules per given volume, the MAF will report the correct mass flow, as it measures the mass of the molecules that actually pass the sensor within a specified time period.

If more molecules pass the sensor, as is the case with denser air, more will be measured and the measured mass will be higher. et vice versa. So we could consider that mass flow is temperature and humidity agnostic.

However... Temperature and humidity WILL influence an objects flow characteristics in terms of volume, which may cloud the users perception of what the mass flow reading means. For example, lets say that we are measuring a super dense charge: Using a MASS measurement will make it appear that the object flows a lot more than the same object with a much less dense charge, as the molar mass will be higher. 

So in this respect one needs to be careful if wanting to use mass flow measurements for determining the _volumetric_ flow characteristics of an object, as they do not really compare very well. The two metrics are not exchangeable. Using mass flow does not overcome the temperature, barometric and humidity shortcomings of volumetric flow. For a mass measurement to make any sense the temperature, barometric pressure and humidity needs to be quoted, which means that If the user is interested in volumetric flow, they should be measuring in CFM which is a much more appropriate metric.

With the most recent version (Build 2501230049 onwards) the bench can be calibrated for mass flow i.e you can store the mass flow value as the calibration offset. This means that anyone who is specifically interested in reading mass flow can both calibrate and read the bench accurately using mass flow readings.

I'm not sure of a use case for mass flow, as in 99% of cases related to ICE we are interested in the volume of air that an object can flow, not the mass. However mass flow could be useful for things like mapping other MAF sensors, or conducting tests related to molecular density.

I will update the documentation relating to using the bench for mass flow.

Regarding the feature request, we could add an inset value into each tile to report the equivalent mass flow, rather than an additional set of mass related tiles.