DIY-Flow-Bench/DIY-Flow-Bench #307: MAF flow value calc incorrect

DeeEmm, 2025-01-05:
Just noticed that the flow value seems to be out of whack. I'm getting a value between 624 - 790 Kg/h on my test setup, nothing lower. 

_Originally posted by @DeeEmm in https://github.com/DeeEmm/DIY-Flow-Bench/issues/293#issuecomment-2571551392_
            
Most likely an issue relating to change from lookup tables to transfer function.

Tests on transfer function returned data in line with the lookup table values, so I suspect that either the transfer function has been implemented incorrectly, or the transfer function data was incorrect.

DeeEmm, 2025-01-05:
A bit of an explanation on how the transfer function works and is created.

All MAF data is now calculated on the fly from a transfer function which takes the voltage reference as an input. The transfer functions can be found in mafData.h along with other data previously found in the MAF.json files.

The transfer functions are calculated using a python file that can be found in the tools folder. There are also a number of other files that I have created that carry out functions related to converting the MAF files.  In all cases the raw data must be in JSON format. For example...

```
{
    "key" : value,
    "23" : 1443,
    "24": 1476,
    "25": 1523,
    "26": 1573,
    "etc" : etc
}
```

## mafScaleJsonValues.py

This program will take the key values in mafData.json and rescale them. All values are recalculated accordingly and a new file created. 

Example: you have a file that has key values from 0-1024 and need to convert them to run from 0-500mv.

The program will rescale the key values and interpolate the data values to align with the new key values.

## mafConvertKGHtoMGS.py

The new default value for all MAF transfer functions is KG/h. This program will convert a mafData.json file in mg/s and create a new file in kg/h.

The program converts all data values from mg/sec to kg/hour

## mafTransferFunctionGenerator.py

This script reads the MAF sensor data from mafData.json file and fits a polynomial, exponential, and power law function to the data. The best fit is then plotted along with the raw data. Coefficients of the best fit function are printed to the console. (generally 6th order polynomial)

This data can then be added to mafData.h

In addition to the above. I've also added a mimic page to monitor sensor values in realtime, which should assist with this issue.

DeeEmm, 2025-01-06:
OK. so my starting point is to replicate the data generated by mafTransferFunctionGenerator.py using excel.

The mafTransferFunctionGenerator.py file was AI generated and not my work. So as we are getting an error from the data generated by it, validating the data is the first logical step.

As a comparison here are the results generated for the BOSCH 0280218067 MAF by the mafTransferFunctionGenerator file

[image]

And here is the implementation of that data in the getMafFlow method in sensors.cpp

```
	// 6th degree polynomial calculation (Coefficients stored in mafData class)
    for(int i = 0; i < 6; i++) {
        flowRateKGH += _maf.getCoefficient(i) * vPower;
        vPower *= mafVolts;
    }
```

There's also an alternate long hand method that I wrote which achieves the same...

```
flowRateKGH = mafCoeff0 + (mafCoeff1 * mafVolts) + (mafCoeff2 * pow(mafVolts, 2)) + (mafCoeff3 * pow(mafVolts, 3)) + (mafCoeff4 * pow(mafVolts, 4)) + (mafCoeff5 * pow(mafVolts, 5));
```

The polynomials generated by mafTransferFunctionGenerator.py are stored in mafdata.h within a class called mafData as an array of arrays... 

```
    float mafCoeff[NUM_MAF_TYPES][NUM_COEFFICIENTS] = {
        {624.306263f, 25.418081f, 0.365468f, 0.002299f, 0.000003f, -0.000000f, 0.000000f}, // ACDELCO_92281162
        {817.925606f, -15.237634f, 0.091194f, -0.000073f, -0.000000f, 0.000000f, -0.000000f} // BOSCH_0280218067
    };
```

The key here is knowing that the transfer function is generated from 5th order polynomials. We can then replicate this using the LINEST function in excel.

Hopefully this will help to validate the data and give me a better understanding of what's going on.

I suspect that the implementation is the issue, however the AI generated version and my version of the calculation both give the same results. This logically leads to the way that the polynomials are being stored / handled. Hopefully dissecting everything will shed some light on what is going on.

DeeEmm, 2025-01-07:
Initial tests show deviation between mafTransferFunctionGenerator generated polynomials and Excel generated polynomials.

The excel generated code appears to give more realistic values that are in line with expectations, but is still not 100% at the value goes out of range at the top end of the scale. (more below)

[image]

Here's the Excel file I created...

[MAF polynomial calcs V1.xlsx](https://github.com/user-attachments/files/18327017/MAF.polynomial.calcs.V1.xlsx)

I also note that there appears to be an error with the ADC decode. Viewing the mimic data, the MAF voltage tops out at 6.14v, which is the max value for the ADC scale. We are obviously only reaching 5v, so should stop somewhat lower than that. I suspect that this may also be the reason for the flow values going out of range.

[image]

Looking through this issue also forced me to go back to look at the source of the raw MAF data. I think that we definitely need to validate the provenance of the data before we include it in the code. Some of the MAF data sets that were included in the previous version of the software are definitely of questionable quality. 🫤

DeeEmm, 2025-01-07:
Ahh yes. If you want to understand / learn more about using the LINEST function in excel to generate the polynomials, this is a pretty good video explanation. There are a few caveats to using it, such as creating arrays

https://www.youtube.com/watch?v=r7iXa6r5H7k

KirikauKiwi, 2025-01-07:
Happy NY, great work getting everything moved over to NVM, I have updated to the latest version of your DEV branch.

I tried to keep an eye on the work/updates over the holiday period, sorry if I am asking questions again that have been covered already.

Questions?
- The configuration page is it still a WIP i.e still needs to be updated. Currently I have a basic page view, no nice formatting etc.
- How do you access the Mimic Data you have shown above.

I am seeing similar odd data with the ADC with 0-6.144 V being shown in the API when the voltage is 0-5V this is with using a trim pot. 
With a pressure sensor connected and sitting at atmospheric pressure approx 2.5V the API reports about 0.76V
With a trimpot attached to the MAF input the API shows 6.144V when the voltage is about 1.95V

DeeEmm, 2025-01-07:
A fresh pair of eyes on this issue would be handy.

I suspect that the AI generated code to create the polynomials has some errors. The Excel generated code seems to be more accurate and the polynomials can be validated directly against those generated by the trend line (the formula displayed on the Excel graph). 

[image]

The R2 value is essential how good a fit the polynomials are to the data. The closer to 1, the better. You can see that the generated values match those for the trend line (the trend line values are rounded). From the experimentation that I have undertaken, generally the 5th and 6th order polynomials are essentially zero. (E+14, E+18)

To that end, I have tried using 4th degree polynomial calculation and whilst the fit is pretty good, it does tend to veer off at the extremities of the data. So it is clear that the zero values of the 5th / 6th order values do influence the curve.

Not sure about the calculation accuracy / mathematical comparison between Python and Excel. TBH, I have not really looked through the AI code yet. If Excel works as a tool, then thats fine. The important thing is accuracy.

The ACDELCO_92281162 data also may not be accurate. I have found some discrepancies with the data set. Some versions state mg/sec others kg/hour, so need to try and validate the reference data we have. Pretty sure that the BOSCH_0280218067 data is okay however.

KirikauKiwi, 2025-01-07:
Started with a clean sheet and now have the current DEV branch working correctly, i.e. I can now see the new/updated GUI etc.

- If I update the WIFI credentials to my local WIFi network it is failing to connect to it, is this currently a known problem???
- When adjusting the trimpot that I have connected to the PRef input it is adjusting the values for the two other pressure sensor inputs.

DeeEmm, 2025-01-07:
Both of those are likely bugs / issues. 

I'm still working my way through testing all of the settings, there are still many untested.

DeeEmm, 2025-01-07:
A quick look and I cannot see any obvious issues, however I note that if I save a value it is not displayed when I revisit the page.

Doing a quick test I can see that each value is getting stored in its respective NVM variable, but there is nmo way to interrogate the NVM other than using the getVariable method.

```
!:Debug Mode On
!:Verbose Print Mode On
Saving Configuration... 
Writing Configuration: Key: bSD_ENABLED  Value: 0 
Writing Configuration: Key: iMIN_PRESS_PCT  Value: 80 
Writing Configuration: Key: dPIPE_RAD_FT  Value: 0.328084 
Writing Configuration: Key: bFIXED_3_3V  Value: 1 
Writing Configuration: Key: dVCC_3V3_TRIM  Value: 0.0 
Writing Configuration: Key: bFIXED_5V  Value: 1 
Writing Configuration: Key: dVCC_5V_TRIM  Value: 0.0 
Writing Configuration: Key: iBME_TYP  Value: 7 
Writing Configuration: Key: iBME_ADDR  Value: 118 
Writing Configuration: Key: iBME_SCAN_MS  Value: 1000 
Writing Configuration: Key: iADC_TYPE  Value: 11 
Writing Configuration: Key: iADC_I2C_ADDR  Value: 72 
Writing Configuration: Key: iADC_SCAN_DLY  Value: 250 
Writing Configuration: Key: iADC_MAX_RETRY  Value: 10 
Writing Configuration: Key: iADC_RANGE  Value: 32767 
Writing Configuration: Key: dADC_GAIN  Value: 6.144 
Writing Configuration: Key: iMAF_SENS_TYP  Value: 0 
Writing Configuration: Key: dMAF_MV_TRIM  Value: 1.0 
Writing Configuration: Key: iMAF_SRC_TYP  Value: 18 
Writing Configuration: Key: iMAF_ADC_CHAN  Value: 0 
Writing Configuration: Key: iPREF_SENS_TYP  Value: 4 
Writing Configuration: Key: dPREF_MV_TRIM  Value: 2.0 
Writing Configuration: Key: iPREF_SRC_TYP  Value: 18 
Writing Configuration: Key: iPREF_ADC_CHAN  Value: 1 
Writing Configuration: Key: iFIXED_PREF_VAL  Value: 1 
Writing Configuration: Key: dPREF_ALG_SCALE  Value: 1.0 
Writing Configuration: Key: iPDIFF_SENS_TYP  Value: 4 
Writing Configuration: Key: dPDIFF_MV_TRIM  Value: 3.0 
Writing Configuration: Key: iPDIFF_SRC_TYP  Value: 18 
Writing Configuration: Key: iPDIFF_ADC_CHAN  Value: 1 
Writing Configuration: Key: iFIXD_PDIFF_VAL  Value: 1 
Writing Configuration: Key: dPDIFF_SCALE  Value: 1.0 
Writing Configuration: Key: iPITOT_SENS_TYP  Value: 4 
Writing Configuration: Key: dPITOT_MV_TRIM  Value: 4.0 
Writing Configuration: Key: iPITOT_SRC_TYP  Value: 12 
Writing Configuration: Key: iPITOT_ADC_CHAN  Value: 1 
Writing Configuration: Key: dPITOT_SCALE  Value: 1.0 
Writing Configuration: Key: iBARO_SENS_TYP  Value: 7 
Writing Configuration: Key: dBARO_MV_TRIM  Value: 5.0 
Writing Configuration: Key: dBARO_FINE_TUNE  Value: 0.0 
Writing Configuration: Key: dFIXD_BARO_VAL  Value: 101.3529 
Writing Configuration: Key: dBARO_ALG_SCALE  Value: 1.0 
Writing Configuration: Key: dBARO_SCALE  Value: 1.0 
Writing Configuration: Key: dBARO_OFFSET  Value: 0.0 
Writing Configuration: Key: iTEMP_SENS_TYP  Value: 7 
Writing Configuration: Key: dTEMP_MV_TRIM  Value: 6.0 
Writing Configuration: Key: dTEMP_FINE_TUNE  Value: 0.0 
Writing Configuration: Key: dFIXED_TEMP_VAL  Value: 21.0 
Writing Configuration: Key: dTEMP_ALG_SCALE  Value: 1.0 
Writing Configuration: Key: iRELH_SENS_TYP  Value: 7 
Writing Configuration: Key: dRELH_MV_TRIM  Value: 7.0 
Writing Configuration: Key: dRELH_FINE_TUNE  Value: 0.0 
Writing Configuration: Key: dFIXED_RELH_VAL  Value: 36.0 
Writing Configuration: Key: dRELH_ALG_SCALE  Value: 1.0 
Writing Configuration: Key: bSWIRL_ENBLD  Value: 0 
Loading Configuration 
Config NVM Free Entries: 249 
```

You can view this output if you enable debug mode and verbose mode

As you can see, I saved values 1 through 7 in each of the mv trim inputs. They all appear to be in the correct place.

I suspect that there may be an issue with the read / load.

KirikauKiwi, 2025-01-07:
> When adjusting the trimpot that I have connected to the PRef input it is adjusting the values for the two other pressure sensor inputs.

 Clarifing what I mean above!!!!
 When I adjust the 10K potentiometer to vary the input voltage the PRef Voltage it is also adjusting the ADC values for PDiff and Pitot sensors

DeeEmm, 2025-01-07:
Ahhh. okay. 

What values are the ADC channels set to?

DeeEmm, 2025-01-07:
Looks like defaults are all set to '1'. 

I've updated them but you will need to manually save the appropriate values in the GUI

KirikauKiwi, 2025-01-07:
Adjusting the potentiometer this is what I see in the API and it is mirrored in the GUI also.

https://github.com/user-attachments/assets/397fbee5-6d35-44b9-805f-b3e3e59d1110

A quick look in the code it looks like the ADC channel are set correctly out of the places I have looked so far.

I did find this in the datahandler.cpp
```
if (!_prefs.isKey("iMAF_SRC_TYPE")) _prefs.putInt("iMAF_SRC_TYPE", ADS1115);
  if (!_prefs.isKey("iMAF_SENS_TYPE")) _prefs.putString("iMAF_SENS_TYPE", 0);
  if (!_prefs.isKey("dMAF_MV_TRIM")) _prefs.putDouble("dMAF_MV_TRIM", 0.0);
  if (!_prefs.isKey("iMAF_ADC_CHAN")) _prefs.putInt("iMAF_ADC_CHAN", 0);

  if (!_prefs.isKey("iPREF_SENS_TYP")) _prefs.putInt("iPREF_SENS_TYP", MPXV7007);
  if (!_prefs.isKey("iPREF_SRC_TYP")) _prefs.putInt("iPREF_SRC_TYP", ADS1115);
  if (!_prefs.isKey("iFIXED_PREF_VAL")) _prefs.putInt("iFIXED_PREF_VAL", 1);
  if (!_prefs.isKey("dPREF_MV_TRIM")) _prefs.putDouble("dPREF_MV_TRIM", 0.0);
  if (!_prefs.isKey("dPREF_ALG_SCALE")) _prefs.putDouble("dPREF_ALG_SCALE", 0.0);
  if (!_prefs.isKey("iPREF_ADC_CHAN")) _prefs.putInt("iPREF_ADC_CHAN", 1);

  if (!_prefs.isKey("iPDIFF_SENS_TYP")) _prefs.putInt("iPDIFF_SENS_TYP", MPXV7007);
  if (!_prefs.isKey("iPDIFF_SRC_TYP")) _prefs.putInt("iPDIFF_SRC_TYP", ADS1115);
  if (!_prefs.isKey("iFIXD_PDIFF_VAL")) _prefs.putInt("iFIXD_PDIFF_VAL", 1);
  if (!_prefs.isKey("dPDIFF_MV_TRIM")) _prefs.putDouble("dPDIFF_MV_TRIM", 0.0);
  if (!_prefs.isKey("dPDIFF_SCALE")) _prefs.putDouble("dPDIFF_SCALE", 0.0);
  if (!_prefs.isKey("iPDIFF_ADC_CHAN")) _prefs.putInt("iPDIFF_ADC_CHAN", 1);

  if (!_prefs.isKey("iPITOT_SENS_TYP")) _prefs.putInt("iPITOT_SENS_TYP", MPXV7007);
  if (!_prefs.isKey("iPITOT_SRC_TYP")) _prefs.putInt("iPITOT_SRC_TYP", ADS1115);
  if (!_prefs.isKey("dPITOT_MV_TRIM")) _prefs.putDouble("dPITOT_MV_TRIM", 0.0);
  if (!_prefs.isKey("dPITOT_SCALE")) _prefs.putDouble("dPITOT_SCALE", 0.0);
  if (!_prefs.isKey("iPITOT_ADC_CHAN")) _prefs.putInt("iPITOT_ADC_CHAN", 1);
```
changing the ADC channel to the correct values didn't correct the issue I am seeing currently.

DeeEmm, 2025-01-07:
ADC channels are set on the GUI config page.

The default values were all set to '1'

If you change the values on the config page it should resolve the issue

DeeEmm, 2025-01-07:
values are in order - MAF = 0, pRef = 1, pDiff = 2 and Pitot = 3

KirikauKiwi, 2025-01-07:
> Looks like defaults are all set to '1'.
> 
> I've updated them but you will need to manually save the appropriate values in the GUI

I have manually changed them in the GUI, now there appears to be no data being displayed in the GUI, the API now shows the correct inputs regarding the potentiometers for PRef and PDiff, however Pitot is now fixed on 6.144V.

DeeEmm, 2025-01-07:
What sensor type are you using for the pitot?

DeeEmm, 2025-01-07:
NM. Checked though sensor types. 

I cannot recreate here.  6.144V is FSD for the ADC. Not sure why you should see this unless the ADC input is maxed out.

DeeEmm, 2025-01-07:
What ADC value do you see for the pitot?

KirikauKiwi, 2025-01-07:
I have 10K Potentiometers on MAF, PRef, and PDiff and the MPXV7007 connected for the Pitot.

The voltage output for the Pitot sensor is sitting about 2.7V when measured, the ADC value is maxed out at 32767 via the API.
I think the Pitot problem is to do with the ADC scaling as it is sitting above the 1.9V measured previously which for me currently then gives a FS reading on the ADC.

KirikauKiwi, 2025-01-07:
Testing with the oscilloscope and the API, I am getting a FS reading for the ADC with an input voltage at about 1.95-2 volts

DeeEmm, 2025-01-07:
Weird that you are seeing FSD at 2.7V.

There is definitely an issue with the ADC scaling. I suspect that what you are seeing is related to what I an seeing with the MAF and pRef inputs. I don't have anything connected to the pDiff / Pitot inputs but I can see the same thing if I assign them to channel 1 or 2.

[image]

I am unsure if this issue is recent or has always been present but it obviously needs a bit of a looking at 🫤

KirikauKiwi, 2025-01-07:
It wasn't present pre the 205-post-complile update which caused the memory issues. 

I recall that the memory issues did cause a problem with the ADC scaling and that we changed the get volts I think form a two look up values to the fixed formula of 6.144/32767. but that was a different problem to this one.

DeeEmm, 2025-01-07:
Yes that change was to bring the gain and range values into the method rather than calling them as it was failing sue to the memory issue. 

[image]

I will try and revert those changes

DeeEmm, 2025-01-07:
No difference.

KirikauKiwi, 2025-01-07:
It must be the RAW ADC values that are incorrect as the get volts is only a scaling factor to this.

5V should only give a raw ADC value of approx 26600 where as we are getting the FS of 32767

DeeEmm, 2025-01-07:
Certainly seems that way. Although I cannot figure out what might be causing this.

As an observation. Dialling in the displayed value does equate to an ADC value of approx 26600, which would seem to point to the conversion being correct.

KirikauKiwi, 2025-01-07:
I have had a check of the ADS1115 data sheet and it appears the default gain is two (2) so I suspect that the gain is not being set correctly to 2/3

DeeEmm, 2025-01-07:
Yes this is it.

If I mod the library so that the default is ADS1115_REG_CONFIG_PGA_6_144V it works okay.

[image]

Now to work out why this is happening.

The correct value is being set in hardware->begin. This hasn't changed from the previous version. 

```
if (config.iADC_TYPE != SENSOR_DISABLED) {

    _message.serialPrintf("Initialising ADS1115 ( Address: %u ) \n", config.iADC_I2C_ADDR);

    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V); // Set ADC Gain +/-6.144V range = Gain 2/3
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS); // Set ADC Sample Rate - 8 SPS
    
    if (!adc.testConnection()) {
      _message.serialPrintf("ADS1115 Connection failed");
      while(1); // Freeze
    } else {
      _message.serialPrintf("ADS1115 Initialised\n");
    }
  }
```

I will try changing the read frequency back to ADS1115_REG_CONFIG_DR_128SPS

DeeEmm, 2025-01-07:
OK. Found the issue - the class instantiation...

[image]

This works. (before the top version was commented out)

Cannot recall why / when this was changed.

DeeEmm, 2025-01-07:
Updated DEV branch. There's a few other changes in there too such as additional voltage values on mimic page for VCC 5v + 3.3v

Now just need to figure out what the issue with the MAF is. 😃 

Thanks for your help. 👍

KirikauKiwi, 2025-01-07:
I have noticed with the most recent DEV that you just updated to that there appears to be some latency appearing in the GUI & API updating the values.

PRef, PDiff and Pitot are all working correctly, MAF Flow KGH/CFM on the GUI is not updating, the ADC values and voltages in the API appear OK, the API Flow value in KG/H and CFM are not updating when varying the input voltage

KirikauKiwi, 2025-01-07:
> OK. Found the issue - the class instantiation...
> 
> <img alt="Screenshot 2025-01-08 at 12 41 11 am" width="450" src="https://private-user-images.githubusercontent.com/3038710/400794363-0b0480e6-0c06-437e-a95b-929d6392f443.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNjIyNTIsIm5iZiI6MTczNjI2MTk1MiwicGF0aCI6Ii8zMDM4NzEwLzQwMDc5NDM2My0wYjA0ODBlNi0wYzA2LTQzN2UtYTk1Yi05MjlkNjM5MmY0NDMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDdUMTQ1OTEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWRjMmI0MjgwZDFmZTMwYzQ4NWRmY2Y2Y2E2ZWVmNWQ3YmFhMmRiZGQyNWIwNzZkZTgzM2Q3YTNhODA1OGY2MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ylod3HCjYZMd0Q4shPix5uf_QwOlBVVfO-gb50YN3_4">
> This works. (before the top version was commented out)
> 
> Cannot recall why / when this was changed.

I had a bit of a dig around and it seems that there is an issue with the ESP32's ADC 1 & 2 when the WiFi AP is being used, I   cannot see how this should have affected the setup of the external ADC being used though?????? as this problem relates to the ESP32 and its internal ADC???

Or is there some crossed wires (code based) that may have caused the ADS1115 not to have its gain set correctly.
From the hardware.cpp line102
```
 adc = ADS1115_lite(config.iADC_I2C_ADDR);
```

From the hardware.cpp line 44
```
ADS1115_lite adc;
```

This code is before the hardware begin and appears to intertwine with the external ADC setup/use, which doesn't seem like it should.
```
// Kludge to fix ADC2 + WiFi - Source: https://forum.arduino.cc/t/interesting-esp32-issue-cant-use-analogread-in-esp_wifimanager-library/679348/2
#include "soc/sens_reg.h"    // needed for manipulating ADC2 control register
uint64_t reg_b;              // Used to store ADC2 control register
int sensorValue = 0;         // variable to store the value coming from the sensor

ADS1115_lite adc;

/***********************************************************
 * @brief CONSTRUCTOR
 ***/
Hardware::Hardware() {
}

/***********************************************************
 * @brief save_ADC_Reg - Kludge to fix ADC2 + WiFi 
 * @details  Save ADC2 control register value : Do this before begin Wifi/Bluetooth
 * @note Source: https://forum.arduino.cc/t/interesting-esp32-issue-cant-use-analogread-in-esp_wifimanager-library/679348/2
 ***/
void Hardware::save_ADC_Reg(void)
{
  reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
}

/***********************************************************
 * @brief restore_ADC_Reg - Kludge to fix ADC2 + WiFi 
 * @details  Restore ADC2 control register value : Do this after Wifi/Bluetooth
 * @note Source: https://forum.arduino.cc/t/interesting-esp32-issue-cant-use-analogread-in-esp_wifimanager-library/679348/2
 ***/
void Hardware::restore_ADC_Reg(void)
{ 
  WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
  SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
}

/***********************************************************
 * @brief readAnalog - Kludge to fix ADC2 + WiFi 
 * @details Read Analog pin value 
 * @note Source: https://forum.arduino.cc/t/interesting-esp32-issue-cant-use-analogread-in-esp_wifimanager-library/679348/2
 ***/
long Hardware::readAnalog(int sensorPin)
{ 
  restore_ADC_Reg();
  sensorValue = analogRead(sensorPin);
  // Serial.print("Pin " + String(sensorPin) + " = ");
  // Serial.println(sensorValue);

  return sensorValue;

}
```

I cannot make sense of the external ADC being the ADS1115 and running over I2C how it would need to be involved in the code for the internal ADC2 and WiFi problem/work around???

DeeEmm, 2025-01-08:
The 'kludge' is to do with the ESP32 internal ADC's and is not related to the ADS1115. 

On the Wemos board, you cannot use ADC Pin 2 if the WiFi is active as it will throw an error. If you try to assign pin 2 to analog read, it throws up an error. I am not 100% of the reason for this, it is something related to how the code is written that menas that when WiFi is enabled the results from pin 2 are affected.

The kludge is supposed to be a workaround for that particular issue. Essentially it creates a new analogRead() method called readAnalog. This provides access to the registers allowing the regular analogRead to take place, eliminating the error. However it did not work as expected. Sometimes the error would still show when initialising the pins, making it unreliable.

I commented it out when we were fault finding for the ADS1115 ADC issue, just in case it was causing the issue.

A fix for that issue should be possible. Even if pin 2 is required for both aspects, it should be possible to multiplex the data. However I suspect that this requires coding at the ESP_IDF level rather than via the Arduino framework, and interrogating the registers directly. This is a down-the-road problem, at least until the other issues get resolved.

Biggest issue at present is getting the MAF transfer function sorted out.

The generated trendline and polynomials match the data with a good degree of accuracy, but the implementation still has problems. Fixing the voltage issue did not resolve the problem. The flow values still veer off at the top end of the scale. Need to try to understand why this is happening.

I did read that extrapolating data using a polynomial curve does tend to lead to incorrect data when you travel past the end of the dataset, However we are calculating within the confines of the dataset. But the generated value seems to diverge long before we reach 5 volts.

KirikauKiwi, 2025-01-08:
Thanks for clarifying the ADC/WiFI issue, I just wasn't sure as to the formatting with the code relating to the two sections that changed with the commenting out to get the ADS1115 gain to be set correctly.

Are you seeing the latency issue I mentioned [above](https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2575537937) along with the MAF data issue mentioned too.

This has appeared for me after a5cff98d, where the data was updating as expected, to now where there is some latency again.

KirikauKiwi, 2025-01-08:
My issue with the data not displaying for the MAF sensor was due to Source for the MAF sensor on the Hardware configuration page not being set when updating to the current DEV version.

Having cleared the flash on the ESP32 this morning and reloading the current DEV version, the latency is still present but not as slow as it appeared with the same version yesterday afternoon. Going to try a restart on my computer first.

DeeEmm, 2025-01-08:
> Are you seeing the latency issue I mentioned [above](https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2575537937) along with the MAF data issue mentioned too.

Can you clarify where you are seeing the latency. I assume you are referring to the main Dashboard page as prior to this merge commit the mimic page did not exist.

I don't see any latency issues, but then I didn't see them (much) previously. Not sure why they are not so prevalent for me. Perhaps a browser issue? The Server Side Events (SEE) are a two-fold process. The data is packaged up by the ESP32 and sent out to the browser via the websocket connection, but it is the browser that receives, unpacks and updates the data.

Both the mimic and dashboard pages use seperate SSE streams to prevent sending / processing unnecessary data. The currently active page is set by the **status.GUIpage** variable within the webserver.cpp request handlers. This is then used to determine which SSE event to process in the main loop.

I've taken a look through the merge above and nothing really jumps out as a potential cause. The changes relate only to the mimic page, there's nothing there that really impacts the main dashboard page. The mimic page and associated data processing is entirely seperate from the main GUI Dashboard data processing.

I think that adding some kind of latency / cycle timer might be of benefit. We can split this into ESP and browser processing times. It should help to understand where any latency issues lay.

KirikauKiwi, 2025-01-08:
I first witnessed the problem on the main dashboard display with at first intermittent data updates, almost every second with the occasional stumble.
It also appears in the API when you make a request there are long delays often before the data is displayed.
I think I can also see it in the Mimic page.

I have just captured these two screen recordings, the first one I reload the page in the first few seconds, the second one was captures once the data populated the GUI.
The two recording were 1 minute and 16 seconds apart in their start recording times.

Capture one with page reload.

https://github.com/user-attachments/assets/7a928cd3-deca-4ba3-902b-00084cbdccb6

Capture two, once Dashboard data was displayed.

https://github.com/user-attachments/assets/4189076c-0da7-45c3-a52c-595038309b60

DeeEmm, 2025-01-08:
In normal operation, once the page has loaded the browser sits waiting for the SSE event from the server. Once triggered it fires the javascript event listener in the browser that then parses the JSON data passed from the server and then updates the GUI elements. 

In the second video where the values do not update, the timeline freezes indicating that the browser itself has stalled. Obviously when the browser stalls it cannot process any of the SSE events as javascript execution also stalls. 

If you view the events tab, you will see a timestamp on each of the SSE events. If it was the javascript or server code causing the issue, I would expect the timeline to continue whilst the SSE events stall or error. I do not note any errors in the console.

I am seeing a pretty consistent SSE update rate here of about 1/2 a second, which is the rate defined in the code. I do not see the issue that you see.

[image]

The issue could be browser related or potentially WiFi, computer, network or even hardware related, but given that it is working okay here and I am unable to replicate the issue I suspect that it is not code related. I am confident that the memory issues we previously experienced have been addressed with the recent code changes.

Not sure what else to suggest. Maybe try and eliminate potential network issues with a network router reboot. Try a different computer or device, undertake a network benchmark / speed test. Check the system activity monitor to see if there is a memory or CPU issue.

If you are using your local WiFi network, switch the AP mode and see if the issue improves.

KirikauKiwi, 2025-01-08:
I am clearing the flash each time when I make a change rolling back and when uploading the current version 3591f670, no physical or other software changes other than the differences between the versions are being made.

I am seeing very odd behaviour, I rolled back to a5cff98 and had good GUI data refresh etc for both BME and ADS data.
I then uploaded the current version and had good GUI data refresh for BME and ADS data, The MAF data was not updating, it appears in the current version the MAF data is default set to "Disable", I changed this and only this in the configuration GUI to ADC which enabled the MAF data update as expected, however the BME data was then not updating.

I have connected my scope to decode the I2C bus and the data form the BME is sometime not there, unfortunately I cannot make sense of what is happening as there are two many things changing between when I upload the versions to be able to pin point what may actually be causing the problem.

I am also experiencing this on occasion when refreshing the GUI, I get a blank page with the plain text of "Decompression error [Init]" from here I usually have to reset the ESP32 to get it to function again, this appears to be mainly on the Dashboard and mimic pages, this maybe due to the fact they are the pages I am trying to access when this occurs.

Sorry I cannot be of much help currently other than to point out what I am experiencing.

PS, I still seeing the delays in the API to, for example once I type A to request the voltages I am getting a delay of up to approx 40 seconds.

DeeEmm, 2025-01-08:
There's no need to clear the flash. Once you have set your hardware settings, configuration and pins data, there is no need to change it. Just upload the new code using a regular compile + upload. It will pick up the data already present in the NVM.

The 'Decompression Error' means that the HTML code stored in compressed format in PROGMEM could not be re-inflated. The code uses the built in miniz library to do this. For some reason it is failing. This could potentially be a symptom of another issue, but what is unclear.

> Is there any way you know of to be able to interrogate the I2C bus and devices on it to see their actual setup/configs????

Yes, you can use your scope to decode the traffic between devices, but it is not something that I have done.

Regarding the configuration of your board, are you using the env:esp32dev to compile and upload?? If youa re using a development / debugging framework it will load a bunch of additional stuff that will weigh things down

KirikauKiwi, 2025-01-08:
> There's no need to clear the flash. Once you have set your hardware settings, configuration and pins data, there is no need to change it. Just upload the new code using a regular compile + upload. [...]

I have only been using the clear flash to ensure there were not stragglers left behind that may have been causing issues between the versions.

Yes I am using the env:esp32dev

Here is a screen recording of the API response with the current DEV version loaded.
You can see the TX green light flash when I enter a keyboard command.

https://github.com/user-attachments/assets/f83c3ab9-da6f-4681-95ba-32c9d9695d65

DeeEmm, 2025-01-08:
I don't see that here. The response is pretty much instant.

https://github.com/user-attachments/assets/22c975d6-8aef-4e23-9349-59b553cdc294

Do you have a regular ESP32? i.e. not a dev board or MStack?

DeeEmm, 2025-01-08:
I know that my debugging board is laggy as hell. I only crack it out if I need to debug line by line. Otherwise it's just too much of a PITA to use, for very similar reasons as you are experiencing. I'd wager it could be the issue. 

The generic Wemos D1 R32 clones are pretty inexpensive.

KirikauKiwi, 2025-01-08:
Here is a screen recording of DEV version 3591f670

https://github.com/user-attachments/assets/dc04844d-1b82-45b5-974f-2f1fa194fdc5

Here is a screen recording of DEV version a5cff98

https://github.com/user-attachments/assets/3470bb54-fe71-4592-aa1d-66af2b25a77b

KirikauKiwi, 2025-01-08:
> I know that my debugging board is laggy as hell. I only crack it out if I need to debug line by line. Otherwise it's just too much of a PITA to use, for very similar reasons as you are experiencing. I'd wager it could be the issue.
> 
> The generic Wemos D1 R32 clones are pretty inexpensive.

If this is the case, it would put most of the work you have done, down the gurgler as there would be no need to have a variations for the pins etc, I cannot believe that an ESP32 WROOM 32E on one generic board should behave so different to another ESP32 WROOM 32E on another generic board.

[image]

DeeEmm, 2025-01-08:
I thought that you were using a debugging board (with JTAG)

DeeEmm, 2025-01-08:
The main issue is that I cannot replicate your issue here so cannot debug the problem, whatever that may be.

The most logical deduction is that it is environment related as that is the thing that differs between here and there.

KirikauKiwi, 2025-01-08:
> The main issue is that I cannot replicate your issue here so cannot debug the problem, whatever that may be.
> 
> The most logical deduction is that it is environment related as that is the thing that differs between here and there.

Yes this does makes some sense, however I have had stable working versions prior to the 205-post-compile changes that initially caused the memory issues, it also doesn't really explain why between to recent DEV code versions what has affect my setup and not apparently your setup.

What sample rate are you currently using in the code you are running now? I haven't checked mine but you mentioned trying 128SPS again so I need to look at that here to see if I am on 8SPS or 128SPS???

What is set as the sample rate in the DEV version 3591f67????

DeeEmm, 2025-01-08:
I only see a screen shot, the videos do not play.

The I2C addresses are set in the GUI, not in the code.

If the I2C address is incorrect it will need to be set in the GUI.

My advice is use the current version and try to debug that if you have issues.

If you are convinced that there is an error introduced between these two versions, then review the dIffs and try and identify what that issue is. When you can replicate the issue, let me know and can then try and fix it. But at the moment without being able to even see the issue here, there is no way that I can debug it. 

I went through the diffs and did not find anything.

RE I2C sample rate. I have exactly the same code version as you. 

The only difference is the hardware and environment.

KirikauKiwi, 2025-01-08:
> I only see a screen shot, the videos do not play.
>
> The I2C addresses are set in the GUI, not in the code.
>
> If the I2C address is incorrect it will need to be set in the GUI.
>
> My advice is use the [...]

Try here to see the videos,
https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2577431148

DeeEmm, 2025-01-08:
Like I said, the videos do not play. I just see the screen shot.

Even so. it does not help me as I cannot reproduce the issue here.

KirikauKiwi, 2025-01-08:
I know this is handled differently than the BME and ADS data, but as an observation the START BENCH, BENCH BUTTONS are just as responsive as each other between the two DEV versions mentioned, I am only seeing problems with the BME and ADS data being displayed on the GUI and API.

As I have been writing this I have rolled back to DEV version 148f4ab and here is have a responsive API, and both BME and ADS data are updating as expected in the Dashboard GUI.

148f4ab is one verified version before 3591f67, will try the intermediate version next.

DeeEmm, 2025-01-08:
They are two different things.

The start / stop buttons work using XHR. In this case it is the browser that sends the data to the server.

The SSE works in the opposite direction, the server sends the data to the browser.

KirikauKiwi, 2025-01-08:
Ok so the problems appear for me in DEV version ac591a2 which appears to have the changes made to the the ADS setup to correct the GAIN, so I am assuming that there is something there that has adversely affect my ADS1115 and not yours!

So in DEV version ac591a2 I have poor BME and ADS data on the GUI and a sluggish API reponse.
If I decode the I2C bus with this version I see no data from the BME.
When I roll back to DEV version 148f4ab I have a good BME and ADS data updates on the GUI and a responsive API and I can see the BME data when decoding the I2C bus.

KirikauKiwi, 2025-01-08:
> The data pipe for SSE is the socket connection between the server and browser.
> 
> Do you find that the browser eventually hangs. I.e. the 'Uptime' counter on the dashboard stops counting and the GUI values stop updating?

No the browser doesn't hang the uptime counter is updating as expected.

DeeEmm, 2025-01-08:
Diffs... 
https://github.com/github-linguist/linguist/compare/148f4ab0..ac591a2

KirikauKiwi, 2025-01-08:
> Diffs... https://github.com/DeeEmm/DIY-Flow-Bench/compare/148f4ab0..ac591a2

This is not displaying for me, possibly because of the Zero on the end of 148f4ab"0"

DeeEmm, 2025-01-08:
Yeah just edited it. C+P error (nut loose on the keyboard)

DeeEmm, 2025-01-08:
The main change there is the ADS instantiation

[52 more replies]

closed 2025-01-08 by DeeEmm