DIY-Flow-Bench/DIY-Flow-Bench #315: WiFi credentials not working in SSID mode

DeeEmm, 2025-01-09:
> with clear flash and full clean, most things now works except for setting a SSID and Password for wifi !?

_Originally posted by @stefan63 in https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2581457520_

DeeEmm, 2025-01-11:
Issue resolved

closed 2025-01-11 by DeeEmm

stefan63, 2025-01-11:
Yes wifi now works as expected for me.
However I have tested to enable MAF and can confirm that the issue with
killing the BME280 and getting a sluggish UI is still valid.
Only whay to solve it is to clean flash and reflash again. (Using DEV
downloaded @ 21:10 UTC  Jan 11 2025).
However I must admit I didnt actually have a MAF hooked up can that be the
issue?
Kind regards/Stefan

DeeEmm, 2025-01-12:
> However I must admit I didnt actually have a MAF hooked up can that be the issue?

No, it just means that the data it uses is not valid.

Thanks for confirming the behaviour, I cannot see this issue here for some reason. I am at a loss as to why. I need to find some way to replicate this issue here. I will create a new issue.

DeeEmm, 2025-01-12:
I'm moving the discussions relating to GUI sluggishness into a dedicated issue - #316