DIY-Flow-Bench/DIY-Flow-Bench #316: GUI Sluggish when MAF enabled

DeeEmm, 2025-01-12:
I have tested to enable MAF and can confirm that the issue with
killing the BME280 and getting a sluggish UI is still valid.
Only whay to solve it is to clean flash and reflash again. (Using DEV
downloaded @ 21:10 UTC  Jan 11 2025).
Kind regards/Stefan

_Originally posted by @stefan63 in https://github.com/DeeEmm/DIY-Flow-Bench/issues/315#issuecomment-2585426280_

GUI data stops updating on the Dashboard GUI, or becomes visually blocky, stop/stating, sluggish etc like it did since the 205-post-compile-configuration changes.

I also wouldn't expect a single change in the GUI configuration to enable the ADC for the MAF sensor to visually freeze the BME data on the Dashboard GUI.

I would also expect this change to be reversible which for me it is not currently, the only way I can get the BME data updating again is to clear the flash on the ESP32 and upload the code again.

As mentioned earlier today when I reverse this change (disable the MAF ADC) it does appear to improve/restore the API sluggishness.

My current list of issues are then;
- When I change the MAF to enable the ADC, the BME data on the dashboard GUI stops updating.
- I cannot reverse this change via the configuration GUI, I have to clear the flash and upload the code again.
- When I enable the MAF ADC, the API becomes very slow to respond, often more than 30 seconds, this does appear to improves again if the MAF is disable via the configuration GUI.
- The GUI overall has issues loading, the pages are often incomplete or load blank (known issue)
- WiFi for Local network the configuration saves but the Wifi won't connect and rolls back to WiFi AP mode. (known issue)

We know the hardware works so it has to be software related!

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2582625098_

DeeEmm, 2025-01-12:
Here is my test with a WEMOS D1 R32 clone with DEV version e7b8ad52.

I can confirm the WiFi fix has worked for me, which is making testing much less of a PITA.

Nothing else appears different to me other than now occasionally on the sluggish API from the initial upload I am now seeing some data that is like the baud rate is not set correctly.

Wemos Clone tasks

https://github.com/user-attachments/assets/218fc3b7-aecd-4834-ab27-ec4f03630f4a

API snap shot
[image]

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2585347721_

DeeEmm, 2025-01-12:
And since running the code on the WEMOS clone I have had two crashes that are the same.

```
assert failed: heap_caps_free heap_caps.c:381 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40083bcd:0x3ffe8200 0x4008dd75:0x3ffe8220 0x40093585:0x3ffe8240 0x40084136:0x3ffe8370 0x400935b5:0x3ffe8390 0x400df025:0x3ffe83b0 0x401a1c33:0x3ffe83d0 0x400df0c5:0x3ffe83f0 0x400e0520:0x3ffe8410 0x400e07b5:0x3ffe8470 0x400e268e:0x3ffe8490 0x400f0375:0x3ffe84d0 0x400ede7d:0x3ffe8520 0x400edf8d:0x3ffe8570 0x400ee175:0x3ffe85c0 0x40186999:0x3ffe85e0 0x40186a15:0x3ffe8610 0x401870c6:0x3ffe8630
```

_Originally posted by @KirikauKiwi in https://github.com/DeeEmm/DIY-Flow-Bench/issues/307#issuecomment-2585348717_

DeeEmm, 2025-01-12:
Issue now confirmed by @stefan63 and additional test by @KirikauKiwi 

My big issue is that I cannot replicate this issue locally, so i need to try and understand why that is, so that I can begin to fault find. 

Breaking the contributing factors down to try and isolate the root cause we have the following:

- Enabling MAF sensor causes slow scan times
- Change made in ac591a2 seems to be the [origin of issue](https://github.com/DeeEmm/DIY-Flow-Bench/compare/148f4ab..ac591a2)
- After enabling the MAF, disabling the MAF does not recover the issue

Things that do not appear to be a factor:

- ESP32 Board type
- Declaration of ADS1115-lite library
- Semaphore in main loop (controls API calls)

Things that Help / Improve issue:

- My hardware setup??? (Actual factor unknown)

What else have we identified that either contributes to, or reduces the problem?
What else have we eliminated by way of testing?

------------------

I guess I would like to start by analysing differences in hardware. We are after all running the same code. If we can isolate what makes my system impervious to the issue, I can then set my system up to replicate the issue. We may also be able to identify hardware components that are suitable / not suitable and add them to the recommended components list. 

I will sit down later and document my exact setup, we can then try and identify any differences between what I have, and what you guys are running. Here's a basic start...

--------------------

- Wemos D1 R32 clone MCU (I need to formally identify this one)
- PicoDev BME280 BOB
- Sparkfun pca9306 logic level shifter BOB
- Generic ADS1115 ADC BOB (I need to formally identify this one)
- Generic SD card BOB (I need to formally identify this one)

- Official DIYFB shield + components as per [Schematic](https://github.com/DeeEmm/DIY-Flow-Bench/blob/master/PCB/Schematic_DIY-FLOW-BENCH-V2-3.jpg) Component tolerances are general and not high spec.

At the moment I only have two pots hooked up for testing as follows:

A10K pot on MAF input
A10K pot on pRef input

pDiff and Pitot inputs are currently floating.

------------------------

This is my pins data. Note that the '4294967295' value is actually the integer representation of  '-1' (0xffffffff)

Saving Pins data to NVM... 
Writing Pins data: Pin: VAC_SPEED  Value: 25 
Writing Pins data: Pin: VAC_BLEED_VALVE  Value: 26 
Writing Pins data: Pin: VAC_BANK_1  Value: 13 
Writing Pins data: Pin: VAC_BANK_2  Value: 12 
Writing Pins data: Pin: VAC_BANK_3  Value: 14 
Writing Pins data: Pin: AVO_STEP  Value: 32 
Writing Pins data: Pin: AVO_DIR  Value: 33 
Writing Pins data: Pin: FLOW_VALVE_STEP  Value: 27 
Writing Pins data: Pin: FLOW_VALVE_DIR  Value: 4 
Writing Pins data: Pin: VCC_3V3  Value: 4294967295 
Writing Pins data: Pin: VCC_5V  Value: 4294967295 
Writing Pins data: Pin: SPEED_SENS  Value: 15 
Writing Pins data: Pin: SWIRL_ENCODER_A  Value: 4294967295 
Writing Pins data: Pin: SWIRL_ENCODER_B  Value: 4294967295 
Writing Pins data: Pin: ORIFICE_BCD_1  Value: 34 
Writing Pins data: Pin: ORIFICE_BCD_2  Value: 36 
Writing Pins data: Pin: ORIFICE_BCD_3  Value: 39 
Writing Pins data: Pin: MAF  Value: 4294967295 
Writing Pins data: Pin: PREF  Value: 4294967295 
Writing Pins data: Pin: PDIFF  Value: 4294967295 
Writing Pins data: Pin: PITOT  Value: 4294967295 
Writing Pins data: Pin: TEMPERATURE  Value: 4294967295 
Writing Pins data: Pin: REF_BARO  Value: 4294967295 
Writing Pins data: Pin: HUMIDITY  Value: 4294967295 
Writing Pins data: Pin: SERIAL0_TX  Value: 1 
Writing Pins data: Pin: SERIAL0_RX  Value: 3 
Writing Pins data: Pin: SERIAL2_TX  Value: 16 
Writing Pins data: Pin: SERIAL2_RX  Value: 17 
Writing Pins data: Pin: SDA  Value: 21 
Writing Pins data: Pin: SCL  Value: 22 
Writing Pins data: Pin: SD_CS  Value: 5 
Writing Pins data: Pin: SD_MOSI  Value: 23 
Writing Pins data: Pin: SD_MISO  Value: 19 
Writing Pins data: Pin: SD_SCK  Value: 18 
Writing Pins data: Pin: SPARE_PIN_1  Value: 4294967295 
Writing Pins data: Pin: SPARE_PIN_2  Value: 4294967295 
Loading Pins Data 

-----------------------------

Things that I can immediately think of testing that may be different on my setup

- Adding additional pots to pRef an Pitot inputs. - I would not expect this to make a difference but worth testing.
- Removing SD card BOB - SD code is commented out, but is in circuit and pins are assigned.

If anyone can see / think of any other differences between your setup and mine, post below and I will test to see if I can replicate the problem here.

DeeEmm, 2025-01-12:
Some additional info.

The ADS1115 is a generic ebay version - https://www.ebay.com.au/itm/303971402874 However, the version that I received is different from most photos (note the markings - 16 bit i2c adc +p g ahw-198)

[image]

Unsure if this is relevant. St the moment I do not have another version to try. Jaycar do not appear to stock them. Will try Aztronics + Altronics

-------------------------------------

I tested reassigning SD card pins to '-1' and disconnecting the SD card BOB from the main board, however the performance remains the same.

-------------------------

Tried disabling the MAF and rebooting - Performance remained the same

-------------------------

Tried clearing flash and re-uploading to default configuration (MAF disabled) - Performance was different, however it was the opposite to what was expected.

Normal Task frequencies

[image]

After clearing flash + vanilla upload

[image]

The ADS task time has increased five-fold.

To clarify this is the time it takes from the end of previous ADS task to the beginning of next ADS task and should be determined by the ADC / BME scan intervals

So the ADS task scan interval actually increased with the MAF disabled. 

This could indicate that the BME task was greedier and therefore held the task preventing the ADS from running. This is the exact opposite to what you guys see, but may be indicative of the mechanism that causes both. (i.e. task 'greediness'). This gives me an idea to measure how many times each task executes. 

Theoretically, each task should be able to run in accordance with the scan time proportions  i.e. 

```
**_BME scan : ADS scan_** 
```

which is currently **1 : 4** or **1 sec : 250 millisecs**. This is also approximately what I see in the Task frequencies on my setup.

I will try to devise some counters to count how many cycles each task runs for. We can then compare results

DeeEmm, 2025-01-12:
Added a simple scan counter to the mimic page. This takes a rolling average of the number of times each task is run in succession.

KirikauKiwi, 2025-01-12:
I have been able to sudo recreate the problem at a very elementary level with the test code I created to test the BME and ADS in isolation.

By sudo I mean that sorry to mention it @DeeEmm looking at the I2C bus. I can see similar behaviour on the I2C bus in the test code that I am seeing with there I2C bus in the DIYFB code just by varying the delay at the end of my code block.

What I have found in changing a delay in my test code that the I2C bus visibly changes and once I decrease the delay below a certain level the I can no longer see the I2C address for the BME appear in the I2c bus decode.

What I am seeing in my test code when I describe as a visible change is depicted in the two images at the bottom of this post.

When running the code with a 125ms delay the BME data block on the I2C bus takes about 1.8ms to complete.
If I change the delay down to IIRC 50ms or lower that the BME address starts to disappear from the I2C decode.

The test code I created with the help of ChatGPT as my coding skills a very poor used the same libraries as DIYFB, Tiny280 and ADS1115_lite and I have tried to set it up as close as possible to DIYFB by using semaphore handling and the sam board, env top etc.

This is the test code .cpp
```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {
        float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
        float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
        float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

        Serial.print("Temperature: ");
        Serial.print(temperature);
        Serial.println(" C");

        Serial.print("Humidity: ");
        Serial.print(humidity);
        Serial.println(" %");

        Serial.print("Pressure: ");
        Serial.print(pressure);
        Serial.println(" hPa");

        // Release the semaphore after reading BME280
        xSemaphoreGive(i2cSemaphore);
    }

    // Read from ADS1115
    for (int i = 0; i < 4; i++) {
        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {
            // Set the multiplexer for the appropriate channel
            switch (i) {
                case 0:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                    break;
                case 1:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                    break;
                case 2:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                    break;
                case 3:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);
                    break;
            }

            // Trigger the conversion
            adc.triggerConversion();

            // Wait for the conversion to complete (adjust the delay as necessary)
            delay(10); // You can fine-tune this delay based on your needs

            // Read the conversion result for the selected channel
            int16_t adcValue = adc.getConversion();
            float voltage = adcValue * (6.144 / 32767.0);

            Serial.print("Analog Input ");
            Serial.print(i);
            Serial.print(": ");
            Serial.print(voltage);
            Serial.print(" V");
            Serial.print(": ");
            Serial.print(adcValue);
            Serial.println(" R");

            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
        }
    }

    Serial.println(); // Print a newline for better readability

    // Delay for the ADS1115 sample rate
    delay(125); // Wait for 125 ms to respect the 8 SPS rate
}
```

I2C decode with 125ms delay.
[image]

I2C decode with 125ms delay commented out.
[image]

What I speculate from these observations in this isolated test, is that something in the DIYFB code is affecting the implementation or the timings in relation to the I2C bus.

KirikauKiwi, 2025-01-12:
Small update, I am sometimes seeing the BME data when the delay is commented out. The I2C decode appears different to when the delay is implemented.

[image]

DeeEmm, 2025-01-12:
It's not quite the same. There are some errors in your code as well.

The for-loop essentially does nothing it basically incrementally sets each channel, So it will always only read channel 4

There should also be no delay in the ADC reads. It does not need to wait for the conversion to finish as the library implicitly does this within the getConversion method...

From library documentation...
```
      adc.triggerConversion(); //Start a conversion.  This immediately returns
      Raw = adc.getConversion(); //This polls the ADS1115 and wait for conversion to finish, THEN returns the value
```

The loop within the DIYFB code also does not delay. It is the frequency that each task is called that is controlled independently with the scan interval timers. This is not the same as just adding a delay in the loop.

The tasks operate independently of the loop. The tasks are not part of the loop.

The loop itself keeps running. The only delay in the loop is to prevent the watchdog from triggering (1ms) as otherwise it scans too fast and triggers the WDT

```
vTaskDelay( 1 );  //mSec delay to prevent Watch Dog Timer (WDT) triggering for empty task
```

One of the reasons I previously mentioned that watching the I2C bus to try and diagnose code errors is that you only see a very small snapshot of the data. The assumption that the BME data reads are not happening based on not seeing them in the snapshot is folly. To get an idea of the actual bus activity it really needs to be datalogged and analysed. 

It really is much better to try and diagnose this issue from within the code itself.

DeeEmm, 2025-01-12:
My Bad the loop does cycle through all four channels

Remove the timer from the ADC reads as it is not needed and not present in the DIYFB code

```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {
        float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
        float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
        float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

        Serial.print("Temperature: ");
        Serial.print(temperature);
        Serial.println(" C");

        Serial.print("Humidity: ");
        Serial.print(humidity);
        Serial.println(" %");

        Serial.print("Pressure: ");
        Serial.print(pressure);
        Serial.println(" hPa");

        // Release the semaphore after reading BME280
        xSemaphoreGive(i2cSemaphore);
    }

    // Read from ADS1115
    for (int i = 0; i < 4; i++) {
        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {
            // Set the multiplexer for the appropriate channel
            switch (i) {
                case 0:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                    break;
                case 1:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                    break;
                case 2:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                    break;
                case 3:
                    adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);

            // Trigger the conversion
            adc.triggerConversion();

            // Read the conversion result for the selected channel
            int16_t adcValue = adc.getConversion();
            float voltage = adcValue * (6.144 / 32767.0);

            Serial.print("Analog Input ");
            Serial.print(i);
            Serial.print(": ");
            Serial.print(voltage);
            Serial.print(" V");
            Serial.print(": ");
            Serial.print(adcValue);
            Serial.println(" R");

            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
        }
    }

    Serial.println(); // Print a newline for better readability

    // Delay for the ADS1115 sample rate
    delay(125); // Wait for 125 ms to respect the 8 SPS rate
}
```

KirikauKiwi, 2025-01-12:
I am only trying to help the best I can!!!

> The for-loop essentially does nothing it basically incrementally sets each channel, So it will always only read channel 4

It works, I see different voltages and they change independently of one another!!

Do with my observation as you wish, but to my eyes it appears that what ever is happening in the code is inadvertently affecting something that is related to the data going to and coming from the I2C bus.

My test is in isolation and yes will of course be different from what you have implemented in the DIYFB code, my isolation test does however show that by varying the delay is affecting the I2C bus and this is mirroring  what I have seen and been able to measure in when running the DIYFB code.

I don't disagree with you about the data logging of the bus would be a better solution, unfortunately I don't have a logic analysers to be able to do this!!!!

> It really is much better to try and diagnose this issue from within the code itself.

I agree with you on this as I also believe it to be a software issue, I have been reporting what I am seeing able to measure at a hardware level which is obviously affected by the software instructing it. I am sure you would also expect software changes to make changes to how the hardware operates as this is what it is design to do.

KirikauKiwi, 2025-01-12:
I would expect that the the enable of the MAF input is somehow affecting the timings too as you are going from reading three of the analogue inputs on the ADS1115 to four.
What I don't understand is why it is not reversible, as if the code that is being enabled, disabled with the change it should be reversible, but it is not, so this would indicate that the code is not working correctly.

Unfortunately for me trying to work through the DIYFB code is difficult due to only having a very elementary understanding of C++, I get and understand the logic behind it but have difficulty reading the code it self.

DeeEmm, 2025-01-12:
Normal code - Spend 3 hours developing / 30 minutes debugging
AI code - Spend 3 minutes developing / 3 days debugging

There are many issues with the code that has been generated for you.

This is a better approximation of DIYFB

```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

adcPollTimer = 0;
bmePollTimer = 0;
BME_SCAN_DELAY = 1000;
ADC_SCAN_DELAY = 250;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (millis() > bmePollTimer){

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            bmePollTimer = millis() + BME_SCAN_DELAY; 

            float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
            float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
            float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

            Serial.print("Temperature: ");
            Serial.print(temperature);
            Serial.println(" C");

            Serial.print("Humidity: ");
            Serial.print(humidity);
            Serial.println(" %");

            Serial.print("Pressure: ");
            Serial.print(pressure);
            Serial.println(" hPa");

            // Release the semaphore after reading BME280
            xSemaphoreGive(i2cSemaphore);
        }
    }

    // Read from ADS1115
    if (millis() > adcPollTimer){

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            adcPollTimer = millis() + config.ADC_SCAN_DELAY; 

            for (int i = 0; i < 4; i++) {
                // Set the multiplexer for the appropriate channel
                switch (i) {
                    case 0:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                        break;
                    case 1:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                        break;
                    case 2:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                        break;
                    case 3:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);

                // Trigger the conversion
                adc.triggerConversion();

                // Read the conversion result for the selected channel
                int16_t adcValue = adc.getConversion();
                float voltage = adcValue * (6.144 / 32767.0);

                Serial.print("Analog Input ");
                Serial.print(i);
                Serial.print(": ");
                Serial.print(voltage);
                Serial.print(" V");
                Serial.print(": ");
                Serial.print(adcValue);
                Serial.println(" R\n");
          }
            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
        }
    }

    Serial.println(); // Print a newline for better readability

    delay(1);
}
```

DeeEmm, 2025-01-12:
Note: untested

DeeEmm, 2025-01-12:
I've moved the sensor tasks to core 2. Most system tasks such as wifi etc are on core 1. 

See if this makes a difference

KirikauKiwi, 2025-01-12:
> My Bad the loop does cycle through all four channels
>
> Remove the timer from the ADC reads as it is not needed and not present in the DIYFB code
>
> ```
> #include <Wire.h>
> #define TINY_BME280_I2C
> #define [...]

So with the 10ms delay commented out it doesn't change anything as you have mentioned and I note that it is commented that it can be varied.

Test code with 10ms delay commented out.
[image]

Test code with 10ms delay and 125ms delay commented out. BME data appears truncated.
[image]

PS the picsocope software captures 64 pages of waveforms and with the setting I have for the I2C decode set on the oscilloscope it take about 10 seconds to fill the buffer, so I have been searching this each time so essentially it is a 10 second data log and I have not just be relying on a single screen of the oscilloscope with would only be a few milliseconds. Also I have visually live monitoring the buffer which is not that difficult to see a visible difference to when the BME is there regularly or infrequently.

KirikauKiwi, 2025-01-12:
> Normal code - Spend 3 hours developing / 30 minutes debugging AI code - Spend 3 minutes developing / 3 days debugging
> 
> There are many issues with the code that has been generated for you.

I am not surprised by this, however it has enable me to at least in part be able to recreate a similar issue to what me and Stefan are having!!!!, I am not saying it is the same issue, it does behave in a similar fashion.

Thanks for the sample code that is more like how you have implemented it in DIYFB, I will test that now!!

DeeEmm, 2025-01-12:
> BME data appears truncated.

What do you mean by this?

KirikauKiwi, 2025-01-12:
> Normal code - Spend 3 hours developing / 30 minutes debugging AI code - Spend 3 minutes developing / 3 days debugging
>
> There are many issues with the code that has been generated for you.
>
> This is a [...]

I get these compile errors, if you can edit this code to correct the errors I will test it.

```
src/main.cpp:16:1: error: 'adcPollTimer' does not name a type
 adcPollTimer = 0;
 ^~~~~~~~~~~~
src/main.cpp:17:1: error: 'bmePollTimer' does not name a type
 bmePollTimer = 0;
 ^~~~~~~~~~~~
src/main.cpp:18:1: error: 'BME_SCAN_DELAY' does not name a type; did you mean 'portMAX_DELAY'?
 BME_SCAN_DELAY = 1000;
 ^~~~~~~~~~~~~~
 portMAX_DELAY
src/main.cpp:19:1: error: 'ADC_SCAN_DELAY' does not name a type; did you mean 'portMAX_DELAY'?
 ADC_SCAN_DELAY = 250;
 ^~~~~~~~~~~~~~
 portMAX_DELAY
src/main.cpp: In function 'void loop()':
src/main.cpp:48:20: error: 'bmePollTimer' was not declared in this scope
     if (millis() > bmePollTimer){
                    ^~~~~~~~~~~~
src/main.cpp:52:39: error: 'BME_SCAN_DELAY' was not declared in this scope
             bmePollTimer = millis() + BME_SCAN_DELAY;
                                       ^~~~~~~~~~~~~~
src/main.cpp:52:39: note: suggested alternative: 'portMAX_DELAY'
             bmePollTimer = millis() + BME_SCAN_DELAY;
                                       ^~~~~~~~~~~~~~
                                       portMAX_DELAY
src/main.cpp:76:20: error: 'adcPollTimer' was not declared in this scope
     if (millis() > adcPollTimer){
                    ^~~~~~~~~~~~
src/main.cpp:80:39: error: 'config' was not declared in this scope
             adcPollTimer = millis() + config.ADC_SCAN_DELAY;
                                       ^~~~~~
src/main.cpp:80:39: note: suggested alternative: 'confstr'
             adcPollTimer = millis() + config.ADC_SCAN_DELAY;
                                       ^~~~~~
                                       confstr
src/main.cpp:121:1: error: expected '}' at end of input
 }
 ^
src/main.cpp:46:13: note: to match this '{'
 void loop() {
             ^
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1
```

DeeEmm, 2025-01-12:
```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

long adcPollTimer = 0;
long bmePollTimer = 0;
int BME_SCAN_DELAY = 1000;
int ADC_SCAN_DELAY = 250;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (millis() > bmePollTimer){

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            bmePollTimer = millis() + BME_SCAN_DELAY; 

            float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
            float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
            float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

            Serial.print("Temperature: ");
            Serial.print(temperature);
            Serial.println(" C");

            Serial.print("Humidity: ");
            Serial.print(humidity);
            Serial.println(" %");

            Serial.print("Pressure: ");
            Serial.print(pressure);
            Serial.println(" hPa");

            // Release the semaphore after reading BME280
            xSemaphoreGive(i2cSemaphore);
        }
    }

    // Read from ADS1115
    if (millis() > adcPollTimer){

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            adcPollTimer = millis() + config.ADC_SCAN_DELAY; 

            for (int i = 0; i < 4; i++) {
                // Set the multiplexer for the appropriate channel
                switch (i) {
                    case 0:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                        break;
                    case 1:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                        break;
                    case 2:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                        break;
                    case 3:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);

                // Trigger the conversion
                adc.triggerConversion();

                // Read the conversion result for the selected channel
                int16_t adcValue = adc.getConversion();
                float voltage = adcValue * (6.144 / 32767.0);

                Serial.print("Analog Input ");
                Serial.print(i);
                Serial.print(": ");
                Serial.print(voltage);
                Serial.print(" V");
                Serial.print(": ");
                Serial.print(adcValue);
                Serial.println(" R\n");
          }
            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
        }
    }

    Serial.println(); // Print a newline for better readability

    delay(1);
}
```

KirikauKiwi, 2025-01-12:
> > BME data appears truncated.
> 
> What do you mean by this?

Here is what appears to be a full block of BME data when the 125ms delay is present in my test code.
[image]

Here is what what I referred to as truncated BME data.
[image]

In what I believe to be the full block of data from the BME in the first image there are three sequential write/reads in the data, in the truncated image there is only a single write read for the BME.

KirikauKiwi, 2025-01-12:
> ```
> #include <Wire.h>
> #define TINY_BME280_I2C
> #define BME280_I2C_ADDRESS 0x76
> #include <TinyBME280.h>
> #include <ADS1115_lite.h>
> #include <freertos/FreeRTOS.h>
> #include <freertos/semphr.h>
>
> // Create [...]

further compile errors. I am assuming correctly that this is written to run as a stand alone code block?

```
src/main.cpp: In function 'void loop()':
src/main.cpp:80:39: error: 'config' was not declared in this scope
             adcPollTimer = millis() + config.ADC_SCAN_DELAY;
                                       ^~~~~~
src/main.cpp:80:39: note: suggested alternative: 'confstr'
             adcPollTimer = millis() + config.ADC_SCAN_DELAY;
                                       ^~~~~~
                                       confstr
src/main.cpp:121:1: error: expected '}' at end of input
 }
 ^
src/main.cpp:46:13: note: to match this '{'
 void loop() {
             ^
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1
```

DeeEmm, 2025-01-12:
```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

long adcPollTimer = 0;
long bmePollTimer = 0;
int BME_SCAN_DELAY = 1000;
int ADC_SCAN_DELAY = 250;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (millis() > bmePollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            bmePollTimer = millis() + BME_SCAN_DELAY; 

            float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
            float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
            float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

            Serial.print("Temperature: ");
            Serial.print(temperature);
            Serial.println(" C");

            Serial.print("Humidity: ");
            Serial.print(humidity);
            Serial.println(" %");

            Serial.print("Pressure: ");
            Serial.print(pressure);
            Serial.println(" hPa");

            // Release the semaphore after reading BME280
            xSemaphoreGive(i2cSemaphore);
        }
    }

    // Read from ADS1115
    if (millis() > adcPollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            adcPollTimer = millis() + ADC_SCAN_DELAY; 

            for (int i = 0; i < 4; i++) {
                // Set the multiplexer for the appropriate channel
                switch (i) {
                    case 0:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                        break;
                    case 1:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                        break;
                    case 2:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                        break;
                    case 3:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);

                // Trigger the conversion
                adc.triggerConversion();

                // Read the conversion result for the selected channel
                int16_t adcValue = adc.getConversion();
                float voltage = adcValue * (6.144 / 32767.0);

                Serial.print("Analog Input ");
                Serial.print(i);
                Serial.print(": ");
                Serial.print(voltage);
                Serial.print(" V");
                Serial.print(": ");
                Serial.print(adcValue);
                Serial.println(" R\n");
            }
            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
            
        }

    Serial.println(); // Print a newline for better readability

    }

    delay(1);
}
```

DeeEmm, 2025-01-12:
> In what I believe to be the full block of data from the BME in the first image there are three sequential write/reads in the data, in the truncated image there is only a single write read for the BME before the ADS data appears

I'm not going to try and decipher the image as I can only just about see it. My eyes are not what they used to be.

I would recommend that you decode the data stream to establish if the data is correct and whether any data has been lost. 

The protocol is widely documented. Not too difficult to understand.

DeeEmm, 2025-01-12:
[image]

There is a very good primer on I2C comms online, but I cannot find it at the mo. I'll post it up for you if I come across it

DeeEmm, 2025-01-12:
> using the test code you have provided, I have just tested this and it appears, i.e can only see data for the ADS on the I2C decode, so no address visible for the BME at address 76.

So is this an issue with the test code, or indicative of an issue with the DIYFB code?

KirikauKiwi, 2025-01-12:
Correction to my observations is that there serial monitor is showing correctly the 1:4 ratio of the reads in respect to the timers.

KirikauKiwi, 2025-01-12:
> > using the test code you have provided, I have just tested this and it appears, i.e can only see data for the ADS on the I2C decode, so no address visible for the BME at address 76.
> 
> So is this an issue with the test code, or indicative of an issue with the DIYFB code?

It is the test code only that I am referring to here.

KirikauKiwi, 2025-01-12:
@DeeEmm I have just tested you BME, ADS test code with the 125ms delay, changed from the 1ms you had, with this change only, I can then see the BME address and data on the I2C decode with the three sequential Write/Reads.

It also then appears 4-5 times in the 10 second buffer, rather than on occasion or not at all.

[image]

DeeEmm, 2025-01-12:
Sorry can you clarify what you mean here. I am having trouble understanding..

> using the test code you have provided, I have just tested this and **it** appears, i.e can only see data for the ADS on the I2C decode, so **no address visible for the BME at address 76.** I also checked the serial monitor output for your code and it appears to be only reporting analogue 3 of the ADS and sometimes reporting BME data, see screen shot below.

What is *it*
What does **no address visible for the BME at address 76.** mean?

Also I cannot compile / test the code above as I don't have a project set up for it. It's basically the code that you provided but with corrections for the AI errors and the timers added for each device. I think there was a bracket missing for the Select case statement. There's only so much I can do there so you're kinda on your own for the most part.

Have you tested any of the updates that I have published?

Would appreciate some feedback.

```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

long adcPollTimer = 0;
long bmePollTimer = 0;
int BME_SCAN_DELAY = 1000;
int ADC_SCAN_DELAY = 250;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (millis() > bmePollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            bmePollTimer = millis() + BME_SCAN_DELAY; 

            float temperature = _BME280Sensor.readFixedTempC() / 100.0F;
            float humidity = _BME280Sensor.readFixedHumidity() / 1000.0F;
            float pressure = _BME280Sensor.readFixedPressure() / 100.0F;

            Serial.print("Temperature: ");
            Serial.print(temperature);
            Serial.println(" C");

            Serial.print("Humidity: ");
            Serial.print(humidity);
            Serial.println(" %");

            Serial.print("Pressure: ");
            Serial.print(pressure);
            Serial.println(" hPa");

            // Release the semaphore after reading BME280
            xSemaphoreGive(i2cSemaphore);
        }
    }

    // Read from ADS1115
    if (millis() > adcPollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            adcPollTimer = millis() + ADC_SCAN_DELAY; 

            for (int i = 0; i < 4; i++) {
                // Set the multiplexer for the appropriate channel
                switch (i) {
                    case 0:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                        break;
                    case 1:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                        break;
                    case 2:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                        break;
                    case 3:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);
                }

                // Trigger the conversion
                adc.triggerConversion();

                // Read the conversion result for the selected channel
                int16_t adcValue = adc.getConversion();
                float voltage = adcValue * (6.144 / 32767.0);

                Serial.print("Analog Input ");
                Serial.print(i);
                Serial.print(": ");
                Serial.print(voltage);
                Serial.print(" V");
                Serial.print(": ");
                Serial.print(adcValue);
                Serial.println(" R\n");
            }
            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
            
        }

        Serial.println(); // Print a newline for better readability

    }

    delay(1);
}

KirikauKiwi, 2025-01-12:
So with further testing of you BME/ADS test code the three sequential Write/Reads appear 8 times in the approx 10 second buffer the picoscope stores, so this almost lines up with once every second, so my guess is that the buffer on my picoscope is closer to 9 seconds.

If I set the delay to;
- 10ms, the three sequential Write/Reads appear only 2 times in the 9-10 second buffer.
- 20ms, the three sequential Write/Reads appear only 2 times in the 9-10 second buffer.
- 30ms, the three sequential Write/Reads appear only 2 times in the 9-10 second buffer.
- 60ms, the three sequential Write/Reads appear only 4 times in the 9-10 second buffer.
- 90ms, the three sequential Write/Reads appear only 5 times in the 9-10 second buffer.
- 120ms, the three sequential Write/Reads appear only 8 times in the 9-10 second buffer.

There appears to be a direct correlation to the delay and the number of times the BME data appears in the oscilloscopes buffer 9-10 second buffer with the test code.

Data buffer with delay set to 125ms.
[image]

Data buffer with delay set to 1ms
[image]

The first images shows four occurrences of the data for the BME, there are 8 occurrences in the full buffer, each occurrence is 6 lines in the image, being the line for each Write/Read for the Temp, Baro and Humidity.

The second image shows only one occurrences of the data for the BME, the first six lines from the top, this is the three sequential Write/Reads.

Each of the images is for the approx 9-10 second buffer from the oscilloscope.

KirikauKiwi, 2025-01-12:
> Sorry can you clarify what you mean here. I am having trouble understanding..
>
>
> What is _it_ What does **no address visible for the BME at address 76.** mean?
>
> Also I cannot compile / test the code [...]

I have just tested this code now with the delay set to 125ms

So I can now see all four analogue individual voltages on the serial output however the ratio has changed from 1:4 to 1:2 for the outputs, with this versions go the code when I look at the buffer from the oscilloscope, what I mean is when I cannot see the data for the BME on address 76 is that it is not visible in the data buffer, like it has disappeared totally.

I cannot see any changes in this code that should affect the 1:4 ratio. Could this be due to now have the three additional reads from the ADS1115 now being visible in the serial output, as they would add time to take place????

Serial monitor output.
[image]

KirikauKiwi, 2025-01-12:
Will jump over to Github now and have a look at the updates and test.

DeeEmm, 2025-01-12:
> Could this be due to now have the three additional reads from the ADS1115 now being visible in the serial output, as they would add time to take place????

The 1:4 ratio is time based, so it will be affected by the additional serial print commands. These are notoriously slow. 

It would be better to store the values and write them in one go outside of the two code blocks

DeeEmm, 2025-01-12:
```
#include <Wire.h>
#define TINY_BME280_I2C
#define BME280_I2C_ADDRESS 0x76
#include <TinyBME280.h>
#include <ADS1115_lite.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

// Create instances for BME280 and ADS1115
tiny::BME280 _BME280Sensor; // Initialize without address
ADS1115_lite adc(0x48); // Pass the I2C address directly in the constructor

// Create a semaphore for I2C bus access
SemaphoreHandle_t i2cSemaphore;

long adcPollTimer = 0;
long bmePollTimer = 0;
int BME_SCAN_DELAY = 1000;
int ADC_SCAN_DELAY = 250;
double temp = 0.0;
double baro = 0.0;
double relh = 0.0;
double adc1 = 0.0;
double adc2 = 0.0;
double adc3 = 0.0;
double adc4 = 0.0;

void setup() {
    Serial.begin(115200);
    Wire.begin();
    _BME280Sensor.beginI2C((int)118);

    // Create the semaphore
    i2cSemaphore = xSemaphoreCreateMutex();
    if (i2cSemaphore == NULL) {
        Serial.println("Failed to create semaphore!");
        while (1);
    }

    // Initialize BME280 with address 0x76
    if (!_BME280Sensor.begin()) {
        Serial.println("Could not find a valid BME280 sensor, check wiring!");
        while (1);
    }

    // Set gain and sample rate for ADS1115
    adc.setGain(ADS1115_REG_CONFIG_PGA_6_144V);
    adc.setSampleRate(ADS1115_REG_CONFIG_DR_8SPS);

    Serial.println("Setup complete!");
}

void loop() {
    // Take measurements and print BME280 data
    if (millis() > bmePollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            bmePollTimer = millis() + BME_SCAN_DELAY; 

            temp = _BME280Sensor.readFixedTempC() / 100.0F;
            relh = _BME280Sensor.readFixedHumidity() / 1000.0F;
            baro = _BME280Sensor.readFixedPressure() / 100.0F;

            // Release the semaphore after reading BME280
            xSemaphoreGive(i2cSemaphore);
        }
    }

    // Read from ADS1115
    if (millis() > adcPollTimer) {

        if (xSemaphoreTake(i2cSemaphore, portMAX_DELAY)) {

            adcPollTimer = millis() + ADC_SCAN_DELAY; 

            for (int i = 0; i < 4; i++) {
                // Set the multiplexer for the appropriate channel
                switch (i) {
                    case 0:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_0);
                        adc.triggerConversion();
                        int16_t adcValue = adc.getConversion();
                        adc1 = adcValue * (6.144 / 32767.0);
                        break;
                    case 1:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_1);
                        adc.triggerConversion();
                        int16_t adcValue = adc.getConversion();
                        adc2 = adcValue * (6.144 / 32767.0);
                        break;
                    case 2:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_2);
                        adc.triggerConversion();
                        int16_t adcValue = adc.getConversion();
                        adc3 = adcValue * (6.144 / 32767.0);
                        break;
                    case 3:
                        adc.setMux(ADS1115_REG_CONFIG_MUX_SINGLE_3);
                        adc.triggerConversion();
                        int16_t adcValue = adc.getConversion();
                        adc4 = adcValue * (6.144 / 32767.0);
                }

            }
            // Release the semaphore after reading from ADS1115
            xSemaphoreGive(i2cSemaphore);
            
        }

        Serial.println(); // Print a newline for better readability

    }

    Serial.print("Temperature: %d C\n", temp);
    Serial.print("Humidity: %d %\n", relh);
    Serial.print("Pressure: %d hpa\n", pressure);

    Serial.print("Analog Input 1: %d Volts\n", adc1);
    Serial.print("Analog Input 2: %d Volts\n", adc2);
    Serial.print("Analog Input 3: %d Volts\n", adc3);
    Serial.print("Analog Input 4: %d Volts\n", adc4);

    delay(1);
}
```

KirikauKiwi, 2025-01-12:
> The 1:4 ratio is time based, so it will be affected by the additional serial print commands. These are notoriously slow.
>
> It would be better to store the values and write them in one go outside of [...]

So what would be causing the serial monitor to output only one ADS reading between the BME reads then when there was three previous you correcting the ADS reads????

KirikauKiwi, 2025-01-12:
Test DEV version occf258F.

Observations when viewing the Dashboard GUI.
- ADS data does updates but is slow and stumbles
- Rebooting the ESP32 and refreshing the browser ADS Data is populated when the page loads, the BME Data show 0.00 and BME had not updated even after 3 minutes. The API is slow but works.

[image]

Checking the mimic page I see this, not the change in the voltages, as soon as the voltage change was visible I dropped it to zero and when it hit her put it back up to 3.3V.

https://github.com/user-attachments/assets/efd562e9-d275-4b7b-ac0b-aa4b09290364

DeeEmm, 2025-01-12:
> So what would be causing the serial monitor to output only one ADS reading between the BME reads then when there was three previous you correcting the ADS reads????

Generally you cannot include serial print statements within things like tasks and interrupts as they are a blocking command.

But I just realised that in the change that I just made, it now does not show when the code blocks are run, as there is no way to know which block has been fired. This is why I added the scan counts to the main code. To count each time that the block (task) was run.

Like I mentioned a few days back. I suspect that the issue is that one task / device is 'hungrier' than the other. 

When the task has possession of the semaphore, the other task cannot run until it has finished executing and the semaphore is free. So the slower that the task executes, the longer the other task has to wait.

Then when the semaphore is released, there is a race as to which task gets the semaphore. It is not a flip-flop situation where one releases it and the other automagically gets it.

Both tasks have the ability to take the semaphore. So it's basically a race

So which task gets it??

That's the $60k question

That depends on where in the scan, whether the task timer has expired, and a bunch of other things that we are trying to identify.

In your screen shot above it shows that the ADC task is running 48 times to every one BME task.

DeeEmm, 2025-01-12:
> Test DEV version occf258F.

What build number is that?

KirikauKiwi, 2025-01-12:
> > Test DEV version occf258F.
> 
> What build number is that?

It is the latest one you have posted to your DEV.

[image]

KirikauKiwi, 2025-01-12:
> Like I mentioned a few days back. I suspect that the issue is that one task / device is 'hungrier' than the other.

> When the task has possession of the semaphore, the other task cannot run until it has finished executing and the > semaphore is free. So the slower that the task executes, the longer the other task has to wait.

> Then when the semaphore is released, there is a race as to which task gets the semaphore. It is not a flip-flop situation > where one releases it and the other automagically gets it.

> Both tasks have the ability to take the semaphore. So it's basically a race

> So which task gets it??

> That's the $60k question

Yes I would agree with this more now after further testing today that is is most likely a problem with the semaphore handling and that a task may be hogging it as such!!! or that the BME task cannot grab it for what ever reason.

Researching this a bit further the only things that I found that may be helpful would be to create a Unique Label for each of the tasks that make use of the semaphore in DIYFB, create a global variable "Current Owner" that gets set to the current task once a task has the semaphore, then when the semaphore is released the "Current Owner" is cleared.
This is the most basic way I could find of tracking what task has hold of it, the Current Owner could then be printed to the Serial monitor or displayed on the GUI.

How many tasks in DIYFB can use the semaphore?

DeeEmm, 2025-01-13:
There's only the two tasks in DIY-Flow-Bench.ino along with the API call and SSE send in the main loop.

I have considered moving all of the code into the loop and just processing it sequentially. Getting rid of the tasks.

As we no longer try to assign a task to each core, the necessity of using tasks is not imperative. It is fine to block the other tasks from execution. Additionally we already know ahead of time what tasks need to run and in what order / proportion. Whilst we can currently tune this relationship in the GUI by altering the scan delays, it is not actually necessary.

I've considered setting a simple count to process the BME task once to every four ADC task executions. The SSE code must execute in each cycle, along with the API code. Once we set this sequence we can then control the scan time using a single delay in the loop, just as you have with your test code. This would potentially allow people to 'tune out' similar issues by tweaking the delay. The cost of a 100ms or so delay in the main loop is not too great. Although I will say that using timers to mask what is essentially an unidentified issue is a massive kludge and a big no-no as far as coding is concerned. It is very bad practice.

Reality is that the code execution speed is non-critical. ADC sensor scans can take 1 or 2 seconds and it will not make any difference to the bench operation, the BME can be maybe even be 5-30 seconds or more. The environmental (BME) values do not really change that quickly. So we can afford to slow the operation down to ensure that each sensor gets read correctly. (The original code was written to ensure everything happened as fast as possible. ). If we slow sensor reads down and ensure that the SSE and API tasks run correctly we should be back to a responsive UI / GUI. 

In the background I've been working through a bunch of A/B tests to help optimise the current code. Specifically the MAF code, but nothing that I've tried so far seems to make any drastic improvement,  but this is obviously hard to quantify as I do not have the same issues here as you guys are experiencing. If anything many of the changes actually slowed execution down and I started to see some of the same symptoms. I will continue with these tests as it is still a useful exercise. Maybe I will stumble across a solution.

KirikauKiwi, 2025-01-13:
I am wondering if the delay timers are part of what is causing the problem I am seeing.

Can I ask based on the DIYFB delay timers being 1000ms and 250ms are you expecting this to set the ratio for for the Data flow, e.g. 1 BME data set for every 4 ADS data sets?

As I have been working through why I was only getting the two ADS data sets between each BME data set in the isolation test code for the BME and ADS, what I have found is that the BMS is happening in 2ms as far as the ESP32 is concerned and looking at the I2C decode this is about 600us for each BME data set.
The ADS data set take 677ms as far as the ESP32 is concerned and looking at the I2C decode this is about 1.8ms which is 3x 600us which is 600us for each of Temp, Humidity and Baro.

Once I added the Start/End timers in to the test code this revealed the loop times I believe for the BMS and ADS data sets, I then adjusted the delay timers to get the 1:4 ratio and this is then displayed on the serial monitor as expected.
When the timers are set to 1000ms for the BME and 250ms for the ADS you will only get two ADS reads between each BME read.
So could it may be possible that the scan delay timers are what is blocking the BME from being read as if the ADS loop is holding the semaphore and taking 677ms to execute the and it is being asked to run every 250ms would this almost cause  the loop to keep running possibly??? and not be able to release the semaphore.

Screen shot of the serial monitor for the isolation test code.
[image]

KirikauKiwi, 2025-01-13:
> Although I will say that using timers to mask what is essentially an unidentified issue is a massive kludge and a big no-no as far as coding is concerned. It is very bad practice.

I would agree with you here, from everything I have read in the last few days the DIYFB code you have created is following the best practices regarding core allocation, the use of semaphore mutex etc.

This particular issue is being a right PITA to try and resolve and more frustratingly why is it happening to Stefan and I and not with your setup, which doesn't aid in trouble shooting at all.
I am close to banging my head on the wall 🤯

DeeEmm, 2025-01-13:
> I am close to banging my head on the wall 🤯

I feel your pain 😆 

The 1:4 ratio is not the goal, that is just the ratio of ADC scan delay to BME scan delay. The actual time / ratio is unimportant. it is just arbitrary.

The BME Device does not physically update that fast, so it is pointless polling it at a high frequency as the default update is something like 60 secs.

Have you tried setting the BME to a really slow update time?? Or alternatively the ADC to a much slower time than the BME?

I've been playing with this and there are some interesting results in the GUI behaviour.

I will update the DEV version to what I have here as the config values for the scan delays are not updating in the GUI.

DeeEmm, 2025-01-13:
Updated. Try the current DEV version.

DeeEmm, 2025-01-13:
You will note that I've added the loop time and also changes the task timers to include the task execution time instead of the time between executions. This gives a better view of what's happening as now the task time should consist of the task_delay + actual task execution time.

I need to go sand some floors but will check back later.

KirikauKiwi, 2025-01-13:
I have tried DEV version d5ef2af it doesn't appear to be any different compared to the previous version.

Initially the BME data didn't update and display 0.00 it took more than three minutes for it to populate.
[image]

Recording of Scan Count etc, taken at about 1hr 17 mins of uptime.

https://github.com/user-attachments/assets/21f6325d-4a95-4bb9-8d28-81bad262fdad

KirikauKiwi, 2025-01-13:
This is after 1hr 30mins of uptime.

https://github.com/user-attachments/assets/4266c20e-4cc1-4a77-9fca-27a13924af79

stefan63, 2025-01-13:
Looks about the same for me.

DeeEmm, 2025-01-13:
[image]

[image]

DeeEmm, 2025-01-13:
OK. So think I've found something.

[image]

The scan delay for the ADC task was outside of the loop within the task. I'm hopeful that this is the issue. This would cause the ADC task to be 'greedy' as it would always beat the BME task to the semaphore flag.

Note that my previous scan time of 1ms has now increased to 50ms as the loop task is now being blocked by the ADC task.

Let me know if this makes any difference (along with the tests outlined in the previous post).

DeeEmm, 2025-01-13:
I deffo think that this is it.

The loop delays were also set to 500ms for some reason. This meant that the ADC task was scanning faster then the BME loop delay, practically preventing the BME task from running. So not only was the ADC task greedy as there was no ADC loop delay in the ADC loop, but the ADC scan delay was also shorter than the BME loop delay.

Will upload an updated version with the loop delays set to a more reasonable value.

KirikauKiwi, 2025-01-13:
@DeeEmm would it be a difficult task to change the library that is used for the ADS in the DIYFB code, I am asking because swapping the library in the stand alone code I put together just for the ADS had a significant change with the loop time.

- ADS1115_lite library the loop time for all four reads was 677ms, single read were approx 160ms
- Adafruit_ADS1X15.h Library dropped the loop time down to 49ms, single reads were approx 12ms

Analogue reads with the Adafruit ADS1115 library.

[image]

I feel we are close to exhausting the options that may still be available to use to weed out this issue, so would it be worth trying a different library for the ADS to see if it makes a difference or not???

DeeEmm, 2025-01-13:
Please try the fix and tests above and let me know the results.

KirikauKiwi, 2025-01-13:
> Please try the fix and tests above and let me know the results.

Testing with the DEV version 84d85c0 it appears to be working in more of the expected manor, the stat updates on the GUI are a little blocky or stumble, but yes when first loading the dashboard GUI the BME data populated straight away and both BME and ADS data are updating.

Screen recording of the GUI.

https://github.com/user-attachments/assets/01ba5d78-59fa-4c5c-ade7-4bce329ad2b1

Scan recording

https://github.com/user-attachments/assets/b2ed7d75-4b76-4e99-978f-477a9b2b553b

DeeEmm, 2025-01-13:
Excellent.

It still concerns me why your scan times are so much higher than mine...

[image]

I see the GUI update delay plus the loop delay as the reported scan time within a few microseconds.

The BME loop time is the loop delay plus the loop execution time, 

It's a similar story with the ADC loop. 

There's no additional time in any of these tasks.

Whereas yours has 10x the expected time.

It is very bizarre. I suspect that it is also why your hardware is more susceptible to issues such as this one, whereas my setup isn't.

There must be a reason for this.

DeeEmm, 2025-01-13:
Ironically this error was already in the code and was not changed as part of  ac591a2

~~I am unsure exactly when this change was made~~

The error has been present since Build 22112201 on 22nd November 2022 😮

KirikauKiwi, 2025-01-13:
I have an interesting observation, when viewing the dashboard GUI page I can see data for both address 48 and 76, see the first video below, in the second video the only difference here is that I am viewing the Mimic page and it changes what is happening on the I2C bus, I can then only see data for address 48, when I browse to the Dashboard page I can then see data for both address 48 and 76.

Dashboard page

https://github.com/user-attachments/assets/bf897a56-5c2c-451f-8d09-807ec141020f

Mimic page

https://github.com/user-attachments/assets/caaa0a2c-850f-4460-ac8a-459c80c1a2f7

So viewing a different page is affect the data visible on the I2C bus

KirikauKiwi, 2025-01-13:
Update to above post.

Data visible for both addresses 48 and 76 on the I2C bus, when viewing the following pages.
- Dashboard
- Data
- Setting
- Configuration

Data only visible for address 48 on the I2C bus, when viewing the following pages
- Pins
- Mimic

So something is causing a change when viewing the Pins and Mimic pages that then prevents the BME task running, or that is what it appears to be.

KirikauKiwi, 2025-01-13:
> The dashboard and mimic pages use seperate SSE events. There is a lot more data being transmitted on the mimic SSE data object. My guess it that it may be blocking the task. Did you try the experiments changing the scan delays? If you set the BME scan delay shorter than the ADC, it should overcome this issue as the BME task will then have 'priority'
> 
> The pins page does not use SSE, nor does it have any javascript / XHR, so not sure what is going on there.

Thanks for the update, I am currently running through various different delays for the BME and ADS and trying to make sense of my observations as the data ADS and BME data is very different, in loose terms the ADS like a to spam the I2C buffer i.e the ESP32 is Writing every 600us almost like a machine gun spraying bullets every where, the BME seem more refined in the way it works with three sequential Write and reads for its data. 
I will keep trailing the different delays and try and put something useful together about my observations.

[145 more replies]

closed 2025-01-20 by DeeEmm