DIY-Flow-Bench/DIY-Flow-Bench #322: System page does not load correctly

DeeEmm, 2025-01-18:
> System page still not load fully, this happends frequently, need to refresh multiple times to eventally get it fully populated and to show save button. Sometimes crash during refresh so a reboot is the only way to recover. 
> 
> I previously noted this behaviour with the Hardware Configuration page, but after removing all non-essential items from the page, have not seen this since. I believe that the issue relates to the size of data passed to the templating function in the EspAsyncWebServer library. It seems to fall over once the file size gets too big. We can potentially move the calibration data to a dedicated page. This will reduce the data size for the System Settings page. I will take a look at this.

stefan63, 2025-01-18:
> System page still not load fully, this happends frequently, need to refresh multiple times to eventally get it fully populated and to show save button. Sometimes crash during refresh so a reboot is [...]

That sounds like a possible solution.

Also when you are at it, maybe a reduction of tabs on the main menu would make the UI less cluttered.
Pages like System, hardware, pins, mimic will rarely be used after a working bench has been built an configurated!?
Maybe a gear icon link in upper right corner that bring up a sub menu for settings?

KirikauKiwi, 2025-01-19:
I have managed to cause an issue a few times this evening when browsing the Dashboard GUI, it appears to happening when change from from one page to another, e.g. System to Dashboard.
```
assert failed: heap_caps_free heap_caps.c:381 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40083bcd:0x3ffe8200 0x4008dd75:0x3ffe8220 0x40093585:0x3ffe8240 0x40084136:0x3ffe8370 0x400935b5:0x3ffe8390 0x400def2d:0x3ffe83b0 0x401a31c7:0x3ffe83d0 0x400defcd:0x3ffe83f0 0x400e0428:0x3ffe8410 0x400e06f9:0x3ffe8470 0x400e28da:0x3ffe8490 0x400f07c1:0x3ffe84d0 0x400ee2c9:0x3ffe8520 0x400ee3d9:0x3ffe8570 0x400ee5c1:0x3ffe85c0 0x401870e9:0x3ffe85e0 0x40187165:0x3ffe8610 0x40187816:0x3ffe8630
  #0  0x40083bcd in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x4008dd75 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40093585 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x40084136 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:381
      (inlined by) heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:366
  #4  0x400935b5 in free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #5  0x400def2d in miniz_def_free_func at ESP32/DIY-Flow-Bench/miniz.c:175
  #6  0x401a31c7 in mz_deflateEnd at ESP32/DIY-Flow-Bench/miniz.c:306
  #7  0x400defcd in mz_inflateEnd at ??:?
  #8  0x400e0428 in PublicHTML::decompress(unsigned char const*, unsigned int) at ESP32/DIY-Flow-Bench/publichtml.cpp:53
  #9  0x400e06f9 in PublicHTML::dataJs() at ESP32/DIY-Flow-Bench/publichtml.cpp:204
  #10 0x400e28da in std::_Function_handler<void (AsyncWebServerRequest*), Webserver::begin()::{lambda(AsyncWebServerRequest*)#34}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at ESP32/DIY-Flow-Bench/webserver.cpp:424
      (inlined by) _M_invoke at /Users/philipcottle/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #11 0x400f07c1 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at /Users/philipcottle/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at libdeps/esp32dev/ESPAsyncWebServer-esphome/src/WebHandlerImpl.h:132
  #12 0x400ee2c9 in AsyncWebServerRequest::_parseLine() at libdeps/esp32dev/ESPAsyncWebServer-esphome/src/WebRequest.cpp:581 (discriminator 1)
  #13 0x400ee3d9 in AsyncWebServerRequest::_onData(void*, unsigned int) at libdeps/esp32dev/ESPAsyncWebServer-esphome/src/WebRequest.cpp:123
  #14 0x400ee5c1 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at libdeps/esp32dev/ESPAsyncWebServer-esphome/src/WebRequest.cpp:76
      (inlined by) _M_invoke at /Users/philipcottle/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #15 0x401870e9 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/philipcottle/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:971
  #16 0x40187165 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:1291
  #17 0x40187816 in _async_service_task(void*) at libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:164
      (inlined by) _async_service_task at libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:199

ELF file SHA256: 477a86a1661eb54c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Serial started 
```

KirikauKiwi, 2025-01-19:
Another observation related to this. When I click the "Settings" button the page often only partially loads, it has always reloaded fully if I refresh the page.

What is the difference between clicking on the "Setting" and refreshing the page???

KirikauKiwi, 2025-01-19:
Settings page html. page not fully loaded

[image]

Setting page html. when browser refreshed and page load correctly.

[image]

When the pages loads correctly with the browser refresh this bit of html appears, there is are two additional lines of coed appear after the blu lines in the images above.

DeeEmm, 2025-01-20:
The truncated output is a similar issue to the AsyncWebServer problem. A different mechanism but essentially the output gets truncated. Each page is constructed from three separate files - the header, which includes the navbar and banner. The main content and the footer. The HTML is converted and stored as a bytestring in the htmldata.h file. The bytestring is uncompressed by the decompressMultipleToStream method in publichtml.cpp

The call to the javascript file is the last thing on the content page. So it appears that the content is getting truncated. 

Both this and the AsyncWebServer problem are similar issues, and both may be related. The main difference between the two issues is that when the decompressMultipleToStream truncates the content the footer is included, whereas when AsyncWebServer truncates the data the footer is missing

I did try previously to increase memory to try and fix this issue when I had it ont eh configuration page, but memory did not appear to be the issue. 

I will take another look at this and see if I can figure something out.

KirikauKiwi, 2025-01-20:
Testing DEV Build 2501200030 version https://github.com/KirikauKiwi/DIY-Flow-Bench/commit/4c92ad230ad896c79a4b2288014d55f3fab68850

I am now having a similar issue with the dashboard GUI behaving weirdly, whilst I didn't have this problem on the previous build.

Researching the assert failed error above https://github.com/DeeEmm/DIY-Flow-Bench/issues/322#issuecomment-2601024053 I suspect it is minor differences in how the miniz is compressing the page that maybe causing the issues with the GUI.

DeeEmm, 2025-01-20:
Sorry there might be some tests still active in that commit. I've moved my testing for this issue onto a dedicated branch now.

Just working through a bunch of different tests to see if I can isolate the issue.

DeeEmm, 2025-01-20:
Just updated the DEV branch. Should be good now.

DeeEmm, 2025-01-20:
> Another observation related to this. When I click the "Settings" button the page often only partially loads, it has always reloaded fully if I refresh the page.
> 
> What is the difference between clicking on the "Setting" and refreshing the page???

There should be no difference. The same http request is sent

KirikauKiwi, 2025-01-20:
Dashboard GUI, complete page load.

[image]

Setting GUI, partial page load.

[image]

Setting GUI, full page load.

[image]

I note that in the index.html, there is HTML code in green, it appears after the id for the Dashboard GUI
```
    <div id="tabcontainer">
    <!-- Main Tab-->
    <div id="dashboard" class="tabcontent tabcontentMain">        

```
It is in a different line location in the settings.html file it is before the id.it appears before the id for the Settings GUI
```
    <!-- settings tab -->
    <div id="settings">
      <div class="config-panel">
```

Could this possible cause the issue?

DeeEmm, 2025-01-20:
> I note that in the index.html, there is HTML code in green, it appears after the id for the Dashboard GUI

That's just a comment. Should have no effect.

HTML code is essentially treated like text, it is not executable, so has no real effect on the process. It is only 'executed' (if you can call it that) by the browser.

KirikauKiwi, 2025-01-20:
Search in the compressed index.html for <br><br>, I found two instances of this double <br>, I have changed this so each instance has now only a single <br> and the page loads more reliably, but still not 100% of the time.

I think the miniz puts the <br> in and sometimes adds extras.

DeeEmm, 2025-01-22:
> Researching the assert failed error above [#322 (comment)](https://github.com/DeeEmm/DIY-Flow-Bench/issues/322#issuecomment-2601024053) I suspect it is minor differences in how the miniz is compressing the page that maybe causing the issues with the GUI.

the code is not compressed using the miniz code in the main project. It is compressed using the htmlToBytes.py script in the tools folder. This uses the zlib library. Both miniz and zlib use the same format, so there should be no issues relating to the content itself.

I suspect that the issue is memory related, miniz requires that it knows the zie of the data ahead of time, which is proivided within htmldata.h along side the byte array...

[image]

One thing that I want to check is how miniz is handling the uncompressed file size. I.e. it knows the size of the compressed data thta is being sent to it, but how does it know how much space the uncompressed data takes up???

DeeEmm, 2025-01-22:
So there seems to be a ceiling for the HTML file size of approx 21k

The overall compiled page sizes are as follows:

```
index - 11437 bytes
datagraph - 10501 bytes
settings - 21878 bytes
config - 17788 bytes
pins - 134449 bytes
mimic - 7400 bytes
```

Note that the setting page is actually truncated, so this is hitting a restriction somewhere. The page size is indicative of the size of that limit. If we take into account the header file size (1371) the resultant size (size ceiling) is 20507.

The reason for this limit is not readily apparent in the code. It may possibly be a limitation of Miniz or of AsynchWebServer.

The easy fix is to reduce page sizes. The current commit removes the calibration stuff from the settings page, reducing it to 17450, which loads correctly.

The proper fix however is to determine the root cause of the issue.

closed 2025-01-23 by DeeEmm