DIY-Flow-Bench/DIY-Flow-Bench #324: [BUG]MAF Type not changing when selecting each MAF sensor

KirikauKiwi, 2025-01-21:
Version: V2.0-RC8
Build: 2501200041
GUI: 2412140054

When switching MAF sensors via the Hardware GUI page and returning to the dashboard GUI the MAF Flow Kg/h value does not change.
I suspect the the MAF type is not changing in with this selection and the default MAF sensor settings, data set etc are still being used.

The MAF sensor type changes and saves on the hardware page, when viewed via the info page it is displayed as it has been selected from the hardware page. They match each other.

The datas value do not change on the dashboard GUI or mimic pages when changing between the MAF types.

DeeEmm, 2025-01-23:
At the moment MAF data is only loaded at boot which just follows the original way that we processed the MAF data when it resided in a file.. So whilst we can select a different MAF sensor, the system requires a reboot to load the new data into memory

Reloading the MAF data on-the-fly would not be without some penalty, the nature of which would depend on how it was processed...

At the moment when we save settings / config data, it is written to the NVM in a loop, so we do not look at individual values from the submitted form. To update the MAF data on-the-fly we would therefore need to expand the way we process the form data which increases file size.

The other option is to test to see if the MAF data has changed and reload the data if it has, but this adds an additional overhead to each cycle, which has both a memory and time cost.

We can also just reload the data on each cycle without checking. This is obviously the worst of the three options penalty-wise.

At present I don't see an issue with having to reboot to load the MAF data. Needing to change on-the-fly is a fringe case.

Ive added a javascript alert if the value is changed alerting the user that a reboot is needed.

[image]

closed 2025-01-23 by DeeEmm

stefan63, 2025-01-23:
Good solution!