DIY-Flow-Bench/DIY-Flow-Bench #325: I2C Improvements

DeeEmm, 2025-01-22:
> My loop time was around 1000us and now with the timings on the I2C line working correctly, my loop scan time has increase to 51000us, I am guessing this is because it now has time to get the semaphore and run its tasks correctly.

Yes I see the 51000us timing with the above values but it was around 1000us with the previous values you gave. 

I'm assuming that the 51000us timing is not optimal as the loop is being blocked.

If the ADC task execution time is 71ms and the BME task execution time is 3ms, then the SSE task in the loop should still have enough time to execute its 129ms task without being blocked.

> It is a tricky process of refinement with the timers and has taken a while to get where we are currently. Let me know if you find other delays that works and I am happy to test them and verify them on the scope too.

The frequency of the events should not be an issue, there is still some quirk that we are missing. I think that all we are doing with the poll timers is increasing the delay between tasks so that the task tries to grab the semaphore when it is not competing with another task. This brings us back to trying to manage shortcomings in the code using delays and timing, which is inherently unstable, specifically as my hardware is different from yours, which is different to the next persons, which will all have different inherent timings. We cannot expect the end user to have to fine tune poll timers to get a stable system.

I think that a better way to deal with this instability is to prioritise each task so that they do not compete. A bit like a 3 gate flip flop. This way when we release the semaphore, only the task which is supposed to receive it can take the semaphore. 

So we run tasks in a sequence...

```
ADC > SSE > ADC > SSE > BME > SSE > ADC > SSE > ADC > SSE > BME > SSE
```

This way the ADC task runs twice as much as the BME task and the SSE updates every time there is an update to either task.

We can then control the over scan timing using only the SSE task.

When the ADC task runs it runs for a scan (or maybe for a set number of scans to overcome the potential for an initial ghost value) and then completes. This will happen after lets say 142ms, it then releases access back to the SSE task which still has a further 358ms to want until it runs.

SSE task then runs and then alternately releases access back to the ADC / BME tasks.

When the BME task runs it runs once for 129ms before releasing back to the SSE task, which waits a further 371ms

In each case there is no clash between tasks. The SSE task is the main task, its frequency determines the frequency of the other tasks. The only caveat is that the frequency cannot be less than the duration of the longest task (BME task).

If any task takes longer, the next task simply waits until it gets access, however long that is.

> 
> I will have another attempt at refining them to see if I can find the magic number to reduce the Main loop scan time down to its previous 1000us.

I think that this will be an almost impossible task to make work for all scenarios / installations. Current DEV version has your previous values in it. I will revert back to the timing values in this PR as the 51000us scan time at least provides a metric by which we can measure any changes.

I will write some basic logic to implement the sequence above and we can then both test.

As an aside I did end up pulling the trigger on the Analog Discovery 3 DRO, so I am now able to take a deeper dive into the I2C timing. I'm interested to take a look to see if I can now see the same issues that you are seeing here on my setup.

[image]

I will open a seperate issue for I2C improvements / stability and we can chip away at this in parallel to the other issues. It is even possible that this may improve the issues relating to page loading, after all they are directly tied to the SSE task.

We can run with the last values from this PR in the main code at is is at least more stable for others who ay share the same issue you guys are having, but I think that there is still more work to do here

_Originally posted by @DeeEmm in https://github.com/DeeEmm/DIY-Flow-Bench/issues/323#issuecomment-2606118097_

DeeEmm, 2025-01-22:
I'm not going to assign this to a specific milestone / target. If the general consensus is that the poll timer values from PR #323 provide a stable GUI, then we will use these until improvements can be made.

Other development for RC9 / V2 / V3 still needs to move forwards

DeeEmm, 2025-01-22:
OK. Task sequencing added. I also commented out the semaphore transactions as we have essentially replicated this with the code changes.

The GUI seems to be a lot more stable, however that's just a visual impression, I'm not seeing as much variation / fluctuation in the ADC values.

You will note that here is some variation in the ADC task frequency reported on the mimic page. This is because every other cycle that the ADC task fires it also includes the BME task. This needs to be fixed to report correctly but for the purposes of this test are unimportant.

BME values and task timing is rock solid.

Let me know how this appears to you.

I *think* I may have also discovered the cause of the issue that you have been seeing but requires testing to see if I'm correct. We can look at that afterwards.

KirikauKiwi, 2025-01-22:
I will test these later tomorrow.

Check the pull request for the current DEV version and also for the ADS1X15 library, I think after another iteration I have the conversion delay minimised and thus new Poll timers were required.

The performance tile now appears like this.

[image]

DeeEmm, 2025-01-22:
> I will test these later tomorrow.

Only the most recent commit [5f9a9e1](https://github.com/DeeEmm/DIY-Flow-Bench/commit/5f9a9e1d4b4459a3d9c473d3700a13b014f13705) needs testing on this branch.

There are also changes on the branch for issue #322 that need testing as well.

KirikauKiwi, 2025-01-23:
> > I will test these later tomorrow.
> 
> Only the most recent commit [5f9a9e1](https://github.com/DeeEmm/DIY-Flow-Bench/commit/5f9a9e1d4b4459a3d9c473d3700a13b014f13705) needs testing on this branch.
> 
> There are also changes on the branch for issue [#322](https://github.com/DeeEmm/DIY-Flow-Bench/issues/322) that need testing as well.
> 
> Also, Point to note: Poll timers are redundant with the changes on this branch as the poll timers have been removed and replaced with task sequencing

 Tested 6cfe0d8
The only thing I really noticed was that the ADC data didn't change a smoothly on the Dashboard GUI, when varying the voltage of pRef, it had a small amount of lag compared to DEV.

> I think I may have also discovered the cause of the issue that you have been seeing but requires testing to see if I'm correct. We can look at that afterwards.

Curious to test this at some stage.

Will test #322 tomorrow.

DeeEmm, 2025-01-23:
> The only thing I really noticed was that the ADC data didn't change a smoothly on the Dashboard GUI, when varying the voltage of pRef, it had a small amount of lag compared to DEV.

As the SSE event is being run in sequence the GUI will refresh at the specified rate, rather than when it is able

DeeEmm, 2025-01-23:
I think that the update frequency is also currently less than before

DeeEmm, 2025-01-23:
Branch https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements-test2 has an alternative fix.

This does not sequence events and so is still vulnerable to timing issues as per previous versions. I am specifically interested if this change addresses the i2c line spamming that you previously reported.

Branch https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements contains sequenced code.

KirikauKiwi, 2025-01-23:
> There are also changes on the branch for issue https://github.com/DeeEmm/DIY-Flow-Bench/issues/322 that need testing as well.

Took me a while to realise it was in the DEV branch.

This has tested well, with moving some items from the systems page to the new calibration page, I haven't had the partial page load issue.
The only thing to note is sometimes when navigating back the the Dashboard GUI page the ADS1115 Data in the three top tile does not load, the tiles are either blank or 0.00, I can see the data correctly on the Mimic page.

KirikauKiwi, 2025-01-23:
> Branch https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements-test2 has an alternative fix.
>
> This does not sequence events and so is still vulnerable to timing issues as per previous [...]

Will take a look at these soon.

DeeEmm, 2025-01-23:
> Took me a while to realise it was in the DEV branch.

I validated the changes and merged into DEV. The old branch was deleted in the merge.

I've reverted the changes to the ADC library. Can you test with the current version.

KirikauKiwi, 2025-01-23:
Take a look at the I2C bus on e7b8ad5, can you let me know what you see with that version. the data from the ADS1115 is almost constant for me I made no changes in my hardware only swapped the code and I get this.

[image]

KirikauKiwi, 2025-01-23:
Here is the DEV version 666772f7, which is a step backwards for me. It appears OK is on the GUI at an initial look and the performance tile, however when looking at the I2C bus there there are repeated attempts to read the data for the ADS1115.

Packet 1 is the register being set, then there should be the delay period to allow the conversion to take place then Packet 2 should appear which is the write requesting the data followed by the read which is the data being sent.
There is no delay between them here and the Write/Read for the data is happening immediately after Packet 1.

[image]

[image]

KirikauKiwi, 2025-01-23:
This code in the ADS1115_lite library DOES NOT!!!! implement the timing delay the isBusy() relies on the Alert/Ready pin being connected to the ESP32
```
int16_t ADS1X15::_readADC(uint16_t readmode)
{
  //  note readmode includes the channel
  _requestADC(readmode);

  if (_mode == ADS1X15_MODE_SINGLE)
  {
    uint32_t start = millis();
    //  timeout == { 138, 74, 42, 26, 18, 14, 12, 11 }
    //  added 10 ms more than maximum conversion time from datasheet.
    //  to prevent premature timeout in RTOS context.
    //  See #82
    uint8_t timeOut = (128 >> (_datarate >> 5)) + 10;
    while (isBusy())
    {
      if ( (millis() - start) > timeOut)
      {
        _error = ADS1X15_ERROR_TIMEOUT;
        return ADS1X15_ERROR_TIMEOUT;
      }
      yield();   //  wait for conversion; yield for ESP.
    }
  }
  else
  {
    //  needed in continuous mode too, otherwise one get an old value.
    delay(_conversionDelay);
  }
  return getValue();
}
```

DeeEmm, 2025-01-23:
```
while (isBusy())
```

loops until isBusy is true

```
bool ADS1X15::isBusy()
{
  return isReady() == false;
}
```

Checks to see if conversion is ready

```
bool ADS1X15::isReady()
{
  uint16_t val = _readRegister(_address, ADS1X15_REG_CONFIG);
  return ((val & ADS1X15_OS_NOT_BUSY) > 0);
}
```

Reads the register and performs bitwise 'AND' to read single byte.

Checks if this has value '1'

The current library also does same thing. Albeit slightly differently.

KirikauKiwi, 2025-01-23:
> ```
> while (isBusy())
> ```
>
> loops until isBusy is true
>
> ```
> bool ADS1X15::isBusy()
> {
>   return isReady() == false;
> }
> ```
>
> Checks to see if conversion is ready
>
> ```
> bool ADS1X15::isReady()
> {
>   uint16_t [...]

Where does the 1 come from???? it is a digital high!!! have you got your ESP32 connect to the Alert/Ready pin on your ADS1115??? Maybe that is the difference between yours and @Stefan63 and my hardware.

Here is the isBusy
``
bool ADS1X15::isBusy()
{
  return isReady() == false;
}

bool ADS1X15::isReady()
{
  uint16_t val = _readRegister(_address, ADS1X15_REG_CONFIG);
  return ((val & ADS1X15_OS_NOT_BUSY) > 0);
``
it the does _readRegister
``
uint16_t ADS1X15::_readRegister(uint8_t address, uint8_t reg)
{
  _wire->beginTransmission(address);
  _wire->write(reg);
  int rv = _wire->endTransmission();
  if (rv == 0)
  {
    rv = _wire->requestFrom((int) address, (int) 2);
    if (rv == 2)
    {
      uint16_t value = _wire->read() << 8;
      value += _wire->read();
      return value;
    }
  }
  _error =  ADS1X15_ERROR_I2C;
  return 0x0000;
}
``

This cannot be working unless you are wired to the ADS's Alert/Ready pin from the ESP32. 

Can you see the I2C Write/Reads for this????? as this code should be sending a request to the device so a Write requesting the status and the you should see a Read when the data is sent.

DeeEmm, 2025-01-23:
My board is wired as per the schematic. Which should be the same as yours if you followed the schematic correctly

I think that you may need to dissect the code to figure out how it works. Your assumptions are incorrect.

KirikauKiwi, 2025-01-23:
Try the same code with the while isBusy commented out and see what happens on the I2C bus.

int16_t ADS1X15::_readADC(uint16_t readmode)
{
  //  note readmode includes the channel
  _requestADC(readmode);

  if (_mode == ADS1X15_MODE_SINGLE)
  {
    uint32_t start = millis();
    //  timeout == { 138, 74, 42, 26, 18, 14, 12, 11 }
    //  added 10 ms more than maximum conversion time from datasheet.
    //  to prevent premature timeout in RTOS context.
    //  See #82
    uint8_t timeOut = (128 >> (_datarate >> 5)) + 10;
    //while (isBusy())
    {
      if ( (millis() - start) > timeOut)
      {
        _error = ADS1X15_ERROR_TIMEOUT;
        return ADS1X15_ERROR_TIMEOUT;
      }
      yield();   //  wait for conversion; yield for ESP.
    }
  }
  else
  {
    //  needed in continuous mode too, otherwise one get an old value.
    delay(_conversionDelay);
  }
  return getValue();
}

DeeEmm, 2025-01-23:
However. Instead of banging your head against the computer trying to convince me.

Can you validate the tests that I proposed. 👍

DeeEmm, 2025-01-23:
I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements 

With current library https://github.com/DeeEmm/ADS1115_lite

KirikauKiwi, 2025-01-23:
> I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements

This I believe I have respond to already and with further testing the it is a bit clunky due to the sequencing of the data from my oberservations on the GUI, oit appeared to stop start with the data updates. This was then visible on the I2C bus too. 
 
> With current library https://github.com/DeeEmm/ADS1115_lite

Will look at this later.

I am testing the code visually to start with as a ender user would!!! I am not looking at it macroscopically and delving in to the code of using the scope, once I have looked at the visually observable I then take a look at the scope.

From the scope traces I am would then be able to say what has most like changed in the code.

> However. Instead of banging your head against the computer trying to convince me.

I am not banging my head, I am reporting my observations and what I know of the situation and what has worked to resolve the issue I was having.

>I think I may have also discovered the cause of the issue that you have been seeing but requires testing to see if I'm correct. >We can look at that afterwards.

Or is it under wraps until I have done the other validation testing, to keep the testing blind and impartial.

It would also be help if you could answer the questions I put forward too.
I am still learning as far as C++ goes and from what I read of the code it won't work unless it is connected to the Alert/Ready pin, which I know from other comments you have made is not how the DIYFB build is setup, it is not how I have my hardware configured???

Or is this part of trying to establish why your hardware appeared to work and mine and @Stefan63 didn't.

Have you inspected your I2C bus on version e7b8ad5, I am curious to know what you see???

KirikauKiwi, 2025-01-23:
> I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements
> 
> With current library https://github.com/DeeEmm/ADS1115_lite

The GUI appears to have lag/latency, when you change a voltage across the top three tiles, the bottom three tiles look slow to update, it is inline with what the performance tile says of around 2 sec.

Checking the I2C bus I get the same with the ADS1115 having repeated Write requests and data Reads that repeat, I don't see the register being set very often at all.

[image]

[image]

KirikauKiwi, 2025-01-23:
> I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements
> 
> With current library https://github.com/DeeEmm/ADS1115_lite

Does this work for you?

What do you see on your I2C bus????

KirikauKiwi, 2025-01-23:
This is the current DEV version 666772f7, the one single change I have made here is to comment out the `while isBusy()`everything else is unchanged.

[image]

What is happening here is packet one is a register set, then immediately after it packet two is the write requesting the data, packet 3 is the read sending the data, packet 4 is a new register set 5 and 6 are the write/read of the data.
This then results in the top image where all input voltages are from input one.

[image]

KirikauKiwi, 2025-01-23:
This is the current DEV version https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda, the one additional change to the above, is that I have changed the conversion delay, which is set as it was in the define as 8ms.

This is still reading input one.
`
    //uint8_t timeOut = (128 >> (_datarate >> 5)) + 10;
    uint8_t timeOut = (_conversionDelay);
    //while (isBusy())
`

[image]

[image]

KirikauKiwi, 2025-01-23:
This is the current DEV version https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda, the one additional change to the above, is that I have changed the code below from > to <

```
 if ( (millis() - start) < timeOut)
```

[image]

[image]

DeeEmm, 2025-01-23:
> Or is it under wraps until I have done the other validation testing, to keep the testing blind and impartial.
> 

No, this is the code in the 325-i2c-improvements-test2 that I asked you to test previously two days ago.

> It would also be help if you could answer the questions I put forward too. I am still learning as far as C++ goes and from what I read of the code it won't work unless it is connected to the [...]

I do answer all of your questions and spend a lot of time explaining things to you so that you understand how things work. Please understand that this makes everything that I do take 3x as long as it needs to. Whilst I am happy to do this, please consider that this is not a free online C++ / javascript / HTML course.

> 
> Or is this part of trying to establish why your hardware appeared to work and mine and [@stefan63](https://github.com/stefan63) didn't.
> 

The project is designed to be used with the official PCB. It should be possible to breadboard this or make your own PCB by following the schematics as both you and Stefan have done. Given that you both have an issue that does not appear on my system that does not appear to be software related, means that the only logical conclusion is that it is a hardware issue, or related to differences in hardware.

I am suspicious that you may have both replicated a common error, which may be as a result of poor or misunderstood documentation. Unfortunately the dataset is too small to be able to draw any kind of logical conclusion. I also do not have access to your systems so cannot draw any data from there to try and determine where such an error may be.

despite asking several times about the I2C pull-ups you always appear to skirt around the issue.

So this process falls back to one of a slow process of change / test iterations to try and find a solution that works for ALL systems. Currently your solution does not work for me. And given that it is delay based, is unacceptable for all of the reasons that I have already given.

> Have you inspected your I2C bus on version [e7b8ad5](https://github.com/DeeEmm/DIY-Flow-Bench/commit/e7b8ad522fe65e83b93a83d45a74bf26382b3d23), I am curious to know what you see???

I see the previous issue that I fixed relating to the loop delay outside of the loop. I also see the master re-sending data request because it fails to read. This is not as a result of other / incorrect / too much data on the bus as you have assumed, it appears to be an issue with the library. (which we no longer use)

We identified these issues a week or so ago and I put forwards tests to determine the best way to resolve them. However instead of undertaking the tests that I asked, you continued down the rabbit hole playing with delays. You could have spent that time decoding the trace so that we had an actual idea of what was going on, rather than manipulating the symptoms

Please consider that what I propose a change and ask for you to test it the change is considered and part of a larger strategy to fix the issue. It is just basic scientific process. One change at a time, test and validate. Rinse and repeat.

Please just forget delays. They are not a valid solution for anything. If you want to learn. Go and learn why they are not acceptable. I have already given you several explanations, but you elect to ignore them.

> This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one additional change to the above, is that I have changed the code below from > to <
> 
> ```
>  if ( (millis() - start) < timeOut)
> ```

What code where? and why? Please reference your changes with some context. 

My time is very sparse. I have many concurrent projects like this one, I'm also in the middle of a house build as well as running my own company. Every second that I don't have to spend deciphering things is a plus. If this change is important or addresses an issue or improves performance please explain how and why.

> > I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements
> 
> This I believe I have respond to already and with further testing the it is a bit clunky due to the sequencing of the data from my oberservations on the GUI, oit appeared to stop start with the data updates. 

Please define 'clunky'. I am unsure what it means in this context. "clunky: _solid, heavy, and old-fashioned_" ?? It appears to have no relationship to what we are doing here. 

What I see are updates on the bus in line with the sequencing and timing defined in the code. The timing does not vary, the data sequence does not vary. it is very stable and should address the I2C 'spamming' issue that you seem to still have 

I see transaction times that are faster than those in the solution that you put forwards as the code does not wait for a fixed delay but returns after the update register has been updated.

> This was then visible on the I2C bus too.

What was? 

What I need is an objective view of how your system performs and whether this is acceptable in the context of GUI experience. These need to be in context with the test.

Please don't post with no explanation. This applies to screenshots too. What exactly am I looking at?? Is this good or bad? Describe what it is that you are showing me. I don't have the time to try and decypher what it is that you are tying to tell me. I am very time poor.

DeeEmm, 2025-01-24:
[image]

DeeEmm, 2025-01-24:
> > I am interested in https://github.com/DeeEmm/DIY-Flow-Bench/tree/325-i2c-improvements
> > With current library https://github.com/DeeEmm/ADS1115_lite
> 
> Does this work for you?
> 
> What do you see on your I2C bus????

Pretty straightforwards request. I made it specifically unambiguous and linked to the items that I wanted you to test

however you ignore this and modify the code...

> This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one single change I have made here is to comment out the `while isBusy()`everything else is unchanged.

Then you modify it again...

> This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one additional change to the above, is that I have changed the conversion delay, which is set as it was in the define as 8ms.

Then you modify it again...

> This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one additional change to the above, is that I have changed the code below from > to <

I cannot make any progress on the issue that you claim is still present if you are not prepared to help me.

Currently I have proposed two solutions

- One in dev branch which retains xTasks
- One in the branch above (325-i2c-improvements) which changes code operation to sequential

I am not interested in your modifications.

I am only interested in the code above. as-is. With no changes.

Remember - basic scientific process - one change at a time. Test and validate. Rinse and repeat.

I have no idea what you are changing, or whether you are undertaking tests as requested. This has eroded my confidence in any response that you provide.

This is further compounded by what come across as emotional responses that appear to be driven by me not accepting your solution.

KirikauKiwi, 2025-01-24:
> Pretty straightforwards request. I made it specifically unambiguous and linked to the items that I wanted you to test
Which I tested and stated my observations  above.

> however you ignore this and modify the code...
I haven't ignored this at all see above I started to modify the code one change at a time to show you what was working and what was not working with the reversion you made.

> > This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one single change I have made here is to comment out the `while isBusy()`everything else is unchanged.
> 
> Then you modify it again...
Yes I did this, again to show that one change didn't solve the issue for me, and another change didn't solve the issue for me

> > This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one additional change to the above, is that I have changed the conversion delay, which is set as it was in the define as 8ms.
> 
> Then you modify it again...
Yes again I made a single change now three in total and I am still having the issue!!

> > This is the current DEV version [666772f](https://github.com/DeeEmm/DIY-Flow-Bench/commit/666772f7c25bf8f1ce8ae1517b849eefffbe7fda), the one additional change to the above, is that I have changed the code below from > to <

> I cannot make any progress on the issue that you claim is still present if you are not prepared to help me.
I have given you exactly what has resolved the issue for me!!!

> Currently I have proposed two solutions
> 
> * One in dev branch which retains xTasks
> * One in the branch above (325-i2c-improvements) which changes code operation to sequential
> 
> I am not interested in your modifications.
> 
> I am only interested in the code above. as-is. With no changes.
I have tested them both and reported that I have the issue present in both of them!!!!!!

> Remember - basic scientific process - one change at a time. Test and validate. Rinse and repeat.
This is exactly what I did above funnily enough, one single change at a time, in a systematic approach

 
> I have no idea what you are changing, or whether you are undertaking tests as requested. This has eroded my confidence in any response that you provide.
Likewise, I have multiple told you multiple time now that I have tested what you have provided and asked and that the issue still exists for me!!!! I do not understand what you have missed.

> This is further compounded by what come across as emotional responses that appear to be driven by me not accepting your solution.
LIkewise, I have gone in a complete circle today, going backwards from something that was working well for me to something not working well.
I had provided you with the code that works for me, you haven't bothered to tell me if it works for you or improves anything with your setup.
Not an emotional response at all, you have already stated I have wasted your time previously, I asked you to provide some information on this as I don't think I have.
This seems to be going down the same path for me, I have asked today what is the purpose of the tests you have been providing for me to try, with no answers.
I am happy to blind test if that is what you want me to do, but when it appears that you are ignoring what I state as my observations of the tests and that the issue is still present in both of the tests you asked about.

You also haven't yet divulged what you think was the cause of the issue or responded to my question regarding it.

You have gone from expressing frustrations about me, to thanking me for helping resolve the issue to now being frustrated again. I don't really understand and personally not to bothered by it.

I am only trying to help and given what you have expressed previously, about me wasting your time, I am conscious of this and so I am a little surprised to be reading this again when the steps you have taken today appear to be in the reverse of the direction we both want the project to head in.

I get that reverting change is sometimes a good thing and also a valid way of testing, however when it appears that my observations and questions are being ignored doesn't really help either of us.

The one change and one step at a time that I showed above was because it was apparent to me that something was missing in your understanding of it.

So to hopefully avoid this situation for a third time, can you clearly state the objectives that you are trying to achieve with each of the test you want someone to undertake, otherwise it feels like I am trying to throw a dart at a dartboard after I have been blind folded and spun around in circles before throwing it.

DeeEmm, 2025-01-24:
I'm done.

KirikauKiwi, 2025-01-24:
Putting this here for future reference.

The code below is a slight variant of what was working for me previously.
It is based on the one change at a time approach, working back through this as I did, when I first found a solution to the issue that I was having and appeared that @Stefan63 has also. 
- while (isBusy()) this appears to me to be needing to use the Alert/Ready pin, awaiting confirmation 
- uint8_t timeOut = (128 >> (_datarate >> 5)) + 10; I had issues with theses timing.
- yield();   //  wait for conversion; yield for ESP., when outside of the loop issues were still present.
- < changed from >, issue were present if it was greater than
- item above, I had to remove the error code from the loops as it would be true since changing to less than causing the error code to show on the mimic page whilst the data was correct on the I2C bus,  the error codes prevented the correct data from coming in to the ESP32.
- uint8_t timeOut = (10); this was used as the simplest implementation for the required conversion delay of the ADS1115.

This code also resulted in a clean and stable ADS data output for me during testing.
```
int16_t ADS1X15::_readADC(uint16_t readmode)
{
  //  note readmode includes the channel
  _requestADC(readmode);

  if (_mode == ADS1X15_MODE_SINGLE)
  {
    uint32_t start = millis();
    //  timeout == { 138, 74, 42, 26, 18, 14, 12, 11 }
    //  added 10 ms more than maximum conversion time from datasheet.
    //  to prevent premature timeout in RTOS context.
    //  See #82
    //uint8_t timeOut = (128 >> (_datarate >> 5)) + 10;
    uint8_t timeOut = (10);
    //while (isBusy())
    {
      while ( (millis() - start) < timeOut)
      {
        yield();   //  wait for conversion; yield for ESP.
      }
    }
  }
  else
  {
    //  needed in continuous mode too, otherwise one get an old value.
    delay(_conversionDelay);
  }
  return getValue();
}
```
NOTES;
My understanding of the ADS1115 data sequence.

1. ESP32 sends a Write to the ADS to set the register in single shot mode. (Data capture and conversion.)
2. The conversion in the ADS take time, a delay is required here.
3. The ESP32 has to manually wait for the conversion since the Alert/Ready pin is not used with the DIYFB setup
4. Once the delay period has been exceeded the ESP32 can send a Write that requests the data in the register to the ADS.
5. The ADS then sends a Read which contains the data to the ESP32.
6. The ESP32 then sends a Write to set the next register.

This continues in the loop to get the data from the four inputs of the ADS as below.
Start of loop
1. Write to set a register e.g. A0, From ESP32
2. Delay
3. Write to request data from ESP32
4, Read sending data from ADS1115
5. Write to set register the next register e.g. A1, From ESP32
6. Delay.
7. Write to request data from ESP32
8, Read sending data from ADS1115
9. Write to set register the next register e.g. A2, From ESP32
10. Delay
11. Write to request data from ESP32
12, Read sending data from ADS1115
13. Write to set register the next register e.g. A3, From ESP32
14. Delay
15. Write to request data from ESP32
16. Read sending data from ADS1115
End of loop

My observations if the timing is not correct for the ADS delay this results in either the Input voltage for a single input appearing on all all four data outputs, another observation is that the voltage input for one input appears to cycle across the data outputs often alternate in pairs, alternating from one to the other.

DeeEmm, 2025-01-24:
Awesome. Still labouring your POV.

Your understanding of how the code works is incorrect, as is your interpretation of the I2C issue. 

After finally being able to take a look at this myself, I can see that the issue is a failed read and subsequent retries, which appears to be related to the old library but may also be related to incorrect I2C pull ups or both (failed STOP as bus cannot be pulled low due to incorrect pull up value). Given that I do not see the same issue here the latter is entirely plausible, if not actually fairly likely. However I cannot make any kind of diagnosis there as you keep avoiding my requests relating to this. I stated this right back at the beginning and have followed up on this several times. But you have evaded my enquiries.

I changed the library. This fixed the sluggishness issue. That should have been the end of the matter as everything else is just cosmetic. I.e GUI update frequency etc.

However I have just spent the best part of a week trying to fix an issue that is no longer present. The sluggish GUI was fixed with the change to the library, but we have since been basically splitting hairs with what is the best (purely subjective) solution based on how the I2C bus and GUI cosmetically appears to you. 

As I have mentioned numerous times, your changes are only dealing with the symptoms of the issue as you see them and not the root cause. They also do not work for me here. Tuning delays to try and sequence the code is basically playing whack-a-mole. It is the worst type of coding that anyone can write. But don't take my word for it, there is an entire internet out there which will tell you the same thing. 

Despite trying to coach you away from this you still continue to labour this solution. I get that you are emotionally attached to it. but it just doesn't fly. I am unsure what metrics you are using to measure the performance of these changes, but feelings and clunkiness are not valid. From where I'm sitting it just looks like you are only interested in pushing your solution to the exclusion of everything else. What's more I get an emotional response every time I try to discount it.

I have tried to explain to you how the code works, but you elect to ignore my explanations, and then claim that I am being intentionally evasive. I cannot help you to understand if you are not willing to learn. This is something that you need to go away and do on your own time, not mine. I have neither the time nor the wherewithal to keep explaining it in a variety of different ways until you can get your head around it. 

There is only one person being evasive here. Please go back to the beginning of the other thread and read ALL of the posts, specifically mine. And then this thread. Please take note where I have explained something to you or asked you to test something and you have completely ignored the matter. 

Please take the time to consider how many times you have derailed the project - trying to get non-standard + non-supported hardware working (Core2 M5). I spent a LOT of time integrating that into the code for you only for you to not even use it. That is time I could have better spent on other things. Same thing with BME680 integration. It is unsupported hardware. Yet you still push to get it included despite the supported hardware being well documented. Now it is the same thing again disappearing down the rabbit hole for a week because you are insistent on pushing your code changes. This is despite me offering to send you an official shield free of charge (Ive asked you for your address three times now). 

So. If you think I come across as frustrated, you are damn right. I have tried very hard to accommodate you. But I'm now over it.

If you are not happy with the solutions that I am putting forwards, there is nothing stopping you from developing your own. You have the option to fork the project and make it what you want. You can do whatever you want to do there, in whatever way you want and can have as many delays in it as you please with whatever hardware you want to use. I've already done the hard work for you. Go your hardest. Use AI for the rest.

Or, you can continue contributing here, but please understand I am not interested in I2C screen shots or AI generated code, or delays in code or and endless stream of code changes posted to to the forums or in fact any mention of I2C again. I am also not interested in holding your hand through your development journey. 

Please. If you find a problem with the project, raise a ticket. I will endeavor to fix it

If you have a **solution** to an issue, please raise a PR. But understand this. I will only accept it if it follows best practice and is 100% complete. Don't be offended if it gets knocked back. 

Understand that there's nothing personal going on here. I'm not interested in baby sitting or being a social worker. There is work to be done and all of this BS is simply too distracting. Take it somewhere else or accept that this is my project and I call the shots.

KirikauKiwi, 2025-01-24:
> Awesome. Still labouring your POV.

I was consolidating my observations in to a single post.

> Your understanding of how the code works is incorrect, as is your interpretation of the I2C issue.

Thank you for clarifying this for me as it was one of the questions I have asked, as your knowledges is greater than mine with C++, I don't concur with you regarding the I2C bus, more below on this.

> After finally being able to take a look at this myself, I can see that the issue is a failed read and subsequent retries, which appears to be related to the old library but may also be related to [...]

"the issue is a failed read and subsequent retries" this aligns with what I have been reporting. The pullup resistors I have tried both ways, initially as per the DIYFB drawings, and also without the additional pullups (which you recommend) using only the 10K pull-ups on the ADS's BOB. either way did not change the observations nor the issue I was having.

I suspect that the subsequent retries are happening due to the delay advised by the manufacture is not being adhered too. This has been my observation and proven in my testing via a code change.
The delay here I am referring to the delay only in the ADS1115 library and is no way related to the poll timer or task delays. 
Wether you choose to check that is up to you to test and verify, I hope that it doesn't cause issue further down the road  if additional temp, pressure sensors, encoders etc are need to be added to the I2C bus, I was think about the reverse flow here and the Swirl meter.

> I changed the library. This fixed the sluggishness issue. That should have been the end of the matter as everything else is just cosmetic. I.e GUI update frequency etc.

Yes this did change the sluggishness. I still had an issue, which was reported and with further investigation and the help of one of your test requests having continuous mode active made a noticeable difference to the issue I was having. It was the continuous mode that led me to the delay as it was implemented differently in continuous mode than it was in single shot mode in the the library you changed too.

> However I have just spent the best part of a week trying to fix an issue that is no longer present. The sluggish GUI was fixed with the change to the library, but we have since been basically [...]

I wouldn't agree with this as I was still seeing an issue with my setup, it was a great improvement overall taking DIYFB back to a usable state for me.

> As I have mentioned numerous times, your changes are only dealing with the symptoms of the issue as you see them and not the root cause. They also do not work for me here. Tuning delays to try and [...]

I agree with you on this, but from where my issue started and trying to reverse the problem from an observation and work out if I had faulty hardware or it was a code problem which both were unknown at the time, lead to this as there is more than one part of the code interacting with the ADS, it was an iterative process that has taken a long time to resolve due to my systematic approach to make a single change, observe the change and then iterate again. 
I did take your word on it and hence why I now now why you were so adamant about the delay() and learnt independently about non blocking delays.

> Despite trying to coach you away from this you still continue to labour this solution. I get that you are emotionally attached to it. but it just doesn't fly. I am unsure what metrics you are using [...]

I am not emotionally attached to it, simply put I can see a difference in what you were asking me to test and what I was testing with the changes in the library that worked best for me.
The metric I use are observe the GUI and report back what I see, same with the recently added performance tile. Once I had, like you now have the equipment to see what is happening the I2C bus I also reported what was happening there and your thanks for the work involved was appreciated and kindly received.

> I have tried to explain to you how the code works, but you elect to ignore my explanations, and then claim that I am being intentionally evasive. I cannot help you to understand if you are not [...]

I believe I have not ignored your explanations, I have learnt much from you in a short space of time so thank you. I have spent a lot of my time learning independently from this as I don't want to be a thorn in someone side due to my lack of understanding of something.

> There is only one person being evasive here. Please go back to the beginning of the other thread and read ALL of the posts, specifically mine. And then this thread. Please take note where I have explained something to you or asked you to test something and you have completely ignored the matter.

This is not completely true, I have tested almost everything you have asked of me, along side my own testing in parallel. It is true that I may have missed something as I am not always at my computer and no I miss some of the messages on GitHub.
I don't ignore people I can see how my persistent nature may be perceived like that, I was reporting that I was seeing a difference in the testing I was conducting.

> Please take the time to consider how many times you have derailed the project - trying to get non-standard + non-supported hardware working (Core2 M5). I spent a LOT of time integrating that into [...]

[... cut ...] what I think maybe taking place.

I too hope you understand how difficult it is to come up to partial speed on a project, when I first became involved having no prior knowledge of the code you have written.
I felt that I have made reasonable and useful contributions to the project so far.

I am sorry to hear that you feel differently about this, I hope that over time the friction that has happened between us lessens and we can work in a more conducive way for the betterment of the DIYFB project.